crankyoldgit/IRremoteESP8266 #984: Where is the documentation for the library's functions?

Blend3rman, 2019-10-31:
I expected to find a readthedocs page or at least some information in the wiki regarding the various functions in the library (especially the ones in the IRac class). Shouldn't these functions be documented in their behaviour more than just sparse comments in the source files?

crankyoldgit, 2019-11-01:
In an ideal world, yes. Luckily it's an open-source volunteer project and anyone can contribute. I look forward to your contribution & help creating the documentation you want. If you do it four times in the next few hours, you could even earn a t-shirt. See #932 
e.g.
## Easy:
* Improve some of the documentation.
## Medium:
* Add some function description comments. There are plenty of examples in the code where I've done this. Use that as a template for any function etc that doesn't have a description at the start of it.

See also the https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples directory where a lot of the functionality and some of the functions are demonstrated. BTW, ultimately the source code is the best & most up-to-date documentation in a lot of ways.

crankyoldgit, 2020-02-15:
Marking this issue as stale. Closing.

closed 2020-02-15 by crankyoldgit