crankyoldgit/IRremoteESP8266 #991: I do not get anything on the serial monitor using NodeMCU with KSM803

joelgamal, 2019-11-10:
connected Vout of receiver to D5 and using IRrecvDumpV2.
I tried this IR receiver with arduino uno and it works well.
I tried more than 5 remote controls for ACs, TVs and none of them gives any output.
I changed kRawBuf to 512, also no output
I changed kTimeout equals to 15,50,100,200, also no output.

crankyoldgit, 2019-11-10:
Did you follow the instructions? i.e. https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L39-L44

IRrecvDumpV2 **always** prints text to the serial port at start up. See https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L117-L119

If you are not seeing anything, you probably have something wrong with your serial terminal connection.

joelgamal, 2019-11-10:
this is what I see: 
"IRrecvDumpV2 is now running and waiting for IR input on Pin 14", when pressing any button nothing happens

joelgamal, 2019-11-10:
serial monitor is working, I test it, but this if condition"if (irrecv.decode(&results))" is never true, any clue why this happens?

joelgamal, 2019-11-10:
solved, replaced the IR Receiver, Is there recommended min. and max. distance?

crankyoldgit, 2019-11-10:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#basic-1

"""
5. Use an appropriate distance between the transmitter and the receiver. A receiver module can easily be overpowered and produce a corrupt signal if you have the IR source too close. A range of less than ~300mm is often enough to confuse the receiver.
"""

The real answer depends on the receiver module, the transmitting power/led, and the angle as well as a lot of other environmental factors.

Typically 60cm to 1m+ is a good "safe" bet for reliable signal capture/detection.

Depending on the power, the led focus angle/specs, your aim, and environment 5-6 meters is very easy, 10m is not to far a stretch. Some remotes are powerful enough to reflect off walls and still get a reliable signal. i.e. your personal experience will vary.

closed 2019-11-10 by crankyoldgit