crankyoldgit/IRremoteESP8266 #994: Conversion from String to uint64_t

meudayhegde, 2019-11-15:
how can i convert a string, eg "0x5BF2"
to uint64_t so that it can be transmitted using irSend.send(_,_,_) function

crankyoldgit, 2019-11-15:
This is a general "C/C++" problem, nothing to do with our library. i.e. https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

https://lmgtfy.com/?q=convert+a+string+to+uint64_t

However, to see how it is done in some of our example code, look at:
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L2650-L2667

e.g. `uint64_t my_64bit_unsigned_integer = getUInt64fromHex(my_String.c_str());`

closed 2019-11-15 by crankyoldgit

meudayhegde, 2019-11-15:
Thanks, hex string is converting to uint64_t with this method,
but when i send that uint64_t value using, irsend.send(protocol, value, len)
its returning false and appliance not detecting for SHARP protocol
any reason for this to fix? (bcz its working with smartIRRepeater example)

crankyoldgit, 2019-11-15:
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRsend.cpp#L649-L657
```
// Send a simple (up to 64 bits) IR message of a given type.
// An unknown/unsupported type will do nothing.
// Args:
//   type:  Protocol number/type of the message you want to send.
//   data:  The data you want to send (up to 64 bits).
//   nbits: How many bits long the message is to be.
//   repeat: How many repeats to do?
// Returns:
//   bool: True if it is a type we can attempt to send, false if not.
```

That routine will only return `false` when you've asked it to send a protocol it doesn't support.
I suggest you check what values you're passing to `irsend::send()`
e.g.
SHARP = 14

Also check you have `SEND_SHARP` set to `true` in https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRremoteESP8266.h

meudayhegde, 2019-11-16:
Thanks a lot for the help and very fast support, the issue was dumb, i used a bool fn(String) to use state/value method and i was forgot to return value of irsend.send
it was always returnig false!!