crankyoldgit/IRremoteESP8266 #996: Easiest way to send IR code?

komputerboy, 2019-11-23:
### Version/revision of the library used
Latest v2.7

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
```
Timestamp : 000019.608
Library   : v2.7.0

Protocol  : GREE
Code      : 0x4103205011200010 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: Off, Mode: 1 (Cool), Temp: 19C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off
uint16_t rawData[139] = {8930, 4430,  692, 1624,  666, 508,  694, 480,  692, 484,  668, 508,  694, 482,  720, 1594,  694, 480,  668, 1646,  694, 1620,  696, 478,  694, 482,  666, 508,  666, 508,  696, 480,  694, 480,  666, 508,  696, 478,  668, 508,  696, 478,  666, 508,  722, 1594,  692, 482,  668, 506,  696, 480,  694, 482,  692, 482,  694, 482,  694, 1620,  696, 480,  696, 1618,  696, 480,  694, 480,  696, 1618,  666, 508,  694, 20620,  690, 1644,  696, 480,  694, 482,  692, 482,  696, 1616,  706, 474,  722, 450,  694, 480,  692, 480,  718, 454,  720, 452,  690, 482,  694, 478,  666, 1646,  696, 478,  692, 478,  692, 480,  616, 556,  696, 478,  718, 454,  692, 480,  722, 450,  722, 450,  694, 478,  692, 480,  694, 478,  692, 480,  666, 506,  692, 1620,  692, 482,  692, 478,  666, 508,  700};  // GREE
uint8_t state[8] = {0x41, 0x03, 0x20, 0x50, 0x11, 0x20, 0x00, 0x10};
```

### Steps to reproduce the behavior
1. Compile and upload Example code IrRecvDumpV2
2. Get IR data

#### Example code used
```
#include <Arduino.h>
#include <IRrecv.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRtext.h>
#include <IRutils.h>

// ==================== start of TUNEABLE PARAMETERS ====================
const uint16_t kRecvPin = 14;
const uint32_t kBaudRate = 115200;
const uint16_t kCaptureBufferSize = 1024;

#if DECODE_AC
const uint8_t kTimeout = 50;
#else  
const uint8_t kTimeout = 15;
#endif  // DECODE_AC
const uint16_t kMinUnknownSize = 12;

#define LEGACY_TIMING_INFO false
// ==================== end of TUNEABLE PARAMETERS ====================

IRrecv irrecv(kRecvPin, kCaptureBufferSize, kTimeout, true);
decode_results results;  // Somewhere to store the results

void setup() {
#if defined(ESP8266)
  Serial.begin(kBaudRate, SERIAL_8N1, SERIAL_TX_ONLY);
#else  // ESP8266
  Serial.begin(kBaudRate, SERIAL_8N1);
#endif  // ESP8266
  while (!Serial)  // Wait for the serial connection to be establised.
    delay(50);
  Serial.printf("\n" D_STR_IRRECVDUMP_STARTUP "\n", kRecvPin);
#if DECODE_HASH
  irrecv.setUnknownThreshold(kMinUnknownSize);
#endif  // DECODE_HASH
  irrecv.enableIRIn();  // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    uint32_t now = millis();
    Serial.printf(D_STR_TIMESTAMP " : %06u.%03u\n", now / 1000, now % 1000);
    if (results.overflow)
      Serial.printf(D_WARN_BUFFERFULL "\n", kCaptureBufferSize);
    Serial.println(D_STR_LIBRARY "   : v" _IRREMOTEESP8266_VERSION_ "\n");
    Serial.print(resultToHumanReadableBasic(&results));
    String description = IRAcUtils::resultAcToString(&results);
    if (description.length()) Serial.println(D_STR_MESGDESC ": " + description);
    yield();  // Feed the WDT as the text output can take a while to print.
#if LEGACY_TIMING_INFO
    Serial.println(resultToTimingInfo(&results));
    yield();  // Feed the WDT (again)
#endif  // LEGACY_TIMING_INFO
    Serial.println(resultToSourceCode(&results));
    Serial.println();    // Blank line between entries
    yield();             // Feed the WDT (again)
  }
}
```

### Question
That is my Sharp AC IR data from IrrecvDumpV2. Then, i've able to send IR code using raw data with sendRaw(). But i want to ask is there any other easier way to send IR code?
Maybe using code data 
`Code      :   0x4103205011200010 (64 Bits)`
Or using state data
`uint8_t state[8] = {0x41, 0x03, 0x20, 0x50, 0x11, 0x20, 0x00, 0x10};`
Or using commands
`setTemp()` or `on()` or `off()`

Because every state in my Sharp AC is different, for example :
1. On IR data
```
Protocol  : GREE
--
Code        : 0x490E605001200030 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On,   Mode: 1 (Cool), Temp: 30C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off,   XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto),   Timer: Off
uint16_t rawData[139] = {9050, 4428,  710, 1594,    708, 496,  708, 498,  706, 1604,    702, 500,  704, 502,  704, 1604,    702, 504,  702, 504,  700, 1608,    700, 1606,  700, 1606,  698, 506,    698, 506,  698, 508,  696, 508,    694, 512,  692, 514,  690, 518,    686, 518,  686, 516,  686, 1622,    682, 1626,  682, 522,  682, 522,    680, 524,  680, 526,  680, 526,    678, 1630,  676, 528,  678, 1630,    676, 530,  676, 530,  674, 1634,    672, 530,  676, 19968,  674, 1630,    674, 530,  676, 530,  674, 530,    676, 530,  676, 530,  676, 530,    674, 532,  674, 532,  674, 532,    674, 532,  672, 534,  672, 532,    672, 1636,  672, 532,  674, 532,    672, 534,  672, 534,  672, 534,    672, 534,  672, 534,  672, 532,    672, 534,  672, 534,  672, 534,    670, 534,  672, 534,  670, 534,    672, 1636,  670, 1636,  672, 534,    672, 534,  672};  // GREE
uint8_t state[8] = {0x49, 0x0E, 0x60,   0x50, 0x01, 0x20, 0x00, 0x30};
```

2. Off Ir Data
```
Protocol  : GREE
--
Code        : 0x4105205001200020 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: Off,   Mode: 1 (Cool), Temp: 21C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off,   XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto),   Timer: Off
uint16_t rawData[139] = {9048, 4432,  706, 1600,    704, 500,  706, 500,  704, 502,    702, 504,  702, 504,  702, 1606,    702, 502,  702, 1608,  698, 506,    700, 1606,  700, 504,  698, 506,    698, 508,  694, 512,  692, 512,    690, 514,  688, 516,  688, 518,    684, 522,  684, 522,  684, 1624,    680, 522,  682, 524,  678, 526,    678, 528,  678, 528,  678, 526,    678, 1632,  674, 530,  674, 1634,    674, 530,  676, 530,  674, 1634,    672, 532,  674, 19966,  674, 1632,    674, 530,  674, 530,  674, 530,    674, 532,  672, 534,  674, 532,    672, 532,  674, 532,  674, 532,    672, 534,  670, 534,  672, 534,    670, 1636,  672, 534,  672, 534,    672, 534,  672, 534,  670, 534,    672, 534,  672, 532,  672, 534,    672, 536,  668, 538,  668, 536,    670, 532,  672, 534,  672, 534,    670, 536,  670, 1650,  658, 534,    672, 538,  668};  // GREE
uint8_t state[8] = {0x41, 0x05, 0x20,   0x50, 0x01, 0x20, 0x00, 0x20};
```

Also when i try to increase the temperature from 16 C to 30 C, it is always get different IR code. So if i use raw data to send IR code, it will be hard and time consuming because i need to get every IR raw code then save it to my esp8266.

Thank You

crankyoldgit, 2019-11-23:
> That is my Sharp AC IR data from IrrecvDumpV2. Then, i've able to send IR code using raw data with sendRaw(). But i want to ask is there any other easier way to send IR code?

Yes.

If take a look at the ArgoAC example code (https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/TurnOnArgoAC/TurnOnArgoAC.ino) you will see how to use an AirCon class object to create custom messages with the settings you want.
Obviously, you will need to use the `IRGreeAC` class object and `kGree*` constants instead of the Argo ones.
You can read through the code for what functions and constants/values are supported by skimming the code at: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Gree.cpp & https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Gree.h

The above is probably the solution you want.

As for the your other implied questions, yes, you can also send using the `state[]` data supplied by IRrecvDumpV2.

See https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino for example code of how to do it for a Samsung A/C message.

e.g. (Using the data from your first capture)
```
uint8_t state[8] = {0x41, 0x03, 0x20, 0x50, 0x11, 0x20, 0x00, 0x10};
irsend.sendGree(state);
```

In fact, the `GREE` protocol protocol is a rare one that also accepts using the `uint64_t` "code" as well.
e.g.
```
uint64_t data = 0x4103205011200010;
irsend.sendGree(data);
```

closed 2019-11-23 by crankyoldgit

komputerboy, 2019-11-24:
okay, i have another question. What if there is no protocol like this:
```
Timestamp :   000054.350
--
Library     : v2.7.0
 
Code        : 0x81081789 (136 Bits)
uint16_t rawData[271] = {3560, 3404,  944, 808,    922, 2556,  924, 816,  908, 842,    892, 854,  880, 2584,  902, 2576,    902, 838,  890, 852,  886, 2590,    894, 846,  886, 854,  878, 862,    874, 2602,  888, 2590,  888, 850,    880, 858,  872, 2608,  882, 856,    886, 852,  886, 854,  884, 854,    880, 860,  872, 868,  866, 874,    864, 2614,  864, 874,  864, 876,    862, 876,  862, 878,  860, 878,    860, 880,  3466, 3492,  858, 880,    858, 2620,  858, 880,  858, 882,    858, 880,  858, 2620,  858, 2620,    858, 880,  858, 880,  858, 2618,    858, 882,  858, 884,  854, 884,    852, 2622,  858, 2622,  856, 894,    844, 902,  838, 2626,  854, 884,    852, 908,  830, 914,  826, 920,    818, 922,  816, 922,  818, 922,    816, 2636,  844, 902,  838, 918,    820, 924,  814, 924,  814, 924,    814, 926,  3424, 3516,  834, 14226,    3442, 3512,  838, 906,  832, 922,    818, 2638,  840, 902,  836, 2638,    842, 2636,  842, 2638,  840, 2638,    840, 902,  836, 920,  818, 2640,    840, 904,  834, 2640,  840, 2642,    836, 2642,  836, 2642,  836, 910,    828, 2646,  834, 2644,  836, 912,    826, 2644,  834, 2646,  834, 922,    816, 926,  812, 930,  810, 2648,    832, 2648,  830, 930,  806, 2650,    830, 2648,  830, 928,  810, 928,    3418, 3524,  826, 928,  810, 930,    808, 2654,  826, 932,  804, 2656,    824, 2654,  824, 2654,  826, 2654,    824, 932,  806, 932,  804, 2660,    822, 932,  804, 2658,  824, 2654,    824, 2656,  822, 2658,  822, 932,    804, 2660,  822, 2656,  822, 932,    804, 2662,  820, 2658,  820, 932,    802, 1036,  696, 962,  774, 2672,    822, 2666,  812, 932,  798, 2664,    822, 2660,  820, 934,  798, 940,    870, 6076,  820};  // UNKNOWN 81081789
```

how to send the code? can i do like this?
```
int data = 0x81081789 ;
irsend.send(data);
```

crankyoldgit, 2019-11-24:
Interesting ... It should have described that as `UNKNOWN`

See the FAQ on dealing with `UNKNOWN` messages:

https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#My_remotes_signal_is_being_reported_as_an_Unknown_code_What_does_that_mean_How_do_I_send_an_Unknown_code

komputerboy, 2019-11-24:
I'm sorry, there was a protocol. It is described as `UNKNOWN`. 

Okay, I will check the FAQ. 

Thanks