crankyoldgit/IRremoteESP8266 #1000: Add repeat functionality dependant on the protocol

jimmys01, 2019-12-04:
If I send:

> irsend.sendNEC(0x20DF8877,32,**3**);

it is equilevant to 
```
irsend.sendNEC(0x20DF8877);
irsend.sendNEC(0x20DF8877);
irsend.sendNEC(0x20DF8877);
```

The thing is that LG protocol has a special code for "repeat previous signal" which is

> uint16_t rawData[3] = {9092, 2204, 566}; //LG repeat
> irsend.sendRaw(rawData, 3, 38);

Can you add this functionality to do this within the library? (i had to capture the repeat park and replay it as RAW as you saw by my example)

Thanks!

crankyoldgit, 2019-12-04:
Hey @jimmys01,
You are slightly mistaken.

First off:
> _irsend.sendNEC(0x20DF8877, 32, **3**);

The `3` means send the message **AND** then send `3` repeats.

So, to use your example/assumptions, that would be the equiv. of:

```
irsend.sendNEC(0x20DF8877);
irsend.sendNEC(0x20DF8877);
irsend.sendNEC(0x20DF8877);
irsend.sendNEC(0x20DF8877);
```

Secondly, `sendNEC()` has a special repeat mode, it is **NOT** the same as just duplicate single calls to `sendNEC()`. It actually sends the kind of message you are looking for.

See: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_NEC.cpp#L31-L38

```
  // Optional command repeat sequence.
  if (repeat)
    sendGeneric(kNecHdrMark, kNecRptSpace, 0, 0, 0, 0,  // No actual data sent.
                kNecBitMark, kNecMinGap, kNecMinCommandLength, 0,
                0,                     // No data to be sent.
                38, true, repeat - 1,  // We've already sent a one message.
                33);
}
```

where:
```
kNecHdrMark = 8960;
kNecRptSpace = 2240;
kNecBitMark = 560;
kNecMinCommandLength = 108080;
```
i.e. For every NEC repeat, you'd get a `uint16_t rawData[3] = {8960, 2240, 560};` plus a gap of 108,080 usecs.

That is to say, `sendNEC()` does EXACTLY what you want it to do.

> The thing is that LG protocol has a special code for "repeat previous signal" which is
>
> ```
> uint16_t rawData[3] = {9092, 2204, 566}; //LG repeat
> irsend.sendRaw(rawData, 3, 38);
> ```
>
> Can you add this [...]

Beware: LG use two different protocols which totally depend on the number of bits.
I can't tell for sure which LG protocol/size you need from the (lack of) data you have provided.

Both size variants have a similar "repeat" code to `sendNEC()`

e.g. https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_LG.cpp#L79-L104
```
  // Repeat
  // Protocol has a mandatory repeat-specific code sent after every command.
  if (repeat)
    sendGeneric(repeatHeaderMark, kLgRptSpace, 0, 0, 0, 0,  // No data is sent.
                kLgBitMark, kLgMinGap, kLgMinMessageLength, 0, 0,  // No data.
                38, true, repeat - 1, duty);
}
```
where:
```
repeatHeaderMark = 16110 (for >= 32 bit sizes) or 8500 ( < 32 bit sizes)
kLgRptSpace = 2250
kLgBitMark = 550
```

Similar to `sendNEC()` you'd get a `uint16_t rawData[3] = {8500, 2250, 550};` for each repeat.

So, if you called `sendLG(0x12345, 28, 3);` you'd get an LG message and 3 special LG repeat codes sent after it.

The **TL;DR** is if you are calling it with the correct data (and repeats) it should do pretty much what you want.

Or am I missing something?

jimmys01, 2019-12-04:
I clearly had missed something. thanks for clarifying that.

closed 2019-12-04 by jimmys01