crankyoldgit/IRremoteESP8266 #1001: seems a mistake in IRac.cpp

spyzp, 2019-12-04:
sorry for my english...
it seems I've found a bug, but I have no possibility to fix and debug it. 
So, I just leave it for authors:

IRremoteESP8266 2.7.1

in function `IRac::sendAc(const stdAc::state_t desired, const stdAc::state_t *prev) {`
after `stdAc::state_t send = this->handleToggles(desired, prev);`
I guess you should use `send`, but sometimes you use `desired` instead

```
  // Convert the temperature to Celsius.
  float degC;
  if (desired.celsius) 
    degC = send.degrees;
  else
    degC = fahrenheitToCelsius(desired.degrees);
```
it is not a big mistake
but below you should strongly want to use `send`, modified by `handleToggles()`:

```
  bool on = desired.power;
```

And here:
```
  // A hack for Home Assistant, it appears to need/want an Off opmode.
  if (desired.mode == stdAc::opmode_t::kOff) on = false;
```
Just try to imagine how it works with WHIRLPOOL_AC, which toggles power... 
I guess `desired.power` should be modified in this case, and it must be done before it passed `handleToggles()`.

And one more thing, not so important, `IRac::cmpStates` does not count if `next` in celsius but `prev` in fahrenheit or vise versa.

fuseI0, 2019-12-04:
hey !

can you help me out ! how can i use Multiple Receivers on ESP8266
i want to receive 3/4 different IR from different direction !

Suggest me some examples if possible ! Thank you !

crankyoldgit, 2019-12-04:
@fuseI0 The only way to use multiple IR receiver modules with this library is for them to share the same physical input GPIO. The library doesn't support multiple modules in software.
Also, please don't hijack someone else's Issue. Create your own! :)

crankyoldgit, 2019-12-04:
> sorry for my english...

@spyzp No need to apologise. Your English is way way better than my what ever your native language is. ;-)

> it seems I've found a bug, but I have no possibility to fix and debug it.
> So, I just leave it for authors:
>
> IRremoteESP8266 2.7.1
>
> in function `IRac::sendAc(const stdAc::state_t desired, const [...]

Technically it shouldn't matter, but yes. It's inconsistent. I'll fix it.

> but below you should strongly want to use `send`, modified by `handleToggles()`:
>
> ```
>   bool on = desired.power;
> ```
>
> And here:
>
> ```
>   // A hack for Home Assistant, it appears to need/want an Off [...]

Yep. That's a bug. Will fix it in a PR in a few minutes.

> 
> And one more thing, not so important, `IRac::cmpStates` does not count if `next` in celsius but `prev` in fahrenheit or vise versa.

Correct. If one state is in celsius mode, and the other in fahrenheit mode, they are not the same state. Thus it is behaving correctly.

crankyoldgit, 2019-12-05:
@spyzp Can you please download and test https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1001 (#1002) and let me know if that fixes your problems?

spyzp, 2019-12-05:
As I said before, I have no possibility to test it, and this bug is not affected my task.
I just accidentally noticed this problem.

But I've read through your pull request, and it seems to be ok.

fuseI0, 2019-12-05:
> @fuseI0 The only way to use multiple IR receiver modules with this library is for them to share the same physical input GPIO. The library doesn't support multiple modules in software.
> Also, please don't hijack someone else's Issue. Create your own! :)

@crankyoldgit
apologies for the previous comment ! ill create a new issue

crankyoldgit, 2019-12-05:
> As I said before, I have no possibility to test it, and this bug is not affected my task.
> I just accidentally noticed this problem.
> 
> But I've read through your pull request, and it seems to be ok.

Gotcha. Yes, it's an odd corner case, but it should be fixed now when merged. Thanks again for bringing it up. ;-)

spyzp, 2019-12-05:
@crankyoldgit Thank YOU! You've made beautiful library. 
But, waiting a lot for `IRLGAc` class... =)
Is there some dates when you plan to release it?

crankyoldgit, 2019-12-05:
> @crankyoldgit Thank YOU! You've made beautiful library.

Thanks.

> But, waiting a lot for `IRLGAc` class... =)
> Is there some dates when you plan to release it?

No, there isn't. Without data or existing code/analysis, it's not going to happen anytime soon. i.e. I don't have access to an LG aircon.

See also:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#The_libraryexample_code_doesnt_understand_my_Air_Conditioner_remote_
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Can_you_reverse_engineer_my_AirCon_protocol_for_me

spyzp, 2019-12-06:
I don't have access to an LG aircon too. =)

There are many IR remote control applications for Android. Two years ago, I've used one, written by Xiaomi, on Xiaomi Mi4c. It works good. And it sends correct codes for many many models for each of huge number of brands... So, as I understand, we could use this application for reverse engineering, having no access to real devices... 
But I don't like this idea. Somewhere must exist documentation/libraries, which allowed them to release such an application... Did you dig this way?

crankyoldgit, 2019-12-07:
> I don't have access to an LG aircon too. =)

Then why do you want a `IRLGAc` class? :)

> Did you dig this way?

No, I have not gone looking for details of a protocol and/or other implementations. Without someone who has a device, there is no way to know I've coded it up correctly. Plus, I don't have the need/incentive to add support for it. Someone who has one is best to collect the data and confirm it works.

spyzp, 2019-12-07:
> Then why do you want a IRLGAc class? :)
I'm trying to make a "product", and I suppose many people use LG ACs, so I need it for universality.

Ok. If I find one, I will try to dig into reverse engineering... =)

closed 2019-12-11 by crankyoldgit

crankyoldgit, 2020-01-06:
FYI, the aforementioned changes/code has been included in the v2.7.2 release of the library.

PR #1002 (2019-12-11): [Bug] Common AC api: Better handle protocols with power toggles.
* Add a cleaner way of handling HomeAssistant mode "off" -> power "off"
* Add more comments to improve the code readablity.
* Handle A/Cs with power toggles better.
* Add unit tests to confirm correct operation with Whirlpool A/Cs

Fixes #1001
FYI @spyzp
commit 5afbf628 (2019-12-11): [Bug] Common AC api: Better handle protocols with power toggles. (#1002)

* Add a cleaner way of handling HomeAssistant mode "off" -> power "off"
* Add more comments to improve the code redablity.
* Handle A/Cs with power toggles better.
* Add unit tests to confirm correct operation with Whirlpool A/Cs

Fixes #1001
FYI @spyzp