crankyoldgit/IRremoteESP8266 #1003: Support for Multiple Receivers

fuseI0, 2019-12-05:
from @crankyoldgit 

> @fuseI0 The only way to use multiple IR receiver modules with this library is for them to share the same physical input GPIO. The library doesn't support multiple modules in software.

at - https://github.com/crankyoldgit/IRremoteESP8266/issues/1001#issuecomment-561874994

Thanks for the reply ! but i really wanted the multiple receiver support on multiple GPio's
something like this guy did for UNO - https://github.com/Neco777/Arduino-IRremote/commit/4810d99b516fb9783e0083d1df8de1d455925f5a

can it be done ?

Thank you for your library it really helps !

crankyoldgit, 2019-12-05:
> can it be done ?

Technically, yes. It could be done, in software. I've already tried to hack it in once, and it became horrible, fast. I'm not going to go into it here, but the way the library currently does it, it becomes expensive memory-wise, and `IRAM`-wise as well. i.e. A capture array and an distinct interrupt handler, and timer per receiving channel. It just uses too much resources for general use.

As there has only been three requests for this feature that I know of in the several years I've been involved with the library, it's not a hotly requested or needed feature. i.e. Most people don't care which module received the message, they just want the message.

To do it _"like this guy did for UNO"_ would require a major rewrite of the low levels of the library and change the way the library behaves. It's not going to happen anytime soon, if at all.

fuseI0, 2019-12-05:
GOTCHA ,

Thank you for the reply !

if esp8266 is unable to handle such resources then how does UNO ?

and if there's any way i could do this ? i'm weak in basics though but can give a shot if possible !
or maybe only few changes i can do to the https://github.com/Neco777/Arduino-IRremote/commit/4810d99b516fb9783e0083d1df8de1d455925f5a library to work for ESP8266 ?
cause i just need a basic receiving part not processing only NEC would work but for multiple.

also can i connect UNO and esp8266 together like UNO does the multiple IR receiving part and esp does the other ?

waiting for your reply !

crankyoldgit, 2019-12-05:
The ESP can do it, but unlike the UNO, the ESP is trying to do a lot lot lot more other things at the same time. e.g. Wifi etc. People use this library in things like Tasmota (sonoff firmware) and ESPEasy etc. In those and other projects, they are trying to do a gagillion other things all at once. We fuss over using half a kilobyte of ram/flash and are desperate to save even a few bytes of IRam (special ram used by interrupts).

> or maybe only few changes i can do to the Neco777/Arduino-IRremote@4810d99 library to work for ESP8266 ?

You are always welcome to fork your own copy of the library & try. That's the beauty of Open Source Software. If you do, you will find that the original "Arduino IRremote" library uses a different style/system of capturing an IR message than this library, so it will be a major re-work to try to fit it in.

> also can i connect UNO and esp8266 together like UNO does the multiple IR receiving part and esp does the other ?

Yes, you could do that. However, you are going to have to handle passing messages between the two devices yourself. e.g. via a Serial connection.

Stepping back a bit. What are you trying to achieve?

If you can physically block IR signals so they are received only by one IR module at a time, you can easily share the same input GPIO. The nature of IR remote messages are they are very short and you are unlikely to ever be receiving a message by two (or more) IR receiver modules at the same time unless they both can see the same signal.

fuseI0, 2019-12-05:
@crankyoldgit  thank you for the reply !

> The ESP can do it, but unlike the UNO, the ESP is trying to do a lot lot lot more other things at the same time. e.g. Wifi etc. People use this library in things like Tasmota (sonoff firmware) and [...]
ok ! got the point 

> You are always welcome to fork your own copy of the library & try. That's the beauty of Open Source Software. If you do, you will find that the original "Arduino IRremote" library uses a different [...]
will try if i could ! as i said I'm weak in basics but can code

>Yes, you could do that. However, you are going to have to handle passing messages between the two devices yourself. e.g. via a Serial connection.
>
i'll try that. this looks easy then to fork.

>Stepping back a bit. What are you trying to achieve?

If you can physically block IR signals so they are received only by one IR module at a time, you can easily share the same input GPIO. The nature of IR remote messages are they are very short and you are unlikely to ever be receiving a message by two (or more) IR receiver modules at the same time unless they both can see the same signal.
>

I'm trying to place IR receivers on human body eg : head, chest and shoulders (basically like a tagging system) then the receivers will react to the IR signal specifically on the specific parts of body !

although since it'll be tough for me to achieve this on ESP8266 i'll go with UNO+ESP8266 together !

let me know if you have any suggestions on this for me !

fuseI0, 2019-12-05:
@crankyoldgit  thank you for the reply !

> The ESP can do it, but unlike the UNO, the ESP is trying to do a lot lot lot more other things at the same time. e.g. Wifi etc. People use this library in things like Tasmota (sonoff firmware) and [...]
ok ! got the point 

> You are always welcome to fork your own copy of the library & try. That's the beauty of Open Source Software. If you do, you will find that the original "Arduino IRremote" library uses a different [...]
will try if i could ! as i said I'm weak in basics but can code

>Yes, you could do that. However, you are going to have to handle passing messages between the two devices yourself. e.g. via a Serial connection.
>
i'll try that. this looks easy then to fork.

>Stepping back a bit. What are you trying to achieve?

> If you can physically block IR signals so they are received only by one IR module at a time, you can easily share the same input GPIO. The nature of IR remote messages are they are very short and [...]

I'm trying to place IR receivers on human body eg : head, chest and shoulders (basically like a tagging system) then the receivers will react to the IR signal specifically on the specific parts of body !

although since it'll be tough for me to achieve this on ESP8266 i'll go with UNO+ESP8266 together !

let me know if you have any suggestions on this for me !

crankyoldgit, 2019-12-05:
> I'm trying to place IR receivers on human body eg : head, chest and shoulders (basically like a tagging system) then the receivers will react to the IR signal specifically on the specific parts of body !

Yep, I understand your problem now.

> 
> although since it'll be tough for me to achieve this on ESP8266 i'll go with UNO+ESP8266 together !
> let me know if you have any suggestions on this for me !

That seems like the path of least resistance/easiest to implement.
e.g. Have the UNO detect the IR messages & say "chest hit from player X" over serial to the ESP and have the ESP react accordingly.

fuseI0, 2019-12-05:
yes ! Exactly 

Thank you for your support !

closed 2019-12-05 by fuseI0