crankyoldgit/IRremoteESP8266 #1004: ALOKA SleepyLights LED Lamp

JHale716, 2019-12-15:
### Other useful information
_More information is always welcome. Be verbose._

IR codes and structure for kids ALOKA SleepyLights LED Lamp. https://aloka-designs.com/

May be useful for someone wanting to control the baby's lamp without entering the room... Squeaky floors can be a major hassle when baby is a light sleeper.

The lamp is toggled on and off with the same power button
The colour, when selected, is the brightest and there are 4 levels of brightness that decrease on each send of the colour.

Codes captured using the IRrecvDumpV2.ino sketch with no mods on a Wemos d1Mini with a KY-022 infrared receiver.

The remote does send a standard NEC repeat when repeating.

```
// Protocol NEC & buttons defined left to right, top line to bottom line on the remote.
#define ALOKA_BITS  32
#define POWER 0xFF609F     
#define ledWHITE 0xFF906F 
#define ledGREEN 0xFF9867
#define ledBLUE 0xFFD827
#define ledPINKRED 0xFF8877
#define ledRED 0xFFA857
#define ledLTGREEN 0xFFE817
#define ledMIDBLUE 0xFF48B7
#define ledPINK 0xFF6897
#define ledORANGE 0xFFB24D
#define ledYELLOW 0xFF00FF
#define NIGHTFADE 0xFF50AF
#define NIGHTTIMER 0xFF7887
#define ledRAINBOW 0xFF708F
#define ledTREEGROW 0xFF58A7  // (didn't have a better description for it...)
```

crankyoldgit, 2019-12-15:
Thanks for this. I'll update the supported devices and add them as constants etc.

closed 2019-12-26 by crankyoldgit

crankyoldgit, 2020-01-06:
FYI, the aforementioned changes/code has been included in the v2.7.2 release of the library.

JHale716, 2020-01-07:
> FYI, the aforementioned changes/code has been included in the v2.7.2 release of the library.

Awesome! Thanks for the inclusion and updates!

JHale716, 2020-01-07:
Question not to create a new thread. Would you like other makes/models codes to add to the library? Have heaps of IR remote stuff here

crankyoldgit, 2020-01-07:
"It depends".

We could probably create a text file that is known codes etc for people to use. Or ... you could set up a website and create a DB for people to use directly with IRremote etc. i.e. Get your own exposure for your effort.

I often resort/push people to https://irdb.globalcache.com/ which has most things. If it's not in there, then a) It probably should be, and b) we could add it.

The TL;DR: is, let's not re-invent the wheel, but if we do, lets make it easy to add to for all people and not require them to update to the latest version of the library to have it included in the code etc.

JHale716, 2020-01-07:
Cool, fair comment. I’ll scratch my head on that one. I'll check out the link too, as that may well be the best answer.

PR #1005 (2019-12-26): Add remote codes for Aloka LED lamp.
Fixes #1004
commit c5bc792d (2019-12-26): Add remote codes for Aloka LED lamp. (#1005)

Fixes #1004