crankyoldgit/IRremoteESP8266 #1008: LG Hvac Support

pablogventura, 2020-01-03:
### Version of the library used
v2.7.1

### Expected behavior
I want to control LG AC using Hvac, as in Mitsubishi AC Example.

### Actual behavior
I don't know if this library implement HVAC for LG AC. In the past, I used Sonoff Tasmota firmware, that used a adhoc implementation for LG HVAC. Now, Tasmota use only this library and I would know if I can keep using HVAC for LG AC.

Thanks you for your great work!

crankyoldgit, 2020-01-03:
Currently the library (2.7.1) doesn't support deep LG a/c integration. However, if tasmota once supported it, there is probably code out their how the messages are constructed. I'll take a look to see if I can find it and integrate it. Can you please use the `IRrecvDumpV2` example code in 2.7.1 to capture a few messages and document what those messages should be. e.g. "On, Cool, 23 deg C, Fan Auto", etc etc. So I can verify if I have it working correctly.

I'll try to add it shortly.

crankyoldgit, 2020-01-03:
@pablogventura Do you know what (older) version of Tasmota supported your LG a/c?

I can see support for Toshiba & Mitsubishi using something other than IRremoteESP8266, but not anything else.
e.g. https://github.com/arendst/Tasmota/blob/v6.3.0/sonoff/xdrv_05_irremote.ino#L281-L282

pablogventura, 2020-01-03:
It was added in 6.4.1 version. Here you have: https://github.com/arendst/Tasmota/blob/54c2eb283a02e4287640a4595e506bc6eadbd7f2/sonoff/xdrv_05_irremote.ino#L327

Thanks for your interest!

crankyoldgit, 2020-01-03:
Thanks. Looking at it now.

crankyoldgit, 2020-01-03:
P.S. I'm still going to need a few samples captured by `IRrecvDumpV2` as described earlier.

pablogventura, 2020-01-03:
Ok! I'll do it tomorrow! It's 1 am here.

Thanks a lot!

crankyoldgit, 2020-01-03:
@pablogventura Can you please download and try out the `LgAc` branch (https://github.com/crankyoldgit/IRremoteESP8266/tree/LgAc) per PR #1009 and let me know how it goes.

I'll still need some capture/codes from you to verify it is working correct, oh, and the make/model numbers of the AC unit **and** the remote too please.

jimmys01, 2020-01-03:
@crankyoldgit it is not exactly what you wanted (not captured from IRrecvDumpV2) but here, 
[LG AC codes.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4020207/LG.AC.codes.txt)

This is an old GE unit, when I get back to work I can test and or send codes from modern LG units. 
I know from our units that even between LG remotes that are the same looking and work interchangeably between ACs they have a weird thing that you cannot power off one AC with the other ones remote if it is in a certain state...

jimmys01, 2020-01-03:
I can confirm that it works for my GE AC!!
The off command is 

```
{"protocol":"LG","data":"0x88C0051","bits":28}
 Power: Off, Mode: 0 (Cool), Temp: 15C, Fan: 5 (Auto)
```

crankyoldgit, 2020-01-04:
@jimmys01 Thanks for the confirmation and test data. I've added "Fan" mode, and tests based on your data.

I haven't worked out "sleep" mode yet, so if you can work out what is going on bit-wise I'll add it later.

jimmys01, 2020-01-04:
How about the off state situation? My Ac expects this code 88C0051 which is  `Power: Off, Mode: 0 (Cool), Temp: 15C, Fan: 5 (Auto)` or else it wont shut down. I am sure that other AC need different code for that.

crankyoldgit, 2020-01-04:
Good to know. Thanks for the feedback I'll look at updating it to ensure that gets sent always for any "off" state.
Can you please capture any other remotes which have a different "off" code and I'll work out some way of supporting them.

pablogventura, 2020-01-04:
Using IRrecvDumpV2 v2.7.1 all the codes has been detected as "Protocol  : UNKNOWN".

Codes:
On, Cool, 18 deg C, Fan 5

Code      : 0xAFC3034C (30 Bits)
uint16_t rawData[59] = {3272, 9844,  506, 1588,  536, 498,  534, 498,  536, 498,  534, 1540,  534, 506,  534, 498,  534, 500,  532, 500,  534, 498,  534, 498,  534, 506,  534, 500,  534, 498,  534, 498,  534, 498,  534, 500,  534, 498,  534, 1566,  508, 1566,  508, 500,  534, 1540,  534, 506,  534, 500,  534, 500,  534, 1560,  508, 1540,  534, 1558,  508};  // UNKNOWN AFC3034C

Off, Cool, 18 deg C, Fan 5

Code      : 0x1035C9DA (30 Bits)
uint16_t rawData[59] = {3196, 9836,  506, 1580,  534, 498,  536, 498,  534, 498,  534, 1566,  508, 498,  534, 498,  536, 498,  534, 1566,  508, 1566,  508, 508,  534, 498,  534, 500,  534, 500,  532, 500,  534, 498,  534, 498,  534, 500,  534, 506,  534, 506,  534, 500,  534, 1558,  510, 498,  534, 1558,  508, 506,  534, 498,  536, 498,  534, 1556,  510};  // UNKNOWN 1035C9DA

On, Cool, 19 deg C, Fan 5

Code      : 0x4CFFC1A3 (30 Bits)
uint16_t rawData[59] = {3276, 9830,  506, 1580,  534, 500,  534, 498,  534, 500,  532, 1568,  482, 540,  510, 540,  534, 498,  534, 500,  534, 498,  536, 498,  536, 498,  534, 1558,  508, 498,  534, 506,  534, 498,  534, 498,  534, 1576,  508, 506,  534, 500,  534, 500,  534, 1540,  534, 498,  534, 498,  534, 498,  534, 498,  534, 500,  534, 498,  534};  // UNKNOWN 4CFFC1A3

On, Cool, 19 deg C, Fan Auto

Code      : 0x2CFB0AD6 (30 Bits)
uint16_t rawData[59] = {3274, 9850,  562, 1568,  508, 506,  534, 498,  534, 498,  534, 1548,  534, 506,  536, 498,  536, 506,  532, 516,  536, 498,  534, 498,  536, 504,  534, 498,  536, 498,  534, 498,  532, 516,  534, 498,  534, 1558,  510, 498,  536, 498,  536, 498,  534, 1558,  508, 506,  534, 1566,  508, 1566,  506, 516,  536, 498,  534, 1566,  508};  // UNKNOWN 2CFB0AD6

On, Auto, 23 deg C, Fan 5

Code      : 0xF5DF959A (30 Bits)
uint16_t rawData[59] = {3250, 9876,  510, 1588,  534, 500,  534, 498,  534, 500,  534, 1540,  534, 508,  534, 498,  534, 498,  534, 498,  534, 500,  534, 500,  534, 506,  534, 1556,  512, 506,  534, 1554,  510, 1556,  486, 1572,  534, 506,  534, 500,  534, 508,  508, 540,  534, 1564,  486, 540,  536, 498,  534, 498,  534, 1556,  510, 1564,  510, 1564,  510};  // UNKNOWN F5DF959A

On, Dehumidifier, --, Fan 5

Code      : 0x5C44A393 (30 Bits)
uint16_t rawData[59] = {3168, 9856,  508, 1572,  536, 506,  534, 498,  534, 498,  534, 1558,  508, 506,  536, 498,  534, 498,  534, 500,  534, 498,  534, 500,  534, 508,  534, 1558,  508, 506,  534, 498,  534, 1558,  510, 1574,  510, 506,  508, 540,  508, 1582,  534, 500,  534, 1558,  508, 498,  534, 498,  534, 498,  534, 1558,  508, 1566,  508, 508,  534};  // UNKNOWN 5C44A393

On, Heat, 25 deg C, Fan 5

Code      : 0x41F5C34D (30 Bits)
uint16_t rawData[59] = {3276, 9864,  504, 1572,  534, 506,  534, 506,  534, 498,  534, 1566,  508, 498,  534, 500,  534, 506,  534, 506,  534, 498,  534, 500,  534, 500,  532, 1558,  508, 1574,  508, 506,  534, 498,  534, 1558,  510, 498,  534, 1558,  508, 506,  534, 498,  534, 1558,  508, 498,  534, 498,  534, 1558,  508, 506,  534, 1566,  508, 498,  534};  // UNKNOWN 41F5C34D

On, Fan, --, Fan 5

Code      : 0x8F05206C (30 Bits)
uint16_t rawData[59] = {3432, 9864,  560, 1568,  508, 506,  534, 498,  534, 498,  534, 1574,  510, 506,  536, 506,  510, 540,  534, 500,  534, 498,  534, 500,  534, 506,  534, 1558,  508, 498,  508, 1580,  536, 498,  534, 498,  536, 504,  534, 1558,  508, 1548,  534, 506,  534, 1566,  510, 506,  508, 540,  534, 506,  508, 540,  534, 498,  534, 1558,  508};  // UNKNOWN 8F05206C

On, Cool, 22 deg C, Fan 5, ENERGY SAVE MODE ON

Code      : 0x9195CE4A (30 Bits)
uint16_t rawData[59] = {3274, 9846,  562, 1566,  510, 506,  534, 498,  534, 498,  534, 1564,  512, 496,  536, 498,  534, 506,  508, 540,  536, 506,  510, 538,  534, 1556,  510, 506,  532, 500,  510, 540,  534, 498,  534, 498,  534, 500,  534, 500,  534, 498,  534, 498,  534, 1564,  510, 498,  534, 498,  534, 500,  510, 1572,  534, 506,  534, 1556,  510};  // UNKNOWN 9195CE4A

On, Cool, 22 deg C, Fan 5, ENERGY SAVE MODE OFF

Code      : 0x49475271 (30 Bits)
uint16_t rawData[59] = {3274, 9854,  504, 1580,  536, 506,  534, 506,  534, 498,  534, 1558,  508, 498,  534, 498,  534, 500,  534, 498,  536, 498,  534, 498,  534, 1568,  508, 506,  534, 498,  534, 498,  534, 498,  534, 498,  534, 498,  534, 506,  534, 498,  536, 498,  534, 1558,  508, 506,  534, 1558,  508, 506,  534, 1566,  508, 1566,  508, 498,  534};  // UNKNOWN 49475271

On, Cool, --, Fan 5, TURBO/JET MODE ON

Code      : 0x3A398D8A (30 Bits)
uint16_t rawData[59] = {3192, 9844,  530, 1566,  532, 506,  508, 524,  532, 502,  508, 1568,  530, 508,  532, 500,  510, 524,  532, 500,  532, 500,  508, 526,  532, 1542,  532, 500,  532, 500,  532, 510,  532, 508,  532, 502,  532, 500,  508, 540,  534, 508,  506, 1576,  532, 508,  532, 510,  508, 524,  532, 1556,  486, 540,  534, 508,  508, 1576,  532};  // UNKNOWN 3A398D8A

crankyoldgit, 2020-01-05:
@pablogventura thanks for the data. The raw data you provided successfully decodes as `LG2` by the library. By chance, did you disable that protocol?

Can you also provide me with brand/model info for your A/C and remote please?

pablogventura, 2020-01-05:
I didn't disable any protocols. I used the example as it comes in the library. I was surprised that the library took it as "Unknown". Where would I look if it was deactivated? 

The model of the air conditioner is: LG S4-w12ja3aa. The remote control is AKB75215403.

I haven't been able to test the signal sending to the air conditioner yet because the circuit I use for that is embedded in the wall, and I can't easily reprogram it. On Monday, I will be able to get the transistor and the led to test it.

Thank you very much!

crankyoldgit, 2020-01-05:
@pablogventura 

> I didn't disable any protocols. I used the example as it comes in the library. I was surprised that the library took it as "Unknown". Where would I look if it was deactivated?

Have a look at: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L110-L111

That's what controls the LG/LG2 decoding.

You may need to check what **copy** of the library you are using for your compiles of `IRrecvDumpV2`. IIRC, the branch that is shipped with Tasmota disables most of the protocol decodes.

To assist yourself, try changing `_IRREMOTEESP8266_VERSION_ ` in https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L55 to something unique. `IRrecvDumpV2` will show that string and you'll know for sure.

I added a _test_ based on the first raw data snippet you provided, and it successfully decodes here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/LgAc/test/ir_LG_test.cpp#L804-L825

@pablogventura & @jimmys01 
I've tweaked that branch to handle the `LG` & `LG2` protocols from the same `IRLgAc` class.
You can use the `ac.setModel()` command to switch between them.
e.g.

```
IRLgAc ac(kIrLed);

ac.begin();
ac.setPower(true);
ac.setFan(kLgAcHigh);
ac.setMode(kLgAcCool);
ac.setTemp(26);
ac.setModel(GE6711AR2853M);  // Default model. (aka 1)
ac.send();  // Sends a 28bit LG message
delay(5000);
ac.setModel(AKB75215403);  // Alternative model (aka 2).
ac.send();  // Sends a 28bit LG2 message
```

@jimmys01 Per your feedback, it should always send `0x88C0051` now when it's in an _off_ state.

Enjoy!

Feedback welcome

pablogventura, 2020-01-05:
You're right. I was using the version provided by Tasmota. Using version 2.7.1 as it stands in this repository all code is detected as 28-bit LG2.
I'll test sending tomorrow, Monday. Thanks again!

pablogventura, 2020-01-05:
I can confirm that it works! In fact, both GE6711AR2853M and AKB75215403 models work in my air conditioner.
Thank you very much!

crankyoldgit, 2020-01-05:
Excellent! Thanks for the confirmation #1009

closed 2020-01-05 by crankyoldgit

s-hadinger, 2020-01-06:
Great. Do you plan to launch 2.7.2 shortly?

crankyoldgit, 2020-01-06:
@s-hadinger Because you(Tasmota) asked for it, how can I not do it immediately. :)  #1011 

It will happen shortly!

X-ref: https://github.com/arendst/Tasmota/pull/4377

crankyoldgit, 2020-01-06:
FYI, the aforementioned changes/code has been included in the v2.7.2 release of the library.

s-hadinger, 2020-01-06:
Thanks a lot David. 2.7.2 is now part of Tadmota-IR.

PR #1009 (2020-01-05): Experimental detailed support for LG a/c.
* Power, Mode, Temp, & Fan speed control.
* Common a/c support.
* Unit tests coverage.

Fixes #1008
commit ad399c5f (2020-01-05): Experimental detailed support for LG a/c. (#1009)

* Initial detailed support for LG a/c.
* Support LG2 in IRLgAc class.
* Add different model support to handle LG & LG2 protocols.
* Power, Mode, Temp, & Fan speed control.
* Common a/c support.
* Unit tests coverage.
* Update supported devices list.
* Add unit test to check raw LG2 messages work.
* Ensure the Off command is always 0x88C0051

Fixes #1008