crankyoldgit/IRremoteESP8266 #1012: Local include of this library into other projects

Def3nder, 2020-01-09:
Hi David,

I love the IRremoteESP8266 library and the ESP32 support since 2.6.2 😄 

However the number of different mostly air conditioner protocols occupies a lot of program space (like 28k). Changing the #defines in the IRremoteESP8266.h header file from "true" to "false" does resolve this problem.

Now my question: is it allowed to use an exact copy of this library e.g. v2.7.2 and add this to another project (not as include but directly in the src directory) and then make only these #define-changes ?

...or is there any way of getting the compiler to not use any SEND-code and no code for air conditioners ?

crankyoldgit, 2020-01-09:
> Now my question: is it allowed to use an exact copy of this library e.g. v2.7.2 and add this to another project (not as include but directly in the src directory) and then make only these #define-changes ?

As long as you are complying with https://github.com/crankyoldgit/IRremoteESP8266/blob/master/LICENSE.txt then you are free to do so as you please.

[IANAL](https://en.wikipedia.org/wiki/IANAL) but I think the relevant clauses are:
```
  For example, if you distribute copies of the library, whether gratis
or for a fee, you must give the recipients all the rights that we gave
you.  You must make sure that they, too, receive or can get the source
code.  If you link other code with the library, you must provide
complete object files to the recipients, so that they can relink them
with the library after making changes to the library and recompiling
it.  And you must show them these terms so they know their rights.

 We protect your rights with a two-step method: (1) we copyright the
library, and (2) we offer you this license, which gives you legal
permission to copy, distribute and/or modify the library.
```

or TL;DR: As long as you make this library's source & license freely (as in beer & speech) available, & don't claim it as your own (except for your modifications), then you can do with it as you please.

e.g. [Tasmota](https://github.com/arendst/Tasmota) & [ESPEasy](https://github.com/letscontrolit/ESPEasy) use/include/incorporate this library

> ...or is there any way of getting the compiler to not use any SEND-code and no code for air conditioners ?

Not easily. The `#define`s are the way to do it. I could change them to see if they are already defined before it defines them so you could pass a **compiler-time** flag. e.g. `-DDECODE_KELVINATOR false` etc. i.e. I could do it, but it would make code more unreadable than necessary IMHO.
e.g.
Existing:
```
...
#define SEND_NEC true
#define DECODE_NEC true
...
```
would become:
```
...
#ifndef SEND_NEC
#define SEND_NEC true
#endif  // SEND_NEC
#ifndef DECODE_NEC
#define DECODE_NEC true
#endif  // DECODE_NEC
...
```
Not impossible, but it's hard enough getting people to edit that file as is, let alone with the extra lines around them.

FYI, the compiler will only include the code you actually need/gets referenced. i.e. If you don't use an `sendBlah` calls, then it won't include/link them. Even if `SEND_BLAH` is set to `true`.

Decoding however is different. The main [irrecv.decode()](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRrecv.cpp#L322-L695) method will go through/link in EVERY protocol specific decode routine UNLESS you've disabled it via `#define DECODE_BLAH false`

There is no **run-time** way to say "only decode these few protocols" **and** ONLY include the compiled decode code for those protocols. Compile-time is the only way to exclude/reduce the included code.

I hope that answers your questions.

crankyoldgit, 2020-01-09:
Oh, and the `decode()` routine won't pull in all the hip A/C controlling code to bloat your binary either. Unless of course you use it in some way.

Def3nder, 2020-01-09:
Hi @crankyoldgit thanx a lot - yes, that helps !

Would it be another possibility to not locally include the WHOLE library but only the IRremoteESP8266.h file and change only that ?  (and tell platformio to use the specific version where the header was copied from (e.g. @2.7.2) ?

crankyoldgit, 2020-01-10:
Hey @Def3nder , I've created a [branch](https://github.com/crankyoldgit/IRremoteESP8266/tree/compile_flags) that should allow you to use the library without modification and control which protocols are compiled etc via `-D` flags to the compiler.

This should solve your issue and address any concerns w.r.t. use under different licenses.
i.e. You shouldn't need to duplicate the library within your own source tree.

Can you please test it out and let me know how it goes?

Def3nder, 2020-01-10:
Hi David,

thank you so much !
It does actualy work 😄 

I changed the lib_deps_external  from `IRremoteESP8266@2.7.2` to `https://github.com/crankyoldgit/IRremoteESP8266.git#compile_flags` and added the following build_flags:
  -D_IR_ENABLE_DEFAULT_=false 
  -D DECODE_HASH=true 
  -D DECODE_NEC=true 
  -D DECODE_SONY=true 
  -D DECODE_PANASONIC=true 
  -D DECODE_JVC=true 
  -D DECODE_SAMSUNG=true
  -D DECODE_LG=true
  -D DECODE_SANYO=true
  -D DECODE_SHARP=true 
  -D DECODE_DENON=true 
 
And this reduces the .bin size by 2% !
Great ! 

Do you plan to commit this to the master ?

crankyoldgit, 2020-01-10:
Excellent. Thanks for confirming it worked as intended.

> Do you plan to commit this to the master ?

Yes. It is pending code review (PR #1013). Once that is done, it will be merged.

closed 2020-01-16 by crankyoldgit

crankyoldgit, 2020-01-30:
The code changes referenced above have been included in the v2.7.3 release of the library.

PR #1013 (2020-01-16): Allow protocols to be enabled or disabled with compiler flags.
Allow users of the library to externally control which protocols are 
used/enabled/compiled without having to modify the `IRremoteESP8266.h` 
file.
i.e. `SEND_BLAH` & `DECODE_BLAH` can be controled with preprocessor 
compiler flags (see 
https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html)
e.g. `-DDECODE_KELVINATOR=false`

Fixes #1012
commit 48056bcd (2020-01-16): Allow protocols to be enabled or disabled with compiler flags. (#1013)

Allow users of the library to externally control which protocols are 
used/enabled/compiled without having to modify the `IRremoteESP8266.h` 
file.
i.e. `SEND_BLAH` & `DECODE_BLAH` can be controled with preprocessor 
compiler flags (see 
https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html)

Fixes #1012