crankyoldgit/IRremoteESP8266 #1014: How can i decode the IR protocol from these details?

wounder123, 2020-01-16:
From below, response get from IR receiver (using Arduino library )
could you please help me decode the protocol formate?
thanks.

Timestamp : 000657.493
Library : v2.7.2

Protocol : UNKNOWN
Code : 0xB4C0FCA9 (129 Bits)
uint16_t rawData[257] = {462, 1044, 474, 532, 472, 1016, 492, 532, 472, 532, 448, 1038,
492, 514, 490, 1010, 462, 544, 490, 1014, 450, 558, 416, 1090, 474, 1012, 490, 516, 492,
1008, 418, 590, 462, 544, 492, 514, 584, 440, 472, 532, 536, 474, 462, 538, 446, 560,
416, 590, 462, 548, 492, 516, 464, 560, 434, 570, 444, 556, 466, 538, 416, 588, 462,
544, 490, 1014, 472, 534, 462, 548, 486, 518, 492, 514, 492, 536, 560, 440, 468, 534,
472, 532, 416, 590, 462, 544, 492, 514, 492, 532, 472, 532, 472, 532, 472, 532, 444,
560, 416, 590, 486, 520, 490, 516, 490, 534, 470, 538, 446, 554, 470, 534, 416, 590,
462, 544, 490, 516, 492, 534, 470, 534, 470, 534, 470, 534, 470, 536, 416, 590, 462,
544, 490, 516, 490, 534, 470, 534, 470, 534, 470, 534, 418, 588, 460, 546, 488, 518,
490, 536, 470, 532, 472, 534, 470, 534, 468, 536, 416, 590, 462, 548, 486, 520, 492,
478, 518, 536, 470, 534, 470, 534, 416, 588, 460, 546, 490, 518, 490, 534, 470, 534,
470, 534, 470, 534, 444, 560, 416, 600, 454, 544, 490, 518, 488, 538, 470, 530, 468,
536, 470, 536, 416, 590, 434, 578, 482, 520, 490, 534, 470, 536, 474, 528, 470, 538,
458, 542, 416, 592, 460, 546, 490, 518, 488, 538, 470, 536, 444, 556, 446, 560, 442,
562, 482, 530, 476, 530, 478, 544, 462, 540, 468, 536, 468, 536, 466, 538, 416, 590,
436, 570, 488, 518, 488, 536, 468};

crankyoldgit, 2020-01-16:
This looks like a 128bit space encoded message with no headers.

e.g.
```
kBitMark   = 467
kOneSpace  = 1027
kZeroSpace = 540

  Bits: 128
  Hex:  0xA55A0000800000000000000000000000 (MSB first)
        0x00000000000000000000000100005AA5 (LSB first)
  Dec:  219789926061393334772828078412658638848 (MSB first)
        4294990501 (LSB first)
  Bin:  0b10100101010110100000000000000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 (MSB first)
        0b00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100000000000000000101101010100101 (LSB first)

Total Nr. of suspected bits: 128
```

No headers is quite unusual. What devices produced/receive this message?

wounder123, 2020-01-17:
IR transmitter is in the unknown dongle and IR receiver is in RGB LED band.
I try to decode the tranmsmitter side.

wounder123, 2020-01-17:
> This looks like a 128bit space encoded message with no headers.
>
> e.g.
>
> ```
> kBitMark   = 467
> kOneSpace  = 1027
> kZeroSpace = 540
>
>   Bits: 128
>   Hex:  0xA55A0000800000000000000000000000 (MSB [...]
IR transmitter is in the unknown dongle and the IR receiver is in RGB LED band.
I try to decode the transmitter side.

Could you please Which command used to send these decoded result?

I try  

irsend.sendData(0, 1027,0,
                      540, 0xA55A0000800000000000000000000000,129,
                      1);
but its not work (instead if i am sending raw data completely (irsend.sendRaw(rawData,257, 38) its work ))

code::
`
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

uint16_t rawData[257] = {472, 1016,  490, 536,  446, 1038,  464, 544,  490, 516,  492, 1008,  
418, 592,  462, 1042,  476, 532,  444, 1062,  474, 532,  470, 1014,  492, 1010,  446, 562,  460, 
1046,  474, 532,  472, 534,  416, 590,  458, 548,  486, 520,  490, 516,  490, 534,  470, 534,  
470, 534,  470, 536,  416, 590,  460, 546,  488, 518,  490, 536,  468, 536,  470, 534,  470, 
536,  442, 564,  414, 1092,  470, 536,  468, 536,  416, 590,  414, 592,  486, 520,  490, 536,  
470, 534,  470, 534,  468, 536,  416, 590,  432, 574,  486, 520,  490, 536,  470, 534,  468, 
536,  468, 536,  468, 538,  420, 590,  454, 546,  488, 518,  488, 536,  468, 536,  468, 536,  
468, 536,  440, 566,  414, 592,  462, 546,  490, 536,  468, 536,  468, 538,  468, 536,  468, 
538,  414, 592,  460, 546,  488, 518,  490, 536,  468, 536,  470, 536,  468, 536,  442, 564,  
414, 592,  462, 546,  490, 518,  488, 536,  470, 534,  470, 536,  470, 536,  416, 590,  460, 
548,  488, 518,  490, 536,  470, 534,  470, 534,  470, 536,  468, 536,  414, 592,  462, 546,  
490, 518,  488, 534,  470, 536,  468, 536,  468, 536,  414, 590,  462, 546,  488, 518,  466, 
560,  444, 560,  446, 560,  446, 560,  444, 562,  416, 592,  462, 546,  464, 542,  464, 560,  
444, 560,  446, 560,  446, 560,  416, 590,  460, 546,  464, 544,  464, 562,  444, 560,  446, 
560,  446, 560,  444, 560,  416, 592,  462, 1042,  446, 560,  444, 560,  416, 592,  462, 544,  
488, 520,  466, 558,  446, 560,  446};

void setup() {
  irsend.begin();
#if ESP8266
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
#else  // ESP8266
  Serial.begin(115200, SERIAL_8N1);
#endif  // ESP8266
}

void loop() {
   Serial.println("send_data");
  
irsend.sendData(0, 1027,0,
                      540, 0xA55A0000800000000000000000000000,129,
                      1);

  delay(2000);

}`

crankyoldgit, 2020-01-17:
You can't send an Integer of that size. i.e. a `uint128_t`. The compiler/arduino doesn't support Integers of that size. The maximum is `uint64_t`. i.e. 64 bits.

If you want to use `irsend.sendData()` (the integer version, you are going to have to split the 128 bit value into 2 x 64 bit values.

e.g.
```
const uint16_t kBitMark = 467;
const uint16_t kOneSpace = 1027;
const uint16_t kZeroSpace = 540;
irsend.begin();
irsend.enableIROut(38000);
irsend.sendData(kBitMark, kOneSpace, kBitMark, kZeroSpace, 0xA55A000080000000ULL, 64);  // First half of 0xA55A0000800000000000000000000000 (128 bits)
irsend.sendData(kBitMark, kOneSpace, kBitMark, kZeroSpace, 0x0000000000000000ULL, 64);  // Second half of 0xA55A0000800000000000000000000000 (128 bits)

// or

irsend.sendGeneric(0, 0,  // No Header
                   kBitMark, kOneSpace,
                   kBitMark, kZeroSpace,
                   0, 0,  // No Gap
                   0xA55A000080000000ULL, 64,  // First half of 0xA55A0000800000000000000000000000 (128 bits)
                   38000, true,
                   kNoRepeat, kDutyDefault);
irsend.sendGeneric(0, 0,  // No Header as it follows on from the previous sendGeneric()
                   kBitMark, kOneSpace,
                   kBitMark, kZeroSpace,
                   kBitMark, kDefaultMessageGap,  // End the message
                   0x0000000000000000ULL, 64,  // Second half of 0xA55A0000800000000000000000000000 (128 bits)
                   38000, true,
                   kNoRepeat, kDutyDefault);
```

wounder123, 2020-01-17:
const uint16_t kBitMark = 467;
const uint16_t kOneSpace = 1027;
const uint16_t kZeroSpace = 540;

Could you please tell me how its  these values from raw data?

crankyoldgit, 2020-01-17:
Those values are calculated using the average of each bit mark, space values etc.

You can use the updated tool to calculate/hand-decode your raw data: https://github.com/crankyoldgit/IRremoteESP8266/blob/Issue1014/tools/auto_analyse_raw_data.py

wounder123, 2020-01-18:
> You can't send an Integer of that size. i.e. a `uint128_t`. The compiler/arduino doesn't support Integers of that size. The maximum is `uint64_t`. i.e. 64 bits.
>
> If you want to use [...]

Still IR receiver not responding.. What modifications in code that I want to made?

crankyoldgit, 2020-01-18:
Can you please include the raw data for the message you are trying to send **and** all the code you are running?

wounder123, 2020-01-18:
`Raw data i am tried,

Protocol : UNKNOWN
Code : 0xB4C0FCA9 (129 Bits)
uint16_t rawData[257] = {462, 1044, 474, 532, 472, 1016, 492, 532, 472, 532, 448, 1038,
492, 514, 490, 1010, 462, 544, 490, 1014, 450, 558, 416, 1090, 474, 1012, 490, 516, 492,
1008, 418, 590, 462, 544, 492, 514, 584, 440, 472, 532, 536, 474, 462, 538, 446, 560,
416, 590, 462, 548, 492, 516, 464, 560, 434, 570, 444, 556, 466, 538, 416, 588, 462,
544, 490, 1014, 472, 534, 462, 548, 486, 518, 492, 514, 492, 536, 560, 440, 468, 534,
472, 532, 416, 590, 462, 544, 492, 514, 492, 532, 472, 532, 472, 532, 472, 532, 444,
560, 416, 590, 486, 520, 490, 516, 490, 534, 470, 538, 446, 554, 470, 534, 416, 590,
462, 544, 490, 516, 492, 534, 470, 534, 470, 534, 470, 534, 470, 536, 416, 590, 462,
544, 490, 516, 490, 534, 470, 534, 470, 534, 470, 534, 418, 588, 460, 546, 488, 518,
490, 536, 470, 532, 472, 534, 470, 534, 468, 536, 416, 590, 462, 548, 486, 520, 492,
478, 518, 536, 470, 534, 470, 534, 416, 588, 460, 546, 490, 518, 490, 534, 470, 534,
470, 534, 470, 534, 444, 560, 416, 600, 454, 544, 490, 518, 488, 538, 470, 530, 468,
536, 470, 536, 416, 590, 434, 578, 482, 520, 490, 534, 470, 536, 474, 528, 470, 538,
458, 542, 416, 592, 460, 546, 490, 518, 488, 538, 470, 536, 444, 556, 446, 560, 442,
562, 482, 530, 476, 530, 478, 544, 462, 540, 468, 536, 468, 536, 466, 538, 416, 590,
436, 570, 488, 518, 488, 536, 468};     `

code i am tried,

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
const uint16_t kBitMark = 467;
const uint16_t kOneSpace = 1027;
const uint16_t kZeroSpace = 540;
const uint16_t kIrLed = 4; // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed); // Set the GPIO to be used to sending the message.

uint16_t rawData[257] = {462, 1044, 474, 532, 472, 1016, 492, 532, 472, 532, 448, 1038,
492, 514, 490, 1010, 462, 544, 490, 1014, 450, 558, 416, 1090, 474, 1012, 490, 516, 492,
1008, 418, 590, 462, 544, 492, 514, 584, 440, 472, 532, 536, 474, 462, 538, 446, 560,
416, 590, 462, 548, 492, 516, 464, 560, 434, 570, 444, 556, 466, 538, 416, 588, 462,
544, 490, 1014, 472, 534, 462, 548, 486, 518, 492, 514, 492, 536, 560, 440, 468, 534,
472, 532, 416, 590, 462, 544, 492, 514, 492, 532, 472, 532, 472, 532, 472, 532, 444,
560, 416, 590, 486, 520, 490, 516, 490, 534, 470, 538, 446, 554, 470, 534, 416, 590,
462, 544, 490, 516, 492, 534, 470, 534, 470, 534, 470, 534, 470, 536, 416, 590, 462,
544, 490, 516, 490, 534, 470, 534, 470, 534, 470, 534, 418, 588, 460, 546, 488, 518,
490, 536, 470, 532, 472, 534, 470, 534, 468, 536, 416, 590, 462, 548, 486, 520, 492,
478, 518, 536, 470, 534, 470, 534, 416, 588, 460, 546, 490, 518, 490, 534, 470, 534,
470, 534, 470, 534, 444, 560, 416, 600, 454, 544, 490, 518, 488, 538, 470, 530, 468,
536, 470, 536, 416, 590, 434, 578, 482, 520, 490, 534, 470, 536, 474, 528, 470, 538,
458, 542, 416, 592, 460, 546, 490, 518, 488, 538, 470, 536, 444, 556, 446, 560, 442,
562, 482, 530, 476, 530, 478, 544, 462, 540, 468, 536, 468, 536, 466, 538, 416, 590,
436, 570, 488, 518, 488, 536, 468};   
void setup() {

#if ESP8266
Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
#else // ESP8266
Serial.begin(115200, SERIAL_8N1);
#endif // ESP8266
}

void loop() {
Serial.println("send_data");

irsend.begin();
irsend.enableIROut(38000);
irsend.sendData(kBitMark, kOneSpace, kBitMark, kZeroSpace, 0xA55A000080000000ULL, 64);  
irsend.sendData(kBitMark, kOneSpace, kBitMark, kZeroSpace, 0x0000000000000000ULL, 64); 
delay(2000);

}`

crankyoldgit, 2020-01-19:
Try modifying it to:
```
void loop() {
  Serial.println("send_data");

  irsend.begin();
  irsend.enableIROut(38000);
  irsend.sendData(kBitMark, kOneSpace, kBitMark, kZeroSpace, 0xA55A000080000000ULL, 64);  
  irsend.sendData(kBitMark, kOneSpace, kBitMark, kZeroSpace, 0x0000000000000000ULL, 64); 
  irsend.mark(kBitMark);  // <- This is probably the missing part.
  delay(2000);
}
```

Or use the `sendGeneric()` code I suggested earlier. It will do that trailing Bit Mark.

wounder123, 2020-01-19:
> Those values are calculated using the average of each bit mark, space values etc.
> 
> You can use the updated tool to calculate/hand-decode your raw data: https://github.com/crankyoldgit/IRremoteESP8266/blob/Issue1014/tools/auto_analyse_raw_data.py

  arg_group.add_argument(
      "rawdata",
     uint16_t rawData[257] = {462, 1044, 474, 532, 472, 1016, 492, 532, 472, 532, 448, 1038,
492, 514, 490, 1010, 462, 544, 490, 1014, 450, 558, 416, 1090, 474, 1012, 490, 516, 492,
1008, 418, 590, 462, 544, 492, 514, 584, 440, 472, 532, 536, 474, 462, 538, 446, 560,
416, 590, 462, 548, 492, 516, 464, 560, 434, 570, 444, 556, 466, 538, 416, 588, 462,
544, 490, 1014, 472, 534, 462, 548, 486, 518, 492, 514, 492, 536, 560, 440, 468, 534,
472, 532, 416, 590, 462, 544, 492, 514, 492, 532, 472, 532, 472, 532, 472, 532, 444,
560, 416, 590, 486, 520, 490, 516, 490, 534, 470, 538, 446, 554, 470, 534, 416, 590,
462, 544, 490, 516, 492, 534, 470, 534, 470, 534, 470, 534, 470, 536, 416, 590, 462,
544, 490, 516, 490, 534, 470, 534, 470, 534, 470, 534, 418, 588, 460, 546, 488, 518,
490, 536, 470, 532, 472, 534, 470, 534, 468, 536, 416, 590, 462, 548, 486, 520, 492,
478, 518, 536, 470, 534, 470, 534, 416, 588, 460, 546, 490, 518, 490, 534, 470, 534,
470, 534, 470, 534, 444, 560, 416, 600, 454, 544, 490, 518, 488, 538, 470, 530, 468,
536, 470, 536, 416, 590, 434, 578, 482, 520, 490, 534, 470, 536, 474, 528, 470, 538,
458, 542, 416, 592, 460, 546, 490, 518, 488, 538, 470, 536, 444, 556, 446, 560, 442,
562, 482, 530, 476, 530, 478, 544, 462, 540, 468, 536, 468, 536, 466, 538, 416, 590,
436, 570, 488, 518, 488, 536, 468},
      nargs="?")

When i add the raw data,  try to decode it but  , compiler  shows error ??

crankyoldgit, 2020-01-20:
You are calling it incorrectly.

From the built-in help function:
```
IRremoteESP8266/tools$ ./auto_analyse_raw_data.py -h
usage: auto_analyse_raw_data.py [-h] [-g] [-n NAME] [-f FILE] [-r MARGIN]
                                [--stdin]
                                [rawdata]

Read an IRremoteESP8266 rawData declaration and tries to analyse it.

positional arguments:
  rawdata               A rawData line from IRrecvDumpV2. e.g. 'uint16_t
                        rawbuf[37] = {7930, 3952, 494, 1482, 520, 1482, 494,
                        1508, 494, 520, 494, 1482, 494, 520, 494, 1482, 494,
                        1482, 494, 3978, 494, 520, 494, 520, 494, 520, 494,
                        520, 520, 520, 494, 520, 494, 520, 494, 520, 494};'
                        (default: None)
```

Note the `'` marks around the raw data. They are important.
e.g.
```
./auto_analyse_raw_data.py 'uint16_t
                        rawbuf[37] = {7930, 3952, 494, 1482, 520, 1482, 494,
                        1508, 494, 520, 494, 1482, 494, 520, 494, 1482, 494,
                        1482, 494, 3978, 494, 520, 494, 520, 494, 520, 494,
                        520, 520, 520, 494, 520, 494, 520, 494, 520, 494};'
```

Alternatively, put the raw data into a file. e.g. `/tmp/rawdata_sample`
and use `./auto_analyse_raw_data.py -f /tmp/rawdata_sample`

wounder123, 2020-01-20:
> You are calling it incorrectly.
>
> From the built-in help function:
>
> ```
> IRremoteESP8266/tools$ ./auto_analyse_raw_data.py -h
> usage: auto_analyse_raw_data.py [-h] [-g] [-n NAME] [-f FILE] [-r [...]

sorry...still confused from auto_analyse_raw_data.py  where i am put this raw data??

'uint16_t   rawbuf[37] = {7930, 3952, 494, 1482, 520, 1482, 494,
                         1508, 494, 520, 494, 1482, 494, 520, 494, 1482, 494,
                         1482, 494, 3978, 494, 520, 494, 520, 494, 520, 494,
                         520, 520, 520, 494, 520, 494, 520, 494, 520, 494};'

crankyoldgit, 2020-01-21:
You've got two options really.
1) You either put it on the command line.
e.g. `auto_analyse_raw_data.py 'uint16_t rawData[257] = {462, 1044, 474, 532, 472, 1016, ... , 468};'`
Note the `'` quote marks around the raw data info. **_They are needed._**

or

2) You put the raw data into a text file (e.g. _my_raw_data_file.txt_) using your favourite text-file editor (e.g. notepad.exe, or vim/vi/nano etc), and then use the command line: `auto_analyse_raw_data.py -f my_raw_data_file.txt`

wounder123, 2020-01-22:
thanks .. I will check  it and update.

crankyoldgit, 2020-01-22:
As you have all the info now, I'm going to mark this closed. Update if you still have a problem and we can reopen it.

closed 2020-01-22 by crankyoldgit

crankyoldgit, 2020-01-30:
The code changes referenced above have been included in the v2.7.3 release of the library.