crankyoldgit/IRremoteESP8266 #1018: sendSony once again

tomash, 2020-01-19:
Library version 2.7, ESP32-Dev module.

I'm building my own web-to-ir to control both TV and Soundbar via phone in local network. The IRMQTTSample seems to compile, upload but not run, and I got pretty far with my setup anyway.

Baseline: the TV works with it all, it's a pretty old Toshiba 42" LCD using NEC protocol. So I've got it confirmed that my wiring (with transistor for amplifying the IR LED etc.) is working.

The problem: Sony soundbar (HT-CT380, to be exact) does not react to any sendSony() attempts I throw at it. I've recorded-and-decoded all the signals from original remote buttons:

```
  Sony soundbar
   power: 540C
   vol+:  240C = 9228
   vol-:  640C
   input: 4B0D
   menu:  770D
   back:  BE110
   enter: 180C
   up:    F0D
   down:  4F0D
   sw+:   1D0C
   sw-:   5D0C
   mute:  140C
   clear audio+: F4116
   sound field: 3B0D
   pairing: EE114
```

In my Arduino sketch code, whenever I call:
```
irsend.sendSony(code, 15);
```
with or without third `repeat` argument, the soundbar does not react. Yes, I'm passing the codes as integers, so that volume up, `0x240C`, is passed to `sendSony()` as `9228`.

The fun part, and a bit of silver lining which might help in debugging what happens, is that the raw data works properly (!).

```
      // Example of data captured by IRrecvDumpV2.ino
      // Protocol  : SONY
      // Code      : 0x240C (15 Bits)
      // 0x240C = 9228,  (Volume Up)
      uint16_t rawData[127] = {2448, 550,  648, 544,  1250, 546,  648, 548,  648, 550,  1272, 524,  648, 550,  644, 550,  674, 524,  648, 550,  648, 544,  674, 524,  1270, 524,  1246, 550,  674, 524,  648, 22404,  2474, 524,  674, 520,  1250, 548,  648, 544,  674, 524,  1270, 524,  648, 550,  648, 546,  674, 524,  648, 546,  652, 546,  674, 524,  1270, 524,  1272, 522,  674, 520,  648, 22404,  2452, 544,  674, 524,  1270, 524,  674, 518,  674, 522,  1246, 550,  674, 524,  648, 544,  674, 524,  648, 546,  674, 524,  674, 518,  1276, 518,  1276, 524,  648, 546,  674, 22380,  2474, 520,  674, 524,  1250, 544,  674, 524,  674, 518,  1276, 520,  674, 522,  674, 524,  674, 520,  674, 524,  674, 524,  674, 518,  1276, 518,  1276, 524,  672, 524,  648};  // SONY 240C
      irsend.sendRaw(rawData, 127, 38); // works like a charm, volume goes up
```

Which means I'm probably one-two parameters of sendSony() away from getting it to work. Is that the delays (22404?)? Is that something else?

And even if not, I can record raw dumps of all the original remote buttons and just use them.

Thanks for this library, it makes working with IR signals a breeze.

crankyoldgit, 2020-01-20:
Thanks for the excellent issue report. Full of details! Just what I like.

I'll try simulating a few things in a few minutes, but I'd like you to do a few things for me in parallel. 
1) Can you give me model numbers of your Toshiba TV and remote, so I can add them to the supported list?
2) `sendSony` uses the traditional 40kHz modulation. Can you please try sending your `rawData` above, but using `irsend.sendRaw(rawData, 127, 40);  // Use 40kHz instead of 38kHz`. That should tell us if the that might be the problem. i.e. You sending it at 38kHz it seems to work. Wondering if newer Sony stuff expects 38kHz now.
3) Note that your 5-hex-digit codes won't fit into the Sony 15-bit protocol (e.g. "back" etc), so you'll need a different bit size for those (see the IRrecvDumpV2 output for that bit size).

In theory, it should work as you've used it. I'll do so tests while I wait for your results to see if I can see what could be causing the issue.

Looking forward to your response/feedback.

crankyoldgit, 2020-01-20:
On first examination of your `rawData` you provided. I see there are a total of **4** messages in it. i.e. Original plus **3** repeats.
`sendSony()` defaults to 2 repeats (a total of 3 messages) sent. So, try calling it with:
```
irsend.sendSony(code, 15), 3;
```

crankyoldgit, 2020-01-20:
More detailed analysis indicates that your captured data is just a _little_ off spec.
It should be well within expected tolerance ranges of devices, so it's **unlikely** to be the culprit.
My guess is you only need the extra repeat (i.e. `sendSony(data, 15, 3)`) to get the physical device to see it.

**IF and ONLY IF** that doesn't work, then I would suggest you tweak the [library values](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Sony.cpp#L22-L26) as following:
```
const uint16_t kSonySpace = 550;  // or 540 (was 600)
const uint16_t kSonyOneMark = 1250;  // or 1270 (was 1200)
const uint16_t kSonyZeroMark = 650;  // or 675 (was 600)
```

Other than that, I can't see any problems.
I'll upload the tests I carried out in a sec.

crankyoldgit, 2020-01-20:
> The IRMQTTSample seems to compile, upload but not run

I just noticed this line.

What exactly do you mean by "not run"?

tomash, 2020-01-20:
Thanks for the quick and substantial reply. I'll try the 38KHz modulation and repeat=3. I've tried up to repeat=5 but with the default modulation. By the way, the dumps were made with Vishay's TSOP4838, but according to specs it works for signals from 38 to 40 KHz base.

Here are raw dumps of all the commands from Sony soundbar remote:
https://github.com/tomash/esp32-web-ir-remote/blob/master/irrecvdump.txt

Most are 15bit but some are 20 bit, regarding your reply re: code sizes.

The TV is Toshiba 42TL838.

>What exactly do you mean by "not run"? IRMQTTServer doesn't produce any useful serial output by default.

No serial output and not seeing the wifi network to connect to for initial configuration.

crankyoldgit, 2020-01-20:
> No serial output and not seeing the wifi network to connect to for initial configuration.

Sounds like you need to do a full reset/clear of the ESP device. A Google search should tell you how to do it with tools like esptool.py etc

crankyoldgit, 2020-01-21:
> I'll try the 38KHz modulation and repeat=3.

I think you mean 40kHz, when sending raw.
If you want to change Sony's modulation, you need to change `40` to `38` [here](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Sony.cpp#L52).

> I've tried up to repeat=5 but with the default modulation. By the way, the dumps were made with Vishay's TSOP4838, but according to specs it works for signals from 38 to 40 KHz base.

Yep. You signal was captured fine. No issues with you captures. They look fine to me.

> Here are raw dumps of all the commands from Sony soundbar remote:
> https://github.com/tomash/esp32-web-ir-remote/blob/master/irrecvdump.txt

Ta.

If changing the timings don't work, or the modulation, the only thing left to try really changing the duty cycle (`33` on that same code line) to the default `50`.

Alternatively, can you capture with another ESP device the signal your ESP32 is producing when sending via `sendSony`?

crankyoldgit, 2020-01-21:
FYI, to full erase the flash, here are some instructions:
* https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region
* https://github.com/arendst/Tasmota/wiki/Esptool#erase-firmware

It's likely IRMQTTServer is picking up previous wifi config info. Erasing the chip fully should get you into a clean new state. Also, you could enable serial debugging in https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.h#L231-L237 to get more info on what might be going wrong.

crankyoldgit, 2020-01-22:
@tomash Friendly ping/follow up to see how your tweaking/testing went.

tomash, 2020-01-23:
Thanks for the ping. I was out of town for the last three days, I'll resume work on this today or friday.

crankyoldgit, 2020-01-26:
Any luck?

tomash, 2020-01-26:
I finally have some time for hacking!
Your suspicions were correct: `irsend.sendRaw(x240crawData, 127, 40);`, i.e. changing carrier frequency to 40KHz, resulted in the soundbar not picking it up. Trying to make `sendSony` use 38KHz and 4-times message multiplier, will get back to you in a minute...

tomash, 2020-01-26:
All right, great success: after editing ir_Sony.cpp in the library dir:
```
void IRsend::sendSony(uint64_t data, uint16_t nbits, uint16_t repeat) {
  sendGeneric(kSonyHdrMark, kSonySpace, kSonyOneMark, kSonySpace, kSonyZeroMark,
              kSonySpace,
              0,  // No Footer mark.
              kSonyMinGap, kSonyRptLength, data, nbits, 38, true, repeat, 33);
}
```

in the library, `irsend.sendSony(code, bits, 3);` codes finally get picked up by the soundbar. IT WORKS! 👏

tomash, 2020-01-26:
Now, how do I implement the "sendSony but 38 Khz" function in a cleaner way, other than editing directly in the library's folder? I can't add a method to IRsend that wasn't declared in the header file, the used symbols aren't visible either. A pull request adding yet another parameter, base frequency, after `repeat`?

tomash, 2020-01-26:
As for IRMQTTServer -- I'll try your suggestions and if something's broken for me I'll open a separate issue to avoid mixups. Thanks 👌

crankyoldgit, 2020-01-26:
> Now, how do I implement the "sendSony but 38 Khz" function

I'll code something up shortly.

crankyoldgit, 2020-01-26:
@tomash Can you please download/test PR #1029 / branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Sony38kHz and let me know if `sendSony38()` solves your problem?

Thanks in advance.

Btw. Thanks for doing the testing and confirming my theory, because otherwise, I was stumped for reasons why it was failing. ;-)

crankyoldgit, 2020-01-29:
@tomash I haven't heard back from you testing the new branch/PR etc. I'm assuming it will work fine, merged it, and closing this issue. Please re-open if it doesn't work, or open a new issue and reference this one.

;-)

closed 2020-01-29 by crankyoldgit

crankyoldgit, 2020-01-30:
The code changes referenced above have been included in the v2.7.3 release of the library.

tomash, 2020-01-30:
Thanks! I do it in my spare time which I don't have much of this week so I'll test it around saturday and get back to you here :)

AptemCat, 2022-09-05:
This helped me fix my Sony HT-SF150 soundbar. It worked with sendSony38(value,kSony15Bits,3);
Library version 2.8.2

PR #1029 (2020-01-29): Add support for sending Sony at 38Khz
- Add new protocol: `SONY_38K` & `sendSony38()`
- `SONY_38K` uses 3 repeats (4 messages) it seems, so default to that.
- There is no way to reliably tell if an incoming message is using 40kHz 
or 38kHz, so assume the SONY standard (40kHz).
- Unit tests for the above.
- Add supported model info.

Fixes #1018
commit fa97474d (2020-01-29): Add support for sending Sony at 38Khz (#1029)

- Add new protocol: `SONY_38K` & `sendSony38()`
- SONY_38K uses 3 repeats it seems, so default to that.
- There is no way to reliably tell if an incoming message is using 40kHz 
or 38kHz, so assume the SONY standard (40kHz).
- Unit tests for the above.
- Add supported model info.

Fixes #1018