crankyoldgit/IRremoteESP8266 #1023: When to call irrecv.resume()?

egueli, 2020-01-22:
The comment at https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/DumbIRRepeater/DumbIRRepeater.ino#L120 explains why the `irrecv.resume()` line has been put so far away from `irrecv.decode()`. But what does it mean "capture our own message"? Where is this loop realized? In the GPIO pins? In the interrupt handling? In the library itself?

My actual question is: where is the ideal moment to call `irrecv.resume()`? Too late, and we will lose IR messages. Is there a "too soon, and..."? What's the drawback on calling it right after `irrecv.decode()`?

Cheers, Enrico

crankyoldgit, 2020-01-22:
> But what does it mean "capture our own message"?

As the program re-sends what it just received, capture isn't resumed till it is finished sending, so it can't accidentally begin capturing the message the program has just sent.

The capture code happens independently  in the background via interrupts from the receiving GPIO. Hence, in order to stop it recording what it is sending (and causing a loop of sorts) it stops recording after detecting a message **AND** while sending. It only resumes recording after all that when it is safe.

> Where is this loop realized? In the GPIO pins? In the interrupt handling? In the library itself?

I'm not sure what you mean by this. I hope I've explained it enough above.

> My actual question is: where is the ideal moment to call irrecv.resume()? Too late, and we will lose IR messages. Is there a "too soon, and..."? What's the drawback on calling it right after irrecv.decode()?

[IRrecv](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRrecv.cpp#L131-L145) essentially has two modes of operation.

1) The "legacy" low-memory mode used by the original [IRremote](https://github.com/shirriff/Arduino-IRremote/) library. It uses a single buffer to capture the message.  This mode is still useful in some cases, but is primarily for backward compatibility.
2) The "newer" more-memory mode. If you pass it a second buffer, it will use a [technique](https://en.wikipedia.org/wiki/Multiple_buffering) to begin capturing a second message as soon as the other one is finished. Precisely so you can do some processing and hopefully not miss a message.
e.g. See https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino which uses this feature.

Both have advantages and disadvantages. Use which ever one suits you.
Now, lets answer your questions:

> where is the ideal moment to call irrecv.resume()?

Method 1) As soon as you've finished processing the message.
Method 2) It happens for you automatically as soon as it's decoded a message.

> Is there a "too soon, and..."?

Yes. Method 1) only has a single buffer/memory structure. If you `resume()` before you're finished with the first message, resuming can corrupt what was previously captured with the new incoming message.

Also, per the "Repeater" examples, you might want to pause capture if you are sending an IR message which you might not want to capture. i.e. A feedback loop.

Method 2), No. unless you are doing a "Repeater". ;-)

> What's the drawback on calling it right after irrecv.decode()?

Method 1) Message/data corruption. With only a single place to store the data/results, it will start overwriting the results immediately. Due to the "original" design, you have a pointer into the same shared data structure, not a copy of the results that is immutable. Hence the data can change under your feet at any instant.

Method 2) None. In fact it calls `resume()` before the `decode()` call completes. Hence it is unneeded unless you pause/halt capture manually.

I hope his answers your questions.

egueli, 2020-01-24:
Thank you for the insight! Some replies below.

> > Where is this loop realized? In the GPIO pins? In the interrupt handling? In the library itself?
> 
> I'm not sure what you mean by this. I hope I've explained it enough above.
> 
My question was rather why is there such a loop, to justify delaying the resuming after sending. But I think the answer is "because the IR light emitted by the sender would be captured by the receiver".

> Precisely so you can do some processing and hopefully not miss a message.

I suppose you would still miss messages if you're doing processing that lasts a lot more than the typical message duration.

closed 2020-01-24 by egueli

crankyoldgit, 2020-01-24:
> > Precisely so you can do some processing and hopefully not miss a message.
> 
> I suppose you would still miss messages if you're doing processing that lasts a lot more than the typical message duration.

Correct. There is basically only one way you can lose IR data when using Method 2).
i.e. Taking to long before you call `decode()` in your main loop after receiving a new message.
e.g.

- 1st Message comes in.
- Receiver times out listening for more data after message. This will let `decode()` know there is a message.
- 2nd Message starts coming in.
- User's program finally calls `decode()` which copies the 1st message, and re-arms the receiver to listen for new messages.
- 2nd Message will only now begin being captured but is only a partial message at this stage or worse, completely missed.

IR messages typically have a lengthy (50-100+ micro second) minimum gap between them. That equates to a LOT of time for CPU instructions. i.e. in the order of 100s of thousands, to millions.
As long as your main loop/processing allows you to call `decode()` (or `resume()`) around every 1/10th of a second (100ms) or so you should be fine. The message and gap typically take approx 200ms. So as long as your pipeline can process a message with in 200ms, you should be fine. Again, that's a lot of time CPU-wise.

egueli, 2020-01-24:
I'm currently not able to call `decode()` every 100ms, as the IR messages will cause HTTP requests to a webserver which are blocking unfortunately. There's not much CPU processing but it has to stay idle, sometimes even for 2-3 seconds, to wait for an answer. During that time the IR messages are not received indeed. But I found a library that can handle asynchronous HTTP connection, I'll try that out.

crankyoldgit, 2020-01-24:
Assuming you are talking about "AsyncWebserver", you will need to change this setting in the library:
[ALLOW_DELAY_CALLS](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L584-L590)

egueli, 2020-01-24:
That's the one! Thanks!