crankyoldgit/IRremoteESP8266 #1024: Panasonic DKW partially (not) working

pilaGit, 2020-01-22:
My docs say I have Panasonic DKE(W)s and my remote controls are A75C2616-1. I use Tasmota 8.1.0.3 20200106 w. IRremoteESP8266 library v2.7.2 runing on D1 mini with Wemos IRshields.

I can send and receive IRhvac commands, but some DKE functions are ignored: Ionisator and Timers. When IRhvac is sending to the Panasonic, these settings are destroyed. When IRhvac is reading code sent by my remote, they are not shown.

Source code seems to indicate it should hadnle Timers, but Ionisator is not mentioned there. 

I have two questions:

1) Can I send raw command back, the same one as I receive from my remote control, without it being interpreted (and broken in the process)? 

2) Ideally, this missing functionality would be included in the code and handled correctly. Today I took a look at the source code, but am not able to read it fully as I did not use C++. But, I worked a lot in C, so I can adapt. If someone in the know would land me a hand, this should be easily acomplished. 

I am currently writing my own SmartHome automation bash script, without using any of the home automation software, using just Tasmotas and few RPis. Everything is working well. HVAC was the last area to be addded to my SmartHome. I plan to integrate it into my existing central heating automation.

Yesterday I started reverse enging my raw strings. I can interpret and reconstruct parts important for me: temp, on/off, timers active, Ion on/off and current mode. I have just (15 minutes ago) started to look at Timers, so for now I can only read the Curent Time sent from the remote. 

As I am not sure this is a proper place for the above topic, I am keeping it brief. I can provide breakdown of my raw data and raw samples.

crankyoldgit, 2020-01-22:
Our library doesn't support setting the Ion filter for Panasonic A/Cs. It literally doesn't know how to reproduce or understand that bit in the protocol. See https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L956

If you want to do the analysis work, and find what bit in the message controls it, I can support for it.

As for timers, Yes. The native `IRPanasonicAc` class supports the timers. However, timers are not supported via the "common" a/c api, hence Tasmota (which uses that API) doesn't support it.

If an A/Cs feature is supported, then that feature will typically be cleared. Depending on which interface is used. e.g. Ion & Timers are not supported via the `IRac` common API, where as only Ion isn't supported via `IRPanasonicAc`. Both can support Ion if you reverse engineer it. However the common API does not support timers. Honestly, that's best done via home automation software, hence no support.

> Can I send raw command back, the same one as I receive from my remote control, without it being interpreted (and broken in the process)?

Yes, but it depends if you are listening for new IR commands when you send it.
You can even send a "cooked" Panasonic code via this library (i.e. not the raw data) and have it work.
I'm no expert on how Tasmota uses this library, you will need to talk to them to be sure how it operates via them.

> Ideally, this missing functionality would be included in the code and handled correctly. Today I took a look at the source code, but am not able to read it fully as I did not use C++. But, I worked [...]

C++ is a superset of C, so you shouldn't have that much of a problem.
That said, if you can reverse engineer what bit controls the Ion (Filter) setting and do some testing, I'll add the code myself.

See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#working-out-the-details & https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#binary-settings for how to easily do it.
e.g.
Capture (via `IRrecvDumpV2`) your Panasonic Remote's message when the Ion Filter is on, and when it is off. With just those two messages it is likely we can work out which bit in the message controls the filter.

pilaGit, 2020-01-22:
Excellent answer. Many thanks :) Just in case there are differences in raw output, I stress I am using Tasmota.

Ion is a mixed bag.
 
1. Ion can be activated during regular modes (cool, heat...). Then I read Ion easily as (zero based) byte 1 at offset 49
# AC on, On+Off timers, Temp: 25°C heat, Ion: On
0X0X0220E004000000060220E004004F3280AF0D000660000101000631

and when the (zero based) byte 49 turns to 0, Ion is off
# AC on, On+Off timers, Temp: 25°C heat, Ion: Off
0X0X0220E004000000060220E004004F3280AF0D000660000001000630

2) But, I noticed that if AC is Off, if I press Ion button, Ac turns On serving only as an Ionisator. I will investigate tomorrow as I need it at Heat mode now :) since it is -3C outside :)

It will take some effort on my part regarding the timer. In essence, Timer is really not needed if SmartHome system runs things. But, I prefer not to have loose ends. I just added basic HVAC commands to control heating, and since I am unable to read the Timer, my program claims AC is off. If I could read them, I could easily calculate if Timer activated AC - as it did. But, not very important. I will have to study your great answer more.

crankyoldgit, 2020-01-23:
fyi, if you could supply the `uint8_t state[]` line for those messages that would be great.

crankyoldgit, 2020-01-23:
@pilaGit I've created a branch with code that detects/supports the Ion Filter for DKE Panasonic models.
Can you please download test this branch out? (https://github.com/crankyoldgit/IRremoteESP8266/tree/panasonic_ion)

Ref: #1025 for the code changes and tests etc.

pilaGit, 2020-01-23:
I am using Tasmota so I have to find a way to combine them. If you could just direct me which file(s) should I import into my custom Tasmota, I will do it immediatelly.

Regarding Timers, you are right, there is po point waisting time on them. They are mostly if not completely irrelevant in any .

pilaGit, 2020-01-23:
Just to make things clear: Ion is not a filter or something like it. That is a Ioniser device which can be activated to ionise air or deactivated.

crankyoldgit, 2020-01-23:
> I am using Tasmota so I have to find a way to combine them. If you could just direct me which file(s) should I import into my custom Tasmota, I will do it immediatelly.

You will have to talk to the Tasmota folk, or download the branch via the "download zip" link thing on that branch URL I sent you.

As for which files changed/you need, this URL will show you what is included in my proposed change:
https://github.com/crankyoldgit/IRremoteESP8266/pull/1025/files

> Just to make things clear: Ion is not a filter or something like it. That is a Ioniser device which can be activated to ionise air or deactivated.

Ionisers are used by A/Cs to remove some particulate matter from the air. Hence it _is_ a filter of sorts. :-) It's also mostly a placebo health-wise from what I can tell. https://en.wikipedia.org/wiki/Air_ioniser#Ionic_air_purifiers

pilaGit, 2020-01-26:
We are good! As I do not know what is considered standard in this library, I will comment on the logic side and bare minimum request from the user side. I work from Tasmota, D1 minis with Wemos IR shield blaster.

1. Ion is not listed in IRhvac response when I use the Pana remote. It is simply ignored.

2. When I use Tasmota to send IRhvac commands to Panasonic DKE, I use:
# to turn it on:
IRhvac {"Vendor":"PANASONIC_AC","Model":3,"Power":"on","Mode":"heat","FanSpeed":"auto","Temp":25}

# and to turn it off:
IRhvac {"Vendor":"PANASONIC_AC","Model":3,"Power":"off","Mode":"off","FanSpeed":"auto","Temp":25}

From the logic point of view: IRhvac commands now leave Ion activated if it was allready activated with a Pana remote. Does not quash it any more!

From the user point of view: (I think) I can not manipulate Ion and Timers using IRhvac command. But, if Ion and Timers are allready set using the Panasonic remote, IRhvac commands will not quash them but will let them be forever as they are set! So, for me this is perfectly acceptable solution. We really do not need to modify them. Just let them be.

Many thanks for your time and knowledge! I do not know if this will be considered acceptable to go into a release. I hope it will. 

I think this solution fucntions very good and consider the problem solved. I think issue can be closed if this fix will be included in the master branch.

A bit of additional info. I agree that the Timer is not needed when a computer runs the device. But, IRsend command can fail for many reasons: door is open so the beam bounces funny, sun is shining, funny lightbulb.... I use D1mini IRshield with terribly short range (less than 2 meters) and no resistance to any stray IR sources. I prefer to have inbuilt Timer set to Off at certain time so I am sure it will go off should IRhvac and IRblaster fail to deliver the Off command.

As for Ion, when I said it is not a filter, I meant it is not piece of mesh, plastic or paper... we can take out. As for its usefulness, we need not discuss it (helps with the dust somewhat). I wanted to try a simple modification since I will have to make Panasonic MKE system work on another location. And they have several usefull features which will present a real problem should they be quashed by IRhvac.

crankyoldgit, 2020-01-26:
I think you are now venturing into Tasmota implementation issues, and not strictly IRremoteESP8266 problems.
i.e. this `IRhvac` thing is a Tasmota-ism. It may well be a thin layer on top of IRremoteESP8266, but you are probably going to have to take it up with them for those settings being visible or not.

Re: D1 mini IRshield.
Sorry, but I can't account for poor hardware implementations. It's not a fault of the library.
However, do note that no matter how good the IR hardware, you still have to deal with _loss in transmission_ issues. The protocols/devices in questions don't use any form of acknowledgement, so this is a inherent issues that can't be avoided.

Re: Timers.
As I said, the Panasonic AC class supports timer control, you can write custom software to use those features.

RE: Failed to receive Off commands.
You can always re-transmit the message at different times to increase the likelihood that the device is turn off. e.g. You don't need to send it only once, or at the same time. Why not send "Off" every X minutes while the device is meant to be off etc.? :-)

The patch will be applied to the `master` branch after it has been code reviewed or when it 7 days old. New releases tend to come out every month or two. (No promises)

closed 2020-01-29 by crankyoldgit

crankyoldgit, 2020-01-30:
The code changes referenced above have been included in the v2.7.3 release of the library.

PR #1025 (2020-01-29): Panasonic AC: Add Ion Filter support for DKE models.
* Add set/getIon()
* Handle model specifics as it is confirmed only on DKE.
* Add and update unit tests accordingly.
* Update Common A/C API for panasonic.
* Update supported model info

Fixes #1024
commit 5c931e04 (2020-01-29): Panasonic AC: Add Ion Filter support for DKE models. (#1025)

* Panasonic AC: Add Ion Filter support for DKE models.
* Add set/getIon()
* Handle model specifics as it is confirmed only on DKE.
* Add and update unit tests accordingly.
* Update Common A/C API for panasonic.
* Update supported model info

Fixes #1024