crankyoldgit/IRremoteESP8266 #1026: fatal error: gtest/gtest.h: No such file or directory

egueli, 2020-01-24:
### Version/revision of the library used
v2.7.2

### Expected behavior
1. `cd test`
2. `make install-googletest`
3. `cd ..`
4. `pio test`
5. Test compiling and passing

### Actual behavior
5. Tests failed to build with errors. See log:

```
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

Processing * in nodemcuv2 environment
--------------------------------------------------------------------------------
Building...
test/IRrecv_test.cpp:8:25: fatal error: gtest/gtest.h: No such file or directory
 #include "gtest/gtest.h"
                         ^
compilation terminated.
*** [.pio/build/nodemcuv2/test/IRrecv_test.cpp.o] Error 1
```
(same error repeats for other source files)

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes, but they don't talk about running unit tests.

### Has this library/code previously worked as expected for you?
No: it's the first time I'm running the tests.

### Other useful information
The build environment seems otherwise sane, i.e. I can `pio run` without issues.

crankyoldgit, 2020-01-24:
The unit tests & test suite (gtest) I use isn't compatible with PlatformIOs `test` function.

Docs on how to run it here:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Library-Maintainers-Guide#unit-tests

If you can figure out how to get it to work with PlatformIO I would be very grateful and happy. :)

egueli, 2020-01-24:
Thank you. I've followed the documentation and ran `make run_tests`, but I get other build errors:

```
c++ -isystem ../lib/googletest/googletest/include -DUNIT_TEST -D_IR_LOCALE_=en-AU -g -Wall -Wextra -pthread -std=gnu++11 -I../src -I. -c IRrecv_test.cpp
IRrecv_test.cpp:569:37: error: variable-sized object may not be initialized
  uint16_t good_trailing_space_data[good_entries_trailing_space] = {
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```
(same error repeated on other lines in the same file)

Maybe some difference in GCC versions?

crankyoldgit, 2020-01-25:
Could be.

Every merge/PR that gets done to the library goes through a Travis check cycle which compiles and runs all the code & unit tests.
e.g. https://travis-ci.org/crankyoldgit/IRremoteESP8266/jobs/640788673#L386-L3752

FWIW on my machine:
```
$ g++ --version
g++ (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

egueli, 2020-01-25:
Indeed. I'm using clang locally
```
c++ --version
Apple clang version 11.0.0 (clang-1100.0.33.17)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

, and doesn't like arrays whose dimensions are specified by non-const variables. Adding `const` to those variables fixes the issue. I can now run all unit tests.

crankyoldgit, 2020-01-25:
@egueli Thanks for that feedback and solution. Do you have a `diff` I can use and I'll make that change to the library so no one else has that problem in future?

Or was it only that file? If so, I can probably find all occurances.

crankyoldgit, 2020-01-25:
@egueli I've created PR #1027 to address the issue with Apple's standard compiler. As I don't have access to an Apple machine, Could you please download/test this branch (https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1026) and let me know if that fixes the unit tests for you?

I used a regex to try to final all occurances in the code base for that sort of initialisation. I think I got them all. Fingers crossed.

egueli, 2020-01-25:
@crankyoldgit I tried your branch on my mac; tests build and run fine 💯

For the record, this is my diff you asked for: https://github.com/egueli/IRremoteESP8266/commit/612deef473b87cc23478191ea53a45a974871164 (tests will fail as the branch isn't finished yet)
It's pretty much the same changes as yours, except I didn't rename the variable.

Apparently, variable-length arrays are a GCC extension: https://gcc.gnu.org/onlinedocs/gcc/Variable-Length.html interesting that I read about them on SO just a few days ago :)

I'll rebase my branch once your PR #1027 is merged. Cheers!

crankyoldgit, 2020-01-25:
Thanks. If it works, then I got them all! Thanks for confirming.

closed 2020-01-25 by crankyoldgit

crankyoldgit, 2020-01-30:
The code changes referenced above have been included in the v2.7.3 release of the library.

PR #1027 (2020-01-25): Fix clang c++ compiler error in unit tests.
Apple's c++ compiler (clang) has issues with array declarations using a
variable.
e.g.
```
error: variable-sized object may not be initialized
```

g++ doesn't seem to have an issue with it.

Some unit tests used this, changing the variable to a constant fixes it, allegedly.

Fixes #1026
commit 27bb5827 (2020-01-25): Fix clang c++ compiler error in unit tests. (#1027)

Apple's c++ compiler (clang) has issues with array declarations using a
variable.
e.g.
```
error: variable-sized object may not be initialized
```

g++ doesn't seem to have an issue with it.

Some unit tests used this, changing the variable to a constant fixes it.

Fixes #1026