crankyoldgit/IRremoteESP8266 #1028: Failing unit test TestCoolixACClass.toCommon

egueli, 2020-01-26:
### Version/revision of the library used
https://github.com/crankyoldgit/IRremoteESP8266/commit/27bb58271d17c54454d8f007e68d2d65184319ef

### Expected behavior
1. `make clean run_tests`
2. All unit tests passing

### Actual behavior
1. `make clean run_tests`
2. Some unit tests failing:
```
[ RUN      ] TestCoolixACClass.toCommon
ir_Coolix_test.cpp:634: Failure
Value of: ac.toCommon().turbo
  Actual: true
Expected: false
[  FAILED  ] TestCoolixACClass.toCommon (0 ms)
[ RUN      ] TestCoolixACClass.Issue722
[       OK ] TestCoolixACClass.Issue722 (0 ms)
[ RUN      ] TestCoolixACClass.Issue985
[       OK ] TestCoolixACClass.Issue985 (1 ms)
[----------] 14 tests from TestCoolixACClass (2 ms total)

[----------] Global test environment tear-down
[==========] 22 tests from 3 test cases ran. (5 ms total)
[  PASSED  ] 21 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TestCoolixACClass.toCommon
```

### Has this library/code previously worked as expected for you?
Yes; all unit tests were passing before the aforementioned commit.

### Other useful information
Using clang compiler on Mac:
```
$ c++ --version
Apple clang version 11.0.0 (clang-1100.0.33.17)
Target: x86_64-apple-darwin19.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

crankyoldgit, 2020-01-26:
Hey @egueli Thanks for the report.

It certainly wasn't/isn't producing that error under `g++`. :-/

Again, as I can't test the fix myself, can you please download & try PR #1030 / branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1028 and let me know if that quashes it?

egueli, 2020-01-27:
Thank you for the quick fix @crankyoldgit , Issue1028 makes all tests pass on Clang.

crankyoldgit, 2020-01-27:
Thanks for the confirmation!

closed 2020-01-27 by crankyoldgit

crankyoldgit, 2020-01-30:
The code changes referenced above have been included in the v2.7.3 release of the library.

PR #1030 (2020-01-27): Fix Coolix unit test errors when using Apple c++ compiler.
- Possible difference in behaviour between g++ & Apple's c++ memory 
initialisation.

FYI @egueli
Fixes #1028
commit edd4abb7 (2020-01-27): Fix Coolix unit test erros when using Apple c++ compiler. (#1030)

- Possible difference in behaviour between g++ & Apple's c++ memory 
initialisation.

FYI @egueli
Fixes #1028