crankyoldgit/IRremoteESP8266 #1031: [ESP32] Cannot use IRSend along with BLE

playrobotics-alex, 2020-01-29:
### Device used:
ESP32 (nodemcu-32s) with Arduino framework

### Version/revision of the library used
v2.7.2

I can send IR codes from my ESP32 very reliably using this code:
```
void loop() {
        unsigned long t1 = micros(), t2;
        irsend.sendSony(0x9999, 16, 2);  
        delay(1000);  
        t2 = micros();  
        Serial.printf("round of ir takes %ld micros\n", (t2 - t1));          
    }
``` 

it works very good with all the following libraries imported:
```
#include <Arduino.h>  
#include <BLE2902.h>  
#include <BLEDevice.h>  
#include <BLEServer.h>  
#include <BLEUtils.h>  
#include <IRrecv.h>  
#include <IRremoteESP8266.h>  
#include <IRsend.h>  
#include <IRutils.h>
```

IRsend is defined in my code as a global variable:
`IRsend irsend(18);`  (using pin D18)

and my `setup()` method looks like this [relevant code only...]
```
void setup() {
    Serial.begin(38400);
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);
    // BLEDevice::init(NAMESTRING);

   /*
   ... 
   more commented out BLE code
   ...
   */

  irsend.begin();
}
```

this code (with the BLEDevice::init commented out) runs perfectly without any issues.
as soon as I un-comment this line, I cannot seem to send my IR codes in a reliable fashion. (meaning that on average about 5-10% of the times the code is sent correctly, but most of the times garbage values are seems to be sent.)

To check myself for receiving  the correct codes, I use multiple Arduinos and ESP32 with several types of receivers (all work on 38 KHz).

crankyoldgit, 2020-01-29:
> `irsend.sendSony(0x9999, 16, 2);`

FYI, there is no `16` bit Sony protocol that I'm aware of. It's `12`, `15`, or `20` bits per the published docs.

See:  https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L829-L831

**_Re: Bluetooth_**
You're the first person to report issues with BLE on the ESP32. So we are in uncharted waters.
I've never used BLE on the ESP32 so I don't have any references, experience, or expectations either.

So here are some things to try:
1) Try a different GPIO pin? GPIO18 is `VSPI CLK`. No idea if BLE is using VSPI. (Unlikely, but worth a shot)
2) Try setting `ALLOW_DELAY_CALLS` to `false`. See https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L589 (This might making timing more reliable)
3) Download and try the latest `master` branch and try the new `sendSony38()` (See issue with Sony reception on some new(er) Sony devices in Issue #1018) (Unlikely, but worth a shot)

Also, can you provide me with `rawData` samples of a "successful" message and an "unsuccessful" message captured using `IRrecvDumpV2` on a separate ESP32/ESP8266 please?

_**What I think is happening ...**_

I think Bluetooth is interfering with the timing routines used by IRremoteES8266's send functions.
Perhaps disable Bluetooth when you go to send, and re-enable afterwards?
Doing some quick research, it looks the Bluetooth code uses Core 0 on the ESP32, thus it's competing for CPU time/resources with the IR library. We could explore using hardware PWM for generating the IR signal. This would solve timing issues with the software PWM implementation, but it won't solve issues with how long the `mark` & `space` IR pulses are. Hence I'm going to need more capture data (or data/analysis from a Oscilloscope) to see what's going on.

crankyoldgit, 2020-01-29:
For reference, the quick research was some Googling which led me to: https://github.com/nkolban/esp32-snippets/issues/937

playrobotics-alex, 2020-01-30:
Thank you for the detailed and fast reply!
#1 and #2 didn't work but I moved the ir send functionality to CORE 1 by using xTaskCreatePinnedToCore 
and everything seems to be working now!

closed 2020-01-30 by playrobotics-alex

crankyoldgit, 2020-01-30:
Excellent. I'm glad you solved it. Care to give a snippet of example code so I can add it to an FAQ so someone else can find it?

playrobotics-alex, 2020-02-02:
@crankyoldgit 

```
TaskHandle_t send_ir_task;
IRsend irsend(18);

void send_ir_f(void *param) {
    irsend.begin();
    delay(100);
    while (1) {
        unsigned long t1 = micros(), t2;
        irsend.sendSony(0x6666, 16, 2);
        delay(100);
        t2 = micros();
        Serial.printf("round of ir takes %ld micros\n", (t2 - t1));        
    }
}

void setup() {
    ... 
    xTaskCreatePinnedToCore(
        send_ir_f,     /* Task function. */
        "Send IR code",    /* name of task. */
        2000,           /* Stack size of task */
        NULL,           /* parameter of the task */
        1,              /* priority of the task */
        &send_ir_task, /* Task handle to keep track of created task */
        1); 
    ...
}
```

crankyoldgit, 2020-02-02:
@playrobotics-alex Many thanks. It's now in the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Some_other_library_eg_Bluetooth_is_interfering_with_IRsend_What_can_I_do_ESP32).