crankyoldgit/IRremoteESP8266 #1033: Electra A/C aircon control

ShonP40, 2020-01-30:
I recently updated my WeMos D1 Mini from v0.5.6 to v2.7.3 because I saw that it has an Home Assistant integration.
After I updated, I noticed that none of my IR commands that I used before to control my Electra A/C work.
I tried the new aircon menu and the main functions in it are working great but the Quiet, Turbo, Econo, Light, Filter, Clean, Beep functions do nothing.
My A/C has an option to toggle the display, turbo mode and clean mode and I was able to use them in the past with the old version.

So, can you fix those commands please (in the aircon menu)?
And do I need to re-record the old commands I have so they would work again?

And is there a way to toggle options in the aircon menu with a URL?

crankyoldgit, 2020-01-31:
FYI, v0.5.6 was the version of IRMQTTServer, not the library version (currently v2.7.3), Back then, the library was v2.5.1.

In the [v2.6.0 release notes](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/ReleaseNotes.md#v260-20190430) we had to change the bit ordering of the Electra protocol.
e.g. _"Change per byte bit-order in Electra protocol. (#648)"_

That's probably why your old codes no longer work. If you reverse the bit order of _each_ **byte** you send, you will probably get a working message again.
e.g. 0x01 (0b00000001)-> 0x80 (0b10000000), 0x02 (0b00000010) -> 0x40 (0b01000000), etc.
0x0102 -> 0x4080. i.e. Bit's with in the each byte are reversed, _not_ the order of the bytes.

"Detailed" Electra support wasn't added until [v2.6.3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/ReleaseNotes.md#v263-20190704) of the library. (#788)

You are now attempting to use the "Common A/C" API to control the Electra device. Different models and brands have different levels of support.

This is what is currently supported by the library for Electra: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L464-L488
i.e. Quiet, Turbo, Econo, Light, Filter, Clean & Beep are not (yet) supported for Electra.

So the library/code/example is working as intended at the moment.

> My A/C has an option to toggle the display, turbo mode and clean mode and I was able to use them in > the past with the old version (via A/C complex commands).
>
> So, can you fix those commands please (in the aircon menu)?

If you can work out what parts/bits of the `uint8_t state[]` change for those given modes, I can add code to support them. e.g. Follow the instructions/guide here: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol for how to work out which bits do what etc.
 
> And do I need to re-record the old "complex" commands I have so they would work again?

Per earlier, yes (**or** no). The bit ordering changed for the Electra protocol, so you can either re-capture the messages with the current `IRrecvDumpV2` example, **or** do the bit-math to reverse them yourself.

> And is there a way to toggle options in the aircon menu with a URL?

It is just a html form. You should be able to construct & submit a URL with the correct arguments to make the changes you want.
Do some Googling and you will work out how to do that.
e.g.
- https://developer.mozilla.org/en-US/docs/Learn/Forms/Sending_and_retrieving_form_data
- https://stackoverflow.com/questions/1977390/html-forms-input-type-submit-problem-with-action-url-when-url-contains-index-a/26894613

ShonP40, 2020-01-31:
How many codes will I need to record so you’ll add a more detailed support for my A/C?

Will it be enough to record a couple (with those options toggled on or off) or will I have to record everything?

crankyoldgit, 2020-01-31:
> How many codes will I need to record so you’ll add a more detailed support for my A/C?

How long is a piece of string? ;-)

Read: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#working-out-the-details & https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#binary-settings

As you're probably looking at "binary" controls/settings, you are probably only looking at a couple of messages per setting.

> Will it be enough to record a couple (with those options toggled on or off) or will I have to record everything?

It all depends. It will only take you as many as it takes you to understand what bits are changing & why.

crankyoldgit, 2020-01-31:
Also, can you please provide the Brand & Model numbers for your A/C head unit **and** your remote?

ShonP40, 2020-01-31:
A/C and remote brand: Electra
A/C Model: ELECTRA Classic INV 17 / AXW12DCS
Remote Model: YKR-M/003E

crankyoldgit, 2020-01-31:
Thanks. Looking forward to your reverse-engineering/analysis of your capture data so I can add your missing features.

ShonP40, 2020-01-31:
Btw, How can I select Electra A/C when controlling it through Home Assistant using MQTT?

crankyoldgit, 2020-01-31:
Select `ELECTRA` on the HTML page, and the device (via MQTT retain) will remember it even beyond reboots.

via MQTT-only, read: https://github.com/crankyoldgit/IRremoteESP8266/issues/1033#issuecomment-580676862
e.g. send `ELECTRA` to the MQTT topic `ir_server/ac/cmnd/protocol` etc.
Again, it only needs to be done once.

ShonP40, 2020-01-31:
oh, ok, ty.

And last thing: How can I add my temperature sensor to it so it will show up correctly and know the room temp? (my temperature sensor is already connected and working in home assistant).

crankyoldgit, 2020-01-31:
You will have to add the code to `IRMQTTServer` yourself for that if you want to do it on the same device.

Or use another device that reports the temp sensor results to where ever you've told HA's MQTT climate to look for it. e.g. https://www.home-assistant.io/integrations/climate.mqtt/#current_temperature_topic

FYI, all of that is out of scope for this library/support channel.
e.g. https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#i-want-to-change-the-example-code-to-make-it-do-something-different-but-i-dont-know-how-please-help-me
and it's really a Home Assistant issue. ;-)

ShonP40, 2020-01-31:
My temperature sensor is connected to my RPI's GPIO pins with the onewire protocol.

crankyoldgit, 2020-01-31:
Then have it report it to MQTT to the topic you've setup in that HA doc I linked.

ShonP40, 2020-01-31:
ok, I'll try that

ShonP40, 2020-01-31:
Hmm, MQTT commands are doing nothing when I try to turn on the A/C (for example).

Here is the MQTT command `wemos_d1_mini_ac_remote_server/ac/cmnd/ELECTRA_AC/power`

crankyoldgit, 2020-01-31:
Sorry, I gave you the wrong link to the docs earlier. See: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L182-L189

e.g. Try sending `on` to `wemos_d1_mini_ac_remote/ac/cmnd/power`

ShonP40, 2020-01-31:
still nothing

crankyoldgit, 2020-01-31:
What does the `ac` page on the ESP running `IRMQTTServer` say?

crankyoldgit, 2020-01-31:
i.e. http://your_esp.local/aircon

ShonP40, 2020-01-31:
[image]

crankyoldgit, 2020-01-31:
And what is the "Climate Information" section say on the "info" tab/page?

ShonP40, 2020-01-31:
[image]

crankyoldgit, 2020-01-31:
According to that, the MQTT topic you need to send to is `homeassistant/ac/cmnd/power` and send it with a payload of `on`

ShonP40, 2020-01-31:
ok, its working.
But can I change that topic?

ShonP40, 2020-01-31:
and why doesn't it work if I add `ELECTRA_AC`?

crankyoldgit, 2020-01-31:
> But can I change that topic?

You set it in your initial setup. To reset it, you need to reset the config on the ESP device. e.g. https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L58-L59

```
 * If you need to reset the WiFi and saved settings to go back to "First Boot",
 * visit:  http://<your_esp's_ip_address>/reset
```
It will use the MQTT prefix. If it isn't set, it will use the hostname as the start of the MQTT topic.

> and why doesn't it work if I add `ELECTRA_AC`?

Because that isn't how MQTT topics work. It's only listening on certain ones. You just used a different topic that it isn't programmed to listen too.

ShonP40, 2020-01-31:
and how do I update it? Do I need to create a bin file?

crankyoldgit, 2020-01-31:
Go to the reset page (as I listed, or under the "Admin" page.) It will take the ESP device back to first boot of this software. i.e. It will give you the wifi & settings menu when the ESP access as a wifi access point. You can configure it there, and there only. You don't need to upload a new bin file.

ShonP40, 2020-01-31:
Ok, I just wanted to know how to upgrade it OTA

ShonP40, 2020-01-31:
Is this the line that sets the prefix?

`char MqttPrefix[kHostnameLength + 1] = "";`

crankyoldgit, 2020-01-31:
No. It is set in the first boot wifi menu page.

ShonP40, 2020-01-31:
oh, ok.

Btw, no one from the Home Assistant Discord server wants to help me integrate the temp sensor with the climate integration.
I had to ask the same question 4 times to get an answer (I waited the whole day) and the answer was that no one added that option to a mqtt climate integration.

crankyoldgit, 2020-01-31:
Read the docs, and try it yourself. Also try posting on their (HA) forums.

ShonP40, 2020-01-31:
They told me to publish the temp sensor to mqtt

https://www.home-assistant.io/docs/mqtt/service/

Do you have a clue on how to do it?

ShonP40, 2020-01-31:
I managed to add it to the config and send the packets but I have no clue on how to make it automatic

ShonP40, 2020-01-31:
got it to work with this automation:

```trigger:
  - platform: state
    entity_id: sensor.something
action:
  - service: mqtt.publish
    data:
      topic: homeassistant/ac/stat/temp
      payload_template: "{{ states('sensor.something') }}"

crankyoldgit, 2020-01-31:
That won't do what you want. You are using the stat topic of the temp command (cmnd). You need a different topic for the ambient temperature and you need to add it to your HA climate configuration.

This is beyond the scope for this issue/forum/library

ShonP40, 2020-01-31:
its working fine. I am not sending the command to the ESP

ShonP40, 2020-01-31:
I'll record the A/C commands tomorrow because its late rn.

crankyoldgit, 2020-02-02:
@ShonP40 Friendly ping for that data/analysis.

ShonP40, 2020-02-02:
Sorry for the delay.
I wasn't home yesterday.

ShonP40, 2020-02-02:
**A/C on - Cooling mode - 24° - Fan speed set to 1 - Light set to on - flaps off - turbo off**

```Protocol  : ELECTRA_AC
Code      : 0xC387E0006000200000200005CF (104 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off
uint16_t rawData[211] = {9190, 4448,  666, 1604,  642, 1610,  666, 510,  666, 510,  720, 514,  640, 508,  664, 1608,  640, 1622,  692, 1606,  666, 1610,  692, 1610,  666, 514,  692, 508,  666, 516,  686, 512,  640, 1648,  684, 514,  694, 532,  666, 534,  666, 534,  666, 532,  666, 1634,  614, 1636,  666, 1646,  640, 540,  666, 534,  640, 540,  666, 534,  640, 540,  666, 536,  638, 540,  666, 546,  640, 538,  666, 534,  638, 542,  664, 536,  638, 540,  664, 1632,  640, 1640,  692, 546,  664, 536,  664, 534,  666, 536,  664, 534,  666, 536,  664, 536,  664, 536,  666, 546,  664, 536,  664, 536,  664, 534,  664, 536,  666, 534,  666, 1636,  638, 536,  638, 552,  664, 536,  638, 540,  664, 536,  638, 540,  664, 536,  638, 542,  664, 538,  638, 550,  664, 536,  638, 540,  664, 536,  640, 542,  662, 562,  612, 542,  664, 538,  636, 554,  662, 560,  612, 544,  662, 562,  612, 542,  662, 562,  612, 1642,  664, 562,  664, 572,  638, 536,  664, 538,  662, 538,  662, 562,  638, 562,  638, 564,  638, 538,  662, 572,  638, 1638,  610, 566,  636, 1638,  636, 540,  662, 562,  664, 538,  662, 540,  660, 546,  662, 1636,  610, 1638,  664, 1636,  636, 1642,  662, 562,  664, 562,  638, 1638,  610, 1642,  742};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x00, 0x20, 0x00, 0x00, 0x20, 0x00, 0x05, 0xCF};
```
**A/C off - Cooling mode - 24° - Fan speed set to 1 - Light set to on - flaps off - turbo off**

```Protocol  : ELECTRA_AC
Code      : 0xC387E0006000200000000005AF (104 Bits)
Mesg Desc.: Power: Off, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off
uint16_t rawData[211] = {9170, 4466,  638, 1634,  638, 1636,  692, 508,  692, 508,  664, 536,  612, 540,  666, 1612,  638, 1648,  690, 1636,  638, 1632,  636, 1640,  718, 510,  664, 536,  666, 536,  664, 534,  670, 1642,  638, 536,  636, 542,  664, 536,  646, 532,  664, 536,  638, 1638,  690, 1610,  664, 1622,  664, 536,  690, 538,  662, 536,  664, 536,  664, 536,  664, 538,  664, 538,  662, 546,  664, 536,  662, 536,  664, 536,  664, 536,  662, 536,  662, 1638,  610, 1642,  662, 548,  690, 534,  664, 538,  660, 538,  660, 538,  664, 538,  660, 540,  662, 538,  662, 550,  662, 538,  660, 540,  662, 538,  660, 562,  640, 560,  640, 1640,  608, 544,  662, 548,  638, 542,  664, 538,  632, 544,  664, 540,  634, 544,  658, 542,  610, 592,  642, 548,  634, 544,  636, 566,  608, 572,  662, 540,  606, 570,  662, 540,  634, 568,  612, 598,  614, 568,  612, 586,  588, 570,  662, 540,  608, 594,  612, 568,  634, 568,  638, 552,  608, 572,  656, 566,  612, 568,  636, 564,  612, 546,  652, 570,  610, 570,  612, 596,  606, 1672,  610, 586,  664, 1664,  558, 594,  612, 566,  632, 570,  626, 576,  584, 582,  658, 1660,  584, 1692,  638, 1688,  584, 1688,  586, 590,  638, 1664,  586, 592,  612, 1690,  718};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x05, 0xAF};
```

**A/C on - Cooling mode - 24° - Fan speed set to 1 - Light set to off - flaps off - turbo off**

```Protocol  : ELECTRA_AC
Code      : 0xC387E0006000200000200015DF (104 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off
uint16_t rawData[211] = {9192, 4448,  638, 1634,  640, 1610,  692, 508,  718, 508,  666, 536,  690, 508,  666, 1636,  638, 1622,  690, 1608,  638, 1638,  690, 1612,  638, 542,  678, 522,  664, 514,  666, 534,  638, 1650,  664, 536,  690, 534,  664, 536,  664, 534,  672, 528,  666, 1634,  610, 1640,  664, 1648,  638, 540,  692, 508,  666, 514,  664, 536,  640, 540,  664, 538,  638, 576,  640, 534,  664, 536,  664, 536,  664, 536,  664, 536,  664, 536,  662, 1638,  610, 1642,  662, 548,  690, 534,  664, 536,  664, 536,  672, 528,  690, 512,  664, 538,  662, 538,  664, 546,  664, 538,  662, 536,  662, 536,  664, 538,  662, 536,  660, 1642,  610, 544,  662, 550,  634, 542,  664, 538,  636, 544,  660, 540,  636, 544,  662, 538,  610, 570,  650, 560,  636, 542,  636, 564,  636, 564,  638, 542,  634, 546,  636, 588,  614, 562,  586, 554,  636, 544,  608, 594,  634, 564,  588, 572,  660, 538,  632, 1668,  636, 550,  654, 576,  656, 566,  612, 584,  640, 562,  638, 562,  640, 540,  634, 590,  612, 590,  634, 574,  612, 1666,  582, 568,  634, 1688,  584, 590,  638, 1664,  608, 570,  612, 590,  610, 578,  612, 1686,  586, 1688,  664, 1642,  618, 1652,  608, 1692,  664, 540,  660, 1642,  634, 1664,  716};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x00, 0x20, 0x00, 0x00, 0x20, 0x00, 0x15, 0xDF};
```

**A/C off - Cooling mode - 24° - Fan speed set to 1 - Light set to off - flaps off - turbo off**

```Protocol  : ELECTRA_AC
Code      : 0xC387E0006000200000000005AF (104 Bits)
Mesg Desc.: Power: Off, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off
uint16_t rawData[211] = {9198, 4440,  674, 1600,  648, 1604,  700, 500,  728, 500,  700, 500,  704, 496,  700, 1602,  646, 1614,  698, 1600,  674, 1602,  702, 1600,  698, 480,  702, 500,  674, 506,  698, 502,  690, 1600,  724, 500,  724, 476,  726, 476,  722, 476,  724, 474,  700, 1602,  696, 1576,  698, 1590,  750, 474,  724, 476,  726, 474,  724, 478,  724, 476,  724, 478,  724, 478,  718, 490,  726, 476,  722, 476,  724, 476,  [... cut ...] 06,  696, 516,  668, 510,  716, 482,  670, 510,  696, 504,  670, 508,  696, 508,  668, 512,  694, 514,  670, 510,  696, 504,  668, 510,  696, 532,  642, 510,  696, 506,  668, 512,  694, 514,  668, 1610,  692, 504,  720, 1608,  642, 512,  694, 532,  642, 512,  694, 508,  666, 520,  694, 1604,  666, 1610,  718, 1606,  668, 1606,  668, 506,  694, 1608,  668, 512,  692, 1608,  798};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x05, 0xAF};
```

ShonP40, 2020-02-02:
A/C on - Cooling mode - 24° - Fan speed set to 1 - Light set to on - flaps off - turbo on

```Protocol  : ELECTRA_AC
Code      : 0xC387E00060402000002000050F (104 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off
uint16_t rawData[211] = {9190, 4448,  666, 1604,  640, 1610,  718, 534,  666, 534,  668, 508,  692, 508,  690, 1610,  640, 1620,  666, 1610,  718, 1606,  668, 1606,  666, 512,  692, 510,  716, 534,  666, 510,  676, 1634,  630, 526,  664, 534,  640, 564,  640, 538,  638, 540,  664, 1632,  638, 1638,  692, 1646,  638, 536,  638, 542,  664, 536,  638, 542,  664, 536,  638, 540,  666, 536,  638, 552,  664, 534,  638, 540,  666, 536,  640, 540,  666, 534,  640, 1636,  666, 1636,  640, 550,  666, 560,  614, 542,  664, 536,  638, 562,  644, 536,  638, 540,  666, 1634,  640, 548,  664, 534,  692, 560,  640, 536,  664, 536,  664, 536,  662, 1638,  612, 542,  664, 546,  640, 540,  664, 536,  638, 542,  664, 534,  640, 542,  664, 538,  664, 538,  638, 546,  666, 560,  638, 536,  664, 536,  664, 536,  664, 562,  640, 538,  662, 562,  640, 548,  662, 560,  638, 562,  638, 538,  662, 562,  638, 538,  662, 1638,  612, 544,  662, 572,  612, 542,  664, 538,  636, 542,  664, 562,  612, 568,  638, 540,  636, 568,  638, 548,  636, 1640,  664, 536,  688, 1638,  610, 568,  638, 562,  612, 568,  638, 564,  610, 576,  638, 1634,  638, 1638,  690, 1636,  636, 1636,  636, 566,  638, 562,  664, 564,  636, 566,  690};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x40, 0x20, 0x00, 0x00, 0x20, 0x00, 0x05, 0x0F};
```

A/C on - Cooling mode - 24° - Fan speed set to 1 - Light set to on - flaps off - turbo off

```Protocol  : ELECTRA_AC
Code      : 0xC387E0006000200000200008D2 (104 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off
uint16_t rawData[213] = {9222, 4418,  696, 1578,  670, 1582,  696, 480,  696, 480,  724, 478,  748, 476,  722, 1578,  696, 1564,  722, 1576,  696, 1578,  724, 1578,  698, 482,  722, 476,  698, 482,  724, 476,  696, 1592,  724, 476,  750, 478,  722, 478,  722, 478,  722, 476,  722, 1580,  670, 1582,  720, 1590,  696, 482,  724, 478,  696, 484,  722, 478,  696, 484,  722, 480,  696, 484,  722, 488,  696, 484,  722, 478,  694, 484,  720, 480,  694, 484,  722, 1578,  722, 1608,  668, 494,  722, 478,  694, 484,  722, 478,  694, 486,  694, 506,  694, 486,  720, 482,  694, 496,  720, 480,  668, 510,  720, 480,  694, 484,  696, 504,  668, 1612,  718, 484,  720, 516,  696, 504,  694, 506,  694, 506,  696, 506,  694, 506,  694, 508,  694, 508,  692, 518,  694, 506,  694, 506,  694, 506,  692, 508,  692, 508,  692, 510,  692, 508,  692, 518,  692, 508,  692, 508,  692, 508,  692, 510,  690, 506,  666, 1614,  690, 510,  690, 520,  638, 512,  664, 512,  664, 512,  690, 534,  692, 510,  690, 512,  690, 510,  690, 520,  690, 510,  690, 510,  688, 510,  688, 1612,  638, 516,  688, 512,  662, 518,  690, 522,  664, 514,  688, 1610,  662, 514,  666, 558,  690, 1614,  608, 568,  662, 1610,  662, 1618,  766, 42516,  122};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x00, 0x20, 0x00, 0x00, 0x20, 0x00, 0x08, 0xD2};
```

ShonP40, 2020-02-02:
**A/C on - Cooling mode - 24° - Fan speed set to 1 - Light set to on - flaps off - turbo off - clean off**

```Protocol  : ELECTRA_AC
Code      : 0xC387E0006000200000200019E3 (104 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off
uint16_t rawData[211] = {9162, 4476,  662, 1610,  610, 1642,  662, 536,  718, 512,  666, 532,  690, 510,  662, 1640,  610, 1650,  660, 1640,  634, 1636,  610, 1642,  686, 540,  664, 538,  662, 538,  660, 538,  664, 1652,  608, 564,  612, 544,  662, 540,  634, 566,  640, 560,  614, 1642,  660, 1640,  634, 1654,  662, 538,  660, 588,  638, 562,  612, 568,  660, 562,  638, 564,  638, 562,  638, 572,  638, 562,  638, 562,  612, 582,  644, 562,  638, 560,  638, 1662,  566, 1686,  638, 572,  666, 560,  612, 588,  638, 562,  638, 560,  638, 564,  630, 572,  612, 588,  612, 598,  638, 562,  638, 560,  638, 562,  640, 560,  638, 560,  638, 1664,  586, 568,  612, 598,  614, 566,  638, 562,  598, 582,  612, 588,  586, 594,  612, 590,  586, 594,  640, 572,  584, 594,  610, 588,  586, 594,  612, 588,  586, 594,  612, 590,  586, 594,  612, 600,  608, 570,  638, 562,  610, 570,  636, 564,  610, 566,  612, 1688,  586, 592,  610, 600,  636, 590,  636, 564,  560, 594,  610, 590,  612, 568,  612, 590,  586, 594,  636, 572,  584, 1694,  610, 588,  664, 560,  612, 1690,  558, 1694,  636, 562,  640, 588,  612, 596,  612, 1688,  586, 1666,  638, 562,  638, 588,  610, 588,  612, 1688,  558, 1692,  610, 1694,  716};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x00, 0x20, 0x00, 0x00, 0x20, 0x00, 0x19, 0xE3};
```

**A/C on - Cooling mode - 24° - Fan speed set to 1 - Light set to on - flaps off - turbo off - clean on**

```

ShonP40, 2020-02-02:
My A/C doesn't have the other options: Quiet, Econo, Filter and Beep.
Everything else except Light, Turbo and Clean works as intended.

crankyoldgit, 2020-02-02:
I've managed to work out Clean & Light, but your Turbo data has two bytes (other than the last byte which is the checksum) changing. You are going to need to be sure why & when they change before I'll add them. Or be confident that they **always** change at the same time to the same values.
e.g. `state[5]` & `state[11]` in https://github.com/crankyoldgit/IRremoteESP8266/issues/1033#issuecomment-581132529

crankyoldgit, 2020-02-02:
Please download & test the following branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Electra_improvements for the new Light & Clean support for Electra.

When you've fully understood Turbo, let me know & I'll add it too.

ShonP40, 2020-02-02:
Ty, I’ll be able to test it in ~2 hours.

ShonP40, 2020-02-02:
Clean mode works but I still can't toggle the light

crankyoldgit, 2020-02-03:
I _thought_ 'light' was being controlled by first nibble of the 2nd last byte of the state code. i.e. `state[11]`. e.g. `0x1n` for on, and` 0x0n` for off in `state[11]`.
But re-reading your own descriptions of the settings that now seems incorrect.

Can you please recompile `IRrecvDumpV2` and capture and annotate new data concentrating on the light setting please? Or preferably & better still, you can isolate which bits/bytes control the light function and tell me.

crankyoldgit, 2020-02-03:
Actually, the more I look at it, the more I need you to explain the behaviour of the second last byte of the state code. i.e. `state[11]`
You **need** to be able to explain what that does fully in all cases/values. If it changes, you need to understand why it changes. i.e. What settings affect it and how,

ShonP40, 2020-02-03:
Ok, I’ll try to do it tomorrow.

The problem is that my ESP in not in an easy accessible place.
And every time I need to re-flash it, I need to bring a ladder to get it.

crankyoldgit, 2020-02-04:
Thanks, and appreciated. Can I suggest buying a second ESP and IR receiver module? Should cost less than $5USD, and has less fall risk than ladders. ;-)

ShonP40, 2020-02-04:
Well, if I do order one rn, I’ll receive it in a month, and I don’t need ESP’s lying around.

ShonP40, 2020-02-07:
Sorry for the delay @crankyoldgit.
Here are the light commands:

A/C on - Cooling mode - 24° - Fan speed set to 1 - Light set to on - flaps off - turbo off - clean off

```Protocol  : ELECTRA_AC
Code      : 0xC387E0006000200000200015DF (104 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off, Light: On, Clean: Off
uint16_t rawData[211] = {9188, 4450,  666, 1606,  640, 1610,  666, 510,  668, 510,  692, 508,  718, 506,  692, 1608,  640, 1620,  694, 1606,  666, 1610,  692, 1608,  666, 514,  692, 526,  666, 514,  692, 508,  664, 1624,  692, 510,  710, 514,  668, 532,  666, 534,  666, 532,  666, 1634,  614, 1638,  666, 1646,  642, 536,  668, 534,  640, 538,  666, 534,  640, 540,  668, 534,  640, 540,  666, 544,  640, 538,  666, 534,  640, 540,  666, 536,  638, 538,  666, 1632,  640, 1640,  692, 544,  666, 534,  666, 534,  666, 534,  666, 534,  666, 536,  664, 538,  664, 536,  666, 544,  666, 534,  664, 534,  666, 534,  666, 534,  666, 534,  666, 1638,  638, 536,  640, 550,  666, 536,  638, 540,  666, 536,  638, 542,  664, 536,  638, 542,  664, 536,  638, 552,  664, 536,  638, 540,  664, 536,  638, 540,  666, 536,  638, 542,  664, 538,  638, 550,  666, 536,  638, 542,  664, 536,  636, 542,  664, 534,  638, 1642,  662, 538,  690, 546,  664, 536,  664, 536,  666, 536,  664, 536,  664, 536,  664, 538,  662, 538,  664, 546,  662, 1638,  610, 542,  662, 1636,  636, 536,  664, 1638,  638, 542,  664, 566,  610, 550,  664, 1632,  638, 1638,  690, 1636,  638, 1634,  636, 1640,  690, 536,  664, 1636,  638, 1638,  770};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x00, 0x20, 0x00, 0x00, 0x20, 0x00, 0x15, 0xDF};
```

A/C off - Cooling mode - 24° - Fan speed set to 1 - Light set to off - flaps off - turbo off - clean off

```Protocol  : ELECTRA_AC
Code      : 0xC387E0006000200000200015DF (104 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off, Light: On, Clean: Off
uint16_t rawData[211] = {9190, 4446,  668, 1604,  640, 1610,  666, 508,  668, 508,  694, 508,  720, 506,  692, 1608,  640, 1620,  692, 1606,  666, 1608,  694, 1608,  666, 512,  694, 508,  666, 512,  694, 506,  666, 1624,  678, 522,  692, 532,  640, 534,  666, 536,  664, 532,  614, 1636,  640, 1636,  692, 1646,  614, 538,  666, 534,  640, 538,  666, 534,  640, 540,  666, 534,  640, 540,  666, 544,  640, 540,  666, 534,  640, 540,  666, 534,  640, 538,  666, 1632,  640, 1638,  694, 544,  666, 534,  664, 534,  666, 534,  666, 534,  666, 534,  666, 536,  666, 536,  666, 544,  666, 534,  666, 534,  666, 534,  666, 534,  666, 534,  664, 1636,  666, 534,  638, 544,  666, 534,  666, 534,  612, 540,  640, 540,  640, 542,  664, 536,  640, 540,  666, 546,  640, 540,  664, 536,  638, 540,  666, 536,  638, 542,  664, 536,  638, 542,  664, 546,  640, 540,  664, 536,  638, 540,  666, 538,  636, 542,  662, 1636,  638, 540,  664, 546,  690, 534,  666, 536,  664, 538,  662, 566,  636, 536,  664, 566,  636, 538,  664, 544,  664, 1640,  610, 540,  662, 1636,  638, 564,  636, 1640,  636, 542,  662, 538,  638, 550,  664, 1632,  638, 1640,  688, 1638,  636, 1634,  638, 1642,  690, 536,  664, 1638,  638, 1638,  768};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x00, 0x20, 0x00, 0x00, 0x20, 0x00, 0x15, 0xDF};

```

ShonP40, 2020-02-07:
I've noticed that my A/C is using the same command to turn on or off the light

crankyoldgit, 2020-02-08:
Please download & test the following updated branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Electra_improvements for the new Light (toggle) support for Electra.

ShonP40, 2020-02-09:
Ok, ty, I will download and test it today

ShonP40, 2020-02-09:
1. Its still not toggling the light (the A/C beeps and nothing happens).

2. I re-recorded the Turbo mode commands

A/C on - Cooling mode - 24° - Fan speed set to 1 - flaps off - turbo off - clean off

```Protocol  : ELECTRA_AC
Code      : 0xC387E0006000200000200008D2 (104 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off, Light: -, Clean: Off
uint16_t rawData[211] = {9192, 4446,  668, 1602,  644, 1608,  668, 506,  670, 508,  694, 506,  722, 504,  694, 1608,  642, 1618,  694, 1604,  668, 1606,  696, 1608,  668, 510,  694, 506,  668, 512,  692, 508,  646, 1644,  690, 506,  696, 530,  670, 530,  670, 532,  668, 530,  668, 1632,  616, 1634,  668, 1644,  642, 536,  668, 532,  646, 534,  668, 532,  642, 538,  670, 532,  642, 538,  668, 542,  642, 536,  668, 532,  642, 538,  668, 532,  642, 536,  668, 1630,  642, 1638,  694, 542,  668, 532,  668, 532,  668, 532,  668, 532,  668, 532,  668, 534,  668, 534,  668, 542,  668, 532,  668, 532,  668, 532,  668, 532,  668, 532,  668, 1634,  642, 534,  642, 548,  666, 532,  642, 538,  668, 534,  640, 538,  668, 534,  640, 540,  666, 534,  640, 550,  666, 534,  640, 538,  692, 558,  666, 534,  666, 534,  668, 534,  666, 534,  666, 544,  666, 534,  666, 534,  666, 534,  666, 534,  666, 532,  666, 1638,  638, 536,  640, 550,  666, 534,  638, 540,  666, 534,  640, 540,  666, 536,  638, 540,  666, 536,  640, 550,  666, 534,  638, 540,  664, 534,  638, 1640,  664, 536,  690, 536,  664, 536,  664, 546,  666, 532,  664, 1636,  612, 540,  664, 536,  638, 1640,  664, 534,  690, 1638,  612, 1642,  742};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x00, 0x20, 0x00, 0x00, 0x20, 0x00, 0x08, 0xD2};
```

A/C on - Cooling mode - 24° - Fan speed set to 1 - flaps off - turbo on - clean off

```Protocol  : ELECTRA_AC
Code      : 0xC387E000604020000020000812 (104 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 24C, Fan: 3 (Low), Swing(V): Off, Swing(H): Off, Light: -, Clean: Off
uint16_t rawData[211] = {9188, 4442,  718, 1608,  692, 1608,  694, 506,  640, 512,  666, 514,  666, 512,  692, 1606,  666, 1620,  720, 1606,  666, 1604,  666, 1612,  718, 508,  690, 510,  684, 516,  666, 534,  668, 1644,  640, 532,  646, 532,  666, 534,  640, 540,  666, 534,  640, 1636,  666, 1634,  640, 1648,  614, 538,  640, 540,  640, 540,  666, 534,  640, 540,  666, 536,  640, 542,  664, 546,  638, 540,  666, 534,  638, 540,  666, 536,  638, 540,  664, 1632,  640, 1638,  692, 546,  664, 534,  666, 534,  666, 534,  664, 536,  664, 536,  666, 534,  664, 1638,  638, 548,  638, 540,  664, 536,  638, 542,  664, 538,  638, 540,  664, 1634,  638, 540,  664, 570,  666, 538,  662, 538,  662, 538,  662, 560,  666, 560,  588, 562,  612, 562,  586, 576,  614, 564,  614, 566,  640, 560,  614, 566,  640, 560,  614, 566,  640, 562,  612, 576,  642, 560,  614, 568,  638, 560,  614, 566,  640, 558,  614, 1642,  662, 562,  666, 570,  640, 560,  640, 560,  640, 560,  640, 560,  640, 560,  640, 562,  640, 562,  640, 572,  638, 560,  638, 562,  638, 562,  620, 1658,  608, 566,  618, 582,  612, 568,  614, 598,  588, 590,  614, 1660,  614, 586,  614, 586,  640, 1666,  582, 592,  614, 588,  586, 594,  692};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, 0x40, 0x20, 0x00, 0x00, 0x20, 0x00, 0x08, 0x12};
```

crankyoldgit, 2020-02-10:
> 1. Its still not toggling the light (the A/C beeps and nothing happens).

Then you are going to need to do more analysis on exactly what changes, and why between the "changing the light" messages and other "normal" messages. e.g. A temp change.

> 2. I re-recorded the Turbo mode commands

Thanks.
_Differences marked in **bold**._

turbo off:
_uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, **0x00**, 0x20, 0x00, 0x00, 0x20, 0x00, 0x08, **0xD2**};_

turbo on:
_uint8_t state[13] = {0xC3, 0x87, 0xE0, 0x00, 0x60, **0x40**, 0x20, 0x00, 0x00, 0x20, 0x00, 0x08, **0x12**};_

The last byte is the checksum, so we can ignore that.
That leaves `state[5]` controlling turbo. Specifically the _5th_ Least Significant Bit(LSB) of `state[5]`.

I'll code something up based on that soon.

crankyoldgit, 2020-02-10:
@ShonP40 Turbo support added. Please redownload and test.

[17 more replies]

closed 2020-03-26 by crankyoldgit