crankyoldgit/IRremoteESP8266 #1034: sendSANYO howto

dariocdj, 2020-01-31:
### Version/revision of the library used
branch master downloaded 31/01

### Expected behavior
tried to send unsuccessfully sanyo code as raw 

unsigned codiceAMPLIALZA[] = {500,1600, 500,1550, 500,600, 450,550, 500,550, 500,1600, 450,600, 450,1600, 500,550, 500,1550, 500,1600, 500,1600, 450,1600, 500,600, 450,550, 500,550, 450};  // SANYO B11111111111111111111111111111111
irsend.sendRaw(codiceAMPLIALZA, sizeof(codiceAMPLIALZA)/sizeof(int), khz);

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
12:24:12.507 -> Ricevo
12:24:12.507 -> Encoding  : SANYO
12:24:12.542 -> Code      : B11111111111111111111111111111111 (0 bits)
12:24:12.542 -> unsigned int  rawData[33] = {500,1600, 500,1550, 500,600, 450,550, 500,550, 500,1600, 450,600, 450,1600, 500,550, 500,1550, 500,1600, 500,1600, 450,1600, 500,600, 450,550, 500,550, 450};  // SANYO B11111111111111111111111111111111
12:24:12.577 -> unsigned int  data = 0xFFFFFFFF;
12:24:12.577 -> 
12:24:12.577 -> {B10001010,B1111000}
12:24:12.577 -> 

#### Circuit diagram and hardware used (if applicable)
Used an esp8266 gpio 13 (always used with simple irled) with 3 led drived by a 2n2222. code is trasmitted regularry.

### Has this library/code previously worked as expected for you?
Absolutely YES, always used with nec and rawdata for AIR cond.

I can't understand syntax for sendSanyo and probably code was not recognized well from my irdump? (0 bits) 0xFFFFFFFF .... 

What can i do? thanks
Codes comes from JVC RM-SRX888R remote for JVC RX888 Audio receiver/amplifier.

crankyoldgit, 2020-01-31:
> #### Output of raw data from IRrecvDumpV2.ino (if applicable)
> 12:24:12.507 -> Ricevo
> 12:24:12.507 -> Encoding : SANYO
> 12:24:12.542 -> Code : B11111111111111111111111111111111 (0 bits)
> 12:24:12.542 [...]

This is **not** the output of this library's `IRrecvDumpV2.ino` (or if it was, it is very very old)

Can you please recapture your JVC RM-SRX888R remote message with the current library and it's `IRrecvDumpV2` example code, and on an ESP8266 or an ESP32 please?

dariocdj, 2020-01-31:
Ok i try to upgrade my old arduino nano receiver ......

crankyoldgit, 2020-01-31:
An `arduino nano` won't run this library. It will only run the https://github.com/shirriff/Arduino-IRremote/ library, of which this is a very distant fork.

dariocdj, 2020-01-31:
Ok i see. I'm tryingto connect an ir receiver to an esp and use your library with IRrecvDumpV2.ino. Thanks a lot for helping...

crankyoldgit, 2020-01-31:
I massaged the data you provided into something that looks okay.

The `Arduino-IRremote` library on your `nano` is actually failing to decode it. It is certainly not a Sanyo protocol that I can tell.

Our library detects it (after some data adjustment) as a `JVC` protocol with a code/data of `0xc578`.
i.e. `irsend.sendJVC(0xc578);` should reproduce it for you.

TL;DR: This library is far far better and more correct and reliable at decoding than the library for the nano etc. Do your captures using this library and on an ESP microcontroller.

dariocdj, 2020-01-31:
Ok with your library that's result for same key : 

11:32:10.352 -> Timestamp : 000232.055
11:32:10.352 -> Library   : v2.7.3
11:32:10.352 -> 
11:32:10.352 -> Protocol  : UNKNOWN
11:32:10.352 -> Code      : 0x5295D310 (106 Bits)
11:32:10.352 -> uint16_t rawData[211] = {4032, 426,  3856, 4308,  466, 1792,  304, 1662,  388, 742,  152, 978,  116, 1012,  178, 1762,  356, 770,  176, 2078,  116, 752,  252, 1660,  494, 3732,  178, 2080,  156, 972,  180, 656,  410, 724,  136, 116,  184, 20750,  384, 1612,  412, 1792,  380, 1876,  176, 632,  180, 2106,  152, 686,  468, 1762,  362, 738,  208, 1790,  384, 1836,  252, 1752,  388, 1838,  210, 632,  464, 692,  358, 746,  152, 20574,  360, 1898,  178, 1790,  332, 1634,  438, 662,  156, 2102,  174, 662,  466, 1736,  308, 876,  180, 1764,  384, 1872,  206, 3730,  486, 642,  382, 746,  150, 21706,  310, 1630,  470, 1786,  182, 868,  226, 718,  412, 688,  152, 2106,  180, 632,  436, 1766,  206, 716,  422, 1806,  334, 1670,  96, 2126,  314, 1652,  438, 690,  318, 782,  258, 606,  464, 20492,  238, 1764,  334, 1898,  204, 658,  442, 662,  410, 716,  360, 1606,  466, 662,  386, 1638,  438, 604,  438, 1822,  206, 1758,  440, 1818,  202, 1764,  408, 720,  358, 536,  414, 632,  498, 20514,  212, 12446,  156, 684,  178, 4306,  144, 6922,  148, 20808,  182, 1822,  106, 118,  162, 1576,  414, 660,  436, 718,  152, 948,  176, 1820,  382, 746,  262, 1678,  468, 662,  150, 1842,  440, 1790,  356, 1610,  466, 1764,  364, 764,  204, 662,  464, 636,  412};  // UNKNOWN 5295D310
11:32:10.491 -> 
11:32:10.491 -> 
11:32:10.902 -> Timestamp : 000232.619
11:32:10.902 -> Library   : v2.7.3
11:32:10.902 -> 
11:32:10.902 -> Protocol  : UNKNOWN
11:32:10.936 -> Code      : 0x20FFA3A0 (92 Bits)
11:32:10.936 -> uint16_t rawData[183] = {704, 976,  152, 394,  626, 784,  406, 120,  208, 400,  1516, 406,  98, 238,  1282, 4598,  150, 4890,  152, 2074,  204, 1794,  150, 2946,  150, 976,  308, 1660,  422, 1810,  282, 1742,  104, 74,  174, 2972,  178, 692,  358, 742,  334, 20944,  272, 3836,  358, 770,  180, 688,  438, 662,  384, 1848,  200, 898,  226, 1770,  330, 746,  224, 1794,  150, 1872,  410, 1766,  386, 1634,  414, 720,  88, 986,  144, 946,  230, 20530,  332, 1634,  412, 1822,  314, 812,  180, 658,  462, 852,  176, 1858,  212, 724,  102, 72,  178, 1844,  150, 952,  182, 2040,  160, 1840,  152, 1816,  156, 1838,  414, 658,  412, 746,  178, 924,  178, 20524,  314, 1678,  442, 1790,  204, 688,  412, 634,  468, 690,  356, 1642,  108, 68,  256, 662,  412, 1822,  152, 710,  466, 1764,  318, 1652,  464, 1758,  290, 1710,  440, 690,  332, 770,  260, 604,  466, 20512,  184, 9092,  156, 974,  152, 3206,  180, 1788,  152, 34330,  118, 746,  362, 766,  116, 114,  158, 2710,  412, 690,  150, 2134,  152, 658,  152, 2074,  204, 2054,  152, 3788,  148, 978,  152, 950,  180, 686,  410};  // UNKNOWN 20FFA3A0
11:32:11.038 -> 
11:32:11.038 -> 
11:32:11.654 -> Timestamp : 000233.368
11:32:11.654 -> Library   : v2.7.3
11:32:11.654 -> 
11:32:11.654 -> Protocol  : UNKNOWN
11:32:11.654 -> Code      : 0x39A5CA9C (43 Bits)
11:32:11.689 -> uint16_t rawData[85] = {706, 416,  1264, 426,  180, 370,  180, 396,  410, 458,  114, 108,  1288, 428,  1398, 4500,  208, 1766,  358, 1670,  300, 712,  414, 768,  318, 754,  208, 1790,  406, 722,  332, 1634,  410, 2940,  212, 1766,  148, 2028,  230, 1844,  328, 1894,  212, 608,  178, 30626,  112, 14478,  150, 21652,  178, 5150,  152, 4998,  436, 2866,  148, 6208,  220, 2714,  438, 690,  150, 23676,  438, 1796,  104, 2086,  210, 4052,  152, 1814,  262, 834,  236, 1766,  150, 2134,  112, 1774,  152, 2972,  410, 926,  152};  // UNKNOWN 39A5CA9C
11:32:11.723 -> 

This is vol + key.... how i can send this code? sendJVC? send what ? Thanks

dariocdj, 2020-01-31:
Thanks for 0xc578 that i can send with irsend.sendJVC(0xc578); 
But unfortunately i need lots of other codes, so i can use your new IRrecvDumpV2.ino but result is always different code? and Protocol : UNKNOWN ?

dariocdj, 2020-01-31:
Or if you explain me i can use old result to massage data and convert it as jvc ?

dariocdj, 2020-01-31:
OK SORRY, running receiver at 5v give me correct code
11:47:27.618 -> Timestamp : 000089.710
11:47:27.618 -> Library   : v2.7.3
11:47:27.618 -> 
11:47:27.618 -> Protocol  : JVC
11:47:27.618 -> Code      : 0xC578 (16 Bits)
11:47:27.665 -> uint16_t rawData[205] = {8432, 4228,  496, 1588,  520, 1584,  522, 556,  496, 560,  520, 504,  522, 1586,  520, 556,  470, 1638,  496, 586,  466, 1610,  468, 1640,  468, 1640,  492, 1614,  520, 506,  520, 584,  468, 560,  490, 20526,  496, 1642,  520, 1610,  474, 528,  516, 560,  498, 558,  518, 1588,  494, 534,  514, 1616,  498, 556,  494, 1616,  516, 1590,  516, 1588,  496, 1608,  496, 588,  486, 538,  492, 588,  468, 20560,  518, 1590,  520, 1586,  494, 560,  520, 504,  520, 588,  466, 1612,  492, 562,  518, 1588,  496, 556,  492, 1590,  544, 1612,  470, 1610,  496, 1612,  494, 586,  470, 558,  490, 560,  494, 20556,  494, 1612,  494, 1612,  492, 586,  446, 610,  464, 560,  494, 1614,  468, 610,  466, 1614,  476, 630,  442, 1612,  492, 1614,  498, 1584,  494, 1616,  544, 504,  520, 590,  470, 554,  470, 20556,  514, 1620,  496, 1612,  520, 530,  494, 588,  468, 556,  492, 1592,  542, 532,  468, 1640,  492, 590,  464, 1614,  496, 1586,  516, 1590,  520, 1612,  496, 532,  520, 584,  466, 562,  516, 20508,  522, 1612,  492, 1616,  494, 532,  520, 584,  470, 558,  488, 1618,  520, 532,  496, 1586,  518, 558,  468, 1640,  496, 1584,  520, 1612,  496, 1614,  492, 532,  516, 560,  468, 616,  468};  // JVC C578
11:47:27.758 -> uint32_t address = 0xA3;
11:47:27.758 -> uint32_t command = 0x1E;
11:47:27.758 -> uint64_t data = 0xC578;
11:47:27.758 -> 

So i will try with sendJVC, thanks a lot.

crankyoldgit, 2020-01-31:
> OK SORRY, running receiver at 5v give me correct code

Phew. I was going to say "Your data looks crap! Try a different receiver!"

Glad you've got it working now!

crankyoldgit, 2020-01-31:
FYI, I'm guessing you held the button down for a bit, as I see the JVC message plus 5 repeats in that raw data.
e.g. `sendJVC(0xC578, 16, 5);` would reproduce effectively the same thing as the `rawData[205]` etc.

dariocdj, 2020-01-31:
Ok now i've decoded all the JVC remote code and send by sendJVC(0xC578); give me a good result, receiver seems to decode JVC. 
BUT, 
my epson remote is decoded as : 
Protocol  : NEC
14:20:14.613 -> Code      : 0xC1AA09F6 (32 Bits)
14:20:14.648 -> uint16_t rawData[205] = {9006, 4496,  548, 1684,  548, 1682,  544, 612,  518, 586,  542, 584,  546, 582,  544, 612,  520, 1688,  542, 1686,  548, 584,  568, 1660,  568, 560,  570, 1636,  572, 582,  546, 1658,  572, 558,  598, 532,  598, 530,  598, 530,  574, 582,  548, 1656,  572, 582,  570, 532,  574, 1658,  598, 1658,  544, 1688,  544, 1688,  544, 1686,  546, 610,  522, 1684,  542, 1688,  546, 584,  546, 40866,  8976, 4472,  592, 1638,  598, 1632,  572, 582,  568, 534,  596, 532,  576, 550,  592, 564,  570, 1632,  572, 1686,  546, 584,  548, 1664,  590, 556,  546, 1686,  544, 584,  544, 1688,  546, 582,  544, 586,  542, 584,  546, 582,  544, 584,  548, 1682,  544, 586,  548, 580,  546, 1684,  546, 1712,  520, 1686,  546, 1684,  544, 1688,  548, 580,  546, 1686,  546, 1684,  546, 582,  564, 40824,  8376, 310,  264, 4522,  596, 1660,  548, 1684,  546, 584,  544, 582,  546, 584,  544, 584,  546, 582,  546, 1686,  546, 1688,  546, 582,  546, 1686,  570, 558,  564, 1668,  548, 606,  522, 1684,  546, 608,  516, 612,  518, 612,  516, 610,  518, 610,  520, 1688,  546, 582,  544, 586,  546, 1684,  546, 1688,  572, 1658,  572, 1634,  598, 1658,  560, 544,  596, 1634,  570, 1686,  546, 582,  546};  // NEC C1AA09F6
14:20:14.747 -> uint32_t address = 0x5583;
14:20:14.747 -> uint32_t command = 0x90;
14:20:14.747 -> uint64_t data = 0xC1AA09F6;

and if try to send with : 
long codicePROIETTOREON=0xC1AA09F6 ;
          irsend.sendNEC(codicePROIETTOREON, 36);

receiver respond me with unknow code..... what can be ?

dariocdj, 2020-01-31:
OK  SORRY AGAIN irsend.sendNEC(codicePROIETTOREON); seems to work.
Thanks a lot again for your help !
Dario

crankyoldgit, 2020-01-31:
> 14:20:14.613 -> Code : 0xC1AA09F6 (**32** Bits)
> ...
> irsend.sendNEC(codicePROIETTOREON, 36);

NEC is **32** bits.
i.e.
`irsend.sendNEC(codicePROIETTOREON, 32);`

dariocdj, 2020-02-02:
Sorry to reopen it, but neither nec neither jvc actually works.
OK i will try irsend.sendNEC(codicePROIETTOREON, 32); where codicePROIETTOREON is a long var?
For sendJVC no parameters is needed? 
Is it codes case sensitive? 0xc5b8 and 0xC5B8 are the same?
Is it ok  irsend.sendJVC(codiceAMPLIALZA); where codiceAMPLIALZA is a uint16_t var?
Thanks a lot and sorry, but i must solve it!

crankyoldgit, 2020-02-02:
> OK i will try irsend.sendNEC(codicePROIETTOREON, 32); where codicePROIETTOREON is a long var?

`irsend.sendNEC(codicePROIETTOREON);` is all you need. All you need to do is pass it an integer of size `uint64_t` or smaller. It will work out what to do.

> Is it codes case sensitive?

No, integers are not case sensitive.

> 0xc5b8 and 0xC5B8 are the same?

If they are integers, Yes.

> Is it ok irsend.sendJVC(codiceAMPLIALZA); where codiceAMPLIALZA is a uint16_t var?

Yes.

dariocdj, 2020-02-04:
Little update after one day. Put 2 82ohm res give me 50mA and sending with 3 repeat my jvc amplifier work!!! Projector SendNec at the moment works only sending continuously in a 50ms loop. Tomorrow I want to put 47ohm res to have 80mA. I will update you!

crankyoldgit, 2020-02-04:
Looking back at your rawData from your Epson remote, it appears to contain three(3) copies of a single NEC message in it. However, it is not using the standard NEC repeat message protocol/standard. The _real_ NEC protocol uses a special repeat code.

Because of that, to emulate what you need, you can't use the `repeat` argument to the `sendNEC()` function. That is `irsend.sendNEC(codicePROIETTOREON, 2);` will probably not work. Try it though, you might be lucky.

You'll need to do this instead:
```
irsend.sendNEC(codicePROIETTOREON);
irsend.sendNEC(codicePROIETTOREON);
irsend.sendNEC(codicePROIETTOREON);
```
or
```
for (int i = 0; i < 3; i++) {
  irsend.sendNEC(codicePROIETTOREON);
}
```

dariocdj, 2020-02-04:
Ok, 
Original NEC code decoded is : 
Timestamp : 002060.405
Library   : v2.7.3

Protocol  : NEC
Code      : 0xC1AA09F6 (32 Bits)
uint16_t rawData[135] = {8978, 4494,  548, 1686,  544, 1686,  546, 610,  518, 584,  544, 582,  546, 610,  518, 610,  520, 1686,  544, 1686,  544, 586,  542, 1688,  546, 584,  546, 1684,  546, 608,  518, 1688,  544, 586,  544, 582,  546, 608,  516, 612,  518, 610,  520, 1686,  542, 586,  544, 610,  520, 1686,  544, 1688,  544, 1686,  544, 1688,  544, 1686,  546, 584,  544, 1684,  546, 1688,  544, 584,  542, 40870,  8978, 4494,  544, 1688,  546, 1686,  544, 584,  544, 584,  544, 586,  542, 586,  542, 586,  544, 1686,  542, 1688,  544, 584,  544, 1686,  544, 586,  542, 1688,  546, 610,  516, 1688,  544, 610,  518, 612,  516, 612,  518, 610,  520, 586,  544, 1660,  568, 584,  544, 588,  544, 1684,  546, 1688,  544, 1686,  544, 1688,  546, 1684,  546, 610,  518, 1688,  544, 1688,  542, 586,  544};  // NEC C1AA09F6
uint32_t address = 0x5583;
uint32_t command = 0x90;
uint64_t data = 0xC1AA09F6;

With "for cycle" this is result : 

Timestamp : 001967.357
Library   : v2.7.3

Protocol  : UNKNOWN
Code      : 0x12F09CF9 (102 Bits)
uint16_t rawData[203] = {8916, 4534,  498, 1748,  520, 1718,  524, 628,  492, 626,  472, 650,  494, 628,  470, 652,  494, 1720,  520, 1720,  496, 652,  496, 1718,  522, 624,  494, 1724,  520, 626,  472, 1744,  524, 598,  522, 1718,  522, 626,  470, 652,  496, 628,  470, 1744,  496, 650,  496, 624,  470, 1746,  522, 626,  470, 1770,  468, 1748,  520, 1720,  522, 626,  470, 1744,  520, 1720,  522, 632,  494, 41452,  8942, 4506,  496, 1770,  472, 1744,  522, 602,  522, 600,  520, 626,  496, 626,  470, 652,  496, 1720,  522, 1720,  506, 640,  470, 1744,  522, 628,  470, 1770,  472, 650,  494, 1720,  522, 624,  470, 1746,  520, 602,  522, 598,  520, 600,  520, 1720,  496, 652,  496, 602,  520, 1720,  496, 650,  498, 1716,  520, 1722,  498, 1744,  496, 652,  468, 1772,  472, 1770,  472, 650,  470, 41482,  8940, 4506,  524, 1720,  520, 1720,  518, 628,  496, 628,  470, 650,  470, 652,  496, 628,  470, 1770,  470, 1770,  498, 626,  470, 1744,  520, 624,  474, 1742,  522, 624,  470, 1772,  468, 652,  494, 1722,  524, 624,  470, 624,  520, 626,  470, 1770,  472, 650,  470, 650,  496, 1722,  520, 626,  472, 1746,  520, 1716,  498, 1744,  520, 626,  470, 1770,  472, 1746,  520, 602,  520};  // UNKNOWN 12F09CF9

And with 3 line irsend.sendNEC :

Protocol  : UNKNOWN
Code      : 0x47BAE575 (102 Bits)
uint16_t rawData[203] = {8832, 4584,  468, 1754,  468, 1800,  440, 678,  442, 680,  468, 654,  444, 650,  470, 678,  468, 1774,  444, 1798,  442, 680,  468, 1776,  444, 676,  444, 1798,  442, 678,  442, 1798,  442, 678,  442, 678,  466, 658,  442, 678,  444, 678,  442, 1800,  444, 650,  468, 680,  442, 1800,  444, 1798,  444, 1796,  444, 1798,  468, 1750,  468, 678,  446, 1794,  444, 1798,  444, 680,  440, 41536,  8858, 4586,  468, 1778,  442, 1798,  444, 678,  442, 678,  446, 678,  446, 676,  444, 678,  468, 1774,  446, 1820,  418, 680,  440, 1800,  442, 680,  442, 1798,  442, 680,  442, 1826,  418, 676,  442, 706,  416, 680,  442, 704,  416, 654,  468, 1800,  442, 678,  468, 656,  444, 1822,  418, 1822,  416, 1798,  446, 1796,  444, 1796,  444, 702,  420, 1824,  418, 1798,  446, 676,  442, 41552,  8836, 4610,  464, 1780,  444, 1798,  442, 678,  468, 656,  442, 678,  444, 676,  468, 656,  444, 1796,  444, 1796,  468, 656,  444, 1798,  442, 678,  468, 1774,  446, 676,  444, 1796,  442, 678,  444, 676,  444, 678,  440, 678,  470, 656,  442, 1800,  442, 676,  470, 654,  444, 1798,  440, 1800,  442, 1798,  444, 1772,  470, 1796,  442, 678,  444, 1796,  446, 1796,  442, 680,  442};  // UNKNOWN 47BAE575

So falling back to 1 time send with irsend.sendNEC : 
FIRST TIME GIVE ME 👍 CORRECT CODE !!! 
Timestamp : 002053.692
Library   : v2.7.3

Protocol  : NEC
Code      : 0xC1AA09F6 (32 Bits)
uint16_t rawData[203] = {8976, 4496,  542, 1690,  542, 1688,  542, 588,  518, 612,  542, 586,  518, 610,  518, 610,  542, 1688,  516, 1714,  542, 614,  516, 1688,  542, 612,  516, 1690,  540, 614,  514, 1690,  520, 638,  516, 584,  518, 612,  542, 586,  540, 588,  542, 1688,  520, 610,  540, 586,  520, 1712,  540, 1692,  542, 1688,  542, 1690,  520, 1710,  518, 610,  518, 1712,  518, 1716,  544, 584,  544, 40870,  8978, 4496,  542, 1690,  542, 1688,  520, 610,  542, 584,  516, 638,  514, 588,  540, 588,  540, 1690,  516, 1716,  516, 614,  542, 1688,  518, 612,  542, 1688,  540, 614,  514, 1692,  542, 588,  538, 590,  542, 586,  542, 612,  516, 612,  514, 1692,  516, 612,  540, 586,  520, 1712,  518, 1714,  542, 1688,  518, 1716,  542, 1688,  516, 614,  542, 1686,  518, 1714,  516, 612,  532, 40882,  8976, 4498,  516, 1716,  542, 1690,  540, 614,  516, 614,  516, 614,  514, 588,  540, 588,  544, 1688,  522, 1710,  518, 638,  516, 1688,  520, 636,  492, 1714,  542, 586,  542, 1690,  542, 588,  542, 584,  542, 612,  516, 612,  514, 614,  516, 1690,  542, 586,  542, 586,  540, 1690,  518, 1712,  518, 1716,  518, 1714,  542, 1690,  542, 586,  540, 1690,  542, 1690,  542, 588,  540};  // NEC C1AA09F6
uint32_t address = 0x5583;
uint32_t command = 0x90;
uint64_t data = 0xC1AA09F6;

but if i retry after some seconds : 
Protocol  : NEC
Code      : 0xC1AA09F6 (32 Bits)
uint16_t rawData[67] = {8916, 4534,  522, 1720,  526, 1716,  522, 624,  522, 602,  496, 622,  498, 624,  524, 600,  498, 1718,  524, 1716,  550, 600,  496, 1720,  522, 596,  548, 1696,  524, 622,  494, 1722,  522, 598,  522, 598,  522, 624,  49 [... cut ...]  524, 600,  496, 1720,  522, 598,  548, 576,  520, 1722,  524, 1740,  498, 1718,  524, 1716,  524, 1718,  524, 598,  524, 1716,  526, 1716,  522, 624,  500};  // NEC C1AA09F6
uint32_t address = 0x5583;
uint32_t command = 0x90;
uint64_t data = 0xC1AA09F6;

Why Rawdata change??? i can't test it in lab with projector and receiver togheter :(
I'm at 50cm from receiver. And now i'm using 47ohm resistor with a 100mA led and 5v (so it must be around 80mA transmitting power). 

Thanks for futher helping

crankyoldgit, 2020-02-05:
What is the brand & model info for both the projector and the remote?

crankyoldgit, 2020-02-05:
There could be a number of reasons that the rawData is changing.
For starters, read the FAQ on this: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#why-is-the-raw-data-for-a-button-or-ac-state-always-different-for-each-capture

The reason(s) why it is not being recognised as NEC is because the timings are slightly out of spec.
For some of the captures, they are with in the expected range and thus detected successfully. The `UNKNOWN`s are just out of the expected ranges, and hence it thinks "This isn't NEC!".

That could be cause by a number of environmental things, the quality of your receiver, the quality of the remote's signal. e.g. Check it's batteries etc.
Or being to close/to far from the receiver etc etc.

dariocdj, 2020-02-05:
Wow, you've made a commit ! Thanks a lot, also if before i want to retest it, i've adjusted resistor and add new led and want to try if it works before updating. 
Last question. My original jvc remote is died. And i've just 1 code that doesn't work. 
Searching on remote central i've found the code in this format:

0000 006c 0001 0011 0140 00a0 0014 003c 0014 003c 0014 0014 0014 0014 0014 0014 0014 003c 0014 0014 0014 003c 0014 0014 0014 0014 0014 0014 0014 0014 0014 003c 0014 0014 0014 0014 0014 0014 0014 0384

can you translate me in "jvc mode" ? Thanks a lot !
Dario

crankyoldgit, 2020-02-05:
I suggest you look at https://irdb.globalcache.com/ as a source for IR codes too.
That is Pronto Code. You can use `sendPronto` (see example code for how to use it).
e.g.
```
  uint16_t pronto_test[38] = {
      0x0000, 0x006c, 0x0011, 0x0011, 0x0014, 0x003c, 0x0014, 0x003c, 0x0014,
      0x0014, 0x0014, 0x0014, 0x0014, 0x0014, 0x0014, 0x003c, 0x0014, 0x0014,
      0x0014, 0x003c, 0x0014, 0x0014, 0x0014, 0x0014, 0x0014, 0x0014, 0x0014,
      0x0014, 0x0014, 0x003c, 0x0014, 0x0014, 0x0014, 0x0014, 0x0014, 0x0014,
      0x0014, 0x0384};

  irsend.sendPronto(pronto_test, 38);
```

We have a tool called `gc_decode` in the tools directory that _normally_ converts Pronto code into our system, however it doesn't seem to work on this one. (Please don't ask for me to convert another one for you)

I think I hacked it convert your code, here is the result:
```
Code type      6 (JVC)
Code bits      16
Code value     0xc508
Code address   0xa3
Code command   0x10
```

crankyoldgit, 2020-02-05:
i.e. The hack was to change the first `0001` in the pronto code to `0011`

dariocdj, 2020-02-06:
Finally with pronto code it seems to work quite good. Don't understand why with this jvc and blaster sometime no one code work and need to restart ampli receiver... Boh. Thanks man

closed 2020-02-06 by dariocdj

crankyoldgit, 2020-02-26:
The changes mentioned above have been included in the latest [IRremoteESP8266 v2.7.4](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.4) release.

crankyoldgit, 2020-06-04:
@dariocdj It's been a while since https://github.com/crankyoldgit/IRremoteESP8266/issues/1034#issuecomment-582206406, but I'm re-asking this:
"What is the brand & model info for both the projector and the remote?"

FYI @NiKiZe X-Ref: https://github.com/crankyoldgit/IRremoteESP8266/pull/1167#pullrequestreview-424081485

dariocdj, 2020-06-04:
Sorry, everything work from month perfectly..... Epson EN-TW9100W.

crankyoldgit, 2020-06-04:
Thanks for that. Is there a model number on the remote at all?