crankyoldgit/IRremoteESP8266 #1035: Power Stat doesn't reflect received command, JSON does

sheppy99, 2020-01-31:
I'm debugging my Daikin2 system and have discovered that Received topic 
>ir_server/ac/stat/power 
doesn't show "off" when an off command is sent by the original remote, 
the topic
>ir_server/ac/stat/json
works correctly, and the other individual topics such as mode and temperature work correctly.
This is present in V1.4.5 and also v1.4.4-alpha

crankyoldgit, 2020-01-31:
What is the library version you are currently using?
Can you give me the code for the off message from the remote please?

sheppy99, 2020-02-01:
The current one downloaded this morning, 

ON >53,0x11DA270001154390290C8004C016240000BEC12D11DA270000092A00B0000006600000C190600C
OFF
>53,0x11DA27000115C390290C8004C016240000BED1BD11DA270000082A00B0000006600000C190600B

crankyoldgit, 2020-02-01:
I'll start looking at it now.

BTW, which version did it last work, do you know? and as I know you do so, what extra options etc have you tweaked so I can reconstruct your setup.

crankyoldgit, 2020-02-01:
@sheppy99 I've looked over the code, and constructed some tests to simulate as best I can with the info available so far. I can't find anything wrong or that jumps out at me.

I'm going to need a shirt-tonne of more info to work this one out I think.
Can I get you to turn on debugging and give me the serial output for these interactions please?

Also, a description of the setup (e.g. Yes, there is a DAIKIN2 a/c unit controlled by channel 0 etc) and what flags/settings you've altered from "normal".

sheppy99, 2020-02-01:
@crankyoldgit Its a pity its not straight forward, I'll test with other remotes to see what I can reproduce.
regarding settings I've changed the following in IRMQTTServer.h

>line 35 const uint8_t kNrOfIrTxGpios changed to 7
>line 39 const int8_t kDefaultIrLed  changed to 16
>line 104 define MQTT_CLIMATE_JSON changed to true
>line 109 #define MQTT_CLIMATE_HA_MODE changed to false
>line 114 #define MQTT_CLIMATE_IR_SEND_ON_RESTART changed to true
>line 122 #define MQTT_DISCOVERY_ENABLE false

Its running on an ESP with a IR detector attached to it and nothing else. I don't know whether this has ever worked as I don't think I've specifically tested it. As summer has finally arrived across the ditch I need to modify my bedroom cooling rule and figured I'd see if I could move to using IRMQTTServer to generate the full IR string instead of using OpenHAB hence the sudden activity.

sheppy99, 2020-02-01:
Further testing, Fujitsu, and 2 different Mitsubishi remotes behave the same and the fault is one way - send an 'on' command and both json and power are updates, send an 'off' command and only json is updated. Send an off command to >ir_server/ac/cmnd/power
and both power and json go off, remote can then turn them on again

crankyoldgit, 2020-02-01:
> line 35 const uint8_t kNrOfIrTxGpios changed to 7

Wowsers! Way to work that little $5 chip! ;-)

> As summer has finally arrived across the ditch

I understand. I've been heavily using mine for months now.

Thanks for the data/feedback, but I am going to need to you turn on debugging and capture the messages for those transitions. I continued looking at the code after I sent that update etc. Still couldn't see what was going wrong, and all my tests to simulate using your data didn't seem to behave the way you were describing it. Hence, more info you is going to be required for me to duplicate it/find it.

Just stating the obvious, inbound IR commands captured by the ESP chip will only work for the "channel 0" (first) A/C unit. It can't control the other ones.

Knowing your env. I had already guess you had JSON on and the HA_MODE off.

A left-field idea/thought, I remember you had issues once with your MQTT server retaining a bad value once. Might be worth resetting/clearing (then checking) the `ir_server/ac/stat/power` topic.

Anyway, the serial debug output should tell me what it things it's sending/producing etc. and that will give me a good idea what is going on & where to look etc.

Thanks in advance.

sheppy99, 2020-02-01:
changing it now!

sheppy99, 2020-02-01:
Just sent a Daikin2 ON and then OFF message, hope it helps!

>0000108.489: Incoming IR message sent to MQTT:
0000108.490: 53,0x11DA2700018D4290290C8004C016240000BEC1A411DA270000092A00B0000006600000C190600C
0000108.491: Converting inbound IR A/C message to common A/C
0000108.498: Difference in common A/C state detected.
0000110.858: Incoming IR message sent to MQTT:
0000110.859: 53,0x11DA2700018DC290290C8004C016240000BED13411DA270000082A00B0000006600000C190600B
0000110.860: Converting inbound IR A/C message to common A/C
0000110.867: Difference in common A/C state detected.

sheppy99, 2020-02-01:
OK, weird stuff time, its now working with all 4 different remotes. I'll recompile without the debugging

crankyoldgit, 2020-02-01:
Great :-/ Worst kind of bug, an intermittent one. :-(

sheppy99, 2020-02-01:
You'll love what just happened - NOT
Recompiled with debug false. pressed Daikin2 remote Auto Button and both topics showed power ON, pressed OFF button and now Power says off and json says ON - sigh....
Suggestions....

crankyoldgit, 2020-02-01:
I'll add some more debug messages and see what we can see.

sheppy99, 2020-02-01:
and one other thing to add a little spice on top - I've got 
>#define MQTT_CLIMATE_HA_MODE false

yet json has "mode":"heat" and the straight mode topic has mode = off

I'll update a few libraries just in case a bug has crept in to the Ardionojson

crankyoldgit, 2020-02-01:
That's weird.

But first things first. Can you please download & try https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1035 ?

It should have mode debug messages in it and may help us see what is going on under the hood better.

You'll know you're running the correct version when you see the different IRMQTTServer version string. ;-)

Again, debug messages please. ;-)

crankyoldgit, 2020-02-01:
Oh, I changed the settings to match what you indicated earlier, so you shouldn't need to do anything but compile and upload etc.

sheppy99, 2020-02-01:
Just playing around I see that a message is sent to
>ir_server/ac/stat/power

when I turn it on, but not when I turn it off.
json works correctly
ALSO there's odd behaviour with the mode
when I switch from cool to auto with power ON both json and mode are updated
when I switch from auto to heat with power ON json is correct, no message is sent to mode
when I switch from heat to auto with power ON both json and mode are updated

I'll compile your added debugging and let you know the states shortly. I'll do a broader test using power, mode, and fan and send you what stands out as wrong. Hopefully its an easy fix

crankyoldgit, 2020-02-01:
The debugging message may confirm what I fear, that is the MQTT message are being sent from the ESP but are being lost on the way to your MQTT server/broker.
We might need to change the QoS setting(s) for outbound messages. Or the problem could be on your MQTT broker side. :-(

crankyoldgit, 2020-02-01:
P.S. Check what the web "Aircon" page thinks the state is too, compared to what you're seeing the state as in your MQTT broker. remember to refresh it as it doesn't update.

crankyoldgit, 2020-02-01:
i.e. The "Aircon" page's state should be authoritative for what the ESP has seen/knows about.

crankyoldgit, 2020-02-01:
The more I look at your results and the code etc. The more I'm leaning to mqtt messages are going missing. Adding debug messages may make it better by adding a small delay while it prints it to serial and thus, be hammering your mqtt server/broker less.

sheppy99, 2020-02-01:
starting from OFF,
pressing 'cool' then OFF

>0000143.006: Incoming IR message sent to MQTT:
0000143.006: 53,0x11DA270001CC4290290C8004C016240000BEC5E711DA270000392A00B0000006600000C190603C
0000143.007: Converting inbound IR A/C message to common A/C
0000143.013: MQTT publish (with retain):
0000143.016: Topic:
0000143.018: Testir_server/ac/stat/mode
0000143.021: Payload:
0000143.023: cool
0000143.025: MQTT publish (with retain):
0000143.028: Topic:
0000143.030: Testir_server/ac/stat/temp
0000143.034: Payload:
0000143.036: 21.0
0000143.037: Difference in common A/C state detected.
0000143.043: MQTT publish:
0000143.044: Topic:
0000143.046: Testir_server/ac/stat/json
0000143.049: Payload:
0000143.051: {"protocol":"DAIKIN2","model":-1,"power":"On","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}
0000145.930: Incoming IR message sent to MQTT:
0000145.930: 53,0x11DA270001CCC290290C8004C016240000BED57711DA270000382A00B0000006600000C190603B
0000145.932: Converting inbound IR A/C message to common A/C
0000145.937: MQTT publish (with retain):
0000145.940: Topic:
0000145.942: Testir_server/ac/stat/power
0000145.946: Payload:
0000145.948: off
0000145.949: MQTT publish (with retain):
0000145.953: Topic:
0000145.954: Testir_server/ac/stat/mode
0000145.958: Payload:
0000145.960: cool
0000145.962: MQTT publish (with retain):
0000145.965: Topic:
0000145.967: Testir_server/ac/stat/temp
0000145.970: Payload:
0000145.972: 21.0
0000145.974: Difference in common A/C state detected.
0000145.979: MQTT publish:
0000145.981: Topic:
0000145.982: Testir_server/ac/stat/json
0000145.986: Payload:
0000145.988: {"protocol":"DAIKIN2","model":-1,"power":"Off","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}

starting from OFF,
pressing 'heat' then OFF

>0000186.543: Incoming IR message sent to MQTT:
0000186.543: 53,0x11DA270001CD4290290C8004C016240000BEC5E811DA270000492600B0000006600000C1906048
0000186.544: Converting inbound IR A/C message to common A/C
0000186.550: NO difference in common A/C state detected.
0000190.815: Incoming IR message sent to MQTT:
0000190.815: 53,0x11DA270001CDC290290C8004C016240000BED57811DA270000482600B0000006600000C1906047
0000190.816: Converting inbound IR A/C message to common A/C
0000190.822: MQTT publish (with retain):
0000190.825: Topic:
0000190.827: Testir_server/ac/stat/power
0000190.830: Payload:
0000190.832: off
0000190.834: Difference in common A/C state detected.
0000190.840: MQTT publish:
0000190.841: Topic:
0000190.842: Testir_server/ac/stat/json
0000190.846: Payload:
0000190.848: {"protocol":"DAIKIN2","model":-1,"power":"Off","mode":"Heat","use_celsius":"On","temp":19,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}

starting from OFF,
pressing 'auto' then OFF

>0000231.402: Incoming IR message sent to MQTT:
0000231.403: 53,0x11DA270001CE4290290C8004C016240000BEC1E511DA270000092A00B0000006600000C190600C
0000231.404: Converting inbound IR A/C message to common A/C
0000231.410: MQTT publish (with retain):
0000231.413: Topic:
0000231.415: Testir_server/ac/stat/mode
0000231.418: Payload:
0000231.420: auto
0000231.422: MQTT publish (with retain):
0000231.425: Topic:
0000231.427: Testir_server/ac/stat/temp
0000231.430: Payload:
0000231.432: 21.0
0000231.434: MQTT publish (with retain):
0000231.437: Topic:
0000231.439: Testir_server/ac/stat/swingv
0000231.443: Payload:
0000231.445: Highest
0000231.447: Difference in common A/C state detected.
0000231.452: MQTT publish:
0000231.453: Topic:
0000231.455: Testir_server/ac/stat/json
0000231.459: Payload:
0000231.461: {"protocol":"DAIKIN2","model":-1,"power":"On","mode":"Auto","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Highest","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}
0000233.538: Incoming IR message sent to MQTT:
0000233.539: 53,0x11DA270001CEC290290C8004C016240000BED17511DA270000082A00B0000006600000C190600B
0000233.540: Converting inbound IR A/C message to common A/C
0000233.546: MQTT publish (with retain):
0000233.549: Topic:
0000233.550: Testir_server/ac/stat/power
0000233.554: Payload:
0000233.556: off
0000233.558: MQTT publish (with retain):
0000233.561: Topic:
0000233.563: Testir_server/ac/stat/mode
0000233.566: Payload:
0000233.568: auto
0000233.570: MQTT publish (with retain):
0000233.573: Topic:
0000233.575: Testir_server/ac/stat/temp
0000233.578: Payload:
0000233.580: 21.0
0000233.582: MQTT publish (with retain):
0000233.585: Topic:
0000233.587: Testir_server/ac/stat/swingv
0000233.591: Payload:
0000233.593: Highest
0000233.595: Difference in common A/C state detected.
0000233.600: MQTT publish:
0000233.602: Topic:
0000233.603: Testir_server/ac/stat/json
0000233.607: Payload:
0000233.609: {"protocol":"DAIKIN2","model":-1,"power":"Off","mode":"Auto","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Highest","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}

sheppy99, 2020-02-01:
Looking at the 'heat' debug in the middle is interesting as it worked the first time I tried it. One thing I have done is updated to buster recently in case its relevant. I'll start up a copy of mosquitto on another port and run it against that for the rest of the tests to rule out congestion

crankyoldgit, 2020-02-01:
I'm confused looking at the log and comparing what I'm seeing to your descriptions.
e.g. the device/esp seems to going to "on" mode, yet you don't seem to be describing that action as a remote button press.

sheppy99, 2020-02-01:
New test - mosquitto on a new port, only thing on it is the test receive setup and MQTT Explorer
Using original Daikin2 remote pressed cool button and then off button. json updated to both settings correctly, aircon page checked at end and shows power off. Debug output from restart on new broker below - there seems to be a missing power off publish
>0000040.295: Unsubscribed OK from:
0000040.295: Testir_server/ac/stat/+
0000040.295: Unsubscribed OK from:
0000040.296: Testir_server/ac_1/stat/+
0000040.297: Unsubscribed OK from:
0000040.299: Testir_server/ac_2/stat/+
0000040.303: Unsubscribed OK from:
0000040.306: Testir_server/ac_3/stat/+
0000040.309: Unsubscribed OK from:
0000040.312: Testir_server/ac_4/stat/+
0000040.316: Unsubscribed OK from:
0000040.319: Testir_server/ac_5/stat/+
0000040.322: Unsubscribed OK from:
0000040.325: Testir_server/ac_6/stat/+
0000040.329: Finished listening for previous state.
0000055.716: Incoming IR message sent to MQTT:
0000055.716: 53,0x11DA270001D54290290C8004C016240000BEC5F011DA270000392A00B0000006600000C190603C
0000055.718: Converting inbound IR A/C message to common A/C
0000055.723: MQTT publish (with retain):
0000055.727: Topic:
0000055.728: Testir_server/ac/stat/protocol
0000055.732: Payload:
0000055.734: DAIKIN2
0000055.736: MQTT publish (with retain):
0000055.739: Topic:
0000055.741: Testir_server/ac/stat/power
0000055.745: Payload:
0000055.747: on
0000055.748: MQTT publish (with retain):
0000055.752: Topic:
0000055.753: Testir_server/ac/stat/mode
0000055.757: Payload:
0000055.759: cool
0000055.761: MQTT publish (with retain):
0000055.764: Topic:
0000055.766: Testir_server/ac/stat/temp
0000055.769: Payload:
0000055.771: 21.0
0000055.773: MQTT publish (with retain):
0000055.776: Topic:
0000055.778: Testir_server/ac/stat/fanspeed
0000055.782: Payload:
0000055.784: Min
0000055.785: MQTT publish (with retain):
0000055.789: Topic:
0000055.790: Testir_server/ac/stat/swingv
0000055.794: Payload:
0000055.796: Low
0000055.797: MQTT publish (with retain):
0000055.801: Topic:
0000055.803: Testir_server/ac/stat/light
0000055.806: Payload:
0000055.808: on
0000055.810: MQTT publish (with retain):
0000055.813: Topic:
0000055.815: Testir_server/ac/stat/filter
0000055.818: Payload:
0000055.820: on
0000055.822: MQTT publish (with retain):
0000055.825: Topic:
0000055.827: Testir_server/ac/stat/clean
0000055.831: Payload:
0000055.832: on
0000055.834: MQTT publish (with retain):
0000055.837: Topic:
0000055.839: Testir_server/ac/stat/beep
0000055.843: Payload:
0000055.844: on
0000055.846: Difference in common A/C state detected.
0000055.852: MQTT publish:
0000055.853: Topic:
0000055.855: Testir_server/ac/stat/json
0000055.858: Payload:
0000055.860: {"protocol":"DAIKIN2","model":-1,"power":"On","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}
0000066.753: Incoming IR message sent to MQTT:
0000066.753: 53,0x11DA270001D6C290290C8004C016240000BED58111DA270000382A00B0000006600000C190603B
0000066.755: Converting inbound IR A/C message to common A/C
0000066.760: MQTT publish (with retain):
0000066.763: Topic:
0000066.765: Testir_server/ac/stat/protocol
0000066.769: Payload:
0000066.771: DAIKIN2
0000066.773: MQTT publish (with retain):
0000066.776: Topic:
0000066.778: Testir_server/ac/stat/mode
0000066.782: Payload:
0000066.783: cool
0000066.785: MQTT publish (with retain):
0000066.789: Topic:
0000066.790: Testir_server/ac/stat/temp
0000066.794: Payload:
0000066.796: 21.0
0000066.797: MQTT publish (with retain):
0000066.801: Topic:
0000066.802: Testir_server/ac/stat/fanspeed
0000066.806: Payload:
0000066.808: Min
0000066.810: MQTT publish (with retain):
0000066.813: Topic:
0000066.815: Testir_server/ac/stat/swingv
0000066.819: Payload:
0000066.821: Low
0000066.822: MQTT publish (with retain):
0000066.826: Topic:
0000066.827: Testir_server/ac/stat/light
0000066.831: Payload:
0000066.833: on
0000066.834: MQTT publish (with retain):
0000066.838: Topic:
0000066.839: Testir_server/ac/stat/filter
0000066.843: Payload:
0000066.845: on
0000066.846: MQTT publish (with retain):
0000066.850: Topic:
0000066.852: Testir_server/ac/stat/clean
0000066.855: Payload:
0000066.857: on
0000066.859: MQTT publish (with retain):
0000066.862: Topic:
0000066.864: Testir_server/ac/stat/beep
0000066.867: Payload:
0000066.869: on
0000066.871: Difference in common A/C state detected.
0000066.876: MQTT publish:
0000066.878: Topic:
0000066.879: Testir_server/ac/stat/json
0000066.883: Payload:
0000066.885: {"protocol":"DAIKIN2","model":-1,"power":"Off","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}

crankyoldgit, 2020-02-01:
I concur, looking...

crankyoldgit, 2020-02-01:
Still hunting for what is causing it, but a new version with some hopeful (not really) changes and more debugging.

Please try reproducing your last test sequence.

sheppy99, 2020-02-01:
Looks like we have no 

>Testir_server/ac/stat/power on 

but we do do have 

>Testir_server/ac/stat/power off

debug output

>0000049.249: Unsubscribed OK from:
0000049.249: Testir_server/ac/stat/+
0000049.249: MQTT publish (with retain):
0000049.250: Topic:
0000049.250: Testir_server/ac/stat/protocol
0000049.253: Payload:
0000049.255: DAIKIN2
0000049.257: MQTT publish (with retain):
0000049.261: Topic:
0000049.262: Testir_server/ac/stat/power
0000049.266: Payload:
0000049.268: on
0000049.269: MQTT publish (with retain):
0000049.273: Topic:
0000049.274: Testir_server/ac/stat/mode
0000049.278: Payload:
0000049.280: auto
0000049.282: MQTT publish (with retain):
0000049.285: Topic:
0000049.287: Testir_server/ac/stat/temp
0000049.290: Payload:
0000049.292: 21.0
0000049.294: MQTT publish (with retain):
0000049.297: Topic:
0000049.299: Testir_server/ac/stat/fanspeed
0000049.303: Payload:
0000049.305: Min
0000049.306: MQTT publish (with retain):
0000049.310: Topic:
0000049.311: Testir_server/ac/stat/swingv
0000049.315: Payload:
0000049.317: Highest
0000049.319: MQTT publish (with retain):
0000049.322: Topic:
0000049.324: Testir_server/ac/stat/light
0000049.328: Payload:
0000049.330: on
0000049.331: MQTT publish (with retain):
0000049.334: Topic:
0000049.336: Testir_server/ac/stat/filter
0000049.340: Payload:
0000049.342: on
0000049.343: MQTT publish (with retain):
0000049.347: Topic:
0000049.348: Testir_server/ac/stat/clean
0000049.352: Payload:
0000049.354: on
0000049.355: MQTT publish (with retain):
0000049.359: Topic:
0000049.361: Testir_server/ac/stat/beep
0000049.364: Payload:
0000049.366: on
0000049.367: Difference in common A/C state detected.
0000049.373: MQTT publish:
0000049.374: Topic:
0000049.376: Testir_server/ac/stat/json
0000049.380: Payload:
0000049.382: {"protocol":"DAIKIN2","model":-1,"power":"On","mode":"Auto","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Highest","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}
0000049.405: Sending common A/C state via IR.
0000049.866: The state was recovered from MQTT broker.
[... 74 lines cut ...]
0000097.234: {"protocol":"DAIKIN2","model":-1,"power":"On","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}
0000103.818: Incoming IR message sent to MQTT:
0000103.818: 53,0x11DA27000100C390290C8004C016240000BED5AC11DA270000382A00B0000006600000C190603B
0000103.819: Converting inbound IR A/C message to common A/C
0000103.825: MQTT publish (with retain):
0000103.828: Topic:
0000103.830: Testir_server/ac/stat/power
0000103.834: Payload:
0000103.835: off
0000103.837: MQTT publish (with retain):
0000103.841: Topic:
0000103.842: Testir_server/ac/stat/mode
0000103.846: Payload:
0000103.848: cool
0000103.849: MQTT publish (with retain):
0000103.853: Topic:
0000103.854: Testir_server/ac/stat/swingv
0000103.858: Payload:
0000103.860: Low
0000103.862: Difference in common A/C state detected.
0000103.867: MQTT publish:
0000103.869: Topic:
0000103.870: Testir_server/ac/stat/json
0000103.874: Payload:
0000103.876: {"protocol":"DAIKIN2","model":-1,"power":"Off","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}

crankyoldgit, 2020-02-02:
No worries. Just remember when looking at it, it's going to remember what ever the last state was after reboot (via the mqtt server), so if it was "on" in MQTT prior to reboot, it will not notice a new "on" IR message after it has restored it's old state etc.

Also, there is an updated version with even more debug info.

sheppy99, 2020-02-02:
Now it seems to have a Testir_server/as/stat/power on but no off
json works correctly

>0000306.302: Incoming IR message sent to MQTT:
0000306.302: 53,0x11DA270001144390290C8004C016240000BEC53011DA270000392A00B0000006600000C190603C
0000306.303: Converting inbound IR A/C message to common A/C
0000306.309: MQTT publish (with retain):
0000306.312: Topic:
0000306.314: Testir_server/ac/stat/power
0000306.317: Payload:
0000306.319: on
0000306.321: MQTT publish (with retain):
0000306.324: Topic:
0000306.326: Testir_server/ac/stat/temp
0000306.329: Payload:
0000306.331: 21.0
0000306.333: Difference in common A/C state detected.
0000306.339: MQTT publish:
0000306.340: Topic:
0000306.342: Testir_server/ac/stat/json
0000306.345: Payload:
0000306.347: {"protocol":"DAIKIN2","model":-1,"power":"On","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}
0000308.926: Incoming IR message sent to MQTT:
0000308.926: 53,0x11DA27000114C390290C8004C016240000BED5C011DA270000382A00B0000006600000C190603B
0000308.927: Converting inbound IR A/C message to common A/C
0000308.933: MQTT publish (with retain):
0000308.936: Topic:
0000308.938: Testir_server/ac/stat/temp
0000308.941: Payload:
0000308.943: 21.0
0000308.945: Difference in common A/C state detected.
0000308.951: MQTT publish:
0000308.952: Topic:
0000308.954: Testir_server/ac/stat/json
0000308.957: Payload:
0000308.959: {"protocol":"DAIKIN2","model":-1,"power":"Off","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}

sheppy99, 2020-02-02:
I'll now redo that with the extra debugging code from "add more debugging code"

crankyoldgit, 2020-02-02:
That'd be great

crankyoldgit, 2020-02-02:
Interesting. It thinks the temp is changing but it isn't/shouldn't be.

sheppy99, 2020-02-02:
Here's the trace

>0000046.411: Incoming IR message sent to MQTT:
0000046.411: 53,0x11DA2700011D4390290C8004C016240000BEC53911DA270000392A00B0000006600000C190603C
0000046.413: Converting inbound IR A/C message to common A/C
0000046.418: MQTT publish (with retain):
0000046.422: Topic:
0000046.423: Testir_server/ac/stat/protocol
0000046.427: Payload:
0000046.429: DAIKIN2
0000046.431: sendClimate debug:
0000046.434: Prev power is off
0000046.436: Next power is on
0000046.439: MQTT publish (with retain):
0000046.443: Topic:
0000046.444: Testir_server/ac/stat/power
0000046.448: Payload:
0000046.450: on
0000046.451: MQTT publish (with retain):
0000046.455: Topic:
0000046.456: Testir_server/ac/stat/mode
0000046.460: Payload:
0000046.462: cool
0000046.464: MQTT publish (with retain):
0000046.467: Topic:
0000046.469: Testir_server/ac/stat/temp
0000046.472: Payload:
0000046.474: 21.0
0000046.476: MQTT publish (with retain):
0000046.479: Topic:
0000046.481: Testir_server/ac/stat/fanspeed
0000046.485: Payload:
0000046.487: Min
0000046.488: MQTT publish (with retain):
0000046.492: Topic:
0000046.493: Testir_server/ac/stat/swingv
0000046.497: Payload:
0000046.499: Low
0000046.501: MQTT publish (with retain):
0000046.504: Topic:
0000046.506: Testir_server/ac/stat/light
0000046.509: Payload:
0000046.511: on
0000046.513: MQTT publish (with retain):
0000046.516: Topic:
0000046.518: Testir_server/ac/stat/filter
0000046.522: Payload:
0000046.523: on
0000046.525: MQTT publish (with retain):
0000046.528: Topic:
0000046.530: Testir_server/ac/stat/clean
0000046.534: Payload:
0000046.536: on
0000046.537: MQTT publish (with retain):
0000046.541: Topic:
0000046.542: Testir_server/ac/stat/beep
0000046.546: Payload:
0000046.548: on
0000046.549: Difference in common A/C state detected.
0000046.555: MQTT publish:
0000046.556: Topic:
0000046.558: Testir_server/ac/stat/json
0000046.561: Payload:
0000046.563: {"protocol":"DAIKIN2","model":-1,"power":"On","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}
0000046.687: Receiving data by MQTT topic:
0000046.687: Testir_server/ac/stat/protocol
0000046.687: with payload:
0000046.687: DAIKIN2
0000046.688: Checking for channel number in Testir_server/ac/stat/protocol
0000046.695: Channel = 0
[... 151 lines cut ...]
0000051.829: Prev power is on
0000051.831: Next power is off
0000051.834: MQTT publish (with retain):
0000051.837: Topic:
0000051.839: Testir_server/ac/stat/power
0000051.843: Payload:
0000051.845: off
0000051.846: Difference in common A/C state detected.
0000051.852: MQTT publish:
0000051.853: Topic:
0000051.855: Testir_server/ac/stat/json
0000051.858: Payload:
0000051.860: {"protocol":"DAIKIN2","model":-1,"power":"Off","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}

but we have more weirdness. From AC OFF to COOLING

0000236.631: Incoming IR message sent to MQTT:
0000236.631: 53,0x11DA270001214390290C8004C016240000BEC53D11DA270000392A00B0000006600000C190603C
0000236.633: Converting inbound IR A/C message to common A/C
0000236.638: sendClimate debug:
0000236.641: Prev power is on
0000236.644: Next power is on
0000236.646: NO difference in common A/C state detected.

Aircon page says its on cooling, mqtt json and power topics say its off

crankyoldgit, 2020-02-02:
You're hitting the IR button a little to fast.
You need to wait till you see:
`0000049.859: Finished listening for previous state.`

It takes about a minute for it to settle down (from flicking the power on to the ESP) and recover the previous state.

Once you see that line, you should be okay-ish to play pressing the remote buttons.

i.e. Please re-do. I don't want to have to try to unscramble that mess.

P.S. New revision to download and try too.

sheppy99, 2020-02-02:
OK, latest revision, after that message, 'cool' button then 'off' button. End result is both json power field and power topic is set to ON

>0000076.425: Incoming IR message sent to MQTT:
0000076.425: 53,0x11DA2700012F4390290C8004C016240000BEC54B11DA270000392A00B0000006600000C190603C
0000076.426: Converting inbound IR A/C message to common A/C
0000076.432: sendClimate debug:
0000076.434: Prev power is off
0000076.437: Next power is on
0000076.440: MQTT publish (with retain):
0000076.443: Topic:
0000076.445: Testir_server/ac/stat/power
0000076.449: Payload:
0000076.450: on
0000076.452: Difference in common A/C state detected.
0000076.458: MQTT publish:
0000076.459: Topic:
0000076.461: Testir_server/ac/stat/json
0000076.464: Payload:
0000076.466: {"protocol":"DAIKIN2","model":-1,"power":"On","mode":"Cool","use_celsius":"On","temp":21,"fanspeed":"Min","swingv":"Low","swingh":"Off","quiet":"Off","turbo":"Off","econo":"Off","light":"On","filter":"On","clean":"On","beep":"On","sleep":-1}
0000080.449: Incoming IR message sent to MQTT:
0000080.449: 53,0x11DA27000130C390290C8004C016240000BED5DC11DA270000382A00B0000006600000C190603B
0000080.450: Converting inbound IR A/C message to common A/C
0000080.456: sendClimate debug:
0000080.458: Prev power is off
0000080.461: Next power is off
0000080.464: NO difference in common A/C state detected.

crankyoldgit, 2020-02-02:
I think I have an idea ...

sheppy99, 2020-02-02:
One other question, should it recognise 0.5 degree increments in temperature such as 20.5C ? I think it does send them???

crankyoldgit, 2020-02-02:
Okay .. try that one. Fingers crossed! (an actual attempt to fix it, not just **moar** debugging!)

crankyoldgit, 2020-02-02:
> One other question, should it recognise 0.5 degree increments in temperature such as 20.5C ? I think it does send them???

Yes & no. IRMQTTServer should recognise and accept them. If the A/C unit accepts them, it will use half degrees. However, if it doesn't. You'll probably get a duplicate IR message being created.

e.g.
Assume AC only understands whole degrees.

1. ESP set to 25C.
2. ESP sends 25C command to A/C
3. ESP gets set to 25.5C.
4. ESP thinks 25.5C is different to 25C, so send.
5. ESP (re)sends 25C command to A/C because A/C doesn't know half degrees.
6. ESP gets set to 26C
7. ESP thinks 26C is different to 25.5C, so send.
8. ESP sends 26C command to A/C.

sheppy99, 2020-02-02:
The power command now works, thanks very much. I'll turn off debug, and if it works I'll run it overnight and check that its working.
Regarding the degrees the unit should accept them, indeed I've been sending it as IR for some time now and I found it in the protocol, the decoding side seems to show whole degrees, but as I can't send them from the remote I can only see what OpenHAB is sending via IR.

crankyoldgit, 2020-02-02:
**_Huz-Forking-zah!_**

sheppy99, 2020-02-02:
Yep, sorry for dropping that one on ya! Very well done

crankyoldgit, 2020-02-02:
I'll update the code in a bit with a new version with some of the debug/other changes backed out.
i.e. So it is just the fix, nothing else.

sheppy99, 2020-02-02:
It is, I plan on migrating my Daikin2 from a full hex string generated by OpenHAB to individual items bound to a common framework topic such as fanspeed, mode and power which is how I use the rest of my units. For now I've got a receiver monitoring whats sent and recording the json so I can be sure its doing what I think it is. It's is how I discovered the bug, tomorrow I'll work on recoding the AC control rule in OpenHAB. Hopefully everything will function perfectly as the other channels do. I use 5 sending channels plus a 6th for testing on the main ESP.
Or to put it another way, hopefully that is all :-)

crankyoldgit, 2020-02-02:
FYI, that branch is now _"Just the fix & a version number bump"_, all your settings & extra debug messages are gone.

sheppy99, 2020-02-02:
tell you what, I’ll run it as is overnight, give me a link to whatever is the final version and I’ll update it tomorrow, testing as I go

crankyoldgit, 2020-02-02:
Same link as before. And that sounds like a wise plan.

sheppy99, 2020-02-03:
Everything seemed to work fine last night with the decoding side, I've now switched using the framework for sending to the main Daikin2 unit. I'll run it overnight and then update the firmware to the #1038 version tomorrow. I'm following the only change so many things at once principle!

crankyoldgit, 2020-02-03:
Sounds reasonable to me

crankyoldgit, 2020-02-04:
Did you test the final version?

sheppy99, 2020-02-04:
Nope, I recently broke my foot and yesterday together with this morning was spent enjoying the joys of Fracture Clinics. I’ll take a look at the logs later and assuming all is well I’ll update

crankyoldgit, 2020-02-04:
Eeeek! Rest up and have a swift recovery.

sheppy99, 2020-02-05:
Thanks.
I've loaded it and initial tests suggest all is OK. I have my suspicions that Mold Guard on the Daikin2 isn't working but I'll see what happens in the morning after its been on all night. As I've only recently switched to using the framework with Daikin2 this could be down to anything. Once I've worked out whats going on I'll file an issue if necessary. It may be something weird like it doesn't work if the time isn't set as I set the time when I was using the full hex string. In the next few days I'll compare everything at a hex level and see whats different.

crankyoldgit, 2020-02-05:
FWIW, Mold Guard (i.e. `setMold()`) on `DAIKIN2` is tied to the "clean" setting.

sheppy99, 2020-02-05:
Yes in my distant memory I remember something like that and much confusion about it’s many modes. I’ll see what I set in the original he’s stream and see what’s missing. Is there an easy way of tracing what byte / bit in the stream is matched to what function? Given the high possibility of a screwup by myself I’d rather trace everything through before I bother you if possible.

crankyoldgit, 2020-02-05:
> Is there an easy way of tracing what byte / bit in the stream is matched to what function?

You will have to go through the source code I'm afraid.
e,g, For `IRDaikin2::setMold()` https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Daikin.cpp#L915-L917
which indicates it's using `remote_state[8]` thus it's the **9th** byte in the array. i.e. the 17th & 18th nibble(s)/character(s) in the hex code.

sheppy99, 2020-02-05:
Thanks, the current issue with the decoding can be closed as all seems well. I'm working on the Mold Guard problem now.

closed 2020-02-06 by crankyoldgit

sheppy99, 2020-02-06:
FYI I think I've found the Mold Guard problem and its a combination of the way Daikin works together with the way the Framework works. If the Daikin receives another OFF command of any sort after its turned OFF and in Mold Guard it goes straight to OFF. As my rule was adapted from the old create a hex string way of working it was sending OFF, then Temperature and then Mode as separate topics which in turn generated 3 IR commands. This meant the unit went from ON > Mold Guard > OFF as it was closing its vanes, and as beep is turned OFF I didn't notice it. I found it by comparing the hex codes and once I found only minor differences I created a rule that used just the power topic which worked correctly.

crankyoldgit, 2020-02-06:
If you send via separate mqtt topics, you'll simulate the send individual commands approach. I think if you send via the single json topic, I think it will send it in a single IR message. That may help you out

[4 more replies]

PR #1038 (2020-02-06): IRMQTTServer: Fix bug when receiving an A/C message and not re-transmitting it.
* When an incoming IR A/C message changed the state, we didn't always update the "previous" state once it was dealt with.
* Misc. code style & typo fixes.
* Version bump.
* Extra unit tests used along the way.

Fixes #1035
commit 0fc83792 (2020-02-06): IRMQTTServer: Fix bug when receiving an A/C message and not re-transmitting it. (#1038)

* When an incoming IR A/C message changed the state, we didn't always update the previous state once it was dealt with.
* Misc. code style & typo fixes.
* Version bump.
* Extra unit tests used along the way.

Fixes #1035