crankyoldgit/IRremoteESP8266 #1036: Coolix saga continues

jimmys01, 2020-02-01:
```
Protocol  : COOLIX
Code      : 0xB25F7C (24 Bits)
Mesg Desc.: Power: On, Mode: 3 (Heat), Fan: 2 (Med), Temp: 22C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4766, 4382,  642, 1556,  644, 492,  614, 1560,  620, 1584,  640, 496,  592, 516,  616, 1558,  642, 496,  590, 518,  590, 1608,  614, 496,  616, 494,  614, 1584,  614, 1584,  612, 496,  592, 1608,  614, 494,  614, 1584,  618, 492,  616, 1582,  592, 1580,  642, 1582,  616, 1582,  592, 1606,  592, 1582,  640, 496,  614, 1584,  640, 470,  638, 470,  590, 518,  640, 470,  638, 470,  614, 494,  614, 1560,  640, 1584,  592, 1608,  610, 1584,  610, 1560,  636, 496,  610, 494,  614, 1580,  638, 468,  612, 494,  588, 516,  612, 494,  612, 492,  614, 1582,  588, 1608,  612, 5184,  4708, 4464,  534, 1634,  582, 552,  552, 1642,  532, 1662,  532, 572,  532, 574,  532, 1662,  532, 572,  536, 570,  554, 1640,  534, 546,  580, 552,  532, 1662,  554, 1642,  534, 572,  558, 1634,  556, 550,  554, 1640,  534, 572,  554, 1640,  554, 1614,  560, 1662,  554, 1640,  534, 1662,  534, 1636,  558, 572,  532, 1662,  554, 550,  578, 526,  532, 574,  532, 572,  532, 572,  534, 570,  554, 1614,  560, 1662,  554, 1616,  580, 1614,  560, 1636,  580, 550,  534, 572,  534, 1636,  558, 572,  532, 572,  534, 572,  532, 572,  534, 548,  578, 1640,  554, 1614,  580};  // COOLIX B25F7C
uint64_t data = 0xB25F7C;
```

```
Library   : v2.7.2

Protocol  : COOLIX
Code      : 0xB25F7C (24 Bits)
Mesg Desc.: Power: On, Mode: 3 (Heat), Fan: 2 (Med), Temp: 22C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4482, 4518,  614, 1662,  590, 538,  590, 1656,  588, 1658,  590, 536,  588, 538,  588, 1656,  590, 548,  588, 538,  588, 1658,  590, 538,  588, 538,  588, 1630,  616, 1632,  614, 514,  612, 1658,  590, 538,  586, 1632,  596, 554,  588, 1632,  614, 1656,  590, 1630,  614, 1632,  614, 1658,  588, 1658,  590, 538,  588, 1658,  590, 536,  590, 536,  588, 538,  588, 538,  590, 540,  588, 510,  618, 1656,  590, 1656,  588, 1658,  588, 1658,  588, 1658,  574, 550,  588, 538,  588, 1632,  616, 536,  588, 512,  616, 538,  588, 538,  588, 538,  588, 1656,  590, 1634,  614, 4990,  4554, 4418,  614, 1658,  590, 536,  588, 1632,  614, 1656,  590, 538,  588, 538,  588, 1656,  590, 540,  588, 538,  588, 1656,  588, 538,  588, 538,  590, 1656,  590, 1634,  612, 538,  588, 1658,  590, 538,  590, 1630,  614, 538,  588, 1658,  590, 1656,  590, 1656,  590, 1656,  590, 1658,  590, 1656,  588, 538,  588, 1658,  590, 536,  590, 536,  590, 512,  614, 538,  588, 540,  588, 538,  588, 1632,  614, 1634,  614, 1656,  590, 1656,  590, 1656,  590, 510,  616, 538,  588, 1632,  616, 538,  588, 538,  588, 538,  590, 536,  588, 538,  588, 1632,  616, 1656,  590};  // COOLIX B25F7C
uint64_t data = 0xB25F7C;
```

The TOP one is recognized by the AC and it is a capture from the remote. 
The bottom one is a replay from the library and the AC is not accepting it.
Oh mighty @crankyoldgit show us the way and release us from this damn protocol

crankyoldgit, 2020-02-01:
Oh padawan, [I find your lack of faith disturbing!](https://www.youtube.com/watch?v=Zzs-OvfG8tE)

[I am altering the timings. Pray I don't alter it any further.](https://www.youtube.com/watch?v=jsW9MlYu31g)

@jimmys01 Please come to the Dark Side\<tm> and download PR #1037 / https://github.com/crankyoldgit/IRremoteESP8266/tree/ohcoolixwhywhywhy and test it out on a real device.

I've tweaked the timings subtly to better match the _real device_ raw data you captured. It **may** work, it may not.

It could easily be the _gap_ between repeats on the Coolix device. That value is completely a made-up guess. (i.e. approx 100ms).

**If and only if** that branch/PR #1037 doesn't work, then it is probably that gap we need to tweak next.
e.g. https://github.com/crankyoldgit/IRremoteESP8266/blob/ohcoolixwhywhywhy/src/ir_Coolix.cpp#L93
```
  space(kDefaultMessageGap);
```

The best way to determine this gap would be with an Oscilloscope. The next best would be some horrible hacks to extended the default capture timeout and force the library to capture the "repeat" message along with the first message, then analyse the raw data. This may or may not work, hence I don't want to try this unless we absolutely have to.
Next is to give me the full output of IRrecvDumpV2 (inc the timestamp & version messages it produces) for the first and second(repeat) message. I can then make an attempt to make a more educated **guess** at the gap between repeat messages etc.

Alternatively, you could try changing that line to see if you can get it to work/respond.
e.g.
`space(30000);`, `space(50000);`, `space(70000);`, or `space(150000);`
Note: `kDefaultMessageGap` is 100,000.

That is, **_as is_** that inter-message gap may be too long or too short for a real device to understand it, and we know Coolix requires a repeat (i.e. two messages)

So go forth and find me those droids!

jimmys01, 2020-02-03:
I can confirm it works (beeps) now!
Thanks a million!

crankyoldgit, 2020-02-03:
Thanks for confirming

closed 2020-02-03 by crankyoldgit

crankyoldgit, 2020-02-26:
The changes mentioned above have been included in the latest [IRremoteESP8266 v2.7.4](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.4) release.

PR #1037 (2020-02-03): Coolix: Subtle protocol timing adjustments
Try to adjust timings to better match real-world captured data and work on a Coolix device.

For @jimmys01

Fixes #1036 (Hopefully)
commit 4600d916 (2020-02-03): Coolix: Subtle protocol timing adjustments (#1037)

Based on real world capture data.

Fixes #1036