crankyoldgit/IRremoteESP8266 #1044: No luck with ESP32 WROOVER

camelator, 2020-02-07:
Hi,
Something looks wrong, please can someone help me?

I have an ESP32 WROOVER
I am using studio code
I am using PlateformeIO
library version is 2.7.3
I have this set of IR receiver and IR transmitter

https://www.aliexpress.com/item/1972257325.html?spm=a2g0o.productlist.0.0.9b887edcMVSYEa&algo_pvid=68ef9d5f-cb8c-48f5-ab79-ab8c6a8b9e54&algo_expid=68ef9d5f-cb8c-48f5-ab79-ab8c6a8b9e54-0&btsid=42793d68-d110-41d4-807f-fd274e25bb9b&ws_ab_test=searchweb0_0,searchweb201602_1,searchweb201603_53

I am sure the IR receiver working fine as I just test it with my old arduino mega with IRRemote lib. I can successfully receive the right NEC codes from my remote.

Switching to my ESP32 :
Firstly I remove the previous IRRemote lib and install IRremoteESP8266
Then I started a new project
Copy and paste the code from :
https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV2
compilation works fine and upoad as well

my wires are as follow :

Usb to ESP32
external power to IR receiver
grounds are linked (IR Receiver / ESP32)
Signal is on pin 14 (like the example)
No changes on the code.
It looks working  as I have output on console when I press a key on the remote
but unfortunately there are random data every time I press a remote, and protocol is UNKNOWN instead of NEC

Timestamp : 000010.011
Library   : v2.7.3

Protocol  : UNKNOWN
Code      : 0xCD57532C (37 Bits)
uint16_t rawData[73] = {9132, 4432,  640, 496,  644, 470,  628, 532,  614, 490,  646, 492,  604, 526,  606, 486,  616, 506,  598, 1630,  228, 436,  1614, 634,  1618, 654,  1598, 666,  1628, 616,  1628, 606,  1632, 648,  1592, 626,  488, 610,  492, 216,  218, 2050,  230, 1820,  636, 502,  646, 474,  652, 508,  616, 496,  646, 1626,  618, 1626,  606, 532,  614, 492,  644, 1610,  602, 1808,  412, 1862,  434, 1612,  636, 39830,  9154, 2158,  664};  // UNKNOWN CD57532C

Any idea what can be wrong or what can I do to debug it?

crankyoldgit, 2020-02-08:
The "arduino mega" uses and handles 5V from memory. The ESP family uses 3.3V. It may accept 5V inputs on the GPIOs, I'm not sure. Anyway, the point I'm trying to make is to make sure you are powering the IR receiver module correctly and it's output(s) so that it is all in spec. e.g. Some IR receiver devices require being powered at 5V to work reliably. Some work fine at 3V or both. Some Don't work well with or at 3V. Your's could be the later.

The AliExpress link you provided indicates these boards require 5V. Also, the transmit board is fairly rubbish if it only has a 1.3M range. I have found through other peoples experience that these types of boards are rubbish in general. You are much better off using the discrete individual parts and making a circuit like our examples on the wiki page(s).

The reason your message was not successfully decoded is because the data collected doesn't match what an NEC message is supposed to look like.
e.g. The following values are either too short (small) or too long (large) for the timings require for a valid NEC message. _..., 1630, **228**, 436, ..., 610, 492, **216**, **218**, **2050**, **230**, 1820, 636, ..._. Something is causing these odd values external to the ESP chip.

Using the incorrect voltage for your IR receiver module is one thing that can cause this, other factors like having the IR source to close to the receiver is another. Check out our [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#receiving-an-ir-signal) for some suggestions.

In short, the ESP/code/library is doing everything correctly. The problem lays on the circuit/IR receiver demodulator/hardware, or environment side. Concentrate your efforts there.

crankyoldgit, 2020-02-12:
Friendly chase up. How are you going?

camelator, 2020-02-13:
Hi,
I had the same conclusion, issue is from the IR receiver. Work fine with +5v, no luck with +3.3v

closed 2020-02-13 by camelator