crankyoldgit/IRremoteESP8266 #1045: Adding Display and Virus Doctor for Samsung AC

rg2k, 2020-02-08:
Requesting an enhancement to add the ability to:

1. Turn the display on/off
2. Turn VirusDoctor on/off

rg2k, 2020-02-10:
I have the code to add the ability for the Samsung AC to:

1. Turn the display on/off
2. Turn VirusDoctor on/off

Can I get a Pull request for this, or just paste the code here in this thread?

crankyoldgit, 2020-02-10:
Yes, you can send us a Pull Request (PR). See https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request or https://help.github.com/en/desktop/contributing-to-projects/creating-a-pull-request and do a [google search](https://www.google.com/search?q=how+to+create+a+PR+in+gi) etc.

If the code change is small, you can just past the _diff_ here too.

BTW, what model is your Samsung A/C **and** remote?

**and**

What is _"VirusDoctor"_? Is it an Ion mode, a filter mode, or a cleaning mode? (Which of those describes it best?)

rg2k, 2020-02-11:
Great. I've completed the PR.

Samsung AC model: AR09FSSDAWKNFA with remote: DB63-03556X003.

VirusDoctor is basically an Ion and filter mode combined into one feature.

crankyoldgit, 2020-02-12:
Thanks! Per PR review comments ... Yeah, Virus Doctor is just an Ion Filter. We've named this type of mode/function on the A/Cs just "Ion" or "Filter" so, lets use "set/getIon()" etc instead.
There are some marketing/sales people working for these manufacturers who just love to come up with different names for the same function just so they can say their product is different/better/more speshal! ;-) I'll add code after this is merged to support it via the common A/C api/framework, where it will be under "filter" for VirusDoctor/Ion, and "light" for Display etc.

I'll add that model info to the list of supported devices! Thanks again.

closed 2020-02-14 by crankyoldgit

crankyoldgit, 2020-02-26:
The changes mentioned above have been included in the latest [IRremoteESP8266 v2.7.4](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.4) release.

PR #1049 (2020-02-14): SamsungAc: Add full support for light & ion.
* Add full support for Light (Display) & Ion (VirusDoctor) for Samsung 
A/Cs
* Adjust unit tests accordingly.
* Add unit tests for Light & Ion respectively.
* Update common A/C api

Fixes #1045
Ref: #1048

FYI @rg2k
commit 39c9f86e (2020-02-14): SamsungAc: Add full support for light & ion. (#1049)

* Add full support for Light (Display) & Ion (VirusDoctor) for Samsung 
A/Cs
* Adjust unit tests accordingly.
* Add unit tests for Light & Ion respectively.
* Update common A/C api

Fixes #1045
Ref: #1048

FYI @rg2k