crankyoldgit/IRremoteESP8266 #1052: #include <algorithm> need a lot of SRAM memory!!

mahdinikan75, 2020-02-23:
hi all!
#include <algorithm> that need to #include <arduinoSTL> allocate 420 Byte of SRAM for global varible and that is near 20% of arduino nano SRAM size!! and this is a problem
can anyone tell me what function of that library used in this library and can i remove that function from this library or not?

NiKiZe, 2020-02-23:
This lib is for ESP8266 or ESP32
Unlikely to work good on nano

mahdinikan75, 2020-02-23:
> This lib is for ESP8266 or ESP32
> Unlikely to work good on nano

yes i know
i changed it to work on nano because i want to use its big A/C remote database

crankyoldgit, 2020-02-23:
`min()` & `max()` are the primary functions that this library uses from `<algorithm>`.
You could probably remove the `#include <algorithm>` line safely. It's mainly there to keep the linter happy.

mahdinikan75, 2020-02-24:
> `min()` & `max()` are the primary functions that this library uses from `<algorithm>`.
> You could probably remove the `#include <algorithm>` line safely. It's mainly there to keep the linter happy.

thanks for your response
i comment all of #include <algorithm> in all library file then when i compile library there is lot of error like "expected unqualified-id before '(' token"  on the code  " #define max(a,b) ((a)>(b)?(a):(b))"
how can i correct this errors??

crankyoldgit, 2020-02-24:
You could create a proper `max()` function rather than rely on the `#define` version that Arduino uses by default.
e.g.
```
// Note: Untested code, but probably works for most cases.
// Probably needs to be put into IRremoteESP8266.h to catch everything.
uint64_t max(const uint64_t a, const uint64_t b) {
  if (a > b) return a;
  return b;
}
```

But what you are attempting is really in "unsupported" territory trying to back port this to a Nano etc etc. I'll try to help where I can but it's beyond the scope of this library.
e.g. See the last line of the [README.md](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/README.md) file that displays when you go to https://github.com/crankyoldgit/IRremoteESP8266

i.e. _"As of v2.0, the library was almost entirely re-written with the ESP8266's resources in mind."_

crankyoldgit, 2020-02-24:
Alternatively ...
Following https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Arduino.h#L255-L256

Try:
```
#undef max
#undef min
#define min(a,b) ({ decltype(a) _a = (a); decltype(b) _b = (b); _a < _b? _a : _b; })
#define max(a,b) ({ decltype(a) _a = (a); decltype(b) _b = (b); _a > _b? _a : _b; })
```
etc ...

Or ask the Arduino folk why "#include <algorithm>" wastes so much SRAM. :-/

crankyoldgit, 2020-02-24:
Or: (from http://www.cplusplus.com/reference/algorithm/max/)
```
template <class T> const T& max (const T& a, const T& b) {
  return (a<b)?b:a;
```

crankyoldgit, 2020-02-27:
@mahdinikan75 How did you go with those options?

crankyoldgit, 2020-02-29:
It's the weekend. Time for another friendly ping.

mahdinikan75, 2020-02-29:
> @mahdinikan75 How did you go with those options?

unfortunately it's not good for me and i can't give good response from that solution.
thanks a lot for your consistency

crankyoldgit, 2020-03-10:
I'm not sure what you mean by that. I've offered all the porting advice I can easily think of at this stage.
Not sure what I can do to help you further.

crankyoldgit, 2020-03-17:
As I've not heard anything back from you in two+ weeks, I'm marking this issue closed/stale until you respond.

closed 2020-03-17 by crankyoldgit