crankyoldgit/IRremoteESP8266 #1053: Denon PMA-50 does not react on any IR codes

robot22, 2020-02-25:
### Version/revision of the library used

Built: Feb 25 2020 17:51:40
IR Lib Version: 2.7.3

### Expected behavior

I try to control "Denon PMA-50" by sending IR commands via MQTT.
The expected bahaviour is that the device reacts on the IR signals.

### Actual behavior

The device doesn't react on any IR code which is send by the ESP8266.

Basically I did following tests. 
- Send IR Command by vendor remote. IRremoteESP8266 received it and also MQTT shows the received code (e.g. 17,0x2A4C0E80800E,48 for Power ON/OFF)
- Send the exact same code via MQTT to ESP -> The Denon Amplifier doen't react on it
- Build second ESP8266 to send IR commands. The Denon Amplifier doesn't react while the original ESP received the same command as it was seen from the vendor IR remote.
- Tried to send codes via GlobalCache and direct input on the ESP webserver -> nothing worked
- Also tried different distances to the Amplifier with same result.

The setup in general is working fine. I can control my LG TV as expected by sending NEC Commands.

### Steps to reproduce the behavior

Installed IRremoteESP8266 - IRMQTTServer serveral times to different Chips. All do have the same bahaviour.

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)

Cabling of the ESP with IR LED and IR Receiver is like the published ones on (https://github.com/crankyoldgit/IRremoteESP8266/wiki#ir-sending)

I use also standard PINs

### I have followed the steps in the [Troubleshooting Guide]
Yes

### Has this library/code previously worked as expected for you?
Yes, controlling other devices works like expected. 

### Other useful information

Since I use also a univeral remote that learned the IR Code from the vendor remote. I'm struggling about what can be the issue. (Codes from orginal remote, univeral remote and ESP looks the same when I check IR received messages)

Hope you can help me.

crankyoldgit, 2020-02-25:
Can you please capture the original remote message(s) using IRrecvDumpV2.ino and copy the entire output here. I.e. I'll need the raw data to see what is going on.

Have you tried specifying/adding some repeats to it?

robot22, 2020-02-25:
Hi,
thanks a lot for your quick response and the hint on specifiying repeats.

With the IRrecvDump I could see that the original remote sends 2 repeats. (Which I could not see on MQTT received output somehow).
I got it working, anyways it's little unstable since it does not recognize it all time. Will play around with it little more during the next days.

Just for reference, this is the messages comming from the remote.

```
00:02:02.649 -> Timestamp : 000568.838
00:02:02.649 -> Library   : v2.7.3
00:02:02.649 -> 
00:02:02.649 -> Protocol  : DENON
00:02:02.649 -> Code      : 0x2A4C0E80800E (48 Bits)
00:02:02.649 -> uint16_t rawData[99] = {3410, 1620,  484, 368,  482, 366,  482, 1204,  484, 366,  482, 1202,  484, 368,  482, 1202,  526, 330,  482, 366,  482, 1202,  484, 368,  482, 368,  480, 1202,  484, 1202,  484, 368,  482, 372,  522, 326,  482, 368,  482, 368,  482, 368,  482, 1202,  486, 1200,  484, 1202,  486, 378,  480, 1204,  484, 368,  482, 368,  482, 368,  482, 368,  480, 366,  482, 368,  482, 376,  482, 1202,  484, 368,  482, 366,  482, 368,  482, 368,  482, 368,  482, 368,  528, 330,  480, 368,  524, 326,  480, 368,  482, 368,  482, 1204,  482, 1202,  530, 1156,  484, 368,  454};  // DENON 2A4C0E80800E
00:02:02.696 -> uint32_t address = 0x2A4C;
00:02:02.696 -> uint32_t command = 0xE80800E;
00:02:02.696 -> uint64_t data = 0x2A4C0E80800E;
00:02:02.696 -> 
00:02:02.696 -> 
00:02:02.790 -> Timestamp : 000568.970
00:02:02.790 -> Library   : v2.7.3
00:02:02.790 -> 
00:02:02.790 -> Protocol  : DENON
00:02:02.790 -> Code      : 0x2A4C0E80800E (48 Bits)
00:02:02.790 -> uint16_t rawData[99] = {3410, 1620,  484, 366,  482, 368,  482, 1202,  484, 366,  482, 1204,  484, 366,  530, 1156,  484, 372,  482, 368,  480, 1184,  504, 368,  482, 368,  482, 1204,  484, 1202,  484, 368,  482, 372,  482, 368,  482, 366,  482, 368,  482, 368,  482, 1202,  484, 1202,  484, 1202,  484, 376,  482, 1202,  526, 324,  482, 366,  482, 366,  482, 368,  482, 368,  482, 368,  482, 376,  480, 1202,  484, 368,  482, 368,  480, 368,  482, 368,  482, 368,  482, 366,  526, 330,  482, 368,  482, 368,  482, 368,  480, 368,  482, 1202,  486, 1204,  484, 1202,  486, 366,  454};  // DENON 2A4C0E80800E
00:02:02.852 -> uint32_t address = 0x2A4C;
00:02:02.852 -> uint32_t command = 0xE80800E;
00:02:02.852 -> uint64_t data = 0x2A4C0E80800E;
00:02:02.852 -> 
00:02:02.852 -> 
```

Again thanks a lot.

closed 2020-02-25 by robot22