crankyoldgit/IRremoteESP8266 #1057: Symphony fan remote without Bit mark

soumaxetuirk, 2020-03-06:
I have a 5 button Fan remote which sends code without bit marks. only two timing values across the signal. Is there any protocol present for such type signal ??
button 1 
`

1296,412,1294,386,420,1224,1322,390,1290,390,420,1224,452,1220,1314,394,420,1222,482,1190,480,1192,452,7960,1290,420,1290,390,418,1226,1318,394,1262,416,420,1224,454,1220,1292,416,422,1222,450,1222,452,1218,454,8208,1296,414,1292,386,418,1226,1292,422,1260,420,424,1218,454,1226,1312,390,420,1224,454,1220,482,1186,454,7960,1318,392,1264,416,392,1252,1318,394,1288,394,418,1224,452,1224,1292,422,414,1222,458,1214,450,1222,454
--

`

Button 2:
`

1102,756,926,590,178,1470,1094,548,1184,514,324,1378,294,1294,302,1394,228,1498,1114,604,240,1350,346,8084,1090,536,1196,512,302,1370,1116,570,1194,592,246,1294,328,1396,270,1406,350,1268,1220,454,364,1344,354,8442,972,594,1168,486,354,1344,1142,546,1164,446,368,1406,292,1322,326,1348,352,1352,1162,486,380,1316,356,8006,1218,516,1168,588,178,1392,1192,486,1172,568,302,1344,300,1332,342,1318,378,1322,1190,490,352,1300,348
--

`

crankyoldgit, 2020-03-07:
Please supply:
a) The make and model of the fan **and** the remote.
b) The full text of the captures via this library's `IRrecvDumpV2` program.

This data you've provided indicates to me it uses "~1200usec mark & ~400usec space" or "~400us mark & ~1200us space" for the values of a bit. There have been similar protocols that have used this encoding that the library supports.

There also appears to be an ~8000us gap between parts of the message.

It looks simple enough to add support for this protocol.

soumaxetuirk, 2020-03-07:
Manufacturer: Symphony
Model : Air Cooler 3Di
Remote : (No model number provided)
after ~8000us the same data gets repeated. It gets repeated almost for a sec, depending on how long I pressed the button.
So the data before first ~8000us gap is enough for device to understand (tested with raw signal 38khz).

POWER: 
1308,368,1310,368,448,1222,1310,372,1308,400,442,1198,472,1198,1284,396,444,1224,470,1200,470,1198,472
FAN SPEED:
1288,424,1250,424,442,1228,1262,422,1260,420,420,1226,446,1248,418,1252,446,1226,1260,420,420,1222,448
SWING:
1290,418,1286,392,422,1248,1284,400,1294,386,422,1248,422,1250,444,1228,424,1248,446,1226,1258,420,446
3D COOL :
1290,420,1262,418,420,1224,1320,392,1294,386,444,1202,450,1222,448,1224,1292,416,418,1224,474,1200,448

crankyoldgit, 2020-03-07:
Please paste the entire exact text please.

crankyoldgit, 2020-03-07:
and just for a short press. If it includes repeats, that's fine. We want to capture/emulate the remote exactly. e.g. The protocol may have a default minimum repeat. It may still work with a single part of the message but it's best to be exactly the same.

crankyoldgit, 2020-03-07:
@soumaxetuirk I've made an attempt to add support for the protocol based on the info you've provide so far.

Can you please download and try the branch referenced in PR #1058 / https://github.com/crankyoldgit/IRremoteESP8266/tree/symphony **and** let me know how it goes for you?

soumaxetuirk, 2020-03-08:
> and just for a short press. If it includes repeats, that's fine. We want to capture/emulate the remote exactly. e.g. The protocol may have a default minimum repeat. It may still work with a single part of the message but it's best to be exactly the same.

the signals gets repeated till I remove my finger.The minimum is 24bits and max I have seen is 512bits. So that's makes 2 times of data with a 8000us gap.
I am testing it right now.Apologies for the delay.

soumaxetuirk, 2020-03-08:
Signals sent using SYMPHONY protocols are working perfectly with the device but not decoded with the same protocol. (it is showing as unknown protocol, 12bit ). Where my remote signals were decoded successfully.
Ex-
Signal sent by SYMPHONY protocol: (device registered the signal)
{"Protocol":"UNKNOWN","Bits":12,"Hash":"0x39D41DC6"}
Signal sent by Original Remote: 
 ({"Protocol":"SYMPHONY","Bits":11,"Data":"0x137","DataLSB":"0x80EC","Repeat":0})

_Sending command has no issues. tried every button's decoded data with more then 50 times testing._

crankyoldgit, 2020-03-08:
> So that's makes a 2 times of data with a 8000us gap as minimum length.

Thanks. I'll adjust the protocol to match that.

> Signals sent using SYMPHONY protocols are working perfectly with the device but not decoded with the same protocol. (it is showing as unknown protocol, 12bit ). Where my remote signals were decoded successfully.
Ex-
Signal sent by SYMPHONY protocol: (device registered the signal)
{"Protocol":"UNKNOWN","Bits":12,"Hash":"0x39D41DC6"}
Signal sent by Original Remote:
({"Protocol":"SYMPHONY","Bits":11,"Data":"0x137","DataLSB":"0x80EC","Repeat":0})

I'm not sure where you are getting that output, but it isn't something the library produces natively.
**PLEASE** include the full output of `IRrecvDumpV2.ino`. Without the full raw data I can't be certain what is going on.

The Unit Test include a self-decode so I'm not sure what is going wrong.
(See [TEST(TestDecodeSymphony, SyntheticSelfDecode)](https://github.com/crankyoldgit/IRremoteESP8266/blob/symphony/test/ir_Symphony_test.cpp#L28-L44)

soumaxetuirk, 2020-03-09:
I don't have ESP8266 chip with me right now. I am using few sonoff devices with custom compiled tasmota on it.So I don't have the IRRecvDump. 
But I sent a signal remote first then sent exactly the same using raw data send & SYMPHONY Protocol. Checked the timings manually. high & low are exactly same.remote data was decoded but others are not. please have a look.

`{"IrReceived":{"Protocol":"SYMPHONY","Bits":11,"Data":"0x137","DataLSB":"0x80EC","Repeat":0,"RawData":[1286,396,1286,396,446,1226,1288,400,1294,388,444,1228,446,1226,1286,396,444,1226,448,1226,468,1204,448,7968,1286,396,1286,396,470,1202,1286,400,1286,396,446,1224,448,1226,1288,396,446,1226,472,1200,448,1226,472]}}

{"IrReceived":{"Protocol":"UNKNOWN","Bits":24,"Hash":"0xBF4AF983","Repeat":0,"RawData":[1222,430,1250,436,410,1242,1274,446,1274,412,414,1266,410,1264,1252,436,436,1240,410,1270,438,1242,410,8012,1254,434,1258,432,438,1240,1276,416,1252,434,438,1234,412,1270,1250,442,412,1264,438,1238,410,1270,438]}}`

crankyoldgit, 2020-03-10:
Thanks for the data. The value _"1222,430,1250,436,410,1242,1274,**446**,1274,..."_ was just out of the expected range for a _space_ value. I've tweaked what it expects and it now matches the data immediately above, and existing test cases I've already included.

closed 2020-03-10 by crankyoldgit

soumaxetuirk, 2020-03-10:
Tested the recent commit.The protocol is now working perfectly.Thank you guys!

crankyoldgit, 2020-04-09:
FYI, This has been included in the new v2.7.5 release of the library.

PR #1058 (2020-03-10): Add support for `SYMPHONY` 11 bit protocol.
* basic `sendSymphony()`/`decodeSymphony()` added.
* Update basic housekeeping to support new protocol.
* Unit tests for new protocol.

Fixes #1057
commit 26d30f96 (2020-03-10): Add support for `SYMPHONY` 11 bit protocol. (#1058)

* basic `sendSymphony()`/`decodeSymphony()` added.
* Update basic housekeeping to support new protocol.
* Unit tests for new protocol.

Fixes #1057