crankyoldgit/IRremoteESP8266 #1059: libraries for AC AUX

akdri, 2020-03-09:
Hi, I have libraries to control the air conditioners of the AUX brand.
I don't know how to adapt these libraries to work with Tasmota.
I attach the liberias in case someone could integrate them into Tasmota.

Thank you

[AUX_IR.zip](https://github.com/crankyoldgit/IRremoteESP8266/files/4308178/AUX_IR.zip)

crankyoldgit, 2020-03-10:
Is this your own code or have you copied it from somewhere?

It suspiciously looks like: https://github.com/ToniA/arduino-heatpumpir/blob/master/AUXHeatpumpIR.cpp etc

If you have one of these devices, please give me the make and model numbers of the A/C Unit **and** the remote.

Also, some "rawData" captures of the protocol using `IRrecvDumpV2`

i.e. Please follow these steps: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

crankyoldgit, 2020-03-14:
@akdri Friendly chase up to answer the questions and to provide some data.

crankyoldgit, 2020-03-26:
@akdri Another chase up to answer the questions and to provide some data. It's been two weeks and we haven't heard from you. I may close this issue due to lack of a response.

crankyoldgit, 2020-04-02:
Closing due to no response from the reporter.

closed 2020-04-02 by crankyoldgit