crankyoldgit/IRremoteESP8266 #1060: Help decoding Hitachi PC-LH3B

MinePlugins, 2020-03-10:
Hello, I attemp to decode the IR of my remote for the A/C with PC-LH3B remote,
I created a [Google Sheet](https://docs.google.com/spreadsheets/d/1x82Cb3-GtOMM5oTVOFdNY9mqwLnJ-qElhOj8MId7mrA/edit?usp=sharing) like recommended and user Auto_analyse_raw_data.py:
`Found 275 timing entries.
Potential Mark Candidates:
[3396, 516]
Potential Space Candidates:
[1660, 1268, 428]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 3396
kHdrSpace  = 1660
kBitMark   = 476
kOneSpace  = 1236
kZeroSpace = 395

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+1000000000001000000000000000001011111101111111110000000011000111001110001001000101101110000100001110111111111100000000111011010001001011
  Bits: 136`

Can I have some help or explaination for this protocole

crankyoldgit, 2020-03-11:
Looking at your spreadsheet, it looks like the bit order needs to be switched. I worked that out from your temperature range data. i.e. https://docs.google.com/spreadsheets/d/1x82Cb3-GtOMM5oTVOFdNY9mqwLnJ-qElhOj8MId7mrA/edit#gid=0&range=Q6:Q18

For starters, lets get the basic(simple) sending/decoding of a message working.
So can you please collect the `rawData` output from `IRrecvDumpV2.ino` for this protocol and put it here.
i.e. Let's follow the steps here: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

MinePlugins, 2020-03-11:
Here is the rawData for POWER OFF `uint16_t rawData[371] = {3422, 1660,  464, 1264,  438, 426,  438, 402,  486, 426,  440, 402,  462, 402,  488, 428,  438, 402,  460, 404,  488, 426,  440, 424,  436, 428,  462, 1240,  466, 398,  462, 404,  486, 426,  438, 402,  464, 400,  490, 400,  464, 426,  438, 402,  488, 428,  436, 428,  462, 376,  486, 404,  474, 416,  438, 400,  488, 402,  462, 426,  438, 426,  458, 1246,  464, 426,  436, 1244,  488, 1266,  434, 1268,  436, 1242,  486, 1242,  462, 1240,  464, 426,  462, 1266,  438, 1242,  462, 1240,  514, 1214,  466, 1236,  462, 1266,  464, 1264,  438, 1266,  438, 1240,  488, 400,  466, 424,  440, 402,  486, 404,  462, 402,  464, 402,  488, 426,  438, 402,  462, 402,  488, 1264,  438, 1242,  460, 428,  462, 428,  436, 1264,  440, 1240,  488, 1240,  464, 1240,  460, 402,  490, 424,  440, 1264,  438, 1242,  512, 402,  438, 426,  466, 398,  464, 1266,  440, 400,  462, 402,  488, 1264,  438, 402,  462, 402,  482, 432,  438, 1264,  436, 404,  488, 1240,  462, 1264,  438, 402,  486, 1246,  456, 1266,  438, 1238,  488, 402,  462, 1240,  462, 402,  512, 402,  438, 428,  438, 426,  462, 430,  434, 428,  438, 402,  512, 376,  462, 1240,  464, 1264,  458, 1270,  436, 1242,  462, 1240,  486, 1242,  460, 1242,  462, 1242,  486, 1240,  464, 1240,  464, 1238,  492, 1264,  436, 1266,  440, 400,  488, 426,  436, 430,  434, 430,  460, 404,  462, 426,  438, 402,  488, 426,  436, 1264,  466, 1238,  462, 1242,  462, 1266,  438, 1238,  490, 1264,  438, 426,  438, 1240,  486, 406,  462, 402,  462, 400,  488, 428,  436, 426,  438, 402,  484, 1268,  438, 426,  436, 1242,  488, 1266,  436, 402,  462, 402,  502, 386,  464, 1240,  464, 1240,  488, 426,  438, 426,  438, 402,  488, 1240,  462, 1266,  438, 1242,  486, 428,  440, 424,  438, 1266,  460, 1268,  438, 1264,  438, 404,  486, 428,  438, 426,  438, 402,  490, 400,  462, 426,  436, 402,  490, 426,  438, 1240,  462, 1268,  460, 1268,  434, 1266,  436, 1240,  514, 1238,  438, 1240,  488, 1240,  460, 406,  464, 426,  436, 402,  488, 402,  462, 402,  462, 1264,  462, 404,  462, 1266,  434, 1268,  464, 1264,  438, 1242,  464, 1238,  488, 1266,  438, 402,  462, 1268,  458, 430,  410};`

AND here for POWER ON 30C LOUVER 1 MODE HEAT `uint16_t rawData[371] = {3400, 1682,  436, 1242,  462, 402,  488, 430,  428, 434,  434, 430,  434, 428,  460, 404,  486, 378,  488, 430,  434, 432,  434, 426,  438, 454,  432, 1270,  438, 402,  462, 450,  440, 428,  436, 402,  488, 428,  432, 406,  460, 404,  462, 454,  434, 428,  438, 430,  456, 430,  434, 404,  464, 400,  482, 434,  432, 408,  460, 428,  440, 426,  462, 1240,  486, 402,  436, 1292,  436, 1242,  466, 1238,  488, 1266,  436, 1242,  486, 1242,  464, 426,  462, 1216,  486, 1216,  488, 1240,  462, 1242,  488, 1238,  440, 1290,  436, 1242,  488, 1214,  516, 1214,  460, 404,  488, 376,  488, 426,  436, 404,  488, 374,  490, 428,  436, 426,  462, 402,  438, 446,  444, 1242,  488, 1238,  438, 454,  434, 430,  464, 1240,  462, 1266,  466, 1212,  490, 1238,  462, 404,  460, 428,  464, 1238,  464, 1240,  486, 402,  464, 402,  460, 406,  484, 1242,  466, 374,  492, 400,  486, 1216,  488, 400,  464, 404,  458, 430,  466, 1238,  464, 426,  438, 1240,  490, 1212,  490, 402,  486, 1242,  464, 1238,  464, 1240,  490, 374,  490, 1212,  492, 400,  484, 404,  464, 398,  466, 402,  484, 382,  488, 374,  520, 396,  434, 406,  490, 1236,  466, 1238,  486, 1216,  490, 1214,  488, 1240,  486, 1218,  488, 1214,  490, 1264,  434, 1268,  466, 1236,  466, 1264,  434, 1244,  488, 1214,  490, 400,  490, 376,  488, 376,  488, 424,  434, 432,  464, 376,  488, 426,  460, 406,  464, 1212,  516, 1238,  462, 1216,  490, 1238,  464, 1240,  486, 1244,  466, 398,  462, 1266,  436, 426,  466, 400,  462, 404,  460, 404,  490, 374,  490, 400,  484, 1218,  490, 372,  488, 1240,  490, 1214,  490, 372,  488, 402,  488, 378,  490, 1236,  466, 1266,  432, 406,  490, 398,  494, 374,  458, 1270,  464, 1236,  466, 1268,  432, 430,  464, 374,  490, 1238,  486, 1216,  490, 1214,  492, 398,  484, 380,  490, 398,  466, 400,  484, 380,  488, 400,  464, 402,  488, 398,  466, 1236,  466, 1240,  484, 1244,  462, 1238,  464, 1240,  486, 1216,  490, 1214,  516, 1238,  462, 400,  466, 400,  466, 400,  460, 404,  488, 1238,  466, 400,  484, 404,  466, 1238,  466, 1238,  486, 1242,  466, 1214,  492, 1238,  486, 378,  490, 1238,  466, 1262,  466, 374,  462}; //0x874AB821 (186 Bits)`

crankyoldgit, 2020-03-11:
Thanks. I'll knock up something to handle the basic part of the protocol in a few.

crankyoldgit, 2020-03-11:
@MinePlugins Can you please download & test this branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/hitachi_PC-LH3B

It should decode your remote and covert it to a 23-byte array/hex code, Hopefully with the bits in the correct order (LSB). e.g. The byte containing the temp should noticeably increase linearly.

If it does, you can update your spreadsheet to use that data. It might some things easier to decode etc.

MinePlugins, 2020-03-11:
Thank's I'll try soon !!

MinePlugins, 2020-03-11:
`uint16_t rawData[371] = {3424, 1662,  438, 1266,  462, 404,  490, 400,  436, 428,  456, 408,  486, 404,  462, 378,  512, 402,  436, 402,  488, 404,  460, 428,  440, 400,  486, 1216,  484, 406,  460, 430,  464, 376,  486, 404,  488, 402,  436, 428,  460, 404,  488, 378,  488, 376,  488, 428,  438, 428,  436, 428,  462, 410,  454, 428,  436, 402,  488, 402,  464, 424,  438, 1292,  464, 376,  486, 1216,  486, 1242,  462, 1266,  438, 1266,  436, 1294,  434, 1244,  486, 378,  514, 1238,  432, 1272,  440, 1262,  436, 1244,  488, 1266,  466, 1234,  440, 1290,  436, 1268,  438, 1240,  488, 376,  512, 404,  436, 430,  460, 428,  434, 430,  438, 400,  490, 374,  514, 400,  436, 428,  436, 1296,  430, 1270,  460, 402,  440, 400,  486, 1268,  436, 1240,  490, 1214,  486, 1268,  462, 400,  438, 428,  460, 1268,  436, 1266,  460, 380,  512, 378,  488, 376,  460, 1268,  460, 428,  436, 428,  436, 1242,  486, 428,  436, 404,  488, 428,  438, 1240,  488, 376,  488, 1216,  486, 1242,  488, 402,  438, 1266,  458, 1246,  488, 1214,  514, 402,  436, 1242,  488, 402,  436, 404,  486, 428,  436, 428,  436, 428,  462, 428,  462, 402,  436, 402,  486, 1242,  486, 1216,  462, 1292,  434, 1244,  488, 1216,  476, 1252,  464, 1264,  440, 1240,  462, 1292,  434, 1268,  438, 1240,  464, 1240,  482, 1272,  434, 430,  460, 378,  488, 402,  486, 402,  440, 452,  436, 428,  436, 402,  464, 402,  488, 1240,  488, 1216,  488, 1266,  434, 1244,  490, 1214,  488, 1216,  488, 402,  486, 1216,  464, 426,  490, 376,  486, 402,  462, 378,  486, 430,  462, 376,  486, 1242,  458, 432,  438, 1264,  488, 1242,  436, 404,  486, 378,  488, 376,  488, 1266,  436, 1268,  438, 454,  430, 408,  488, 402,  460, 1216,  486, 1268,  464, 1214,  486, 430,  436, 428,  436, 1266,  440, 1240,  514, 1240,  436, 404,  486, 404,  460, 428,  464, 400,  438, 402,  486, 428,  436, 428,  488, 350,  486, 1268,  462, 1240,  436, 1290,  434, 1270,  436, 1242,  488, 1214,  488, 1242,  488, 1216,  488, 402,  458, 406,  486, 376,  488, 376,  510, 1244,  446, 420,  436, 428,  464, 1240,  462, 1240,  490, 1214,  486, 1268,  436, 1266,  462, 428,  434, 1270,  452, 1226,  514, 350,  488};  // HITACHI_AC184
                                                                                                       uint8_t state[23] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xE6, 0x19, 0x89, 0x76, 0x01, 0xFE, 0x3F, 0xC0, 0x2F, 0xD0, 0x18, 0xE7, 0x00, 0xFF, 0x90, 0x6F};`

crankyoldgit, 2020-03-11:
Interesting. The "UNKNOWN" one is significantly shorter. I'll try to work out how best to handle it.

crankyoldgit, 2020-03-12:
I've updated that branch to hopefully handle the shorter (136 bit / 17 byte) version.
Let me know how it goes please.

MinePlugins, 2020-03-12:
WHen I change from 17C to 18C`Protocol  : HITACHI_AC184
                         Code      : 0x01100040BFFF00E31C897609F63FC017E8 (136 Bits)
                                                                                    uint16_t rawData[275] = {3396, 1662,  464, 1240,  462, 402,  490, 400,  440, 426,  460, 404,  488, 402,  460, 378,  514, 376,  486, 378,  490, 400,  460, 404,  488, 402,  460, 1216,  488, 402,  490, 376,  514, 350,  486, 428,  438, 402,  462, 452,  434, 404,  488, 400,  440, 400,  512, 378,  486, 376,  488, 426,  434, 430,  464, 402,  438, 400,  488, 404,  486, 402,  440, 1266,  488, 376,  488, 1214,  488, 1216,  486, 1242,  490, 1214,  490, 1240,  458, 1268,  438, 428,  488, 1238,  436, 1242,  486, 1216,  490, 1214,  488, 1240,  488, 1220,  458, 1292,  434, 1268,  438, 1266,  438, 402,  486, 404,  490, 374,  490, 426,  432, 430,  464, 400,  464, 376,  514, 400,  464, 1214,  490, 1264,  434, 432,  462, 376,  488, 376,  484, 1268,  456, 1224,  514, 1188,  486, 432,  436, 428,  432, 1268,  460, 1242,  490, 1238,  464, 376,  484, 406,  488, 400,  464, 1268,  430, 434,  436, 428,  460, 1216,  484, 432,  462, 400,  464, 404,  486, 1214,  488, 376,  490, 1238,  460, 1268,  464, 400,  466, 1240,  460, 1242,  490, 1238,  490, 376,  462, 1242,  490, 400,  462, 378,  484, 1244,  488, 400,  462, 404,  460, 404,  490, 374,  488, 376,  486, 1268,  438, 1240,  490, 400,  460, 1268,  464, 1238,  464, 1240,  492, 1236,  462, 1240,  462, 1242,  486, 1216,  490, 1212,  488, 1216,  486, 1266,  466, 374,  516, 348,  510, 378,  490, 374,  488, 404,  484, 404,  464, 400,  464, 378,  508, 1242,  464, 1240,  464, 1264,  436, 1266,  464, 1214,  488, 376,  512, 1244,  436, 402,  486, 404,  488, 376,  490, 398,  464, 376,  486, 428,  464, 1238,  466, 426,  436, 1266,  466, 1238,  492, 1210,  464};  // HITACHI_AC184
                                                                               uint8_t state[17] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xE3, 0x1C, 0x89, 0x76, 0x09, 0xF6, 0x3F, 0xC0, 0x17, 0xE8};`

crankyoldgit, 2020-03-12:
Wow. This is going to get very messy at this rate. This is yet another size/length. Can you please test/capture different operations on the remote and report (Inc raw data) for any different bit lengths you find. Rather than me adding them one by one. I may need to rethink how we try to manage this protocol. This is certainly in bizarre territory now.

MinePlugins, 2020-03-13:
```                                                                                                        IRrecvDumpV2 is now running and waiting for IR input on Pin 14
                                                                                                                                                                      Timestamp : 000005.746
                                                                                                                                                                                            Library   : v2.7.4

Protocol  : UNKNOWN
                   Code      : 0x2CF317C6 (122 Bits)
                                                    uint16_t rawData[243] = {3368, 1688,  460, 1244,  456, 434,  458, 406,  456, 408,  480, 410,  458, 406,  458, 406,  484, 406,  458, 406,  458, 432,  428, 464,  456, 380,  458, 1242,  458, 432,  456, 406,  486, 378,  430, 458,  456, 408,  458, 406,  456, 434,  456, 410,  454, 408,  454, 434,  460, 404,  458, 406,  454, 436,  456, 408,  458, 406,  458, 432,  456, 434,  430, 1246,  484, 404,  484, 1218,  458, 1244,  484, 1244,  460, 1244,  456, 1270,  430, 1274,  460, 404,  484, 1218,  486, 1244,  456, 1244,  456, 1248,  454, 1272,  460, 1268,  432, 1246,  484, 1244,  458, 1244,  462, 404,  456, 434,  458, 432,  432, 408,  456, 434,  460, 404,  460, 404,  484, 406,  432, 432,  458, 1298,  402, 432,  460, 404,  460, 404,  482, 1244,  460, 1270,  462, 1214,  456, 1272,  460, 404,  456, 1246,  486, 1244,  458, 1244,  458, 406,  454, 460,  434, 404,  460, 1268,  430, 460,  432, 406,  458, 1244,  456, 434,  458, 406,  456, 432,  460, 1268,  432, 406,  458, 1244,  484, 1272,  432, 430,  432, 1298,  378, 1298,  458, 1244,  484, 406,  460, 1242,  458, 406,  458, 1246,  456, 1272,  458, 406,  458, 432,  458, 404,  460, 404,  456, 408,  484, 1272,  432, 432,  432, 406,  454, 1272,  458, 1246,  458, 1244,  484, 1268,  436, 1242,  458, 1298,  402, 1274,  456, 1244,  458, 1244,  486, 1242,  458, 408,  486, 378,  454, 460,  434, 406,  458, 406,  484, 406,  458, 406,  458, 408,  452, 1302,  434, 1244,  430};```

```Protocol  : UNKNOWN
                   Code      : 0xD3A5A0BA (218 Bits)
                                                    uint16_t rawData[435] = {3364, 1668,  486, 1244,  458, 406,  456, 436,  430, 434,  430, 434,  456, 408,  484, 432,  428, 432,  414, 450,  458, 432,  432, 432,  430, 436,  458, 1270,  430, 408,  456, 408,  454, 462,  404, 460,  428, 408,  486, 404,  456, 408,  458, 406,  484, 406,  456, 434,  432, 408,  484, 406,  454, 436,  404, 432,  486, 430,  430, 408,  456, 432,  458, 1270,  406, 434,  456, 1246,  484, 1244,  456, 1274,  432, 1244,  486, 1268,  432, 1244,  454, 412,  454, 1272,  430, 1272,  456, 1274,  456, 1246,  456, 1244,  458, 1270,  458, 1246,  456, 1254,  450, 1246,  484, 408,  456, 434,  430, 408,  486, 430,  404, 460,  430, 408,  486, 404,  458, 406,  458, 406,  486, 404,  458, 432,  430, 1272,  456, 408,  458, 1244,  458, 1244,  456, 1274,  456, 1246,  456, 1270,  460, 1250,  426, 460,  404, 1270,  484, 432,  404, 462,  402, 460,  458, 1244,  456, 386,  452, 460,  458, 1244,  458, 406,  458, 406,  486, 430,  416, 1260,  458, 406,  484, 1270,  406, 1296,  430, 410,  482, 1246,  456, 1246,  456, 1246,  456, 434,  458, 1244,  456, 1272,  458, 408,  458, 1244,  456, 410,  484, 404,  454, 410,  456, 408,  484, 406,  458, 432,  430, 1246,  484, 406,  454, 1248,  458, 1244,  486, 1248,  428, 1270,  456, 1266,  464, 1244,  458, 1246,  456, 1246,  484, 1270,  430, 1246,  456, 434,  430, 460,  406, 432,  430, 432,  484, 432,  406, 432,  458, 406,  484, 432,  432, 1266,  434, 1274,  430, 1298,  428, 408,  456, 1246,  486, 430,  432, 1270,  404, 434,  484, 432,  430, 408,  456, 406,  460, 1268,  458, 408,  456, 1246,  486, 406,  456, 1244,  456, 1248,  456, 1300,  430, 408,  456, 408,  484, 432,  430, 434,  428, 434,  460, 430,  404, 434,  458, 406,  486, 1242,  458, 1270,  430, 1246,  484, 1244,  456, 1270,  432, 1248,  484, 1244,  456, 1246,  456, 408,  486, 404,  456, 408,  456, 432,  462, 430,  406, 458,  432, 432,  428, 436,  456, 1246,  456, 1246,  484, 1244,  454, 1268,  438, 1244,  488, 1266,  430, 1246,  458, 1244,  486, 1244,  430, 1298,  430, 434,  460, 1242,  458, 430,  430, 408,  486, 1242,  456, 434,  404, 462,  456, 430,  432, 1246,  456, 408,  486, 1270,  430, 1246,  460, 404,  482, 1246,  458, 406,  456, 408,  484, 404,  458, 1270,  430, 1274,  458, 432,  430, 408,  456, 406,  486, 1244,  458, 1242,  458, 1246,  484, 406,  458, 432,  430, 1246,  486, 1242,  456, 1272,  430, 434,  458, 406,  458, 406,  458, 406,  486, 404,  458, 406,  458, 430,  430, 460,  430, 1270,  430, 1248,  484, 1244,  456, 1244,  458, 1246,  484, 1244,  456, 1246,  460, 1244,  458};  // UNKNOWN D3A5A0BA```

crankyoldgit, 2020-03-14:
Thanks for that. I think that's a record. I've never come across an A/C remote which produces 5 distinct different length codes.

I'll try to work something out soon. I think I'm going to have to rename the protocol, removing "184" from it as it clearly doesn't use that bit length all the time.

Do you know what buttons/actions those three raw data entries correspond to per chance?

MinePlugins, 2020-03-14:
1 = cancel timer
2 = set timer / change timer
3 = change mode

crankyoldgit, 2020-03-15:
Thanks. I'll try to get to it soon.

crankyoldgit, 2020-03-17:
@MinePlugins Can you please download and test the latest version in that branch?

I think I've covered all the different sizes of messages you've reported.

Also, I think I've worked out the checksum alg. for this protocol.
After about byte 6, every second byte of the state is followed by that same byte, but bit-inverted.
e.g. `0x12 (0b00010010)` is followed by `0xEC (0b11101101)`.

Can you confirm that in your findings?

crankyoldgit, 2020-03-17:
Branch updated to add the integrity checks.

MinePlugins, 2020-03-19:
Hello due to the Corona virus a don't have the remote so I have only the data in sheet to work with, thanks you a lot

MinePlugins, 2020-03-19:
And yes it's what I'm finding

crankyoldgit, 2020-03-19:
Understood. Be safe. Update us when you can.

crankyoldgit, 2020-04-02:
Just a friendly ping to see if you've been able to get access to the remote yet.

MinePlugins, 2020-04-03:
No, no for the moment the remote is in a local that been closed until coronavirus end...

crankyoldgit, 2020-04-03:
As the PR to add support has been merged, I'm going to mark this closed for now. The virus will be with us for many months it seems. When you eventually get around to it, please give it a test and let me know. You can still update this issue and I'll see it. Or create a new one.

MinePlugins, 2020-04-03:
No problem sir

crankyoldgit, 2020-04-03:
Good luck with Covid-19 in your corner of the world

closed 2020-04-03 by crankyoldgit

crankyoldgit, 2020-04-09:
FYI, This has been included in the new v2.7.5 release of the library.

MinePlugins, 2020-12-15:
Hi, I'm back, I have the remote !

crankyoldgit, 2020-12-15:
Congrats. Try the latest version of the library. It should send & decode the hex values.
Follow the steps here: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#create-a-spreadsheet

When you've worked out how the functions map to the codes and collected the data etc, you can create a new issue and we can work on adding detailed support for it, together.

MinePlugins, 2020-12-23:
Thank's I will do that the next week.