crankyoldgit/IRremoteESP8266 #1061: I write program using hitachi ac liberary but it is not working

MarkEvens, 2020-03-13:
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Hitachi.h>
const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRHitachiAc ac(kIrLed);
void setup() {
  ac.begin();
  Serial.begin(115200);
  delay(200);
  ac.on();
  ac.setPower(true);
  ac.setMode(kHitachiAc424Cool);
  ac.setTemp(24);  // 24C
}
void loop() {
  ac.send();
  delay(5000);
}

this thing is not working remote is 13 byte code IRrecevDumpV2 show hitachi ac1 protocol

MarkEvens, 2020-03-13:
it is sending 28 byte code instead of 13 byte

crankyoldgit, 2020-03-13:
It's a different protocol. The protocol you want to use does not have have detailed support yet.
See #1056

crankyoldgit, 2020-03-13:
I.e. HITACHI_AC != HITACHI_AC1

MarkEvens, 2020-03-13:
my remote has same output as this sheet
https://docs.google.com/spreadsheets/d/10eKpJEWJppUYktPRLCcAIwzfFXjtkOZNyn1reh5MFfU/edit#gid=0

crankyoldgit, 2020-03-13:
That's still in progress. You can help by working out the checksum algorithm.

crankyoldgit, 2020-03-14:
In the meantime, you can still send with the `state[] = {...};` data provided by `IRrecvDumpV2` and using `irsend.sendHitachiAc1(state);`

To be clear, detail support (i.e. setting custom temp/modes/speeds/etc using the A/C class object) is dependent on #1056 being completed. Hence marking this a dup of #1056

closed 2020-04-09 by crankyoldgit

crankyoldgit, 2020-04-09:
FYI, This has been included in the new v2.7.5 release of the library.

commit b44de679 (2020-04-02): Add HITACHI_AC1 to Common A/C api.
* Add IRac::hitachi1
* Adjust `sendAc()`
* Adjust `resultAcToString()`
* Add/update unit tests.

Fixes #1056
Fixes #1061
commit 6f9f93f0 (2020-04-02): Add HITACHI_AC1 to Common A/C api.
* Add IRac::hitachi1
* Adjust `sendAc()`
* Adjust `resultAcToString()`
* Add/update unit tests.

Fixes #1056
Fixes #1061
PR #1072 (2020-04-09): Detailed support for `HITACHI_AC1` protocol.
* Add settings for `HITACHI_AC1` A/C protocol
  - Power, Fan, Mode, Temp, Swing, Sleep, On & Off Timers.
  - Kudos to @camilloaddis for working out the alg, & @soumaxetuirk for reverse engineering settings.
* Add support for it in the Common A/C api
* Add unit tests.

Fixes #1056
Fixes #1061
commit 23697d8e (2020-04-09): Detailed support for `HITACHI_AC1` protocol. (#1072)

* Add settings for `HITACHI_AC1` A/C protocol
  - Power, Fan, Mode, Temp, Swing, Sleep, On & Off Timers.
  - Kudos to @camilloaddis for working out the alg, & @soumaxetuirk for reverse engineering settings.
* Add support for it in the Common A/C api
* Add unit tests.

Fixes #1056
Fixes #1061