crankyoldgit/IRremoteESP8266 #1062: WindFree on/off option for Samsung AC

jacek-rozen, 2020-03-14:
Hi,

I am kindly requesting to add an option to turn on/ off WindFree option for Samsung A/C. 

This is a feature that allows to close a flap of the A/C when the device is still on. The flap has hundreds of small holes so the air is still blowing, but there is no wind. 

This feature is present in series Samsung Wind Free.

crankyoldgit, 2020-03-15:
Hi, You'll need to collect the `state[]` data to work out what bit/byte etc changes between the setting being On & Off. e.g. Read this https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#binary-settings (and read the rest of that document too will probably help)

You should only need the `state[]` data as Samsung support has already been added.

I can't add the support without that data. I don't have access to your device etc.
Also, please include the make & model numbers of your AC unit **and** the remote!

crankyoldgit, 2020-03-20:
@jacek-rozen Just a reminder. We can't move this forward until _**you**_ collect the data.

jacek-rozen, 2020-03-20:
@crankyoldgit : Collecting the HW to perform data capture.

crankyoldgit, 2020-03-21:
Cool. I'll chase up in a few weeks then.

jacek-rozen, 2020-03-29:
Hi, @crankyoldgit 

Starting from the hardware, below serial numbers:
- AC Samsung Wind Free, Model: AR12NXCXAWKXEU
- Remote Control: DB93-16761C.

In order to turn WindFree, which is to set the flap to the position when it covers the exhaust, the following code is sent from the remote:

uint8_t state[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xB2, 0xFE, 0x7B, 0x80, 0x31, 0xF0};

When I turn off this mode, it just resumes to the previous state like - FAN MODE, FAN LOW.

It seems that code sent is similar to one generated by kSamsungAcFanAuto, but then state[8] and state[10] are different. In case of sending this from software based on you lib, state[8] = 0xD2 and state[10] = 0x71 (as you see, remote has these set to state[8]=0xB2 and state[10]=0x7B).

crankyoldgit, 2020-03-30:
> Starting from the hardware, below serial numbers:
> 
> * AC Samsung Wind Free, Model: AR12NXCXAWKXEU
> * Remote Control: DB93-16761C.

@jacek-rozen You don't need to provide the serial numbers, just the _model_ numbers etc. So someone else with a similar device knows if it works with it or not.

Thanks, I'll take a closer look at the data you provided soon.

My questions based on your feedback so far:
1) Is that `state[14]` code always the same no mater what mode/temp/etc the unit is set to? i.e. is it a special code, or are the setting in every code sent?
2) Can you adjust the fan speed and still keep this setting (windfree) on?
3) Can you supply the two `state[14]` lines, one when you enable windfree, then one when you immediately turn it off?

Thanks in advance.

jacek-rozen, 2020-03-30:
As it comes to HW, models are mentioned, not serial - my bad.

Regarding your questions:

1. No, state[] is not always the same. I noticed that it is not possible to order wind free when Mode = {Heat, Auto}. Whem mode = Cool I can still adjust temperature, so it is not like it is constant all the time, however I cannot adjust FAN speed, it seems it is set to default/ AUTO speed.

Some examples below:

- MODE COOL, 23C WIND FREE = ON
uint8_t state[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xA2, 0xFE, **0x7B**, 0x70, 0x11, 0xF0};

- MODE COOL, 23C WIND FREE = OFF, FAN=TURBO
uint8_t state[14] =   {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0x92, 0xFE, **0x71**, 0x70,   0x1F, 0xF0};

- MODE COOL, 22C WIND FREE = ON
uint8_t state[14] =   {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xB2, 0xFE, **0x7B**, 0x60,   0x11, 0xF0};

- MODE COOL, 22C WINDFREE OFF, FAN = TURBO
uint8_t state[14] =   {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xA2, 0xFE, **0x71**, 0x60,   0x1F, 0xF0};

- MODE FAN, 24C WINDREE ON
uint8_t state[14] =   {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xB2, 0xFE, **0x7B**, 0x80,   0x31, 0xF0};

- MODE FAN, 24C WINDFREE OFF, FAN = LOW
uint8_t state[14] =   {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xC2, 0xFE, **0x71**, 0x80,   0x35, 0xF0};

It seems that remote_state[10] needs to be adjusted to value **0x7B** to be in WindFree Mode. Would you agree?

2. No, it is not possible to adjust fan speed, it is set to default/ AUTO.

3. These examples you will find in first point of this comment. 

Cheers.

crankyoldgit, 2020-03-31:
Thanks for that. That level of data/work is awesome.

Looking at the code, part of that state change overlaps with "Powerful" (Turbo) mode.
Can you please do some testing to see what happens with that mode & windfree? i.e. How do they react together etc?

crankyoldgit, 2020-03-31:
D'oh! I should look at your descriptions more. You include Turbo. My bad.

crankyoldgit, 2020-03-31:
@jacek-rozen  [Q] How does it interact with Quiet mode?

crankyoldgit, 2020-03-31:
Oh, Does swing get turned off in this mode?

crankyoldgit, 2020-03-31:
Hey @jacek-rozen I've added code to the `windfree` branch (https://github.com/crankyoldgit/IRremoteESP8266/tree/windfree) / PR #1071 

Can you please download and try it out, and give me feedback?

**Note:** Breeze (_WindFree_) is not controllable via the commonAC api. You will have to use the `IRSamsungAc` class directly. e.g. `ac.setBreeze(true);`, `ac.getBreeze();`
It will show up in `IRrecvDumpV2` though.

closed 2020-04-01 by crankyoldgit

jacek-rozen, 2020-04-05:
Assumptions are correct. I checked it with my A/C unit - works well.
I have still some findings on additional functionality (also regarding swing) to share, but I will open another ticket.

jacek-rozen, 2020-04-05:
And btw - thanks :)

crankyoldgit, 2020-04-09:
FYI, This has been included in the new v2.7.5 release of the library.

juniorbocchini, 2024-01-20:
Good morning, how do I use Breeze:"On" ?

daniele-athome, 2024-06-06:
Is there any way to use `setBreeze` through common API? Tasmota doesn't seem to support it.

crankyoldgit, 2024-06-07:
I just checked:
https://github.com/crankyoldgit/IRremoteESP8266/blob/9785cb910d2b8c81785afb58d76090cb57c5b2df/src/IRac.cpp#L2245-L2277
There is no call to `setBreeze()` in that code, so, basically, no. There isn't access via the common API

daniele-athome, 2024-06-17:
As a first patch for this, would it make sense to link fan speed mode "auto" to `setBreeze(true)` only for `SAMSUNG_AC`?

Or maybe create a new fan speed mode altogether? Although that would need patching Tasmota too...

NiKiZe, 2024-06-17:
> As a first patch for this, would it make sense to link fan speed mode "auto" to `setBreeze(true)` only for `SAMSUNG_AC`?
> 
> Or maybe create a new fan speed mode altogether? Although that would need patching Tasmota too...

I don't think this really fits with common..
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#user-content-The_IRac_class_used_by_IRMQTTServer_Tasmota_etc_doesnt_support_a_feature_of_my_AC_Why

If there is a common unused fan speed or mode not used by the Samsung code you could use that, but anything else will probably be a hack?

daniele-athome, 2024-06-17:
At a first read at the code, I believe the only unused fan speed by the Samsung module is `kMediumHigh`. But I don't know, using it for this sounds like a hack as well... it would really fit with `kAuto` except that is also a real fan speed mode for these A/Cs as well, so it would mean losing control of that to allow controlling breeze mode.

I'll research if something could be done on the Tasmota side maybe.

crankyoldgit, 2024-06-17:
While I understand you think this is a good idea (auto ==breeze), some might not. E.g. not all Samsung ACs support breeze etc.

Using an unused higher speed for breeze is not wise.

As @NiKiZe and the FAQ says, this interface is not for fine-grained control of Samsung device. 

You can always record and send your desired (fixed) state for the AC using the sendSamsungAC() method. 
Tasmota can do that. 
Otherwise, you can maintain your own branch of this library with your patch for your own/others use, or use ESPHome or write your own code for more control. 

In short, I will not add it to the IRAc API at this point.

daniele-athome, 2024-06-21:
For anyone coming here in the future: I ended up [forking Tasmota and making a little patch](https://github.com/daniele-athome/Tasmota/commit/development). It works for my needs (fan speed auto == breeze is fine for me) and maybe whoever else has a Samsung Windfree A/C :-)

PR #1071 (2020-04-01): SamsungAc: Add Breeze (Aka WindFree) control
* Add `setBreeze()`/`getBreeze()`
* Adjust Powerful(Turbo) control as it overlaps.
* Unit tests.
* Clean up & organise the `ir_Samsung.h` file.

Fixes #1062
commit a98de7f3 (2020-04-01): SamsungAC: Add Breeze (Aka WindFree) control (#1071)

* Add `setBreeze()`/`getBreeze()`
* Adjust Powerful(Turbo) control as it overlaps.
* Unit tests.
* Clean up & organise the `ir_Samsung.h` file.

Fixes #1062