crankyoldgit/IRremoteESP8266 #1064: Need Help of Coding about DAIKIN NEW MODULE FFN-C

chongkk, 2020-03-19:
Hi thanks about this project, and Really helps a lot.  I don't have any experience about c++, Then I found it's a difficult to add the support about new daikin model FFN-C.(https://www.daikin.com.my/daikin_products/ffn-c-fcn-f-series-r410a/)

I have found the LSB each positon meaning, (https://drive.google.com/file/d/1P_GdkeGFPwjZvXOO2efrbbnJEHZ7kBzo/view?usp=sharing), 
Can you help me add this support or advise about how to add this?

If need any more information please tell me. Thanks a lot.

crankyoldgit, 2020-03-19:
Instructions for how to add (or get added) an new A/C Protocol, read these two pages: 
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

Your Google Sheets spreadsheet does not allow anyone to read it. Please make it read-only to "everyone"/public. It would be good to allow anyone to add comments too.

If you follow the information on those pages, you should learn what is needed and how to collect the data to get it added.

chongkk, 2020-03-19:
Sorry About Set wrongly Access right. I modifted that.
https://drive.google.com/file/d/1P_GdkeGFPwjZvXOO2efrbbnJEHZ7kBzo/view?usp=sharing

crankyoldgit, 2020-03-19:
Thanks. Any chance you can convert the Excel spreadsheet to a Google Spreadsheet. That way if/when you update it, everyone can see it instantly etc, and add comments etc.

chongkk, 2020-03-19:
Hi thanks, 
I transfer that to google sheet and set everyone can comment that. Thanks.
https://docs.google.com/spreadsheets/d/1sxjLQCRLMFM1FQpttBXsye2JG5hHIe2BrKnKDuPV9Bw/edit?usp=sharing

crankyoldgit, 2020-03-19:
Excellent. I'll need some `rawData` examples captured via `IRrecvDumpV2`.
See: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol#capture-some-raw-ir-messages-from-the-remote

chongkk, 2020-03-19:
Hi thanks, I update some rawdata to the sheet already. If you need more information please tell me. Thanks very much.

crankyoldgit, 2020-03-19:
Hey @chongkk 
  Can you please download and test the following branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Daikin64

I've added basic support for `DAIKIN64` which should decode and send (via hex code only at this point) your A/C unit.

Can you let me know if `IRrecvDumpV2` is recompiled with this branch of the library if it detects it etc? and if you can control your a/c by using `irsend.sendDaikin64(hex_code_you_get_from_DumpV2);` works?

If so, I'll move on to adding detailed controls.

chongkk, 2020-03-19:
Hi I try download the Daikin64 branch, then use arduino to open /examples/IRrecvDumpV2/IRrecvDumpV2.ino, Then I compile and upload, When I Testing use the "Serial Monitor" it still show as "Unknown".
Is there any step did I make mistake? Thanks

crankyoldgit, 2020-03-19:
Without the full output from `IRrecvDumpV2` I can't tell what is going on.
My guess is your Arduino environment is still using the Release version of the library.

You can confirm this by changing this line: https://github.com/crankyoldgit/IRremoteESP8266/blob/Daikin64/src/IRremoteESP8266.h#L55
in the branch you downloaded from:
```
#define _IRREMOTEESP8266_VERSION_ "2.7.4"
```
to:
```
#define _IRREMOTEESP8266_VERSION_ "2.7.4-Daikin64"
```

That will show up in the output of your`IRrecvDumpV2` program after you re-compile/build/upload it, if it is using the new/downloaded branch.

chongkk, 2020-03-20:
Hi Thanks for your great work.
I tried and IRrecvDumpV2 output like this:
Timestamp : 000004.998
Library   : v2.7.4-Daikin64

Protocol  : DAIKIN64
Code      : 0x3C16101001301216 (64 Bits)
uint16_t rawData[137] = {9818, 9774,  9816, 9726,  4664, 2480,  390, 340,  386, 922,  386, 928,  388, 348,  390, 920,  388, 348,  382, 320,  416, 356,  388, 338,  388, 920,  390, 346,  384, 346,  388, 922,  388, 348,  386, 346,  384, 360,  386, 338,  388, 348,  386, 348,  388, 342,  384, 926,  386, 928,  390, 346,  388, 354,  384, 920,  384, 350,  388, 320,  414, 346,  386, 346,  382, 352,  386, 318,  414, 356,  384, 342,  388, 346,  386, 348,  388, 316,  414, 922,  386, 350,  386, 344,  388, 356,  384, 342,  388, 346,  382, 352,  388, 342,  386, 924,  390, 344,  388, 344,  388, 354,  390, 338,  386, 924,  388, 928,  390, 344,  386, 924,  386, 350,  388, 344,  388, 354,  386, 342,  390, 344,  388, 926,  384, 928,  386, 896,  418, 898,  418, 346,  388, 350,  388, 20284,  4646};  // DAIKIN64 3C16101001301216
uint64_t data = 0x3C16101001301216;

The code is correct, thanks again.
The AC is in my office and these days I can't go to office. I will test that when I back to office.

crankyoldgit, 2020-03-20:
Excellent news, I'll start work on the more advanced feature support shortly.

crankyoldgit, 2020-03-20:
@chongkk I've updated the branch again. It should now support all the features I can work out from your spreadsheet _except_ for the timer/clock ones. I'll add them later.

Please download and test, and let me know if you find any issues.

crankyoldgit, 2020-03-21:
@chongkk with the recent update to the branch (PR #1065), it now should support clock & timers, which I believe is everything on your spreadsheet.

Please test and let me know if it decodes correctly (and when you can, if it works sending to the real device).

chongkk, 2020-03-21:
@crankyoldgit Hi Thanks again, I think all future is correct. Because these days I am not able to back to  office to test control the real AC, but following the IR controller screen, I tested all function is correct. 

Timestamp : 001536.777
Library   : v2.7.4-Daikin64

Protocol  : DAIKIN64
Code      : 0x9416901011261216 (64 Bits)
Mesg Desc.: Power Toggle: Off, Mode: 2 (Cool), Temp: 16C, Fan: 1 (Auto), Turbo: Off, Quiet: Off, Swing(V): Off, Sleep: Off, Clock: 11:26, On Timer: Off, Off Timer: 10:00
uint16_t rawData[137] = {9870, 9742,  9846, 9748,  4646, 2480,  392, 336,  396, 914,  390, 896,  420, 316,  422, 888,  420, 324,  412, 308,  426, 346,  388, 310,  422, 890,  416, 342,  394, 314,  422, 916,  394, 314,  422, 308,  422, 320,  420, 306,  420, 890,  424, 892,  422, 314,  422, 312,  422, 892,  422, 314,  420, 322,  422, 882,  422, 312,  418, 316,  422, 310,  420, 890,  420, 314,  422, 308,  418, 324,  422, 306,  420, 314,  418, 316,  422, 310,  422, 888,  420, 342,  390, 314,  416, 326,  418, 310,  422, 312,  416, 318,  424, 308,  424, 910,  394, 318,  418, 314,  420, 902,  418, 314,  418, 892,  416, 926,  392, 314,  422, 916,  394, 312,  418, 314,  418, 324,  424, 304,  418, 316,  416, 924,  394, 314,  422, 886,  422, 342,  392, 310,  422, 896,  422, 20306,  4620};  // DAIKIN64 9416901011261216
uint64_t data = 0x9416901011261216;

Timestamp : 000374.215
Library   : v2.7.4-Daikin64

Protocol  : DAIKIN64
Code      : 0x8416109011062216 (64 Bits)
Mesg Desc.: Power Toggle: Off, Mode: 2 (Cool), Temp: 16C, Fan: 2 (High), Turbo: Off, Quiet: Off, Swing(V): Off, Sleep: Off, Clock: 11:06, On Timer: 10:00, Off Timer: Off
uint16_t rawData[137] = {9872, 9718,  9872, 9744,  4648, 2450,  420, 308,  418, 892,  422, 894,  422, 312,  418, 894,  416, 318,  418, 342,  394, 342,  396, 308,  418, 892,  418, 318,  418, 314,  420, 314,  422, 894,  418, 334,  402, 346,  394, 308,  420, 890,  418, 896,  416, 318,  416, 346,  394, 314,  418, 312,  422, 320,  422, 882,  422, 340,  396, 312,  424, 308,  420, 916,  390, 318,  418, 314,  416, 324,  420, 308,  420, 314,  420, 342,  394, 338,  396, 886,  418, 346,  394, 340,  394, 924,  394, 310,  420, 314,  422, 312,  422, 310,  422, 888,  422, 312,  418, 314,  416, 326,  422, 304,  422, 888,  422, 918,  398, 338,  396, 888,  424, 312,  416, 314,  420, 322,  422, 306,  418, 318,  420, 892,  424, 312,  418, 316,  422, 312,  418, 340,  394, 898,  418, 20256,  4670};  // DAIKIN64 8416109011062216
uint64_t data = 0x8416109011062216;

crankyoldgit, 2020-03-21:
Excellent! I'm really surprised there is no heating or auto operation mode.

Looking forward to you being able to test it on the real device eventually,

chongkk, 2020-03-21:
Yeah because I'm living in a equatorial country, all AC here are only function is cool dry and fan, some of them got auto mode. I will update when I can control the real AC.

chongkk, 2020-03-23:
@crankyoldgit Hi, if I use IRMQTTServer, it seems like doesn't transmit to the correct mqtt topic, so the home assistant will not update the cliamte state.

ir_server/received 77,0xAC22808016352216,64
ir_server/received 77,0x5522808016372216,64
ir_server/received 77,0x6523808016372216,64

crankyoldgit, 2020-03-23:
@chongkk Thanks for the report. I missed adding something. Please re-download and try again. I believe I've fixed it.

See: https://github.com/crankyoldgit/IRremoteESP8266/commit/59d3bb583684b832a053b90f07f3c7d8fa627270

chongkk, 2020-03-23:
@crankyoldgit I tested in home assistant. It have been fixed transmit the mqtt topic. So now I can use remoter send code and check it behavior in home assistant.
Another problem is the "Power Toggle", sorry I just realize that you ask me about the power toggle in google sheet.
If now the aircon is power off, then I send power on, fan auto, cool mode ,temperature is 16 use remoter then is correct show in home assistant. but when I increase 1 degree to 17, then in home assistant it will show as power off. The expected behavior is should continue "power on as cool mode".
Mqtt topic like this:
Turn on AC:
```
ir_server/received 77,0x9C16000020551216,64
ir_server/ac/stat/power on
ir_server/ac/stat/mode cool
ir_server/ac/stat/temp 16.0
```
Increase to 17 degree(Shouldn't show power off):
```
ir_server/received 77,0x3417000020561216,64
ir_server/ac/stat/power off
ir_server/ac/stat/mode off
ir_server/ac/stat/temp 17.0
```

Power Toggle: 1100
Without Power Toggle: 0100
SWING with Power: 1101
SWING: 0101
SLEEP: 0110

I am a beginer about IR AC. I think the first 1 is for toggle the power, if power off, turn on, if power on, turn off. 0 is without power toggle.
The second is always 1.
The third is for sleep, if 1 sleep on, if 0 sleep off.
The fourth is for swing, if 1 swing on, if 0 swing off.

Sorry For my bad English that maybe cause the misunderstanding and these days can't control the real AC then feedback more quickly.

Thanks, have a nice day.

crankyoldgit, 2020-03-23:
I _think_ I may have fixed it in https://github.com/crankyoldgit/IRremoteESP8266/commit/be2bc2d8bb6d0e25e2ad445456f27256c08e5914 which I just pushed to that branch.

Please try that. I haven't been able to do any testing on that so I'm not 100% confident it will fix it. So please test and let me know etc.
I'll try to look later on but I'm busy with other things today so won't be able to get to it for many hours or a day.

chongkk, 2020-03-24:
@crankyoldgit It fixed already, thanks crankyoldgit.

crankyoldgit, 2020-03-24:
@chongkk Excellent! I was just about to take another look at it, so you've saved me from that. Thanks.

Let me know if you find any other problems.

crankyoldgit, 2020-03-25:
@chongkk That branch has now been merged into the `master` branch.
If you find any more issues with it. Please create a new issue, rather than add to this one.

closed 2020-03-25 by crankyoldgit

crankyoldgit, 2020-04-09:
FYI, This has been included in the new v2.7.5 release of the library.

commit c4bcc097 (2020-03-21): Add clock & timers to DAIKIN64
* Add support for clock, On time, & off timer etc.

That should be everything that has been reverse engineered on the 
spreadsheet by @chongkk

Fixes #1064
commit 4c45e74a (2020-03-25): Add clock & timers to DAIKIN64
* Add support for clock, On time, & off timer etc.

That should be everything that has been reverse engineered on the 
spreadsheet by @chongkk

Fixes #1064
PR #1065 (2020-03-25): Support for Daikin FFN-C A/C
* `sendDaikin64()`, `decodeDaikin64()`, & `IRDaikin64` class
* Add checksum support.
* Add Power Toggle, Mode, Temp, Fan speed, Swing, Turbo, Quiet, Sleep, Clock, On & Off timer control.
* Unit tests.
* Full support for common A/C api.
* Update support routines.

Fixes #1064
commit bdd0e64c (2020-03-25): Support for Daikin FFN-C A/C (#1065)

* `sendDaikin64()`, `decodeDaikin64()`, & `IRDaikin64` class
* Add checksum support.
* Add Power Toggle, Mode, Temp, Fan speed, Swing, Turbo, Quiet, Sleep, Clock, On & Off timer control.
* Unit tests.
* Full support for common A/C api.
* Update support routines.

Fixes #1064