crankyoldgit/IRremoteESP8266 #1069: Cannot Copy Airwell RC08W ( two LED )

skyderman, 2020-03-25:
I Try to copy RC08W wich have 2 LED on it.

When I try to "dump" I always have UNKNOWN with random data ( for the same bouton ) 

here a extract from console ( Code work well with RC08B reconnize has COOLIX ): 

Protocol  : UNKNOWN
Code      : 0x20B8D8F4 (58 Bits)
uint16_t rawData[115] = {2862, 3864,  1920, 1956,  1920, 1958,  950, 988,  1950, 1004,  942, 964,  984, 956,  982, 1882,  1016, 954,  1954, 1000,  948, 958,  980, 962,  976, 960,  976, 960,  978, 964,  984, 952,  986, 954,  972, 1892,  1016, 950,  1956, 996,  952, 956,  982, 960,  976, 960,  978, 960,  976, 1886,  1012, 958,  1960, 1918,  980, 962,  3924, 3830,  1954, 1928,  1948, 1932,  986, 952,  1924, 1028,  918, 992,  946, 990,  948, 1916,  1012, 958,  1928, 1024,  924, 984,  954, 986,  952, 986,  950, 986,  952, 1020,  918, 990,  948, 992,  944, 1916,  980, 986,  1920, 1032,  926, 980,  958, 984,  952, 982,  956, 982,  956, 1908,  1008, 958,  1464, 286,  228, 1902,  256, 13344,  192};  // UNKNOWN 20B8D8F4

Timestamp : 000082.005
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0xBB3E58D1 (89 Bits)
uint16_t rawData[177] = {2866, 3858,  1956, 1920,  1956, 1920,  986, 954,  1952, 1000,  946, 960,  946, 992,  976, 1886,  1010, 958,  1930, 1024,  924, 986,  982, 954,  954, 986,  952, 988,  980, 958,  980, 960,  958, 980,  978, 1886,  990, 980,  1958, 994,  954, 954,  984, 958,  978, 956,  980, 958,  980, 1882,  1012, 958,  1950, 1926,  980, 960,  3926, 3830,  1954, 1922,  1952, 1926,  980, 956,  1950, 1094,  832, 982,  978, 962,  976, 1888,  1010, 958,  1948, 1008,  920, 984,  952, 986,  982, 958,  950, 988,  980, 960,  948, 986,  982, 958,  978, 1882,  1014, 956,  1952, 1000,  946, 964,  954, 982,  956, 984,  952, 990,  958, 1908,  988, 982,  1926, 1948,  958, 980,  3916, 3866,  736, 292,  898, 1924,  256, 10450,  918, 1020,  948, 1918,  980, 990,  1896, 1056,  892, 1016,  920, 1018,  920, 1018,  920, 1018,  920, 1024,  924, 1010,  928, 1012,  924, 1938,  948, 1022,  804, 284,  442, 1418,  498, 1412,  496, 1444,  494, 1446,  502, 1436,  502, 2362,  496, 288,  196, 988,  504, 282,  202, 282,  200, 2404,  502, 1436,  442, 284,  200, 104,  228};  // UNKNOWN BB3E58D1

Timestamp : 000083.524
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0xDFE746E1 (53 Bits)
uint16_t rawData[105] = {2892, 3832,  1954, 1922,  1924, 1954,  952, 986,  1920, 1032,  926, 982,  976, 962,  956, 1908,  988, 980,  1956, 998,  950, 958,  978, 958,  980, 958,  980, 962,  976, 960,  978, 962,  976, 964,  954, 1908,  1008, 962,  1956, 996,  952, 956,  982, 958,  948, 988,  980, 960,  978, 1886,  1010, 956,  1950, 1928,  978, 958,  3918, 3838,  1946, 1932,  1954, 1922,  984, 952,  1954, 1000,  956, 958,  950, 986,  952, 1912,  986, 986,  1952, 1000,  928, 980,  956, 984,  954, 982,  954, 986,  952, 986,  952, 988,  950, 986,  950, 1912,  986, 982,  1924, 1026,  922, 984,  954, 986,  950, 1014,  530, 1410,  406};  // UNKNOWN DFE746E1

Timestamp : 000085.077
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0x417A225B (56 Bits)
uint16_t rawData[111] = {2894, 3834,  1950, 1924,  1954, 1924,  984, 956,  1920, 1034,  944, 964,  954, 984,  954, 1910,  986, 980,  1926, 1026,  952, 956,  980, 960,  978, 962,  946, 992,  978, 958,  978, 960,  978, 962,  974, 1888,  1010, 962,  1956, 996,  940, 964,  984, 956,  980, 956,  980, 960,  978, 1882,  1014, 958,  1948, 1926,  980, 960,  3918, 3836,  1948, 1932,  1954, 1920,  986, 952,  1956, 998,  948, 960,  978, 960,  948, 1914,  1014, 958,  1950, 1002,  926, 982,  976, 964,  952, 986,  952, 988,  960, 982,  944, 992,  956, 984,  952, 1908,  988, 982,  1924, 1028,  920, 984,  954, 984,  952, 984,  954, 982,  956, 1908,  988, 1008,  496, 290,  192};  // UNKNOWN 417A225B

Timestamp : 000085.972
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0x5280CC2D (55 Bits)
uint16_t rawData[109] = {2892, 3830,  1954, 1924,  1952, 1924,  982, 954,  1922, 1032,  926, 984,  974, 964,  954, 1908,  990, 982,  1926, 1026,  952, 956,  982, 958,  978, 958,  980, 960,  976, 962,  976, 964,  952, 982,  976, 1888,  1010, 960,  1948, 1008,  950, 956,  980, 956,  982, 958,  948, 990,  978, 1884,  1012, 958,  1950, 1926,  980, 958,  3918, 3838,  1946, 1932,  1954, 1922,  984, 954,  1952, 998,  920, 988,  950, 992,  956, 1904,  1014, 960,  1956, 1000,  916, 990,  948, 990,  948, 992,  956, 982,  946, 994,  954, 984,  954, 986,  950, 1910,  988, 980,  1926, 1026,  922, 986,  952, 984,  954, 986,  952, 984,  226, 286,  440, 1938,  200};  // UNKNOWN 5280CC2D

Timestamp : 000086.777
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0xEF4DE025 (56 Bits)
uint16_t rawData[111] = {2894, 3830,  1954, 1922,  1954, 1926,  980, 956,  1922, 1030,  946, 962,  956, 982,  956, 1906,  990, 980,  1928, 1028,  950, 956,  982, 958,  980, 960,  948, 988,  978, 962,  976, 960,  978, 964,  984, 1878,  1010, 960,  1956, 994,  952, 958,  980, 958,  950, 992,  976, 958,  980, 1886,  1012, 956,  1952, 1926,  980, 958,  3918, 3838,  1946, 1930,  1948, 1930,  976, 964,  1954, 998,  950, 960,  948, 992,  956, 1904,  1012, 956,  1960, 998,  920, 992,  946, 990,  948, 990,  946, 994,  954, 982,  954, 986,  952, 986,  952, 1912,  986, 982,  1924, 1024,  922, 986,  952, 986,  952, 988,  950, 986,  952, 1942,  500, 282,  192, 992,  198};  // UNKNOWN EF4DE025

I think it is a wrong setting, but I don't find where :/

Thanks you in advance.

crankyoldgit, 2020-03-26:
There are a number of issues I can see by looking at the data:
1) I think your **RC08W** remote speaks a different protocol than your **RC08B** remote.
2) I don't immediately recognise what protocol you're RC08W remote is using by visually looking at the data. It may be a protocol the library **doesn't** support, yet.
3) The signal/message looks consistent until near the end of the message/data where there is noise/garbage etc. e.g. the values of approx 200usecs at/near the end of the data. They may be cause by a number of reasons. i.e. Low battery, to far/to close to the receiver, reflections, IR noise sources (sunlight, fluorescent lights, etc.), ... You probably need to address them to get an accurate signal.

The only "clean"-ish message I can see in there is:
```
Protocol : UNKNOWN
Code : 0xDFE746E1 (53 Bits)
uint16_t rawData[105] = {2892, 3832, 1954, 1922, 1924, 1954, 952, 986, 1920, 1032, 926, 982, 976, 962, 956, 1908, 988, 980, 1956, 998, 950, 958, 978, 958, 980, 958, 980, 962, 976, 960, 978, 962, 976, 964, 954, 1908, 1008, 962, 1956, 996, 952, 956, 982, 958, 948, 988, 980, 960, 978, 1886, 1010, 956, 1950, 1928, 978, 958, 3918, 3838, 1946, 1932, 1954, 1922, 984, 952, 1954, 1000, 956, 958, 950, 986, 952, 1912, 986, 986, 1952, 1000, 928, 980, 956, 984, 954, 982, 954, 986, 952, 986, 952, 988, 950, 986, 950, 1912, 986, 982, 1924, 1026, 922, 984, 954, 986, 950, 1014, 530, 1410, 406}; // UNKNOWN DFE746E1
```
and I'm not even certain that is a clean/good message/signal either. The values `530, 1410, 406` at the end are inconsistent with the rest of the data, they may be legit, but they are odd none the less.

No matter how I look at this, something is odd/wrong with these captures. As you've got capture working with your other remote I think we can assume your hardware is mostly correct etc.

Have you tried sending this data to your A/C using `sendRaw()` and does it respond?

Other reading you should do:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#my-remotes-signal-is-being-reported-as-an-unknown-code-what-does-that-mean-how-do-i-send-an-unknown-code
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#im-getting-some-random-odd-values--50-usecs-in-my-capture-of-an-ir-message-what-is-up-with-that
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#why-is-the-raw-data-for-a-button-or-ac-state-always-different-for-each-capture

skyderman, 2020-03-26:
Hi, 

First, a big thanks you to answer me.

I try my first dump whithout any succes.
I juste try your clean message without succes BUT I spent about 1 day of testing ( :/ ) , and I finaly find something that work : 

I hide, one of the two LED of the remote ( I don't know if it matter ), and I do lot a time the same button until I see something with 163-170

`
uint16_t rawData[163] = {2860, 3862,  1924, 1952,  1926, 1952,  956, 984,  1924, 1028,  952, 958,  980, 956,  982, 1882,  1016, 950,  1958, 1920,  1948, 1004,  954, 956,  984, 956,  952, 984,  974, 966,  974, 964,  974, 1888,  1010, 960,  1948, 1002,  946, 962,  978, 962,  976, 960,  948, 992,  978, 1886,  982, 984,  1924, 1954,  952, 986,  3892, 3862,  1924, 1954,  1924, 1954,  984, 952,  1956, 996,  954, 990,  948, 958,  980, 1882,  1016, 952,  1958, 1920,  1956, 994,  944, 962,  986, 956,  972, 962,  976, 962,  978, 964,  952, 1908,  1010, 960,  1948, 1002,  946, 960,  978, 962,  946, 990,  978, 960,  978, 1888,  1008, 954,  1952, 1928,  982, 956,  3920, 3830,  1946, 1934,  1954, 1922,  976, 962,  1946, 1006,  922, 990,  948, 988,  950, 1910,  1008, 964,  1922, 1952,  1924, 1030,  920, 984,  954, 986,  952, 986,  952, 984,  954, 986,  952, 1910,  1010, 960,  1928, 1024,  944, 996,  924, 984,  954, 988,  950, 984,  954, 1910,  1008, 960,  1920, 1958,  980, 958,  4800};  // UNKNOWN 565E2BB3
//ON / OFF / 25 degres heat

*/
/*
 
uint16_t rawData[163] = {2862, 3860,  1928, 1952,  1926, 1952,  956, 980,  1928, 1024,  926, 1016,  952, 954,  984, 1878,  990, 980,  1948, 1928,  980, 956,  1952, 1002,  916, 994,  976, 960,  978, 962,  976, 960,  978, 1882,  986, 986,  1922, 1028,  922, 988,  980, 956,  952, 988,  980, 958,  980, 1882,  984, 986,  1924, 1952,  956, 982,  3896, 3862,  1956, 1918,  1950, 1930,  978, 958,  1950, 1004,  914, 992,  978, 962,  976, 1888,  1010, 954,  1954, 1926,  982, 956,  1952, 1002,  948, 958,  980, 960,  948, 990,  948, 988,  980, 1882,  1016, 952,  1948, 1006,  942, 966,  982, 958,  950, 984,  954, 984,  974, 1888,  1010, 958,  1950, 1924,  984, 954,  3924, 3834,  1924, 1958,  1920, 1956,  982, 954,  1922, 1030,  920, 988,  950, 990,  948, 1912,  1006, 964,  1924, 1954,  984, 952,  1926, 1028,  920, 988,  952, 986,  952, 986,  952, 986,  954, 1910,  1008, 964,  1924, 1026,  922, 984,  954, 984,  954, 986,  952, 984,  956, 1908,  1010, 958,  1920, 1962,  978, 958,  4798};  // UNKNOWN EC598435

//26 degres heat
*/

//uint16_t rawData[163] = {2866, 3856,  1922, 1958,  1930, 1946,  952, 984,  1924, 1028,  952, 956,  950, 986,  984, 1880,  1958, 1954,  954, 982,  956, 982,  1958, 996,  954, 954,  984, 952,  976, 968,  952, 984,  954, 1906,  1012, 956,  1952, 1000,  948, 962,  946, 990,  978, 960,  978, 958,  980, 1882,  1016, 954,  1954, 1924,  984, 954,  3922, 3830,  1956, 1922,  1956, 1922,  986, 950,  1958, 994,  944, 964,  986, 954,  984, 1882,  1986, 1922,  988, 950,  1010, 926,  1950, 1004,  926, 980,  978, 960,  978, 960,  980, 990,  918, 1914,  1014, 952,  1956, 1000,  918, 986,  952, 986,  982, 954,  976, 966,  952, 1908,  1010, 956,  1952, 1924,  984, 954,  3924, 3838,  1950, 1924,  1954, 1924,  984, 954,  1954, 1000,  918, 988,  952, 986,  952, 1912,  1986, 1918,  978, 962,  986, 952,  1956, 996,  922, 984,  954, 984,  956, 986,  952, 982,  956, 1908,  1010, 958,  1950, 1002,  926, 984,  956, 980,  958, 982,  958, 980,  980, 1916,  982, 956,  1952, 1924,  986, 954,  4834};  // UNKNOWN 9C46B03B

//ON / OFF / 22 degres heat
//uint16_t rawData[169] = {2864, 3858,  1928, 1024,  924, 1910,  988, 978,  960, 978,  1932, 1022,  926, 984,  954, 982,  976, 1888,  1960, 1948,  960, 976,  1922, 1030,  928, 984,  954, 982,  978, 962,  986, 952,  946, 990,  980, 1886,  1012, 956,  1952, 1002,  948, 958,  980, 958,  950, 990,  948, 988,  952, 1912,  1014, 954,  1956, 1922,  986, 952,  3926, 3832,  1954, 996,  922, 1910,  1018, 950,  988, 952,  1956, 994,  924, 984,  984, 956,  984, 1878,  1978, 1928,  980, 960,  1948, 1002,  976, 932,  974, 962,  1008, 932,  976, 960,  1008, 932,  956, 1906,  1012, 956,  1952, 1004,  924, 980,  948, 990,  978, 958,  980, 962,  956, 1904,  1014, 954,  1954, 1922,  986, 950,  3918, 3840,  1956, 994,  924, 1912,  1016, 954,  986, 958,  1950, 996,  922, 986,  954, 988,  950, 1910,  1978, 1932,  986, 952,  1958, 996,  922, 988,  952, 986,  952, 984,  956, 984,  954, 982,  956, 1910,  1010, 958,  1950, 1004,  924, 986,  952, 984,  956, 984,  954, 984,  954, 1906,  1012, 960,  1948, 1928,  980, 960,  4828};  // UNKNOWN DEA083AC
//21 degres cool
`

But this one don't work : 

`
uint16_t rawData1[175] = {2856, 3866,  1926, 1026,  922, 1910,  980, 986,  954, 986,  1924, 1026,  924, 986,  954, 982,  956, 1906,  984, 984,  956, 982,  1928, 1024,  926, 982,  956, 982,  948, 992,  948, 990,  954, 984,  952, 984,  956, 1906,  984, 988,  1922, 1026,  924, 986,  954, 982,  958, 982,  946, 994,  946, 1914,  986, 984,  1926, 1948,  952, 988,  3892, 3862,  1928, 1022,  946, 1886,  984, 982,  958, 984,  1948, 1004,  946, 962,  978, 956,  982, 1882,  1018, 950,  980, 958,  1950, 998,  952, 958,  982, 958,  972, 964,  974, 964,  976, 960,  978, 958,  982, 1884,  1016, 950,  1950, 1000,  948, 962,  978, 960,  980, 956,  984, 954,  974, 1892,  1008, 958,  1952, 1926,  984, 952,  3918, 3838,  1952, 998,  922, 1912,  1008, 958,  982, 960,  1980, 968,  922, 986,  1006, 932,  976, 1884,  1016, 958,  980, 952,  1948, 1002,  948, 962,  948, 988,  982, 958,  950, 986,  954, 984,  976, 960,  980, 1884,  1016, 954,  1946, 10 [... cut ...] 4, 1014,  1892, 1986,  922, 1016,  3860, 3896,  1898, 1054,  924, 1908,  958, 1012,  926, 1012,  1896, 1056,  942, 966,  952, 988,  950, 1910,  956, 1016,  922, 1016,  952, 988,  950, 986,  1922, 1032,  916, 990,  948, 960,  988, 984,  944, 960,  978, 1918,  978, 990,  1896, 1054,  924, 982,  946, 994,  954, 984,  954, 988,  950, 1910,  956, 1012,  1896, 1984,  922, 1018,  4768};  // UNKNOWN 1002DCC8
`

Do you see something that can help me ? 
( or a setting that help me to catch better dump ? )

crankyoldgit, 2020-03-26:
I _think_ the protocol maybe using Manchester Code (https://en.wikipedia.org/wiki/Manchester_code) to encode the data. The data you have is NOT the typical "space encoding" i.e. the even(gap/"space") values contain the data. The odd("mark"/on) & the "space" values fluctuate between ~1000 & ~2000, so it isn't pure "mark encoding" either.
There are also "header" marks & spaces. i.e. the values >= ~3000. They typically mark the start/end of data segments.

We've got a few protocols that behave like that (Manchester code), I'll look later to see what can be adapted.

As for why some of them don't work. I think it's because they have values that are too far away from the 1000 or 2000 values. e.g. _..., 922, 2180, **740**, 988, **1316**, 2564, 952, ..._

Have you tried replacing the batteries with known fresh ones? And trying different capture distances too, and in a dark room etc?

There are no settings I can think of that you can tweak in the library to help capture better that are going to help here.

crankyoldgit, 2020-03-28:
Hey @skyderman , Can you please download & test out PR #1070 / Branch: "Issue1069" https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1069 and please let me know how it goes for you.

I think I've got it decoding your protocol, but it needs LOTS more testing & data to be sure.
It is VERY experimental (but probably works) and the values it creates/uses etc may change in the future as it gets refined/improved/fixed etc.

skyderman, 2020-03-30:
Hi, 

I just try your improve.
With IRrecvDumpv2, I've got this result :

When I change temperture 

Timestamp : 000183.277
Library   : v2.7.4

Protocol  : UNKNOWN
Code      : 0xFE9A64F4 (89 Bits)
uint16_t rawData[177] = {2860, 2946,  890, 1042,  944, 960,  946, 1920,  986, 984,  954, 984,  1922, 1028,  948, 962,  976, 964,  972, 1892,  1014, 958,  1948, 1928,  980, 960,  1946, 1004,  922, 988,  980, 960,  978, 960,  978, 962,  974, 1888,  1008, 962,  1954, 998,  950, 960,  976, 962,  976, 966,  972, 992,  946, 1922,  984, 958,  1948, 1928,  978, 962,  3922, 2916,  920, 1014,  944, 966,  982, 1880,  1016, 954,  982, 956,  1950, 1004,  944, 966,  950, 986,  950, 1914,  1032, 936,  1980, 1898,  976, 962,  1946, 1008,  950, 962,  946, 990,  946, 992,  974, 962,  974, 1890,  1006, 966,  1940, 1008,  948, 962,  974, 962,  974, 964,  792, 1150,  708, 2154,  792, 1180,  1252, 256,  438, 1932,  974, 962,  1580, 5260,  492, 1436,  500, 1408,  500, 2364,  492, 1476,  502, 1438,  498, 2486,  460, 1418,  500, 1438,  500, 2364,  500, 1470,  498, 3380,  496, 1444,  494, 2490,  466, 1414,  494, 1444,  492, 1446,  492, 1448,  500, 2364,  492, 1478,  500, 2486,  470, 1408,  500, 1438,  500, 1442,  496, 1442,  496, 2368,  498, 1472,  494, 3384,  502, 1436,  500};  // UNKNOWN FE9A64F4

But when I put 'ON/off' ( with this remote, they don't put OFF, but resend the config, and AC turn OFF if is it ON, or turn ON if they are OFF) 

Protocol  : AIRWELL
Code      : 0x70D8181B (32 Bits)
uint16_t rawData[177] = {2864, 3862,  1918, 1034,  922, 1912,  982, 984,  952, 988,  1926, 1026,  920, 990,  976, 960,  956, 1908,  986, 982,  1922, 1956,  950, 990,  1924, 1028,  948, 962,  954, 982,  984, 956,  980, 956,  980, 1916,  978, 960,  1954, 998,  948, 960,  976, 964,  952, 984,  982, 956,  980, 1882,  1012, 960,  1954, 1924,  980, 956,  3926, 3832,  1958, 994,  920, 1912,  1014, 958,  988, 948,  1956, 996,  918, 990,  978, 960,  986, 1878,  1016, 952,  1952, 1930,  986, 950,  1954, 1000,  926, 980,  976, 964,  1004, 934,  982, 960,  946, 1916,  1008, 958,  1946, 1006,  950, 960,  956, 982,  954, 986,  982, 956,  950, 1916,  1010, 960,  1946, 1930,  974, 966,  1466, 256,  228, 256,  228, 5320,  592, 252,  1108, 1004,  922, 1910,  1004, 968,  768, 1168,  676, 256,  1014, 1008,  374, 1532,  374, 1562,  374, 2492,  342, 1624,  342, 3540,  344, 1592,  344, 2610,  346, 1562,  342, 1594,  342, 1598,  348, 1592,  344, 2518,  348, 1624,  342, 2608,  346, 1564,  342, 1596,  350, 1588,  348, 1590,  348, 2520,  344, 1624,  342, 3536,  346, 1590,  346};  // AIRWELL 70D8181B
uint64_t data = 0x70D8181B;

I hope this help you ?

skyderman, 2020-03-30:
I try to resend 

uint16_t rawData[177] = {2864, 3862,  1918, 1034,  922, 1912,  982, 984,  952, 988,  1926, 1026,  920, 990,  976, 960,  956, 1908,  986, 982,  1922, 1956,  950, 990,  1924, 1028,  948, 962,  954, 982,  984, 956,  980, 956,  980, 1916,  978, 960,  1954, 998,  948, 960,  976, 964,  952, 984,  982, 956,  980, 1882,  1012, 960,  1954, 1924,  980, 956,  3926, 3832,  1958, 994,  920, 1912,  1014, 958,  988, 948,  1956, 996,  918, 990,  978, 960,  986, 1878,  1016, 952,  1952, 1930,  986, 950,  1954, 1000,  926, 980,  976, 964,  1004, 934,  982, 960,  946, 1916,  1008, 958,  1946, 1006,  950, 960,  956, 982,  954, 986,  982, 956,  950, 1916,  1010, 960,  1946, 1930,  974, 966,  1466, 256,  228, 256,  228, 5320,  592, 252,  1108, 1004,  922, 1910,  1004, 968,  768, 1168,  676, 256,  1014, 1008,  374, 1532,  374, 1562,  374, 2492,  342, 1624,  342, 3540,  344, 1592,  344, 2610,  346, 1562,  342, 1594,  342, 1598,  348, 1592,  344, 2518,  348, 1624,  342, 2608,  346, 1564,  342, 1596,  350, 1588,  348, 1590,  348, 2520,  344, 1624,  342, 3536,  346, 1590,  346};  // AIRWELL 70D8181B

but the AC not respond

I try an another dump ( 25 ° cool , ventillation Auto )  ( when I put On/off ):
Protocol  : AIRWELL
Code      : 0x70D0181B (32 Bits)
uint16_t rawData[169] = {2860, 3864,  1922, 1032,  928, 1904,  982, 988,  950, 986,  1922, 1034,  956, 954,  944, 992,  978, 1886,  1010, 958,  1950, 1928,  1950, 1002,  946, 962,  976, 964,  984, 952,  986, 954,  974, 966,  974, 1890,  1018, 952,  1956, 994,  954, 958,  980, 956,  982, 956,  982, 956,  982, 1884,  1014, 952,  1954, 1926,  982, 954,  3924, 3836,  1950, 1000,  948, 1890,  1008, 956,  982, 960,  1978, 972,  946, 962,  946, 994,  954, 1906,  1012, 958,  1950, 1928,  1948, 1006,  922, 984,  974, 964,  974, 966,  952, 984,  952, 986,  984, 1880,  1008, 962,  1946, 1008,  920, 986,  972, 966,  974, 966,  952, 986,  972, 1890,  1008, 962,  1946, 1936,  982, 954,  3770, 3990,  1948, 1002,  926, 1912,  1006, 958,  980, 960,  1948, 1004,  924, 986,  952, 984,  954, 1910,  1008, 960,  1946, 1932,  1946, 1008,  922, 988,  950, 990,  948, 988,  950, 990,  948, 992,  948, 1914,  1012, 956,  1952, 1000,  918, 988,  950, 992,  956, 980,  948, 992,  946, 1920,  1008, 958,  1952, 1928,  980, 960,  4828};  // AIRWELL 70D0181B
uint64_t data = 0x70D0181B;

And I worked.

I think only DATA between 163 and 175 work

I hope this will help you ?

crankyoldgit, 2020-03-30:
How are you sending it?
with `sendRaw()` or with `sendAirwell()`?

and, have you changed the batteries on the remote? e.g. the `rawData[177]` mesg in your latest update is corrupted after about 2/3rds of the message. Maybe the remote is running low on power?

The second signal in your last update, on the other hand is fine.

skyderman, 2020-03-30:
I try with sendRaw....

I Changed bettery and I dump  this :

When I push + on temperture from 21 to 22 ( vent: auto, Cool )  : 

Protocol  : UNKNOWN
Code      : 0xD1258DDF (94 Bits)
uint16_t rawData[187] = {2860, 2946,  892, 1036,  922, 986,  952, 1912,  988, 982,  956, 986,  1922, 1026,  922, 988,  950, 986,  954, 1912,  1954, 1952,  958, 980,  958, 984,  1954, 996,  954, 956,  982, 954,  984, 956,  972, 966,  952, 1912,  1016, 952,  1956, 994,  944, 968,  982, 954,  974, 964,  984, 980,  950, 1920,  986, 950,  1948, 1932,  986, 950,  3918, 2922,  926, 1008,  942, 962,  976, 1892,  1016, 952,  988, 950,  1948, 1002,  946, 964,  976, 966,  952, 1910,  1978, 1932,  986, 950,  978, 960,  1948, 1002,  916, 994,  1006, 934,  954, 982,  946, 994,  976, 1886,  1012, 958,  1950, 1000,  948, 964,  944, 992,  946, 988,  980, 958,  980, 1904,  994, 958,  1950, 1924,  984, 956,  1770, 254,  290, 254,  260, 254,  824, 2958,  890, 1008,  922, 1016,  952, 1910,  978, 992,  946, 990,  1312, 1642,  318, 258,  348, 960,  948, 1014,  924, 2152,  1746, 1926,  972, 992,  948, 990,  1282, 1670,  320, 254,  350, 960,  950, 1012,  946, 994,  924, 1012,  948, 2130,  778, 952,  1248, 1728,  322, 254,  342, 968,  970, 992,  926, 1012,  946, 994,  924, 2150,  778, 954,  1246, 2658,  946, 990,  1222};  // UNKNOWN D1258DDF

But When I push On/OFF : 

First time : 

Protocol  : AIRWELL
Code      : 0x70B8181B (32 Bits)
uint16_t rawData[177] = {2862, 3858,  1928, 1024,  924, 1912,  986, 980,  958, 980,  1918, 1034,  924, 984,  956, 984,  954, 1912,  1956, 1950,  958, 980,  958, 980,  1918, 1034,  944, 962,  988, 952,  984, 952,  986, 984,  944, 1888,  1010, 958,  1950, 1002,  948, 964,  974, 960,  978, 962,  978, 964,  974, 1890,  1008, 958,  1950, 1930,  978, 956,  3922, 3836,  1950, 1002,  948, 1888,  1010, 958,  980, 956,  1952, 1000,  948, 960,  978, 960,  978, 1886,  1982, 1928,  980, 956,  982, 956,  1950, 1000,  968, 940,  978, 960,  980, 962,  996, 940,  948, 1916,  1042, 926,  1952, 1000,  918, 988,  950, 990,  978, 958,  950, 988,  980, 1882,  1018, 954,  1942, 1934,  984, 952,  1946, 260,  194, 5362,  1950, 996,  922, 1940,  978, 992,  948, 992,  1320, 1628,  198, 254,  200, 1254,  946, 994,  924, 2120,  1766, 1934,  974, 990,  948, 990,  1312, 1638,  200, 254,  198, 1254,  946, 996,  922, 1016,  924, 1014,  944, 2100,  808, 954,  1428, 1548,  198, 256,  196, 1256,  924, 1014,  946, 992,  946, 994,  924, 2122,  806, 956,  1458, 2444,  948, 990,  1434};  // AIRWELL 70B8181B
uint64_t data = 0x70B8181B;

2 time : 

Protocol  : AIRWELL
Code      : 0x70B8181B (32 Bits)
uint16_t rawData[173] = {2858, 3868,  1930, 1022,  926, 1906,  982, 986,  952, 988,  1920, 1034,  946, 958,  950, 990,  978, 1886,  1952, 1954,  984, 956,  980, 956,  1952, 1000,  948, 958,  980, 960,  978, 960,  978, 960,  980, 1884,  1014, 954,  1954, 1002,  946, 958,  980, 958,  980, 960,  978, 962,  976, 1884,  1014, 954,  1954, 1924,  984, 956,  3922, 3832,  1954, 998,  952, 1914,  984, 954,  984, 952,  1956, 996,  952, 990,  918, 988,  980, 1884,  1984, 1924,  984, 954,  984, 952,  1976, 976,  922, 988,  948, 988,  952, 988,  980, 956,  952, 1908,  1012, 964,  1944, 1006,  922, 984,  984, 954,  954, 982,  976, 964,  976, 1884,  1014, 956,  1952, 1960,  950, 986,  3618, 4110,  1950, 998,  950, 1910,  200, 106,  682, 986,  952, 986,  1922, 1028,  920, 990,  918, 1018,  920, 1940,  230, 104,  1612, 1962,  946, 994,  954, 984,  1914, 1036,  922, 990,  918, 1018,  950, 988,  920, 1018,  950, 1914,  984, 984,  1914, 1040,  890, 1014,  922, 1018,  952, 990,  918, 1020,  918, 1942,  986, 982,  1916, 1962,  946, 996,  4436};  // AIRWELL 70B8181B
uint64_t data = 0x70B8181B;

For + from 22 to 23 : 

Protocol  : UNKNOWN
Code      : 0x80737B65 (96 Bits)
uint16_t rawData[191] = {2862, 2948,  890, 1036,  922, 984,  954, 1910,  988, 984,  954, 980,  1928, 1026,  952, 956,  982, 958,  980, 1884,  1014, 952,  1956, 996,  952, 958,  950, 986,  982, 958,  950, 988,  950, 990,  978, 958,  980, 960,  978, 1884,  1014, 954,  1952, 998,  950, 958,  980, 960,  978, 958,  980, 1050,  888, 1886,  1012, 954,  1954, 1926,  982, 958,  3918, 2918,  918, 1010,  948, 964,  956, 1906,  1010, 958,  980, 958,  1950, 1004,  924, 980,  978, 962,  946, 1918,  1008, 958,  1970, 984,  914, 992,  1008, 934,  944, 992,  946, 992,  976, 964,  944, 992,  946, 992,  976, 1886,  1012, 958,  1950, 1000,  948, 964,  944, 992,  946, 990,  980, 958,  980, 1884,  1002, 970,  1222, 166,  560, 1930,  978, 988,  2496, 4310,  648, 1312,  584, 1324,  554, 2278,  650, 1348,  560, 1378,  560, 2394,  442, 250,  224, 966,  588, 1372,  556, 2280,  646, 1350,  558, 2396,  442, 250,  224, 964,  580, 1386,  564, 1372,  556, 1382,  556, 1384,  564, 244,  198, 932,  562, 1374,  554, 2282,  646, 1350,  558, 2396,  440, 252,  222, 968,  586, 1376,  562, 1376,  562, 1378,  560, 2272,  646, 1354,  554, 3296,  652, 1312,  554};  // UNKNOWN 80737B65

On/off 

Protocol  : AIRWELL
Code      : 0x70C0181B (32 Bits)
uint16_t rawData[179] = {2860, 3862,  1924, 1028,  950, 1882,  986, 1014,  926, 1014,  1924, 998,  948, 990,  948, 958,  980, 1882,  1014, 984,  1924, 1000,  918, 1022,  948, 960,  978, 958,  980, 962,  978, 958,  980, 958,  980, 958,  980, 1884,  1014, 958,  1950, 1002,  946, 958,  980, 958,  980, 962,  978, 962,  976, 1884,  1014, 956,  1952, 1928,  980, 956,  3922, 3836,  1950, 1002,  948, 1884,  1012, 956,  982, 958,  1950, 1002,  946, 964,  944, 992,  946, 1916,  1032, 936,  1950, 1002,  916, 994,  944, 992,  9 [... cut ...]  972, 990,  524, 7204,  582, 2400,  466, 2338,  590, 1408,  530, 1408,  532, 2420,  466, 278,  196, 968,  526, 1412,  556, 2278,  590, 1406,  532, 2420,  466, 278,  196, 966,  528, 1414,  524, 1414,  524, 1412,  526, 1414,  524, 1414,  524, 1414,  524, 2308,  590, 1412,  526, 2422,  466, 1444,  524, 1412,  526, 1416,  522, 1416,  524, 2310,  556, 1442,  530, 3318,  558, 1410,  528};  // AIRWELL 70C0181B
uint64_t data = 0x70C0181B;

I juste try with 

  irsend.sendAirwell(0x70C0181B);
And it's ok

skyderman, 2020-03-30:
What do you need I do to help you ? 

this is more dump ( Temperture / [heat/cool] / ventillation )  ( I skip the part when I change and past only with "on/off", which is detected has "Airwell" : 

30 heat auto

Protocol  : AIRWELL
Code      : 0xB0F8181B (32 Bits)
uint16_t rawData[169] = {2868, 3856,  1928, 1948,  1928, 1946,  950, 992,  1956, 996,  942, 964,  954, 986,  952, 1910,  1018, 952,  986, 950,  978, 964,  986, 950,  1958, 996,  952, 956,  952, 984,  954, 988,  980, 958,  980, 1884,  982, 984,  1924, 1030,  948, 960,  978, 960,  978, 960,  978, 958,  980, 1884,  982, 990,  1948, 1928,  980, 960,  3918, 3836,  1950, 1930,  1956, 1918,  978, 960,  1948, 1004,  954, 954,  982, 956,  972, 1892,  1016, 954,  984, 952,  986, 952,  988, 952,  1956, 996,  922, 986,  972, 968,  980, 956,  982, 958,  950, 1914,  1014, 950,  1946, 1006,  952, 956,  982, 958,  950, 988,  950, 990,  948, 1912,  1016, 954,  1954, 1926,  962, 984,  3894, 3856,  1920, 1958,  1920, 1962,  956, 978,  1920, 1034,  926, 984,  954, 986,  952, 1914,  984, 982,  956, 982,  956, 984,  954, 984,  1924, 1030,  918, 986,  952, 988,  948, 992,  958, 980,  948, 1914,  984, 988,  1918, 1030,  918, 992,  956, 980,  958, 982,  956, 982,  956, 1908,  990, 978,  1928, 1948,  958, 980,  4808};  // AIRWELL B0F8181B
uint64_t data = 0xB0F8181B;

29 / heat / auto 

Protocol  : AIRWELL
Code      : 0xB0F0181B (32 Bits)
uint16_t rawData[169] = {2868, 3860,  1926, 1950,  1928, 1950,  958, 978,  1928, 1024,  944, 962,  986, 954,  974, 1888,  1010, 962,  986, 952,  986, 950,  1948, 1004,  954, 954,  984, 956,  982, 954,  984, 956,  952, 986,  984, 1882,  986, 984,  1924, 1028,  950, 958,  982, 958,  978, 958,  980, 958,  980, 1884,  984, 986,  1920, 1954,  984, 956,  3922, 3836,  1948, 1928,  1950, 1928,  980, 958,  1950, 1004,  954, 956,  974, 962,  974, 1890,  1018, 950,  990, 950,  988, 950,  1958, 996,  920, 984,  986, 956,  952, 986,  952, 986,  984, 954,  974, 1892,  1016, 952,  1946, 1006,  922, 988,  950, 986,  952, 986,  982, 954,  954, 1908,  1010, 962,  1956, 1922,  986, 954,  3894, 3862,  1922, 1956,  1920, 1954,  954, 984,  1922, 1032,  918, 992,  956, 980,  958, 1906,  982, 990,  960, 978,  950, 988,  1920, 1034,  924, 982,  956, 1014,  924, 984,  954, 982,  956, 982,  956, 1910,  988, 984,  1924, 1024,  922, 988,  950, 986,  952, 988,  952, 986,  952, 1910,  986, 986,  1922, 1954,  954, 984,  4802};  // AIRWELL B0F0181B
uint64_t data = 0xB0F0181B;

28 / heat / auto

Protocol  : AIRWELL
Code      : 0xB0E8181B (32 Bits)
uint16_t rawData[163] = {2864, 3858,  1928, 1948,  1950, 1928,  978, 960,  1958, 996,  952, 956,  982, 954,  984, 1880,  1018, 952,  986, 952,  1956, 1922,  1956, 994,  954, 956,  982, 956,  982, 956,  980, 956,  982, 1914,  984, 956,  1952, 1000,  950, 958,  980, 962,  946, 988,  980, 960,  978, 1884,  1012, 958,  1950, 1926,  982, 954,  3922, 3834,  1952, 1926,  1952, 1930,  958, 980,  1928, 1026,  922, 986,  952, 982,  956, 1908,  988, 980,  958, 982,  1926, 1950,  1928, 1024,  924, 988,  950, 986,  954, 984,  954, 988,  950, 1912,  984, 984,  1924, 1028,  922, 988,  950, 986,  952, 988,  950, 988,  950, 1914,  982, 986,  1922, 1954,  954, 986,  3892, 3866,  1920, 1958,  1920, 1956,  950, 990,  1928, 1024,  924, 984,  954, 984,  952, 1910,  990, 980,  956, 980,  1928, 1952,  1926, 1028,  920, 986,  952, 988,  950, 988,  950, 988,  950, 1912,  986, 984,  1924, 1030,  918, 988,  950, 988,  948, 992,  946, 990,  948, 1914,  982, 986,  1922, 1956,  952, 986,  4800};  // AIRWELL B0E8181B
uint64_t data = 0xB0E8181B;

27 / heat / auto 

Protocol  : AIRWELL
Code      : 0xB0E0181B (32 Bits)
uint16_t rawData[169] = {2862, 3860,  1926, 1952,  1956, 1924,  984, 952,  1956, 998,  950, 958,  980, 958,  980, 1882,  1016, 954,  986, 952,  1924, 1030,  918, 988,  980, 958,  950, 988,  950, 992,  976, 960,  978, 958,  978, 1888,  990, 974,  1952, 1004,  946, 960,  988, 954,  974, 960,  978, 962,  976, 1888,  1010, 958,  1948, 1928,  980, 960,  3918, 3840,  1956, 1920,  1956, 1920,  978, 962,  1956, 996,  952, 956,  952, 984,  974, 1892,  1018, 952,  986, 952,  1956, 994,  944, 964,  982, 956,  952, 984,  954, 986,  984, 954,  954, 982,  974, 1890,  1006, 960,  1946, 1004,  954, 956,  952, 988,  952, 986,  952, 986,  952, 1914,  1014, 954,  1932, 1948,  948, 988,  3890, 3868,  1918, 1958,  1920, 1956,  952, 986,  1920, 1036,  922, 984,  954, 984,  954, 1912,  986, 980,  958, 980,  1926, 1026,  922, 988,  950, 986,  952, 990,  948, 988,  950, 988,  950, 988,  950, 1916,  982, 984,  1924, 1030,  916, 988,  950, 990,  950, 992,  956, 982,  956, 1906,  982, 986,  1920, 1956,  952, 992,  4804};  // AIRWELL B0E0181B
uint64_t data = 0xB0E0181B;

26 / heat / auto 

Protocol  : AIRWELL
Code      : 0xB0D8181B (32 Bits)
uint16_t rawData[163] = {2862, 3860,  1958, 1922,  1956, 1922,  984, 952,  1956, 996,  952, 956,  982, 956,  982, 1880,  988, 984,  1924, 1954,  954, 984,  1924, 1030,  948, 958,  980, 960,  978, 958,  980, 958,  980, 1886,  982, 986,  1952, 1000,  948, 964,  976, 960,  978, 960,  978, 960,  978, 1886,  1012, 958,  1950, 1930,  988, 948,  3918, 3838,  1958, 1920,  1948, 1932,  986, 952,  1956, 996,  942, 966,  972, 966,  984, 1880,  1008, 960,  1948, 1932,  986, 950,  1946, 1008,  922, 984,  974, 964,  984, 952,  974, 966,  952, 1910,  1008, 960,  1946, 1006,  922, 988,  950, 988,  950, 990,  958, 982,  946, 1918,  980, 988,  1920, 1954,  954, 986,  3890, 3864,  1922, 1958,  1920, 1960,  958, 978,  1920, 1032,  926, 982,  956, 982 [... cut ...]   1012, 956,  1952, 1000,  918, 990,  978, 960,  948, 988,  980, 960,  946, 1912,  1016, 956,  1952, 1928,  980, 956,  3902, 3856,  1920, 1958,  950, 988,  950, 990,  958, 978,  1920, 1032,  926, 982,  956, 982,  958, 1906,  980, 988,  1920, 1036,  922, 1908,  1958, 1024,  924, 988,  952, 984,  954, 986,  952, 984,  954, 1910,  988, 980,  1928, 1024,  924, 986,  952, 986,  952, 986,  952, 990,  950, 1910,  988, 984,  1924, 1956,  952, 984,  4804};  // AIRWELL F0C8181B
uint64_t data = 0xF0C8181B;

crankyoldgit, 2020-03-30:
Thanks. That's enough data for me for now.

I've made some improvements to the protocol, but I've still got some bugs in it.
I'll let you know when it's ready for you to try. FYI, the codes/`data =` values **will** change when I upload the new code. Don't get to dependant on those values for now.

crankyoldgit, 2020-04-01:
@skyderman I've updated the branch again. This time I think I have decoding fixed, and the 32-bit values that `IRrecvDumpV2` returns should be safe to use going forward.

Please try it and let me know how it goes.

skyderman, 2020-04-02:
Hi @crankyoldgit , I've updated, this is the result : 

When I change from 22° to 23° : 

Protocol  : UNKNOWN
Code      : 0x8E34167B (88 Bits)
uint16_t rawData[175] = {2890, 2918,  924, 1006,  946, 1886,  1986, 1922,  978, 958,  1952, 1000,  952, 956,  974, 964,  976, 1884,  1016, 952,  1948, 1006,  944, 962,  980, 958,  982, 958,  952, 982,  978, 960,  980, 956,  986, 956,  974, 1888,  1014, 954,  1948, 1002,  948, 960,  980, 960,  950, 986,  1006, 962,  980, 1854,  1036, 932,  1948, 1928,  984, 954,  3918, 2920,  922, 1008,  942, 1888,  1984, 1926,  984, 950,  1952, 1000,  952, 956,  984, 956,  954, 1908,  1014, 952,  1948, 1002,  948, 960,  950, 986,  984, 956,  954, 984,  956, 984,  946, 992,  948, 992,  948, 1910,  1012, 958,  1954, 996,  922, 984,  956, 986,  956, 980,  950, 988,  954, 1910,  1010, 956,  1946, 1932,  978, 962,  3922, 2914,  918, 1010,  920, 1912,  1980, 1928,  982, 956,  1944, 1006,  924, 982,  958, 982,  948, 1912,  1008, 958,  1952, 1000,  920, 988,  952, 982,  958, 984,  958, 978,  952, 986,  986, 952,  958, 984,  946, 1912,  1010, 962,  1948, 1004,  926, 980,  950, 988,  952, 984,  956, 982,  948, 1916,  1016, 952,  1950, 1926,  984, 952,  4830};  // UNKNOWN 8E34167B

When On / off  on : ( for next dump, I only past the "on/off" )
23 heat auto

Protocol  : AIRWELL
Code      : 0x4F3FE7E4 (32 Bits)
uint16_t rawData[169] = {2894, 3828,  1954, 1920,  1920, 1960,  982, 954,  1958, 994,  946, 962,  978, 958,  982, 1880,  1010, 958,  1952, 1004,  916, 988,  982, 956,  974, 962,  978, 962,  948, 988,  982, 956,  976, 966,  954, 1906,  1014, 954,  1946, 1004,  946, 960,  980, 960,  950, 986,  1006, 932,  948, 1916,  1034, 932,  1950, 1928,  982, 954,  3920, 3836,  1948, 1928,  1952, 1924,  976, 960,  1952, 1000,  950, 960,  950, 986,  974, 1886,  1004, 964,  1946, 1004,  948, 960,  980, 960,  952, 984,  956, 984,  956, 982,  948, 992,  948, 992,  948, 1912,  1008, 958,  1952, 998,  922, 986,  954, 984,  956, 980,  950, 990,  950, 1914,  1008, 958,  1954, 1924,  976, 958,  3924, 3830,  1952, 1924,  1948, 1932,  978, 956,  1954, 998,  922, 984,  956, 982,  960, 1904,  1008, 960,  1950, 1006,  926, 980,  950, 988,  952, 986,  954, 982,  958, 982,  948, 990,  950, 986,  954, 1908,  1012, 956,  1956, 996,  924, 984,  946, 990,  952, 988,  952, 984,  956, 1910,  1012, 956,  1946, 1930,  980, 956,  4826};  // AIRWELL 4F3FE7E4
uint64_t data = 0x4F3FE7E4;

24 / heat / auto 

Protocol  : AIRWELL
Code      : 0x4F37E7E4 (32 Bits)
uint16_t rawData[163] = {2890, 3828,  1954, 1922,  1950, 1926,  986, 954,  1958, 994,  956, 952,  948, 988,  982, 1880,  1010, 958,  1954, 1000,  920, 1912,  1980, 1002,  948, 960,  952, 988,  952, 984,  976, 962,  978, 1882,  1020, 950,  1952, 1000,  950, 956,  984, 956,  984, 952,  980, 960,  980, 1882,  1018, 948,  1952, 1924,  986, 954,  3950, 3802,  1950, 1926,  1954, 1920,  980, 962,  1950, 996,  952, 956,  984, 952,  978, 1884,  1016, 950,  1952, 1000,  950, 1882,  1978, 1004,  928, 978,  950, 990,  952, 984,  986, 956,  954, 1914,  1008, 956,  1924, 1026,  926, 982,  958, 978,  952, 988,  952, 982,  958, 1906,  1016, 954,  1946, 1928,  982, 956,  3916, 3838,  1954, 1920,  1950, 1926,  984, 954,  1946, 1004,  926, 984,  956, 980,  952, 1912,  1010, 958,  1954, 1000,  920, 1912,  1980, 1002,  928, 978,  952, 988,  952, 988,  954, 982,  958, 1904,  1016, 952,  1950, 1000,  930, 982,  948, 988,  952, 984,  956, 982,  960, 1904,  1018, 954,  1946, 1926,  986, 956,  4834};  // AIRWELL 4F37E7E4
uint64_t data = 0x4F37E7E4;

25 / heat /auto : 

Protocol  : AIRWELL
Code      : 0x4F2FE7E4 (32 Bits)
uint16_t rawData[163] = {2886, 3832,  1952, 1926,  1944, 1930,  980, 960,  1950, 998,  954, 956,  984, 952,  980, 1886,  1016, 952,  1948, 1928,  1952, 1002,  948, 958,  972, 964,  976, 964,  948, 988,  982, 958,  970, 1892,  1008, 958,  1954, 998,  942, 964,  976, 962,  980, 960,  980, 956,  974, 1890,  1010, 956,  1954, 1924,  978, 958,  3912, 3844,  1950, 1926,  1946, 1932,  980, 958,  1942, 1008,  912, 994,  976, 964,  968, 1894,  1006, 962,  858, 3018,  700, 2280,  884, 992,  948, 1022,  918, 1018,  912, 1024,  946, 994,  644, 2188,  522, 1476,  464, 2484,  438, 1476,  464, 1444,  498, 1436,  492, 1474,  466, 2370,  520, 1478,  462, 3382,  520, 1448,  460, 7264,  520, 3354,  528, 3350,  520, 1446,  464, 2486,  436, 1472,  468, 1472,  468, 2364,  528, 1468,  462, 3388,  524, 2454,  436, 1472,  468, 1468,  462, 1478,  462, 1474,  466, 1472,  468, 2366,  524, 1470,  470, 2482,  438, 1468,  462, 1478,  462, 1474,  466, 1502,  468, 2336,  526, 1470,  470, 3378,  524, 1446,  466};  // AIRWELL 4F2FE7E4
uint64_t data = 0x4F2FE7E4;

28 / cool / auto 

Protocol  : AIRWELL
Code      : 0x8F17E7E4 (32 Bits)
uint16_t rawData[169] = {2888, 3834,  1958, 992,  948, 1886,  1014, 950,  978, 962,  1948, 1002,  950, 958,  952, 988,  952, 1908,  1012, 956,  984, 954,  1958, 1920,  1950, 1000,  920, 988,  952, 986,  954, 982,  948, 992,  980, 1882,  1020, 950,  1950, 1000,  920, 988,  952, 986,  984, 952,  978, 960,  980, 1884,  1018, 950,  1950, 1926,  984, 956,  3946, 3804,  1946, 1004,  946, 1886,  1014, 958,  982, 952,  1950, 1002,  948, 960,  980, 958,  952, 1912,  1010, 956,  984, 952,  1948, 1928,  1954, 1000,  920, 986,  986, 954,  956, 980,  950, 990,  950, 1914,  1008, 964,  1946, 1000,  930, 982,  948, 986,  954, 984,  956, 986,  954, 1906,  1014, 954,  1946, 1928,  982, 956,  3916, 3840,  1952, 998,  922, 1908,  1012, 958,  982, 956,  1954, 998,  924, 982,  956, 980,  950, 1914,   [... cut ...]
uint16_t rawData[91] = {2860, 3862,  958, 984,  1956, 996,  952, 952,  978, 960,  978, 962,  948, 988,  980, 956,  986, 954,  950, 1910,  1010, 960,  1948, 1002,  946, 1884,  1984, 996,  952, 988,  952, 954,  984, 958,  952, 988,  952, 1912,  1008, 960,  1948, 1002,  946, 964,  944, 992,  976, 958,  980, 958,  950, 1910,  1008, 962,  1946, 1930,  978, 960,  3920, 3832,  956, 984,  1926, 1026,  922, 984,  956, 984,  954, 984,  944, 994,  944, 992,  948, 988,  950, 1914,  984, 980,  1918, 1032,  918, 1914,  1954, 1026,  924, 1012,  412, 304,  210};  // AIRWELL FF37E7E4
uint64_t data = 0xFF37E7E4;

skyderman, 2020-04-02:
I try
irsend.sendAirwell(0x9F2FE7E4);

And it's work 

What can I do to help you ?

crankyoldgit, 2020-04-02:
@skyderman I've updated the branch again, it should now match that `UNKNOWN` message in your last batch of captures. (i.e. It's value should be `0xB0C0181B`)

> What can I do to help you ?

It looks like we've got matching of the messages mostly sorted for now.
Your next task is reverse engineering what all the bits in side the code do.
e.g.
  https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#capture-some-raw-ir-messages-from-the-remote and onwards. You've already started collecting some data. Time for you to put that data to work.

When you've got it fully worked out, please create a **new** issue, and I'll help code it up and add support for it.

closed 2020-04-04 by crankyoldgit

crankyoldgit, 2020-04-09:
FYI, This has been included in the new v2.7.5 release of the library.

juanbaldovi, 2020-06-17:
Hello David
I have an Airwell HVAC with the RC-4 remote.
Using the current support included on this thread I have captured some data and created an [spreadsheet ](https://docs.google.com/spreadsheets/d/1MeqVsgXQAMSiAx7zNunG8qbFcIT8tqZtcjLmMa0GEng/edit?usp=sharing)with my finings.
Could you please have a look and let me know if that is enough for supporting the protocol within the library?

Also. When I capture the data the AIRWELL protocol is only recognized when the on/off button is pressed. When the fan or temp button is pressed, it is always recognized as UNKNOWN protocol.
Is that because the protocol is not fully supported in the current version?

Thanks!

NiKiZe, 2020-06-17:
> Hello David
> I have an Airwell HVAC with the RC-4 remote.
> Using the current support included on this thread I have captured some data and created an [spreadsheet [...]

Your sheet looks like a great base, more temp values should be great from min to max.

If you get UNKNOWN please post the raw data.
Any more details on model for the AC unit?

juanbaldovi, 2020-06-17:
Hello David
thanks for your quick response.
I have included temperatures from min (16) to max (30)
I have also included a new tab with raw data when the protocol is not detected. The interesting thing is that the protocol is identified when the on/off is pressed.

Regarding the unit.
Remote control is [https://www.amazon.es/dp/B07B4H1B3G/ref=asc_df_B07B4H1B3G1586793600000/?tag=idex05-21&creative=24646&creativeASIN=B07B4H1B3G&linkCode=df0] [[2](https://user-images.githubusercontent.com/29886403/28319721-61d651ea-6bd7-11e7-9141-1afc63ade853.png)]
The AC unit -> Airwell DLS 21 DCI R410 AW

NiKiZe, 2020-06-18:
To use the data in the sheet we would need to convert it into usable code, a process which we already have via DumpV2+
So at least I would prefer the raw text as comment rather than the sheet. But will have another look in a few minutes.

crankyoldgit, 2020-06-18:
@juanbaldovi Please create a new issue, rather than adding to an old **closed** issue. You can refer back to this issue in the new issue. As @NiKiZe has said, if you can provide the `rawData` in it's original format, that would be very helpful.

I'll try to look at the data you've provided thus far in the meantime.

crankyoldgit, 2020-06-18:
For the record, yes, it looks like it's a bug in `matchManchester()`

crankyoldgit, 2020-06-21:
FYI @skyderman You probably want to checkout PR #1200 as it will break existing codes that PR #1070 introduced.

As this is being discussed in newer issue (#1199) I'm going to lock comments on this issue. Please reply there, or open a new issue etc.

PR #1070 (2020-04-04): Experimental support for AIRWELL protocol.
* Add `matchManchester()` to IRrecv class.
* Add `sendManchester()` to IRsend class.
* Add `sendAirwell()` & `decodeAirwell`.
* Update support functions.
* Add unit tests.

Fixes #1069
commit 75726b26 (2020-04-04): Experimental support for AIRWELL protocol. (#1070)

* Add `matchManchester()` to IRrecv class.
* Add `sendManchester()` to IRsend class.
* Add `sendAirwell()` & `decodeAirwell`.
* Update support functions.
* Add unit tests.

Fixes #1069