crankyoldgit/IRremoteESP8266 #1073: IRremoteESP8266 2.7.4 crashes with WiFi :Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

mal19992, 2020-04-03:
It somehow interacts with WiFi. Original IRremote 2.2.3  https://github.com/z3t0/Arduino-IRremote works when myReceiver.enableIRIn(); is called **after** WiFi init, otherwise it crashes  with "Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)", see IRAM_ATTR isn't working at https://github.com/espressif/arduino-esp32/issues/1781 

The IRremoteESP826 library "version": "2.7.4", with a standard receive code:
<pre>static IRrecv myReceiver(IR_reciverPin);
static decode_results myDecode;
in setup:
  pinMode(IR_reciverPin,INPUT);
  myReceiver.enableIRIn();

in loop:
  if (myReceiver.decode(&myDecode)) {
    pult=myDecode.value;
    Serial.print(pult, HEX); Serial.print(F(" protocol=")); Serial.println(myDecode.decode_type);
    myReceiver.resume(); 
  }</pre>

does not work at all. It produces a different **Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.**

<pre>
CREATE WiFi AP YYYYYYY Ready
WIFI IP address = 192.168.4.1
ACCESS THE SITE AS http://192.168.4.1/

Starting web server on port: '80'
Starting stream server on port: '81'
# crash after WiFi
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core 1 register dump:
PC      : 0x4016e48b  PS      : 0x00060130  A0      : 0x800db0fc  A1      : 0x3ffce280  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x400813a0  A5      : 0x00000000  
A6      : 0x3ffc2fdc  A7      : 0x00000000  A8      : 0x800e2cf0  A9      : 0x3ffce230  
A10     : 0x00000105  A11     : 0x0000040e  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x400813a0  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
Backtrace: 0x4016e48b:0x3ffce280 0x400db0f9:0x3ffce2a0 0x400db161:0x3ffce2d0 0x400d4077:0x3ffce2f0 0x400d2505:0x3ffce310 0x400dcb86:0x3ffce3b0 0x400932cd:0x3ffce3d0
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10864
load:0x40080400,len:6432
entry 0x400806b8
</pre>
I also tried to switch the order of WiFi and myReceiver.enableIRIn(); initialization. This does help for original IRremote https://github.com/z3t0/Arduino-IRremote/issues/518 but does not help for IRremoteESP826.

crankyoldgit, 2020-04-03:
You haven't stated it, but your using an ESP32. Correct?
Can you please try some of the example code that uses WiFi and see if that crashes too?
IRMQTTServer has an info page, that lists core library versions etc. Could you please pass that info on.

Can you simplify your code down to the smallest you can that still causes the crash, so I can replicate it and test it at my end?

mal19992, 2020-04-03:
I use ESP32-CAM https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/ The easiest way to reproduce is this:

-  Take https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer sketch (it is among examples)
-  Apply attached patch, it adds IR calls to setup() and loop()
-  Observe constant reboots
- comment the line // myReceiver.enableIRIn(); No reboots, WiFi and everything now works OK 

reboots: <pre>
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core 1 register dump:

PC      : 0x40167ef7  PS      : 0x00060a30  A0      : 0x800d9f98  A1      : 0x3ffb1e80  

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x400813a0  A5      : 0x00000000  

A6      : 0x3ffc1ffc  A7      : 0x00000000  A8      : 0x800dfc68  A9      : 0x3ffb1e30  

A10     : 0x00000105  A11     : 0x0000040e  A12     : 0x00000000  A13     : 0x00000000  

A14     : 0x400813a0  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  

EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40167ef7:0x3ffb1e80 0x400d9f95:0x3ffb1ea0 0x400d9ffd:0x3ffb1ed0 0x400d3633:0x3ffb1ef0 0x400d16ce:0x3ffb1f10 0x400dae9a:0x3ffb1fb0 0x400932cd:0x3ffb1fd0

Rebooting...

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0018,len:4

load:0x3fff001c,len:1216

ho 0 tail 12 room 4

load:0x40078000,len:10864

load:0x40080400,len:6432

entry 0x400806b8

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core 1 register dump:

PC      : 0x40167ef7  PS      : 0x00060a30  A0      : 0x800d9f98  A1      : 0x3ffb1e80  

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x400813a0  A5      : 0x00000000  

A6      : 0x3ffc1ffc  A7      : 0x00000000  A8      : 0x800dfc68  A9      : 0x3ffb1e30  

A10     : 0x00000105  A11     : 0x0000040e  A12     : 0x00000000  A13     : 0x00000000  

A14     : 0x400813a0  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  

EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40167ef7:0x3ffb1e80 0x400d9f95:0x3ffb1ea0 0x400d9ffd:0x3ffb1ed0 0x400d3633:0x3ffb1ef0 0x400d16ce:0x3ffb1f10 0x400dae9a:0x3ffb1fb0 0x400932cd:0x3ffb1fd0

Rebooting...

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0018,len:4

load:0x3fff001c,len:1216

ho 0 tail 12 room 4

load:0x40078000,len:10864

load:0x40080400,len:6432

entry 0x400806b8

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core 1 register dump:

PC      : 0x40167ef7  PS      : 0x00060a30  A0      : 0x800d9f98  A1      : 0x3ffb1e80  

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x400813a0  A5      : 0x00000000  

A6      : 0x3ffc1ffc  A7      : 0x00000000  A8      : 0x800dfc68  A9      : 0x3ffb1e30  

A10     : 0x00000105  A11     : 0x0000040e  A12     : 0x00000000  A13     : 0x00000000  

A14     : 0x400813a0  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  

EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40167ef7:0x3ffb1e80 0x400d9f95:0x3ffb1ea0 0x400d9ffd:0x3ffb1ed0 0x400d3633:0x3ffb1ef0 0x400d16ce:0x3ffb1f10 0x400dae9a:0x3ffb1fb0 0x400932cd:0x3ffb1fd0

Rebooting...

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0018,len:4

load:0x3fff001c,len:1216

ho 0 tail 12 room 4

load:0x40078000,len:10864

load:0x40080400,len:6432

entry 0x400806b8

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core 1 register dump:

PC      : 0x40167ef7  PS      : 0x00060a30  A0      : 0x800d9f98  A1      : 0x3ffb1e80  

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x400813a0  A5      : 0x00000000  

A6      : 0x3ffc1ffc  A7      : 0x00000000  A8      : 0x800dfc68  A9      : 0x3ffb1e30  

A10     : 0x00000105  A11     : 0x0000040e  A12     : 0x00000000  A13     : 0x00000000  

A14     : 0x400813a0  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  

EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40167ef7:0x3ffb1e80 0x400d9f95:0x3ffb1ea0 0x400d9ffd:0x3ffb1ed0 0x400d3633:0x3ffb1ef0 0x400d16ce:0x3ffb1f10 0x400dae9a:0x3ffb1fb0 0x400932cd:0x3ffb1fd0

Rebooting...

</pre>

mal19992, 2020-04-03:
The patch
<pre>
--- CameraWebServer.ino.orig    2020-03-07 22:41:53.633571406 +0300
+++ CameraWebServer.ino 2020-04-04 01:20:07.291554345 +0300
@@ -1,5 +1,9 @@
 #include "esp_camera.h"
 #include <WiFi.h>
+#include <IRremoteESP8266.h>
+#include <IRrecv.h>
+static IRrecv myReceiver(2/*IR_reciverPin*/);
+static decode_results myDecode;
 
 //
 // WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality
@@ -8,11 +12,11 @@
 //
 
 // Select camera model
-#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
+//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
 //#define CAMERA_MODEL_ESP_EYE // Has PSRAM
 //#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
 //#define CAMERA_MODEL_M5STACK_WIDE    // Has PSRAM
-//#define CAMERA_MODEL_AI_THINKER // Has PSRAM
+#define CAMERA_MODEL_AI_THINKER // Has PSRAM
 //#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
 
 #include "camera_pins.h"
@@ -88,6 +92,10 @@
   s->set_hmirror(s, 1);
 #endif
 
+
+  pinMode(2/*IR_reciverPin*/,INPUT);
+  myReceiver.enableIRIn();
+
   WiFi.begin(ssid, password);
 
   while (WiFi.status() != WL_CONNECTED) {
@@ -106,5 +114,13 @@
 
 void loop() {
   // put your main code here, to run repeatedly:
-  delay(10000);
+  //delay(10000);
+  unsigned long pult=0;
+   if (myReceiver.decode(&myDecode)) {
+    pult=myDecode.value;
+    Serial.print(pult, HEX);
+    Serial.print(F(" protocol="));
+    Serial.println(myDecode.decode_type);
+    myReceiver.resume(); // готовы принимать следующий сигнал
+  }
 }
</pre>

mal19992, 2020-04-03:
[diff.patch.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4430226/diff.patch.txt)

crankyoldgit, 2020-04-03:
Again, please try our examples to see if it is broken there too.
I took a quick look at the code and saw they use the timers too. IR receiving requires it's own timer.
Try using a different one.
I.e. https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRrecv.cpp#L138

crankyoldgit, 2020-04-03:
FYI, I don't have an ESP32Cam module to test with, so I can't use that code (I think). I'll try it on bare ESP32, but it's likely not to work and it won't be the same.

crankyoldgit, 2020-04-04:
@mal19992 I read through the bug/issue you referenced in your original message. i.e. the one logged by @marcmerlin (Hi Marc. Long time etc. :-)

It appears there is some funkiness with SPIFFS and hardware timers on the ESP32.
My guess is a) ESP32-Cam software is using SPIFFS or b) as you indicate, via that bug/issue, the IRAM stuff is not working. All the interrupt handles our library uses should be being loaded into Interrupt Ram, and the amount used is tiny. Everything it uses is marked `volatile`, but even still that shouldn't cause the ESP32 to crash.

I'm far from an expert on SPIFFS on the ESP32, but my guess is something is the matter there. @marcmerlin worked around it by kind-of luck with `IRremote`. i.e. He admits he doesn't know why it should fix it, but it did.

My suggestions for work-arounds to try would be to disable the IR capture via `IRrecv.disableIRIn()` in the cam software just before it uses SPIFFS, and re-enable it after it's finished.

`IRremoteESP8266` **needs** that Hardware timer to do IR capture. Without it, it's useless. If SPIFFS on ESP32s doesn't play well with Timer Interrupts, then there is not much I can do other than to say don't use SPIFFS at the same time.

IF you find that I'm not loading the interrupt code correctly into IRAM via using the `IRAM_ATTR` attribute (i.e. That there is some better way) then let me know.

Here is all the code that is used in the interrupt handlers (i.e. IRAM): https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRrecv.cpp#L65-L126

The only non-interrupt/non-timer related function call is `micros()` and that's in the gpio interrupt.

All the timer interrupt related code is just: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRrecv.cpp#L68-L79
That code (cleaned up for the ESP32) is:
```
static IRAM_ATTR read_timeout(void) {
  portENTER_CRITICAL(&irremote_mux);
  if (irparams.rawlen) irparams.rcvstate = kStopState;
  portEXIT_CRITICAL(&irremote_mux);
}
```

If you get crashes with nothing connected to the IRrecv GPIO pin (@mal19992 do you? Have you tried that), then it can only be that code.

If the cam/SPIFFS software can't deal with that tiny bit of interrupt code, then I'm totally flummoxed.

mal19992, 2020-04-04:
If I comment out either section

- esp_err_t err = esp_camera_init(&config); .... to ... Serial.println("Before myReceiver.enableIRIn");
- myReceiver.enableIRIn();

it works without a crash. If both sections are enables --- it crashes with the message

Before esp_camera_init
Before esp_camera_sensor_get
Before myReceiver.enableIRIn
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40167f3f  PS

updated patch:
[diff2.patch.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4431103/diff2.patch.txt)

mal19992, 2020-04-04:
I looked at 

/usr/local/arduino/hardware/espressif/esp32/tools/sdk/include/esp32-camera/*

there seems to be no references to interrupts and SPIFFS. Phe pre-compiled library  seems have no refs to SPIFFS
/usr/local/arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-nm    /usr/local/arduino/hardware/espressif/esp32/tools/sdk/lib/libesp32-camera.a   >/tmp/nm-D.txt

[nm-D.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4431135/nm-D.txt)

crankyoldgit, 2020-04-04:
@mal19992 Can you please try loading https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino in your IDE and upload it to your ESP32?
If that doesn't work (i.e. crashes the ESP), please let me know.

If it does, please move on to trying https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer

That one has wifi, and will also provide useful info to me to work out what versions of libraries you are using, so I can try to investigate.

If it is crashing at the call to `irrecv.enableIRIn()` (https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRrecv.cpp#L201-L238) which is (cleanup):
```
// Set up and (re)start the IR capture mechanism.
//
// Args:
//   pullup: A flag indicating should the GPIO use the internal pullup resistor.
//           (Default: `false`. i.e. No.)
void IRrecv::enableIRIn(const bool pullup) {
  // ESP32's seem to require explicitly setting the GPIO to INPUT etc.
  // This wasn't required on the ESP8266s, but it shouldn't hurt to make sure.
  if (pullup) {
    pinMode(irparams.recvpin, INPUT_PULLUP);
  } else {
    pinMode(irparams.recvpin, INPUT);
  }
  // Initialize the ESP32 timer.
  timer = timerBegin(_timer_num, 80, true);  // 80MHz / 80 = 1 uSec granularity.
  // Set the timer so it only fires once, and set it's trigger in uSeconds.
  timerAlarmWrite(timer, MS_TO_USEC(irparams.timeout), ONCE);
  // Note: Interrupt needs to be attached before it can be enabled or disabled.
  timerAttachInterrupt(timer, &read_timeout, true);

  // Initialize state machine variables
  resume();

  // Attach Interrupt
  attachInterrupt(irparams.recvpin, gpio_intr, CHANGE);
}
```

There is not much to it. Once you've tested/tried this libraries example code on your system and confirmed they work okay. (i.e. don't cause the ESP to crap itself.) we can try to comment out parts of that routine to see what might be causing it in the Cam software. i.e. Try to narrow down if it is the GPIO interrupt or the Timer interrupt that is causing an issue.

crankyoldgit, 2020-04-04:
Also, not that it should make a difference, can you try removinf `static` from your line: `static IRrecv myReceiver(IR_reciverPin);` please?

crankyoldgit, 2020-04-04:
and .. just on the off chance, have you completely removed the `IRremote` library? `IRrecv.h` is common to both I think, just in case there is some mix up in your build env.

mal19992, 2020-04-04:
- I removed "static" and completely removed IRremote --- same thing
- I uploaded  https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino (replaced pin to 2) it works OK
- I did not compile https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer yet, it requires many external librarues

[pin2.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4431264/pin2.txt)

crankyoldgit, 2020-04-04:
Excellent. Then we know it's not an issue with what every core ESP library you are using.
i.e. All the library calls/interrupts etc are working as expected. There hasn't been some change to the ESP core library that has caused this library to break.
As you tested receiving a message, the interrupt routines (both) are working, thus they must be in IRAM. 

Maybe there is not enough IRAM available with the camera stuff loaded?!? Just a theory/guess?!

The `static` and removing the other library were a long shot (but easy to eliminate), thanks for checking.

[Q] What ESP32-Core library version are you using? When did you last update etc?

mal19992, 2020-04-04:
I used 
git clone https://github.com/espressif/arduino-esp32
on Mar, 7

mal19992, 2020-04-04:
I added more code to IRrecvDumpV2.ino (the .h files are from  https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer )
when FLAG_CAMERAT_TO_FAIL=1 it crashes, when FLAG_CAMERAT_TO_FAIL=0 it is not. 
[pin2b.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4431533/pin2b.txt)
[IRrecvDumpV2.ino.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4431535/IRrecvDumpV2.ino.txt)
The only difference is in config settings

crankyoldgit, 2020-04-04:
is `FLAG_CAMERAT_TO_FAIL` correct? A google search, and a git search in their repo doesn't find that string. e.g. https://github.com/espressif/arduino-esp32/search?q=FLAG_CAMERAT_TO_FAIL&unscoped_q=FLAG_CAMERAT_TO_FAIL

mal19992, 2020-04-04:
The FLAG_CAMERAT_TO_FAIL was my (misspelled) variable added to IRrecvDumpV2.ino dependong on which some camera config settings were set/unset, see 
https://github.com/crankyoldgit/IRremoteESP8266/files/4431535/IRrecvDumpV2.ino.txt
After that also pinned it down to 

config.pin_sscb_sda = SIOD_GPIO_NUM;

if one comment this line -- it stops crashing

marcmerlin, 2020-04-04:
Hi @crankyoldgit , so yeah, in https://github.com/espressif/arduino-esp32/issues/1781 I basically found that removing some code made some code cache not overflow and made the problem go away.
Suggestions, I can give:
1) use the RMT ESP32 IR library, and you won't need timers at all: https://github.com/lbernstone/IR32
2) Use FFAT instead of SPIFFS
http://marc.merlins.org/perso/arduino/post_2019-03-30_Using-FatFS-FFat-on-ESP32-Flash-With-Arduino.html

either of those two should fix your problem.

mal19992, 2020-04-04:
I tried  https://github.com/lbernstone/IR32 it does not work. It cannot receive a signal from TV remote. The remote2.available() in https://github.com/lbernstone/IR32/blob/master/examples/Loopback/Loopback.ino always returns 0, no any reaction on TV remote key pressed. A VS1838 is attached to pin 2, RX_PIN is set to 2, and all TX code removed.

marcmerlin, 2020-04-04:
Please ask him for help, I use it and it works for me.  You can find him in gitter.io in the esp32 channel
That said, try this simple code I sent him:
https://github.com/lbernstone/IR32/commit/8c33dc3165f14900f0fe3ea1965f64c7fcd6f06d

mal19992, 2020-04-05:
I did exactly like this 

[LB.ino.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4433438/LB.ino.txt)

remote2.active() always return 1 and remote2.available() always 0, no reaction on TV remote. The 
IRremoteESP8266 and https://github.com/KiLLAAA/Arduino-IRremote work OK

mal19992, 2020-04-05:
marcmerlin, thank you very much. It started to work after adding to setup() <pre>pinMode(RX_PIN,INPUT);</pre>
[LB.ino.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4433466/LB.ino.txt)

crankyoldgit, 2020-04-25:
Closing due to lack of activity.

closed 2020-04-25 by crankyoldgit

crankyoldgit, 2020-12-03:
FYI This may have been fixed with #1351