crankyoldgit/IRremoteESP8266 #1075: couple of questions/suggestions.

kdschlosser, 2020-04-07:
I am currently in the process of porting this library to Python. It is going to be able to run on Windows via EHome USBCIR (MCE Transceiver) and for the various serial/USB/IP transceivers that are available. It is also going to be able to run on an ESP8266 and ESP32 running MicroPython.

One of the things that struck me is I have not come across anything that indicates that the AC decoding and AC classes have an attachment to one another. Maybe I have missed it somewhere? And if so could you point me to where that connection between the receiving and the sending is made?

If there currently is no attachment that would be something worth considering doing. There are quite a few AC's that have 2 way IR communications with the remote control. It displays on the remote any of the following. operating mode, fan speed, swing vertical, swing horizontal, set temperature and any other features the air conditioner has. The AC that I have personally displays the current temperature. I am not sure if this temperature reading is from a sensor in the remote or a sensor in the physical unit But either way that reading has to be sent via IR to the unit or the remote. 

Here is a use case example.
If the remote displays the current set point and the user presses the temp up button on the actual unit that set point then gets sent to the remote.

Right now it appears as tho the data from the unit to the remote would be able to be decoded. There would need to be code added to parse the decoded IR and present it to the user. The receiving and sending would need to be tied together at the AC classes. This would make it easy for a user to use.

I would imaging formatting of the data from the unit to the remote would be similiar to what would be sent to the unit from the remote. I could be wrong in that assumption.

I will be creating a repository for the port to Python. I am still working on how to handle the different send/receive mechanics associated with the ESP8266, ESP32 and Windows. Because I will not be flashing the IR led directly and also not receiving the IR and recording the timings when running on Windows. This is typically done by the transceiver and the received data gets sent over the USB or serial as RLC. I do not believe it would be all to difficult to write some classes for each of the ESP chips that would do the same. and then I would pass the RLC and work with it from there. Would this be a good approach to use?

crankyoldgit, 2020-04-08:
> One of the things that struck me is I have not come across anything that indicates that the AC decoding and AC classes have an attachment to one another. Maybe I have missed it somewhere? And if so could you point me to where that connection between the receiving and the sending is made?

I'm not sure what you mean by "attachment to one another", can you please elaborate?

There is no direct link between receiving and sending. You can see the "repeater" example code for some things.
You can also see in `IRMQTTServer` where it will receive a message from an IR remote and convert it into a corresponding IR message for re-transmission etc. e.g. via [IRAcUtils::resultAcToString()](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L1983), [IRAcUtils::decodeToState()](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L2268-L2269) & [IRac::sendAc()](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L1300-L1307)

> There are quite a few AC's that have 2 way IR communications with the remote control.

[Citation Needed] :-) In fact, my experience has been almost the exact opposite. I think out of all the AirCon protocols I've added to this library, I think only one had a "from the AC unit to the Remote" communication, and it was something minor if I recall correctly. We are talking 30+ aircon types/models/brands compared to maybe 1-2 that have two-way in some form.

> Here is a use case example.
If the remote displays the current set point and the user presses the temp up button on the actual unit that set point then gets sent to the remote.

What do you mean by "the actual unit"? I've never come across an AC unit that has buttons on it that communicates to it's remote. It is almost 95+% the case that it is the IR remote is the sole control of the A/C Unit.

> There would need to be code added to parse the decoded IR and present it to the user. The receiving and sending would need to be tied together at the AC classes. This would make it easy for a user to use.

See: [IRAcUtils::resultAcToString()](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L1983) &[IRAcUtils::decodeToState()](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L2268-L2269)

Note: `IRMQTTServer` kind of already does what I think you're trying to do, using the features I've pointed out.

kdschlosser, 2020-04-08:
I am thinking you are limiting your thoughts as to what kind of an air conditioner.

There are a plethora if Mini Splits that are 2 way, they can also have a hard wired control panel that gets mounted on the wall in a different location then the actual AC. When I was helping a friend out building his house we installed 10 mini splits and they all had control panels and 2 way remotes. 

I have a portable (roll around) that has a 2 way remote. I would have to go and check to see who the manufacturer is, I cannot remember off the top of my head.

From what I have seen most Window AC's do not have 2 way remote controls.
The Sensibo (https://sensibo.com) is a device that allows a user to bring an AC into their Home Automation, I personally do not care for it because of it's size and it's requirement to be cloud attached, so when the internet goes out it stops functioning. I believe it works with just about every AC that has a remote, but it is more useful with mini splits that have 2 way remotes.

If you type in "mini split 2 way universal remote" into google you will see a whole lot of universal remotes supporting 2 way that have LCD screens on them. Too many for it not be a common thing.

What I was trying to explain is I do not see a way for a received IR signal that would be transmitted from the AC to be able to populate any of the data contained within any of the AC classes. The information contained within the AC classes is  based on the last IR code that was transmitted and not what the actual air conditioner is doing. This would only work for 2 way remotes of course.

From your post I am going to assume that this capability is not available.

crankyoldgit, 2020-04-08:
Most of the A/C units & remotes this library supports are split systems. There are a handful of duct ones too.

It seems our experiences differ.

Why don't you try capturing some of the AC-> Remote IR messages to see. My guess is they will the same protocol.

> What I was trying to explain is I do not see a way for a received IR signal that would be transmitted from the AC to be able to populate any of the data contained within any of the AC classes. The [...]

It all depends on the protocol used. Those routines I pointed you to do pretty much what you want.
And if they don't there are other ways. E.g. Each class's `setRaw()` for instance

crankyoldgit, 2020-04-08:
`acclassobj.setRaw(numeric_code_from_IR);`  will take an IR code, and populate the AC's class, enabling you to be able to use any of the `getBlah()` methods the class has.

e.g. https://github.com/crankyoldgit/IRremoteESP8266/blob/master/test/ir_Daikin_test.cpp#L1555-L1576 & https://github.com/crankyoldgit/IRremoteESP8266/blob/master/test/ir_Daikin_test.cpp#L3513-L3515

kdschlosser, 2020-04-08:
OK.. cool I never thought about being able to use that. This is why I asked the question. This mechanism will only work if the format of the IR data coming form the AC and going to the remote is the same as from the remote to the AC. If there is a variation in how the data has been formatted then this way will not work correct?

kdschlosser, 2020-04-08:
Oh and TY for taking the time to explain this to me. I do appreciate it. Once I get the code ported to python I will be able to run tests with it. I do have an ESP32 handy But I do not know if this library in it's current form will run on an ESP32.

BTW here is the repository for the port I am working on. 
https://github.com/kdschlosser/IRDecoder

crankyoldgit, 2020-04-08:
> But I do not know if this library in it's current form will run on an ESP32.

You didn't look very hard. :-)

First line of the README.md:
"This library enables you to send and receive infra-red signals on an ESP8266 or an ESP32 using the Arduino framework using common 940nm IR LEDs and common IR receiver modules."
I.e. the home page of this project/library.

Also, the FAQ: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Does_this_library_work_on_the_ESP32

crankyoldgit, 2020-04-14:
How are you going? Are we done with this issue or is there something I still need to answer?

kdschlosser, 2020-04-14:
Oh sorry about that. I forgot to close this. You did answer my question tho.
It can be done with existing mechanisms but is not "built in" to update the information automatically. It has to be done specifically in the users code.

closed 2020-04-14 by kdschlosser