crankyoldgit/IRremoteESP8266 #1076: generating the raw ir code for other systems (esphome)

chaoranxie, 2020-04-08:
hello there, 

Thank you for creating/improving this library. I have been able to get it and the example TurnOnGreeAC.ino running without issue. It is really nice to be able to use function like `ac.setTemp(20); ` to programmatically set different temperature. Out of all the ways I find controlling my AC, this library is the best for setting different temperatures, so a big thank you.

I am hoping to find a way to use this library to generate raw IR code for other systems. I am hoping you can point me at the right direction.

There are two different systems I am hoping to generate the raw IR code for. 
1. ESPHome https://esphome.io/components/remote_transmitter.html?highlight=dumper#remote-transmitter-transmit-raw-action which use the raw format of a list of integers
```
code (Required, list): The raw code to send as a list of integers. Positive numbers represent a digital high signal and negative numbers a digital low signal. The number itself encodes how long the signal should last (in microseconds).
```

2. Home Assistant https://www.home-assistant.io/integrations/remote.xiaomi_miio/ which use the raw format of either base64 string or pronto
```
- raw:base64:[optional_frequency]
- pronto:pronto_hex:[optional_repeat]
```

I am hoping there is a way to use your library and use method like   `ac.setTemp(20); ` and then maybe generate raw output for each of the above format. 

Each of the 3 systems all have some ways of dumping raw format from capture, though my experience has been that they do not work consistently. some command via the raw dump works while others do not. since your library works consistent in sending the correct signals, I am hoping to generate the raw code.

crankyoldgit, 2020-04-08:
I'd strongly recommend you use a method other than "raw" to do this. See: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Why_is_the_raw_data_for_a_button_or_AC_state_always_different_for_each_capture
You really want to use a higher level abstraction (language) etc.

For example, try the [IRMQTTServer](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer) code. Instructions in the file. I have it controlling my A/C via MQTT via HomeAssistant/Google Home etc.
You can even do it via HTML too.

Tasmota also uses this library to provide a similar JSON interface to control A/Cs.

But as you want to go the "raw" route, I won't stop you. ;-)
Now, to answer your questions.

> I am hoping to find a way to use this library to generate raw IR code for other systems. I am hoping you can point me at the right direction.

There are few ways to covert an `ac` class object into a `raw` format.
* If you look at some of the Unit Tests for some A/C protocols you can see how I convert a "code" by simulating sending it, and then dumping the result to text, using these routines:
```
String resultToSourceCode(const decode_results * const results);
String resultToTimingInfo(const decode_results * const results);
String resultToHumanReadableBasic(const decode_results * const results);
```
e.g. https://github.com/crankyoldgit/IRremoteESP8266/blob/95a2563a1440bdb2427dab9f4c47d1091a9b23c5/test/IRutils_test.cpp#L165-L297
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/test/ir_Daikin_test.cpp#L3450-L3462

* If you have Global Cache or Pronto Codes you can use:
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/gc_decode.cpp

But again, honestly, avoid using "Raw" format if you can, use IRMQTTServer, or the IRHvac json stuff in Tasmota etc. I'm sure ESPEASY has something similar too.

chaoranxie, 2020-04-09:
hi crankyoldgit,

Thanks so much for your fast and detailed response. Looks like I have a lot more resources to look at now to help solve my specific problems. 

Like in your FAQ, I think the trouble I had with other library and system is that they use raw code from capture and because the inconsistently in the analog capture, the "replay" could also be bad.  Totally agree that a higher level abstraction is what I want this is the exact reason I found your library to be helpful with all the ways for selecting brand and setting temperature instead of dealing with codes you find online. I am hoping the generated code via your library will at least be "perfect" digitally before being sent out. 

IRMQTTServer definitely seems like the ideal solution to one of my problems. I tried to get it to run but is not able to, I am using a[ m3stick-c](https://github.com/m5stack/M5Stack) which is based on esp32 pico i think. when i try to upload it as it is I got error like `Multiple libraries were found for "WiFi.h"`. from the documentation in the IRMQTTServer file, looks like i need to specific to use the dev branch but i actually don't know how, installing the library via Arduino IDE only show some beta version, so i guess I might have to git clone the library instead of using the library manager? 

at the moment, it feels like a lot of work to get my m3stick-c to work since I am new to Arduino/ESP8266/ESP32/Pin layout/ESPHome/ESPEasy/Tasmota(did flash my sonoff switch with it before)  ^____^.   I just purchased a nodemcu which I think is esp8266 and i do have a IR adapter that i could connect to it. so I think I will wait for that to arrive and start from there. 

If I get nodemcu to work with IRMQTTServer, i think that will solve half of my problems, so I dont have to use ESPHome/ESPEasy/Tasmota to connect to home assistant. just for completeness though, do you think the info from `resultToTimingInfo` is the input to the ESPHome raw code `a list of integers. Positive numbers represent a digital high signal and negative numbers a digital low signal. The number itself encodes how long the signal should last (in microseconds).`  from what I see it does look to have similar format

I think another issue I ran into when using was maybe my m5stickc/esp32 pico was "slow" in sending out the ir signals. Like I am using ESPHome and one of the well known raw code for my AC/TV, it works sometimes, if i capture the signals send by the device, sometime it is a long list of integers, which is similar to what I put in as raw, sometime i got multiple short list of integers. if my IR receiver is getting inconsistent result, i think thats why my TV and AC is also getting inconsistent results.

This is the second half of my problem, It was because of that I got the https://www.home-assistant.io/integrations/remote.xiaomi_miio/ it captures and replay much better than my m5stickc, there are multiple IR led inside. and it works really well with an phone app and relatively well with Home Assistant, the only downside is that it does not have that abstraction, i still have to deal with its own version of the raw code. 

I guess this is my follow up question, lets say i use `resultToSourceCode` and got a result similar to 
https://github.com/crankyoldgit/IRremoteESP8266/blob/95a2563a1440bdb2427dab9f4c47d1091a9b23c5/test/IRutils_test.cpp#L178

how do I go about converting it to something like `Z6UFANEAAAAjAQAAAwkAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQIAE=` . My assumptions is all of these format are all standard ways of representing a list of integers, i should be able to find a converter online to convert between them, but I might totally be wrong here. 

Again, thank you so much for your help, i will report back on how the nodemcu goes

crankyoldgit, 2020-04-09:
> IRMQTTServer definitely seems like the ideal solution to one of my problems. I tried to get it to run but is not able to, I am using a[ m3stick-c](https://github.com/m5stack/M5Stack) which is based [...]

Probably. It sounds like you've got something odd in your build environment.

> I just purchased a nodemcu which I think is esp8266 and i do have a IR adapter that i could connect to it. so I think I will wait for that to arrive and start from there.

Sounds like a good plan.

> do you think the info from resultToTimingInfo is the input to the ESPHome raw code

I can't be certain without looking at their code. Sounds likely.

> I think another issue I ran into when using was maybe my m5stickc/esp32 pico was "slow" in sending out the ir signals. Like I am using ESPHome and one of the well known raw code for my AC/TV, it [...]

There could be a number of reasons you are having problems. Based on the info you've provided I can't tell from here.
I suggest you try the [troubleshooting guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide).

> how do I go about converting it to something like `Z6UFANEAAAAjAQAAAwkAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQIAE=`.

No idea. It looks like it's [Base64](https://en.wikipedia.org/wiki/Base64) encoded data of some kind. I suggest you try googling the Xiaomi device to see if people have worked out it's format.

chaoranxie, 2020-04-09:
thanks again, I did some googling and it seems like my understanding of raw format is incorrect. Here is my new understanding, do let me know if it sounds reasonable.

I thought raw format is a `standard` format that you could find online from manufacturer or capture via ir receiver and you should be able to use it with all different libraries/platforms.

It turns out for Xiaomi IR transmitter devices and probably others like Broadlink, the raw codes are all different and cant be reused from one another. The standard format I might be looking for is the `pronto format`, which which you can find the code for different commands for your specific device/TV/AC. 

Xiaomi IR device for example, you can ask it to send `pronto` code. it also supports raw format, which is really a special encoding of the pronto code but embedded with other information specifically for Xiaomi device. In fact, Xiaomi has at least 4 different IR devices and the raw code for one do not work for the other. 

The good news is that for each device you have, the raw code you capture should work if you send the raw code from the same device. 

I am closing this issue for now since you have answer all my questions.

reference
1. https://github.com/rytilahti/python-miio/issues/495

closed 2020-04-09 by chaoranxie

crankyoldgit, 2020-04-09:
> thanks again, I did some googling and it seems like my understanding of raw format is incorrect. Here is my new understanding, do let me know if it sounds reasonable.

To me, it sounds like you've got the gist of it.

> I thought raw format is a `standard` format that you could find online from manufacturer or capture via ir receiver and you should be able to use it with all different libraries/platforms.

Correct. "Raw" format is not a standard. The Arduino libraries `IRremote` & `IRremoteESP8266` share the same "raw" format as this library is a fork of that library. It's about as low-level a format as you can have for IR messages. A lot of Arduino/ESP projects use this "raw" format because they are using this library. They even try to compress it, as it is very verbose/wasteful of data etc. Pronto & GlobalCache are two (cough) standards that are more widely used in the IR (learning) remote industry. This library supports sending both of them.