crankyoldgit/IRremoteESP8266 #1079: MQTT coolix remote

cab113, 2020-04-15:
Hi, i dont have a issue with the library or your code but im needing help to make a mqtt IRremote for Coolix unit. i have compiled a example to test coolix using the Library and works perfectly, and have compiled my ESP remote, using the same Library version and the codes are a little off 
### Version/revision of the library used
2.7.5
### Expected behavior
code sent from (attached works perfectly and unit turns on, My IR receiver receives Coolix code "B2BFC0", unit turns on,cool mode,25,auto.
[TurnOn_CoolixAC - Copy.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4479262/TurnOn_CoolixAC.-.Copy.txt)

My MQTT IRremote on same settings sends "B2BBC0", recorded on IRreceiver as coolix "B2BBC0" - No response from the unit. the strange thing is OFF code is correct on IRremote " B27BE0".

it may be my compile is missing a bit of code on coolix IR trasmission. my attached files show my files. Note: it contains two set of codes for DAIKIN and Coolix. daikin code works perfectly on my Daikin unit.
This is not not my work its from OZNU http daikin homebridge remote from github.

[Ac.h.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4479284/Ac.h.txt)
[settings.h.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4479285/settings.h.txt)
[thermostat.ino.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4479286/thermostat.ino.txt)
[Ac.cpp.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4479287/Ac.cpp.txt)

crankyoldgit, 2020-04-15:
Can you upload your code to github in the form of a proper project/repo etc?
That will make the code much easier to read and to download/test etc.

cab113, 2020-04-15:
Ok, will give it a try haven’t done it yet, my codes are attached on original message, just delete the .txt Extensions.

crankyoldgit, 2020-04-15:
> just delete the .txt Extensions.

I know. It just makes it easier for me and anyone else.

FYI: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#things-we-are-unlikely-to-help-you-with
```
Fixing code in some other maintained project that includes our library, unless you're one of the maintainers of that project. You should probably report an issue with them instead of us.
```
You should try contacting the owner of: https://github.com/oznu/homebridge-daikin-esp8266

FYI, code `0xB2BBC0` is `"Power: On, Mode: 0 (Cool), Fan: 5 (Auto), Temp: 25C, Zone Follow: Off, Sensor Temp: 27C"`

It looks like a valid/reasonable code.
How sure are you that is what is being sent?
Have you tried capturing what you're sending with another ESP running `IRrecvDumpV2`?

cab113, 2020-04-18:
Found the missing code for correct IR message i needed to add temp.sensor ignore to be added to IR send

closed 2020-04-18 by cab113