crankyoldgit/IRremoteESP8266 #1081: sendRaw function accepts uint16_t[] not uint32_t[]

arihantdaga, 2020-04-16:
I apologize for not filling the issue in the template. I could not understand what to fill for the context of this problem :) This may not exactly be a bug. 

### Version/revision of the library used
2.7.5

### Expected behavior
I am using irsend.sendRaw function. For some of the remotes which I decoded, the raw sequence has values like 80000. But I saw that sendRaw function accepts uint16_t[] so it'll overflow unsigned int and the signal of 14464 will be sent, and the ir signal is not working, probably because of that. Also, i saw that space() accepts uint32_t but mark accepts uint16_t. Is that intentional and i am missing something? Or this should be accepting uint32_t as well.

crankyoldgit, 2020-04-16:
> the raw sequence has values like 80000

Wow. That's unusual in a number of ways. 

First, Did you change the `kTimeout` value in `IRrecvDumpV2.ino` per chance?

e.g.
```
#if DECODE_AC
// Some A/C units have gaps in their protocols of ~40ms. e.g. Kelvinator
// A value this large may swallow repeats of some protocols
const uint8_t kTimeout = 50;
#else   // DECODE_AC
// Suits most messages, while not swallowing many repeats.
const uint8_t kTimeout = 15;
#endif  // DECODE_AC
```

By default, it should cap at 50ms (i.e. 50,000 usecs) so `80000` shouldn't be possible really.

Second, [String resultToSourceCode(const decode_results * const results)](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRutils.cpp#L587-L664) used by  `IRrecvDumpV2.ino`.

In short, this library's copy of `IRrecvDumpV2.ino` shouldn't produce a value larger than `UINT16_MAX` (65535). It will encode any `value` larger than that into "..., 65535, 0, (`value` - 65535), ..." which the library realises is "keep doing the last thing we did but for longer." `resultToSourceCode()` does that encoding for people already, if a value that big exists.
To see that in action, take a look at the testing code at: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/test/IRutils_test.cpp#L158-L162
In short, the program should never give you a value that large. Something is wrong.

> But I saw that sendRaw function accepts uint16_t[] so it'll overflow unsigned int and the signal of 14464 will be sent, and the ir signal is not working, probably because of that.

Yes, `sendRaw()` is limited to using `uint16_t`s, as that's what the original format was/is. Also, it would mean wasting twice the space to store the array when there is _almost_ never a long mark or space above 65535usecs. i.e. 16bit values vs 32bit values.

You are correct that `space()` accepts 32bit values, because it turns out there are circumstances where there are "gaps"/spaces that are long than 65535 usecs. But we've never seen a `mark()` signal that big before, not even close.

So, the **TL;DR:** is, I think you've screwed up your captures somehow when using this library's `IRrecvDumpV2`. But if for some reason you think you haven't changed anything, and it is producing a value that big, please cut & paste the **entire** text output from `IRrecvDumpV2` for that signal, as it could be a bug. That data may help me track it down.

arihantdaga, 2020-04-16:
@crankyoldgit . Thank you very much for explaining. I am sorry, actually, the raw string i had was coming from another database and not from `IRrecvDumpV2`. I think it's a fault from the source or they could have captured two packets (Usually, some remotes keep repeating the commands untill you don't release the button)  and the gap of 80ms was interval between 2 commands. But somehow it was not working and the receiver was not taking any action on firing part of the raw code before 80ms. I'll assume something is wrong at my end. 
But thank you very much for taking the time to reply :)

closed 2020-04-16 by arihantdaga