crankyoldgit/IRremoteESP8266 #1082: sendAc function will over toggle. When using it without previous state

arihantdaga, 2020-04-16:
### Version/revision of the library used
2.7.5
### Expected behavior
Referring to this line https://github.com/crankyoldgit/IRremoteESP8266/blob/ebaf05d4dd07182c4786ae1aaaae8ecb9d85ac37/src/IRac.cpp#L1346
Shouldn't this be  `this->sendAc(to_send, NULL);` ? Because if we send the same state then by this line https://github.com/crankyoldgit/IRremoteESP8266/blob/ebaf05d4dd07182c4786ae1aaaae8ecb9d85ac37/src/IRac.cpp#L1361

It'll toggle values of the desired state, without the need. 

1. Initialise the IRsend class.

```
// Psuedo code
bool power = true;
decode_type_t vendor DAIKIN128;

irac.sendAc(vendor, model, power,mode, degrees...);
```

### Actual behavior
Even though power was set to true, it'll play power = false; because it will be toggled. 

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
Not Sure

crankyoldgit, 2020-04-16:
I believe that is correct behaviour. For that protocol (`DAIKIN128`) it doesn't have a true power on or off state, but it uses a `setPowerToggle(bool);`. e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Daikin.cpp#L2446-L2454

So, `state.power` for a `DAIKIN128` etc isn't whether the power is on or off, but if it should be `toggled` (`true`) or left alone (`false`).

See also: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L347

i.e.
if `power` was `false` (don't toggle) and the device was already on; then `handleToggles()`'s:
```
    case decode_type_t::DAIKIN128:
        result.power = desired.power ^ prev->power;
        ...
```
then
`false (desired) ^ false (prev)` which is `false` and that means `setPowerToggle(false)` i.e. don't toggle, which is what we want.

If we now want to power off the device, we'd set `state.power` to `true`.
Thus we'd get:
`true (desired) ^ false (prev)` which is `true` thus, `setPowerToggle(true)`. i.e. Please change the power on the device (which was on, will become off)

So it should only "change" when the power state is different from the previous state.

Or am I missing something more fundamental? (If so, can you provide data etc so I can see/understand)

crankyoldgit, 2020-04-16:
> 1. Initialise the IRsend class.
> 
> ```
> // Psuedo code
> bool power = true;
> decode_type_t vendor DAIKIN128;
> 
> irac.sendAc(vendor, model, power,mode, degrees...);
> ```

Oh, I think I get what you are doing now. You are calling it without a "previous" state. Some of these protocols work "best" if you supply/track that. i.e. any protocol that uses a toggle.

i.e. Of the three `sendAc`s
```
  bool sendAc(void);
  bool sendAc(const stdAc::state_t desired, const stdAc::state_t *prev = NULL);
  bool sendAc(const decode_type_t vendor, const int16_t model,
              const bool power, const stdAc::opmode_t mode, const float degrees,
              const bool celsius, const stdAc::fanspeed_t fan,
              const stdAc::swingv_t swingv, const stdAc::swingh_t swingh,
              const bool quiet, const bool turbo, const bool econo,
              const bool light, const bool filter, const bool clean,
              const bool beep, const int16_t sleep = -1,
              const int16_t clock = -1);
```
You should be using the first two (or just the first one) as they handle the complexities better. The last variant assumes you know what you are really doing. As in you know what each parameter really does on the underlying device.

Think of it this way, the first two are cooked, and mostly cooked. The last one is raw (here be dragons!) ;-)

arihantdaga, 2020-04-16:
Thank you very much. I got it.

closed 2020-04-16 by arihantdaga