crankyoldgit/IRremoteESP8266 #1083: MQTT climate discovery FAILED to send

chaoranxie, 2020-04-16:
### Version/revision of the library used
v2.7.5

### Expected behavior and Actual behavior
running "Send MQTT Discovery" return success, but i got the failed message. Note that I am able to connect to MQTT and see the device connected and sent other messages.

#### Example code used
IRMQTTServer.h at master branch with following changes
#define MQTT_CLIMATE_JSON true

### Other useful information
I am able to send IR signal to my AC using the aircon page

crankyoldgit, 2020-04-16:
Have you been able to control your ESP via MQTT?

crankyoldgit, 2020-04-17:
Also, what does the line "Last discovery sent:" say on your info page?

The MQTT discovery message will be sent to the MQTT Topic:
`homeassistant/climate/WhatEverYouSetHostnameToOnSetup/config`
i.e. See: `Hostname:` on the `/info` page for what value of `hostname` it will be using.

Have you tried listening to everything sent to your MQTT Server to see if it is being sent?
i.e.
`$ mosquitto_sub -h your_mqtt_server_address -v -t \#`
e.g.
`mosquitto_sub -h 10.0.0.4 -v -t \#`

crankyoldgit, 2020-04-17:
Heck, please cut & paste the your entire "Info" page please, that may highlight some problems if any.

chaoranxie, 2020-04-17:
I have not tried to control the ESP via mqtt itself, i am not sure what message to post to which topic, but I am able to use the webpage to send command to my AC and it does work. 

```
General
Hostname: ir_server
IP address: 192.168.1.105
MAC address: D8:A0:1D:57:36:11
Booted: 3 Minutes 43 Seconds ago
Version: v1.4.7
Built: Apr 16 2020 22:18:08
Period Offset: -2us
IR Lib Version: 2.7.5
ESP32 SDK Version: v3.2.3-14-gd3e562907
Cpu Freq: 240MHz
IR Send GPIO(s): 9 (default)
Inverting GPIO output: On
Total send requests: 1
Last message sent: FAILED (Never)
Duplicate Wifi networks: Show
Min Wifi signal required: 8%
Serial debugging: On

MQTT Information
Server: 192.168.1.113:1883 (Connected 3 Minutes 2 Seconds ago)
Disconnections: 1
Client id: ir_server561da0d8
Command topic(s): ir_server/send, ir_server/send_0
Acknowledgements topic: ir_server/sent
Log topic: ir_server/log
LWT topic: ir_server/status
QoS: 1
Last MQTT command seen: (topic) 'ir_server/ac/stat/light' (payload) 'off' (3 Minutes 36 Seconds ago)
Total published: 18
Total received: 8

Climate Information
IR Send GPIO: 9
Last update source: HTTP
Total sent: 1
Last send: Ok (2 Minutes 33 Seconds ago)
State listen period: 5 Seconds
State broadcast period: 10 Minutes
Last state broadcast: Never
Last discovery sent: Never
Command topics: ir_server/ac/cmnd/(protocol|model|power|mode|temp|fanspeed|swingv|swingh|quiet|turbo|light|beep|econo|sleep|filter|clean|use_celsius|resend|json)
State topics: ir_server/ac/stat/(protocol|model|power|mode|temp|fanspeed|swingv|swingh|quiet|turbo|light|beep|econo|sleep|filter|clean|use_celsius|resend|json)
```

chaoranxie, 2020-04-17:
This is what I see on the mqtt side
[image]

crankyoldgit, 2020-04-17:
Thanks for the data. I'll take a look to see what the issue might be shortly.

crankyoldgit, 2020-04-17:
@chaoranxie How did you build the `IRMQTTServer` binary/firmware? Arduino IDE or PlatformIO?

Did you follow the steps listed here?:

https://github.com/crankyoldgit/IRremoteESP8266/blob/636c3fe3455a69b7e766a6302e777d904aea960d/examples/IRMQTTServer/IRMQTTServer.ino#L38-L40

chaoranxie, 2020-04-17:
my apologies, i did see that at one point but forgot to change. I just made the change in PubSubClient it is able to send discovery message now. Thank you so much and sorry for the trouble,

closed 2020-04-17 by chaoranxie

crankyoldgit, 2020-04-17:
> my apologies, i did see that at one point but forgot to change.

No worries. I've created PR #1084 to make sure this doesn't happen in the future to anyone else. ;-)

crankyoldgit, 2020-04-25:
FYI, the extra check(s) have been included in the latest release of the library.

chaoranxie, 2020-04-27:
thanks, i tested it using platformio setup, really cool, didn't know it exists. btw I now have your IRMQTTServer running on both M5StickC (ESP32 pico) and NodeMcu with KY-005. Been learning a lot about all this via your library so thanks again

iondarie, 2020-10-12:
Hello, 

I installed the latest version, modified PubSubClient.h with "#define MQTT_MAX_PACKET_SIZE 768" but it refuses to send MQTT Discovery: "MQTT climate discovery FAILED to send."

Is there anything else i can try?

Thank you,

crankyoldgit, 2020-10-12:
> Hello,
> 
> I installed the latest version, modified PubSubClient.h with "#define MQTT_MAX_PACKET_SIZE 768" but it refuses to send MQTT Discovery: "MQTT climate discovery FAILED to send."
> 
> Is there anything else i can try?
> 
> Thank you,

Rather than adding to a long-closed issue, please consider opening a new issue.

If you really are using the latest version(s) of everything, the library no longer requires that parameter changed in the PubSubClient library.

See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/393de0fea74317e5b13b05266b33116e15de64f0/examples/IRMQTTServer/IRMQTTServer.h#L116-L123
& PR #1271

You may need to use a number larger than `768` for discovery to work.

PR #1084 (2020-04-17): Ensure MQTT_MAX_PACKET_SIZE is correctly set.
* Force the compiler to error if the value of MQTT_MAX_PACKET_SIZE is 
too small.
* List the Max MQTT packet size & discovery topic in the info page.
* IRMQTTServer version bump to 1.4.8

Fixes #1083
(Well, it may catch these issues earlier!)
commit cddbdaf4 (2020-04-17): Ensure MQTT_MAX_PACKET_SIZE is correctly set. (#1084)

* Force the compiler to error if the value of MQTT_MAX_PACKET_SIZE is 
too small.
* List the Max MQTT packet size and discovery topic in the info page.
* Version bump to 1.4.8

Fixes #1083
(Well, it may catch these issues earlier!)