crankyoldgit/IRremoteESP8266 #1088: Request, add dependencies for development to wiki

NiKiZe, 2020-04-19:
The guide contains information about code standard, running tests etc.
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Library-Maintainers-Guide

However there is no list of dependencies, for example that `gtest` is required.
also unsure of how to run `cpplint` properly etc.

It would be nice to see a full list of dependencies for tests and python etc.
Also a list of "commands to run before commit" or similar in one easy collected place.

NiKiZe, 2020-04-19:
trying to figure this out 
```
 $ make run_tests
make: *** No rule to make target '../lib/googletest/googletest/include/gtest/*.h', needed by 'IRutils_test.o'.  Stop.
```

this is due to missing submodule
`git submodule init`
`git submodule update`
should get this sorted

crankyoldgit, 2020-04-19:
The guide you reference tells you how to run the tests:
e.g.
```
To run the tests yourself, try the following:

$ cd test
$ make run_tests
```

You are correct on a lot of the other fronts, I could add more documentation on how to install `gtest`
e.g.
```
$ cd test
$ make install-googletest
$ make run_tests
```

A lot of other people figure out what to do by looking at the https://github.com/crankyoldgit/IRremoteESP8266/blob/master/.travis.yml file.

As this runs the tests on EVERY commit that's uploaded to github.
That file contains how to call/run every automatic test.
e.g. for the linters:
https://github.com/crankyoldgit/IRremoteESP8266/blob/1706fa70cf7b6f3bc422d7b3d5a3d9a5838fb9ed/.travis.yml#L44-L45

The google test library isn't listed as an official dependency as it isn't required for the project. Only for testing/development etc. I'm not going to subject typical "Arduino IDE" users to having to install "googletest" and building it (when it doesn't run on/under the Arduino Framework or on the microcontrollers) when it's not required for regular usage.

I'll state it again, yep, the documentation in general needs a lot more work. Especially on the development side.  You raise a fair number of valid points/areas for improvement.

If you are willing to update/tackle that aspect, I'm happy to add you as a contributor and you can go to town etc. ;-)

My personal habits are to have a unix terminal open, that I regularly run `make run` inside of the `test` directory. I typically won't even think of starting a `git commit` before that's passing.
Then before I'm ready to do a `git push` I'll run the linter, and fix any mistakes I find.
i.e. From the `root` git directory for the project (The one where src, test, etc live):
```
cpplint.py --extensions=c,cc,cpp,ino --headers=h,hpp {src,src/locale,test,tools}/*.{h,c,cc,cpp,hpp,ino} examples/*/*.{h,c,cc,cpp,hpp,ino}
```

That catches 95% of the things before I commit, but I'm human and I forget, make last second changes and well, Murphy's law etc. Hence `travis` is there to make sure.

NiKiZe, 2020-04-19:
I don't want to suggest to add this as "project" dependencies, only "development" in connection to how to run tests and all the other housekeeping stuff.

But another suggestion is also to just add a reference to the travis file, "please take a look on the travis file for examples of the below" would be great.

Thanks

crankyoldgit, 2020-04-19:
As I said, I'm offering you "contributor" access (if you want) so you can add it yourself. ;-)

crankyoldgit, 2020-04-25:
Take a look at: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Library-Maintainers-Guide#before-you-create-a-pull-request

I think it answers pretty much all of your points.
I'll look into the `git submodule` stuff shortly.

NiKiZe, 2020-04-25:
Looks good, thanks!

closed 2020-04-25 by NiKiZe