crankyoldgit/IRremoteESP8266 #1092: Home Assistant MQTT wrong behavior about DAIKIN NEW MODULE FFN-C(DAIKIN64)

chongkk, 2020-04-29:
Hi, sorry for delay report, finally now I can access the actual device about DAIKIN FFN-C(DAIKIN64)

Previous issue for reference: 1064, https://github.com/crankyoldgit/IRremoteESP8266/issues/1064
IRremoteESP8266 VERSION: v2.7.6

When I test on home assistant, the behavior step by step will be wrongly like that:
AC origin status: POWER OFF, HA status: POWER OFF,
Action 1:
Change HA AC from "OFF" to "COOL", AC will be beep(received some command), but won't be turn on. HA now status: COOL.
```
mqtt:
ir_server_study_room/ac/cmnd/power ON
ir_server_study_room/ac/cmnd/mode cool
ir_server_study_room/ac/stat/power on
ir_server_study_room/ac/stat/mode off
ir_server_study_room/ac/stat/power on
ir_server_study_room/ac/stat/mode cool
```

Action 2:
Change HA AC FROM "COOL" to "OFF", AC will be turn on. HA now Status: OFF.
```
mqtt:
ir_server_study_room/ac/cmnd/power OFF
ir_server_study_room/ac/cmnd/mode off
ir_server_study_room/ac/stat/power off
ir_server_study_room/ac/stat/mode off
ir_server_study_room/ac/stat/power off
ir_server_study_room/ac/stat/mode off
```

Action 3:
Change HA AC from "OFF" to "COOL", AC will be beep(received some command), still Power On. HA now status: COOL
```
mqtt:
ir_server_study_room/ac/cmnd/power ON
ir_server_study_room/ac/cmnd/mode cool
ir_server_study_room/ac/stat/power on
ir_server_study_room/ac/stat/mode off
ir_server_study_room/ac/stat/power on
ir_server_study_room/ac/stat/mode cool
```

Action 4:
Change HA AC from "COOL" to "OFF", AC Will be turn off. HA now Status: OFF.
```
mqtt:
ir_server_study_room/ac/cmnd/power OFF
ir_server_study_room/ac/cmnd/mode off
ir_server_study_room/ac/stat/power off
ir_server_study_room/ac/stat/mode off
ir_server_study_room/ac/stat/power off
ir_server_study_room/ac/stat/mode off

```
If I use IR Remoter control like last 4 steps, the HA AC status will be show correct.
Thanks, if need more information please tell me.

crankyoldgit, 2020-04-30:
Just so I can better isolate where the issue may be, can you confirm if the AC unit is behaving as expected when controlled solely via the web interface of `IRMQTTServer`?

My initial guess is that because DAIKIN64 uses a power toggle there is something funky with the logic that handles power toggles and the "default"/suggested HA config.
https://github.com/crankyoldgit/IRremoteESP8266/blob/8e06cca94df0455e404244793a3c41cf9663ab9d/src/ir_Daikin.cpp#L3346-L3348
&
https://github.com/crankyoldgit/IRremoteESP8266/blob/8e06cca94df0455e404244793a3c41cf9663ab9d/examples/IRMQTTServer/IRMQTTServer.h#L105-L109
&
https://github.com/crankyoldgit/IRremoteESP8266/blob/8e06cca94df0455e404244793a3c41cf9663ab9d/examples/IRMQTTServer/IRMQTTServer.ino#L231-L270

chongkk, 2020-04-30:
Hi, if I use web interface, the power will be opposite.
[image]
Like that the AC unit will be turn off.

[image]
Like that the AC unit will be turn on.

crankyoldgit, 2020-04-30:
As the DAIKIN64 power is a "Toggle", not a control of it is on or off. You will need to get the AC & the `IRMQTTServer` state in sync first. Otherwise, like you have now, it will be completely out of phase with the actual device. There is no-way I can force the A/C unit to be in sync from the library.

e.g. Set `IRMQTTServer` to "power: off" and then set the A/C unit to "Off" without `IRMQTTServer` knowing about it. i.e. Turn it off via the real remote. so that IRMQTTServer can't see it (if you have an IR receiver on it)

chongkk, 2020-04-30:
Hi, I tryed sync the AC and IRMQTTServer state first then the behavior still wrongly.
But,
From the web interface,
Power have 2 options, OFF and ON. 
Mode have 6 options, OFF, AUTO, COOL, HEAT, DRY, FAN_ONLY.

The AC unit status is off. 
I found when I change the Mode from "OFF" To "COOL", The "power" is still show "OFF"(which should be ON, right?), The AC Unit still off.
But, When I choose power from "OFF"to "ON", Mode from "OFF" to "COOL", The AC will be correctly Turn on.

Is this information help?

crankyoldgit, 2020-04-30:
> Is this information help?

Yes and no.

Lets start at the basics/beginning.

1. Stop HA from controlling the device. (You should only see "stat" MQTT messages, no "cmnd" ones ... IIRC)
1. In IRMQTTServer's AirCon page, set everything as "Off", except "Mode" which you can set to Cool/Heat/Fan, what ever suits your environment. Just not "Off". Set your A/C protocol/model etc to the correct settings.
1. Click "Update & Send".
1. Now, **without** IRMQTTServer seeing it, turn off the A/C via the remote. i.e. get them in sync.,
1. Set Power to On in IRMQTTServer's AirCon page, and click "Update & Send". The A/C unit should turn on to the desired mode et.
1. Change some other setting/mode etc, and click "Update & Send". Confirm the A/C unit is doing what you expect. i.e. changed modes etc, and NOT turned off.
1. Change "Power" to Off, and send it. The unit should now turn off.
1. Change "Power" to On, and send it. The unit should turn on etc.

Let me know how that goes, and if (and where) it starts to break.

Basically, Take both HA &using "Mode: Off" out of the equation.

crankyoldgit, 2020-04-30:
If it all works, I can move the testing to another part of the code (i.e. the HA compatibility bit)

crankyoldgit, 2020-04-30:
Oh, and wait at least a minute before doing any of the above if you've just rebooted the ESP etc.

chongkk, 2020-04-30:
Thanks very much for your working.

1. Stop HA from controlling the device. (You should only see "stat" MQTT messages, no "cmnd" ones ... IIRC)
[image]

2. In IRMQTTServer's AirCon page, set everything as "Off", except "Mode" which you can set to Cool/Heat/Fan, what ever suits your environment. Just not "Off". Set your A/C protocol/model etc to the correct settings.
[image]

3. Click "Update & Send".
[image]
The AC unit still power off. Set MODE to "cool" didn't lead POWER to "ON".

4.Now, without IRMQTTServer seeing it, turn off the A/C via the remote. i.e. get them in sync.,
The AC Unit still Power off, so turn on and then turn off without IRMQTTServer seeing it.

5,Set Power to On in IRMQTTServer's AirCon page, and click "Update & Send". The A/C unit should turn on to the desired mode et.
Yes, The AC unit success turn on to cool mode.
[image]

6, Change some other setting/mode etc, and click "Update & Send". Confirm the A/C unit is doing what you expect. i.e. changed modes etc, and NOT turned off.
Yes, The AC unit success changed to fan_only mode.
[image]

7, Change "Power" to Off, and send it. The unit should now turn off.
Yes, The AC unit success turn off.
[image]

8,Change "Power" to On, and send it. The unit should turn on etc.
Yes, The AC unit success turn on.
[image]

crankyoldgit, 2020-04-30:
So basically, that all worked fine. Correct?

chongkk, 2020-04-30:
The, step 3, set Mode from  "OFF" to "Cool", The POWER will still "OFF"

crankyoldgit, 2020-04-30:
But you shouldn't have been changing "Mode" at all. At step 3, you want the power & the ac unit to be off.

crankyoldgit, 2020-04-30:
receiving "cmnd"s is a different matter when IRMQTTServer HA mode.

chongkk, 2020-04-30:
I was guessing if set Mode to Cool will automatically set Power to on. Because in Ha, there is no button to Power on, All I can set is the mode is "cool or fan or Off", so I guess that's why in HA can't control the AC.

crankyoldgit, 2020-04-30:
When IRMQTTServer is in HA-mode, and it gets a mqtt mode change from "off" -> "cool" etc, it will try to turn on the device. I'm deliberately trying to avoid that control "path" to see what part of the code is causing you problems. We will get to the HA side soon enough.

chongkk, 2020-04-30:
Thanks very much 😊

crankyoldgit, 2020-04-30:
So, with that knowledge, did the AC behave correctly in those 8 steps?
It looks like it did.

FYI, something for you to play with in your HA setup: https://www.home-assistant.io/integrations/climate.mqtt/#send_if_off

Setting that to `false` might help. Don't know. I don't need it on my setup

chongkk, 2020-04-30:
Yes the wrongly behavior is I found in HA, the Web interface I can use Power and Mode correct control AC. Don't have any idea, I have the other brand ac which support already is also toggle power, but that one behavior correctly. Thanks anyway. LoL

crankyoldgit, 2020-04-30:
Yep. With ha mode compiled in, that's correct

chongkk, 2020-04-30:
Thanks. That's mean use web interface, the behavior is correct, I found wrongly in HA. Previous I have another model which have already supportted, The power is also toggle, and in HA is correct. Have no idea about that issue, thanks much much any way. :)

crankyoldgit, 2020-04-30:
What is the other ac type? I can look to see what could be different between the two

chongkk, 2020-04-30:
My another AC is identify as coolix (actual is media, but I think use same IR), all function in HA is correct. My IR remote controller is also toggle only, not a single power on button and power off button. Thanks 😊

crankyoldgit, 2020-04-30:
Coolix doesn't use a toggle for power. There is a specific Off message, and every other message is also an "On" message

chongkk, 2020-04-30:
Hmmm am I wrong remember, I am not sure.  Few days later I can back and access that old AC and check that. Thanks very much.

crankyoldgit, 2020-04-30:
I just checked my HA's [configuration.yaml](https://github.com/crankyoldgit/Home-AssistantConfig/blob/master/configuration.yaml):
Try disabling/removing `power_command_topic`:
e.g.
https://github.com/crankyoldgit/Home-AssistantConfig/blob/e9601834ce67eecdf6463ea247f4d496a37dfa3e/configuration.yaml#L869
and set `send_if_off` to `false`:
e.g.
https://github.com/crankyoldgit/Home-AssistantConfig/blob/e9601834ce67eecdf6463ea247f4d496a37dfa3e/configuration.yaml#L884

Seeing that, I now seem to remember having some problem with it a while ago due to some changes in some HA version or when I was trying to fix "multiple beeps" that my wife was complaining about.

chongkk, 2020-05-01:
I tryed disabling power_command_topic and set send_if_off to false, but still same behavior.
Now I tryed only can control the power correcttly is use mqtt to publish message:
mosquitto_pub -h 127.0.0.1 -t "ir_server/ac/cmnd/power" -m "off"
mosquitto_pub -h 127.0.0.1 -t "ir_server/ac/cmnd/power" -m "on"

crankyoldgit, 2020-05-01:
have you tried just changing `cmnd/mode`?

chongkk, 2020-05-01:
Yeah, I tryed, can't trigger power on.
like that:
`mosquitto_pub -h 127.0.0.1 -t "ir_server/ac/cmnd/mode" -m "cool"
`result like that:
```
ir_server/ac/cmnd/mode dry
ir_server/ac/stat/power off
ir_server/ac/stat/mode off
```

crankyoldgit, 2020-05-01:
Oh. That's not what I'd expect. I'll take a look at that in the morning

chongkk, 2020-05-01:
One weird thing is I found few times (10%), if I use HA change mode from OFF to COOL, The AC Unit will beep 3 times, on that situation, the AC Unit can be successfully power on to COOL mode.
Most time can only beep 2 times, if 2 times beep, the AC unit won't be power on.

Thanks very much for your patient and working. Really helped me a lot. Thanks.

crankyoldgit, 2020-05-02:
So, I think I can replicate the problem now. I just have to engineer how I'm going to fix this so it doesn' break non-HA use of it.

chongkk, 2020-05-02:
Thanks so so much 😘

crankyoldgit, 2020-05-02:
Can you download and try the code in this branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1092

i.e. v1.5.0-alpha of IRMQTTServer.

chongkk, 2020-05-02:
Hi I tested, now whit mqtt message I can successfully control AC Unit.
`
mosquitto_pub -h 127.0.0.1 -t "ir_server/ac/cmnd/mode" -m "cool"

mosquitto_pub -h 127.0.0.1 -t "ir_server/ac/cmnd/mode" -m "off"
`
result like that:
"
ir_server/ac/cmnd/mode cool
ir_server/ac/stat/power on
ir_server/ac/stat/mode cool
ir_server/ac/cmnd/mode off
ir_server/ac/stat/power off
ir_server/ac/stat/mode off
"
If use HA lovelace still can't.
I'm still trying about if I disable power_command_topic and send_if_off see if can work or not.

chongkk, 2020-05-02:
Hi, I tryed, with disable power_command_topic and send_if_off set to false, the HA lovelace can successfully control AC unit.
Great Job, lol, thanks so much.

crankyoldgit, 2020-05-02:
Excellent.

closed 2020-05-09 by crankyoldgit

crankyoldgit, 2020-05-20:
FYI, the changes mentioned above have now been included in the latest release (v2.7.7) of the library.

chongkk, 2020-05-20:
Thanks so much, I will try to use v2.7.7 in few days.

PR #1099 (2020-05-09): IRMQTTServer: Better handle power & mode operations for Home Assistant.
WARNING: Breaking Change!
When `MQTT_CLIMATE_HA_MODE` is enabled (default) this will break previous operation mode resumption when power is changed.

* Increase decoding tolerance for DAIKIN64
* Version bump to v1.5.0

Fixes #1092
commit 9d9ee6cb (2020-05-09): [BREAKING] IRMQTTServer: Better handle power & mode operations for Home Assistant. (#1099)

WARNING: Breaking Change!
When `MQTT_CLIMATE_HA_MODE` is enabled (default) this will break previous operation mode resumption when power is changed.

* Increase decoding tolerance for DAIKIN64
* Version bump to v1.5.0

Fixes #1092