crankyoldgit/IRremoteESP8266 #1100: Library API documentation

wahibmichael, 2020-05-02:
is there a user manual or API documentation for the library function? similar to this one 
https://github.com/cyborg5/IRLib2/blob/master/IRLib2/manuals/IRLibReference.pdf
I

crankyoldgit, 2020-05-02:
Sadly, no. There is no great pdf that tells you how to use this library.

There are however plenty code examples that do the basics with the library.
There is some high level stuff in the Wiki pages.
But if you want a list of all the API functions and what they do and how to use them, then unfortunately at present you have to hunt through the `*.h` & `*.cpp` files for the given protocol(s) e.g. `ir_Protocolname.(h|cpp)`
For more the more "core" API, look through the `IRBlah.(h|cpp)` files.
e.g. `IRsend.(h|cpp)` for the core sending routines, `IRrecv.(h|cpp)` for receiving/decoding, & `IRac.(h|cpp)` for the "common" A/C control APIs.

Almost every file/function/api call has a corresponding Unit Test file under the [test directory](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/test). You can often see/find out how something is used in there.

I find it fairly intuitive, other do too, however, it's not for the feint of heart. Sure, it could easily do with better documentation.

I'm far to busy squashing bugs, and adding features to keep up or write piles of documentation. I am but one person. That is, asking for "how to" docs is unlikely to make it happen. Sorry.

Finally, I won't say no to people willing to contribute documentation! (Hint Hint! :)

wahibmichael, 2020-05-02:
Thank you for your prompt response and great effort over here :)
the reason I'm asking this that I'm trying to decode my AC protocol which isn't identified by the library
I'm receiving a 66-bit message, the protocol is similar to NEC, I need to decode the message in a Binary or HEX format to reverse the protocol. is there a function to do this?
I tried the resultToHexidecimal function from IRutils.h but it returns only 32bit HEX message

crankyoldgit, 2020-05-02:
If it is detected as `UNKNOWN` you *have* to rely on the `rawData` to try to decode it manually/externally.

Please read: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#My_remotes_signal_is_being_reported_as_an_Unknown_code_What_does_that_mean_How_do_I_send_an_Unknown_code & https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol

crankyoldgit, 2020-05-02:
> is there a function to do this?

Yes, there are tools to help with this. See: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol
and https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/auto_analyse_raw_data.py

wahibmichael, 2020-05-02:
Thank you, I already analyzed the RAW data and had the sample phyton code attached
my question is
1- what shall I do next and how to include the protocol to IRUTILS for further analysis?
2- at this point can I use send generic function to send commands to my A/C instead of sending RAW?
excuse me I little bit Noob in this area :)

[python output.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4568654/python.output.txt)
[raw.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/4568655/raw.txt)

crankyoldgit, 2020-05-02:
> 1- what shall I do next and how to include the protocol to IRUTILS for further analysis?

See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol#modifying-the-library-to-add-the-ability-to-send-messages

> 2- at this point can I use send generic function to send commands to my A/C instead of sending RAW?
excuse me I little bit Noob in this area :)

We are all Noobs at some time.
When you get to [this point](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol#congratulations-if-everything-went-well-you-should-now-be-able-to-compile-and-send-using-the-irsendsendtestexample-procedure), you can use your new function.

`sendGeneric()` is really only for internal use within each protocol.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/f4ed9bc9dff38ce3813f93c2a371e6034bb8e3a9/src/ir_Nikai.cpp#L39-L43
However, you can use it (`sendGeneric()`) technically before you've properly added your protocol. You just need to provide the correct parameters. The above example can give you hints on how it's invoked.

crankyoldgit, 2020-05-09:
How are you going? Need any help?

wahibmichael, 2020-05-09:
Hi yes I need some help
I'm working on a generic remote control project for A/C, some remotes I'm testing are not included in the library so I end up sending the RAW timing data to be stored on the cloud which isn't efficient 
is there a receive a generic function that streams the data in an array having (headermark,onemark,data,  zeromark) and use this data later in a send generic function?

crankyoldgit, 2020-05-09:
> is there a receive a generic function that streams the data in an array having (headermark,onemark,data, zeromark) and use this data later in a send generic function?

Not really. there is a `matchGeneric()` function(s) but nothing automatic.

Some downstream projects do a lossey compression on the raw data stream and get significant space savings. You could look into those.

wahibmichael, 2020-05-15:
thank you initial documentation is helping a lot.

crankyoldgit, 2020-05-16:
I've converted some more files to use Doxygen, so that branch/docs html tree is a little better now. Still a mountain to go!

closed 2020-05-29 by crankyoldgit

crankyoldgit, 2020-05-29:
FYI, the Library API documentation can be found via the link in the main `README.md`, & https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/ 

it's not perfect, far from complete, but it will get updated and improved over time.

crankyoldgit, 2020-06-23:
FYI, the code & changes referenced above have been included in the [v2.7.8 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.8) of the library.

PR #1151 (2020-05-29): Update `README.md`'s to point to new API docs.
* Minor cleanup of the files too.

Fixes #1100
commit eb7cf83e (2020-05-29): Update `README.md`'s to point to new API docs. (#1151)

* Minor cleanup of the files too.

Fixes #1100