crankyoldgit/IRremoteESP8266 #1101: Request to add new protocol – Daikin M Series

camilloaddis, 2020-05-02:
Hi, 

This is a request to add support for [Daikin M Series (FTXM-M), remote model ARC466A33](https://www.daikin.it/it_it/cerca-prodotto/FTXM-M.html)

I've already reverse engineered the IR signal, and [used a RaspberryPi to control the AC](https://medium.com/@camilloaddis/smart-air-conditioner-with-raspberry-pi-an-odissey-2a5b438fe984), but now I want to try it with the ESP8266 :)

Thanks!

These are the most important bits (here's a [spreadsheet](https://docs.google.com/spreadsheets/d/1yAZN5y_qO5IkImb1VNM3Z4-shE-fLQuCRtKkMzJKAMU/edit?usp=sharing) used to understand the signal)

spreadsheet cell # | IRprotocol bit # | meaning
-----|-----|----------
59 | 52 | Comfort mode on/off
113-128 | 104-119 | Datetime
179 | 168 | Power on/off
183-186 | 172-175 | Mode (auto:'0000', dry:'0100', cool:'1100', heat:'0010', fan:'0110')
187-194 | 176-183 | Temperature °C (multiplied by 2) (when mode = auto, min/max = 18/30; when mode = cool, min/max = 18/32; when mode = heat, min/max = 10/30)
203-206 | 192-195 | Vertical swing on/off
207-210 | 196-199 | Fan speed (fan1:'1100', fan2:'0010', fan3:'1010', fan4:'0110', fan5:'1110', auto:'0101', night:'1101')
211-214 | 200-203 | Horizontal swing on/off
219-230 | 208-219 | "On timer" minutes
231-242 | 220-231 | "Off timer" minutes
243 | 232 | Powerful mode on/off
248 | 237 | Silent mode on/off
268 | 257 | Sensor mode on/off
269 | 258 | Eco mode on/off
271 | 260 | Clean mode on/off
283-290 | 272-279 | Checksum (modular sum of all previous bits)

These are the values I used as marks and spaces (worked pretty well)
```
kHdrMark   = 3440
kHdrSpace  = 1720
kBitMark   = 430
kOneSpace  = 1320
kZeroSpace = 430
kSpaceGap1 = 35500
kSpaceGap2 = 25000
```

This is the serial output with IRrecvDumpV2
```
Timestamp : 000064.440
Library   : v2.7.6

Protocol  : DAIKIN
Code      : 0x11DA2700C530000711DA27004207086311DA2700003932006F0F0006600000C19000B2 (280 Bits)
Mesg Desc.: Power: On, Mode: 3 (Cool), Temp: 25C, Fan: 4 (UNKNOWN), Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): On, Swing(V): On, Clock: 00:07, Day: 1 (Sun), On Timer: Off, Off Timer: Off, Weekly Timer: Off
uint16_t rawData[583] = {448, 390,  472, 422,  444, 392,  476, 418,  444, 392,  474, 25086,  3502, 1664,  474, 1258,  474, 390,  476, 390,  474, 400,  464, 1256,  476, 418,  444, 392,  474, 422,  442, 392,  472, 1260,  470, 422,  448, 1284,  442, 1256,  476, 388,  472, 1258,  476, 1284,  448, 1254,  476, 1256,  474, 1256,  470, 394,  476, 390,  472, 1258,  474, 420,  444, 390,  474, 390,  474, 390,  472, 392,  474, 392,  472, 394,  470, 422,  446, 390,  476, 390,  472, 1260,  470, 392,  474, 1286,  444, 420,  446, 390,  474, 392,  476, 1252,  476, 1262,  466, 390,  476, 390,  472, 394,  472, 422,  444, 1286,  444, 1286,  446, 388,  474, 390,  474, 392,  472, 392,  474, 418,  448, 390,  472, 394,  472, 392,  474, 392,  472, 392,  472, 1256,  472, 1288,  440, 1260,  474, 390,  472, 392,  476, 388,  476, 420,  442, 420,  446, 35444,  3532, 1694,  446, 1254,  476, 388,  476, 418,  446, 390,  474, 1256,  472, 392,  472, 422,  446, 390,  474, 390,  474, 1256,  474, 400,  464, 1256,  476, 1254,  472, 422,  444, 1256,  476, 1254,  476, 1256,  472, 1258,  474, 1256,  472, 392,  474, 390,  474, 1256,  474, 392,  474, 392,  472, 390,  474, 392,  474, 390,  474, 390,  474, 390,  476, 388,  476, 390,  472, 394,  476, 388,  474, 1258,  472, 396,  468, 422,  444, 420,  446, 418,  442, 1258,  476, 398,  464, 1258,  476, 1252,  476, 1286,  444, 390,  474, 390,  472, 392,  474, 424,  442, 390,  476, 390,  472, 420,  446, 420,  446, 1284,  446, 420,  446, 388,  474, 420,  442, 394,  474, 1254,  474, 1258,  474, 422,  442, 392,  472, 390,  474, 1288,  444, 1286,  444, 422,  442, 35444,  3532, 1662,  476, 1256,  474, 392,  474, 422,  444, 418,  448, 1254,  472, 392,  474, 422,  442, 422,  442, 392,  472, 1288,  442, 422,  448, 1284,  444, 1286,  446, 416,  450, 1254,  472, 1256,  474, 1286,  444, 1288,  440, 1258,  476, 390,  472, 392,  476, 1254,  474, 422,  446, 390,  472, 392,  472, 392,  472, 422,  446, 388,  476, 420,  442, 390,  474, 420,  442, 394,  470, 422,  446, 388,  474, 390,  474, 390,  472, 392,  476, 388,  474, 420,  446, 388,  472, 1258,  474, 390,  472, 392,  474, 1286,  448, 1284,  444, 1256,  474, 390,  476, 388,  474, 392,  470, 1258,  476, 390,  474, 420,  442, 1258,  470, 1258,  476, 398,  468, 388,  474, 392,  474, 390,  476, 418,  446, 420,  442, 392,  476, 422,  442, 390,  474, 394,  472, 1284,  444, 1288,  444, 1286,  444, 1256,  472, 392,  476, 1256,  474, 1254,  474, 392,  472, 1258,  472, 1256,  474, 1284,  446, 1256,  474, 390,  472, 394,  472, 408,  460, 388,  476, 388,  472, 394,  476, 388,  474, 392,  476, 420,  444, 412,  454, 388,  476, 418,  444, 394,  476, 1254,  474, 1256,  476, 390,  472, 392,  474, 422,  444, 388,  476, 420,  446, 418,  444, 390,  476, 418,  444, 392,  474, 420,  444, 1286,  448, 1254,  476, 390,  472, 392,  474, 416,  448, 390,  476, 390,  474, 390,  472, 392,  476, 390,  474, 392,  472, 422,  442, 420,  446, 390,  474, 420,  446, 418,  444, 422,  444, 392,  476, 388,  472, 1288,  442, 394,  474, 390,  472, 392,  474, 422,  444, 390,  474, 1286,  444, 1256,  472, 420,  448, 420,  442, 390,  474, 390,  476, 1278,  450, 420,  448, 388,  472, 1288,  444, 422,  442, 420,  444, 420,  442, 422,  444, 422,  446, 422,  442, 422,  442, 392,  476, 420,  442, 1258,  474, 390,  474, 418,  444, 1286,  442, 1258,  476, 390,  472, 1258,  472};  // DAIKIN
uint8_t state[35] = {0x11, 0xDA, 0x27, 0x00, 0xC5, 0x30, 0x00, 0x07, 0x11, 0xDA, 0x27, 0x00, 0x42, 0x07, 0x08, 0x63, 0x11, 0xDA, 0x27, 0x00, 0x00, 0x39, 0x32, 0x00, 0x6F, 0x0F, 0x00, 0x06, 0x60, 0x00, 0x00, 0xC1, 0x90, 0x00, 0xB2};
```

This is the `auto_analyse_raw_data.py` output
```
$ python auto_analyse_raw_data.py -f test
Found 583 timing entries.
Potential Mark Candidates:
[3532, 476]
Potential Space Candidates:
[35444, 25086, 1694, 1288, 424]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 3522
kHdrSpace  = 1673
kBitMark   = 463
kOneSpace  = 1266
kZeroSpace = 400
[... 349 lines cut ...]
  offset += used;  // Adjust for how much of the message we read.
  pos += 19;  // Adjust by how many bytes of data we read

  // Success
  results->decode_type = decode_type_t::TESTEXAMPLE;
  results->bits = nbits;
  return true;
}
#endif  // DECODE_TESTEXAMPLE
```

Checksum is calculated by adding all the bytes of the third trunk of data (after the second gap), and taking only the last two digits (in base 16, or the last 8 in base 2).
If it's useful here's a JavaScript function to calculate it:
```javascript
var exampleBytes = ["11","DA","27","00","00","09","32","00","A0","00","00","06","60","01","00","C1","80","00"];

function calculateChecksum(lastBytes){
  var s = 0;
  lastBytes.forEach(function(hexByte){
    s += parseInt(hexByte, 16);
  })
  lastBytes.push(s.toString(16).substr(-2));
  return lastBytes;
}
```

crankyoldgit, 2020-05-03:
> This is the serial output with IRrecvDumpV2:
> 
> ```
> Timestamp : 000064.440
> Library   : v2.7.6
> 
> Protocol  : DAIKIN
> Code      : 0x11DA2700C530000711DA27004207086311DA2700003932006F0F0006600000C19000B2 (280 Bits)
> Mesg Desc.: Power: On, Mode: 3 (Cool), Temp: 25C, Fan: 4 (UNKNOWN), Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): On, Swing(V): On, Clock: 00:07, Day: 1 (Sun), On Timer: Off, Off Timer: Off, Weekly Timer: Off
> [SNIP]
> uint8_t state[35] = {0x11, 0xDA, 0x27, 0x00, 0xC5, 0x30, 0x00, 0x07, 0x11, 0xDA, 0x27, 0x00, 0x42, 0x07, 0x08, 0x63, 0x11, 0xDA, 0x27, 0x00, 0x00, 0x39, 0x32, 0x00, 0x6F, 0x0F, 0x00, 0x06, 0x60, 0x00, 0x00, 0xC1, 0x90, 0x00, 0xB2};
> ```
> 

Um, this looks like it **already** is supported. i.e. `DAIKIN` / `class IRDaikinESP`

e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/f4ed9bc9dff38ce3813f93c2a371e6034bb8e3a9/src/ir_Daikin.h#L499-L559

and

https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/TurnOnDaikinAC/TurnOnDaikinAC.ino

FYI @sillyfrog

camilloaddis, 2020-05-03:
Thanks,
i tried to look at the code but couldn't find the same ac/remote model or understand the code to find the right marks for my model.

I'll test it and see if everything works as expected!

crankyoldgit, 2020-05-03:
The short answer is, if `IRrecvDumpV2` detects it, it's supported. If it can decode it into individual settings, then it fully supports the protocol.

closed 2020-05-04 by crankyoldgit

crankyoldgit, 2020-05-20:
FYI, the changes mentioned above have now been included in the latest release (v2.7.7) of the library.