crankyoldgit/IRremoteESP8266 #1103: How to interpret raw codes as binary

dummys, 2020-05-05:
Hi, I'm doing some reverse engineering of a remote to help Logitech engineer to integrate in the Harmony database. The device is a  multibrackets motorized swing arm, url https://products.multibrackets.com/en/motorized/motorized-wallmounts/m-motorized-swing-mount-large. The engineer from logitech asked me to find he HEX/Pronto code of the remote. The remote as stated inside their website should use RC-5. Multibrackets support gave me this PDF: http://info.multibrackets.com/data/common/manuals/4500_code.pdf I'm using your example: IRrecvDumpV2.ino to dump the code from the remote, and I need to change the `MIN_UNKNOWN_SIZE` to 1 to find the correct sequence. I was able to replay the sequence using the `IRsend.ino` and following:

send the right keypress:

`uint16_t rawData[7] = {40204, 5106,  5036, 35834,  40212, 5108,  5038};`

send the 1 + ok keypress:
`uint16_t rawData[7] = {20100, 20472,  15092, 30704,  20102, 20472,  15086};`

I would like to understand exactly the bits esp device is sending to the receiver, and then I would like to translate as pronto code. The problem is that if I use the `resultToHexidecimal` function I never find again the code from the PDF, so I'm wondering how to get the real raw codes, without any decoding first. like with the 11 start bits, 0 togglebit, address etc

crankyoldgit, 2020-05-05:
Looking at your data, and that PDF. I can tell you for certain it is NOT using the `RC5` protocol.
I can write a sender/decoder for this library fairly easily, but that probably won't help you.
I'll see if I can quickly code up a "Raw to Pronto code" python script, that can help you out.

crankyoldgit, 2020-05-05:
If you download the following branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/RawToPronto

There is now a program `RawToPronto.py` which should convert your rawdata line into a workable Pronto code. Hopefully that helps you out.

e.g. https://github.com/crankyoldgit/IRremoteESP8266/blob/RawToPronto/tools/RawToPronto.py

Usage:
```
$ tools/RawToPronto.py -h
usage: RawToPronto.py [-h] --hz HZ [-v] [-f FILE] [--stdin] [rawdata]

Read an IRremoteESP8266 rawData declaration and tries to convert it in to a
Pronto code.

positional arguments:
  rawdata               A rawData line from IRrecvDumpV2. e.g. 'uint16_t
                        rawbuf[37] = {7930, 3952, 494, 1482, 520, 1482, 494,
                        1508, 494, 520, 494, 1482, 494, 520, 494, 1482, 494,
                        1482, 494, 3978, 494, 520, 494, 520, 494, 520, 494,
                        520, 520, 520, 494, 520, 494, 520, 494, 520, 494};'
                        (default: None)

optional arguments:
  -h, --help            show this help message and exit
  --hz HZ               Frequency of the protocol to use in code generation.
                        E.g. 38000Hz (default: None)
  -v, --verbose         Increase output verbosity (default: False)
  -f FILE, --file FILE  Read in a rawData line from the file. (default: None)
  --stdin               Read in a rawData line from STDIN. (default: False)
$ cat /tmp/j
uint16_t rawData[7] = {20100, 20472, 15092, 30704, 20102, 20472, 15086};
$ tools/RawToPronto.py --hz 38000 -f /tmp/j
Pronto code = '0000 006D 0007 0000 02FB 0309 023D 048E 02FB 0309 023D'
$
```

crankyoldgit, 2020-05-06:
Note: Program has been renamed to `raw_to_pronto_code.py` to comply with style guide(s).

dummys, 2020-05-06:
omg man, you are so fast. Thank you so much for the script, and for your work. where can I donate ? Jeez never someone so fast to reply to an open source project. really appreciate. It could be a good idea to implement a sender/decoder for people that want to use this mechanic things for home automation. My goal is to be able to use it with harmony. Can you explain me how, to come from the rawData to a "bits" view like we have in the PDF. As I understand the rawData is represented in useconds right ?

crankyoldgit, 2020-05-06:
Yep. Odd entries are marks (on) and even entries are spaces (off).

Per the pdf:
```
STARTING CODE IS TO KEEP TRANSMITING 15ms, STOPING CODE IS TO STOP TRANSMITING FOR
30MS,
5MS IS THE UNIT FOR EACH CODING (1 STANDS FOR TRANSMITING, 0 STANDS FOR STOPING
TRANSMITING) 
```

Lets take your data:
> send the 1 + ok keypress:
`uint16_t rawData[7] = {20100, 20472, 15092, 30704, 20102, 20472, 15086};`

First number is on/mark for 20100 usecs (20.1msecs).
All of their codes start with a 15000 usec (15msecs) "header". (I.e. not part of the data.
That leaves ~5000 usecs left of that (on/mark) pulse. As a binary `1` is 5000 usecs (5ms) of "on", that means the first bit is a `1` e.g. Data so far is `0b1`.
That's the first On/Mark exhausted, next is an off/Space for 20472 usecs.
That's 4 x 5000 usecs (5ms), so the next four bits are `0`s. We append that. i.e Data so far is `0b10000`.

Back to on/marks, 15092us = 3 x 5000. So that's three `1`s that need to be appended.
e.g. Data is now `0b10000111` or `0x87`.
As the protocol is really only 8 data bits long. We have everything.
A message ends (per pdf) when there is an off/space of more than 30000us/30ms.
Yep, the next entry is a space/off, and it's tada .. 30704us. End of the message.

The code now repeats (`0b10000111` / `0x87`) as the protocol seems to include a single mandatory repeat. As we see the same pattern/data again. e.g. `20102, 20472, 15086};` A mark, a space, and a mark.
The library doesn't both recording leading or trailing Off/Space pulses, as they technically go on forever. They are implied.
e.g. The protocol is looking for a off/space of 30000 or more. There is near-infinite "off/space" at the end of a message, so it would find it.

Ta Da ... Decoded by hand.

/me checks the pdf.
`1 + ok： 000 10000111 //87` .. so yep, `0b10000111` / `0x87` is correct.

crankyoldgit, 2020-05-06:
Is there a Model number on the device/bracket, and/or on the remote? I'll need that if I'm going to add support for it to the library. The brand I'm guessing is "Multibrackets".

From the web site, it looks like the model number could be "4500" but not sure.

dummys, 2020-05-06:
pfiou you are a beast man, nice work. Hmm for the model I had so much trouble finding one to be honest, here is the link of the device: https://products.multibrackets.com/en/motorized/motorized-wallmounts/m-motorized-swing-mount-large They called it "Motorized Swing mount large"... That's so nice for a product name, haha. Their support is very crappy, I asked them to help me on this and they respond a nice "The product is only working with the included remote, period"... For them it could be interesting to be included in the logitech harmony database, but they just don't care. You are helping me a lot with pronto codes, I have done some of them, see below:
```
RIGHT:
Raw:
uint16_t rawData[7] = {40204, 5106,  5036, 35834,  40212, 5108,  5038}
Pronto:
0000 006D 0007 0000 05F7 00C2 00BF 0551 05F8 00C2 00BF
LEFT:
Raw:
uint16_t rawData[11] = {35152, 5110,  5030, 5104,  5030, 30666,  35148, 5110,  5058, 5082,  5032};
Pronto
0000 006D 000B 0000 0537 00C2 00BF 00C1 00BF 048D 0537 00C2 00C0 00C1 00BF
OK:
Raw:
uint16_t rawData[11] = {25124, 5108,  5038, 5110,  5034, 40940,  25132, 5108,  5036, 5110,  5036};
Pronto:
0000 006D 000B 0000 03BA 00C2 00BF 00C2 00BF 0613 03BB 00C2 00BF 00C2 00BF
1+ok:
Raw:
uint16_t rawData[7] = {20100, 20472,  15092, 30704,  20102, 20472,  15086};
Pronto:
0000 006D 0007 0000 02FB 0309 023D 048E 02FB 0309 023D
2+ok:
Raw:
uint16_t rawData[11] = {20122, 10190,  5064, 10194,  5036, 35788,  20086, 10220,  5034, 10222,  5034};
Pronto:
0000 006D 000B 0000 02FC 0183 00C0 0183 00BF 054F 02FB 0184 00BF 0184 00BF
```
Now I have another concern, why for example for the button `right` I got raw entry and for the `left` button I got 11 ? As Logitech is asking me pronto code, I don't want to give them crappy/bogus one.

crankyoldgit, 2020-05-06:
> Now I have another concern, why for example for the button right I got raw entry and for the left button I got 11 ?

That's because if there are `1`s or `0`s next to each other, it sends a _longer_ single pulse for the group of them. If they alternate. ie. 010101 etc then there are more "short" pulses.

I'd suggest you try out the `sendPronto()` function in the library, and confirm your codes work on your device before you send them to Logitech. I'm fairly confident they are correct, but that's one way to double check/verify them. e.g. https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendProntoDemo/IRsendProntoDemo.ino

Some of the remotes/protocols repeat the message twice or more. It isn't uncommon. It's done to give a better probability of the device seeing the message. i.e. in case there is interference etc.

They typically only need to see the a "single" message out of the two to function.

dummys, 2020-05-06:
tried with pronto and it's not working. Here is my code:

```c
include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

#define IR_LED D3  

IRsend irsend(IR_LED);  // Set the GPIO to be used to sending the message.

uint16_t samsungProntoCode[11] = {
    0x0000, 0x006D, 0x0007, 0x0000, 0x02FB, 0x0309, 0x023D,0x048E, 0x02FB,
    0x0309, 0x023D
};

void setup() {
  irsend.begin();
  #if defined(ESP8266)
    Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
  #else  // ESP8266
    Serial.begin(115200, SERIAL_8N1);
  #endif  // ESP8266
}

void loop() {
  Serial.println("Sending a Samsung TV 'on' command.");
  irsend.sendPronto(samsungProntoCode, 11);
  delay(2000);
  
}
```

dummys, 2020-05-06:
testing the same script with a sony TV, and I'm not able to translate rawcode to pronto code working.
Raw is: `uint16_t rawData[77] = {2360, 624,  1164, 626,  566, 624,  1168, 624,  592, 600,  1162, 626,  566, 626,  566, 626,  1166, 624,  566, 628,  564, 628,  566, 626,  566, 25750,  2362, 622,  1166, 622,  566, 628,  1164, 624,  568, 622,  1166, 624,  568, 624,  566, 626,  1166, 624,  566, 630,  562, 628,  568, 624,  568, 25732,  2362, 622,  1164, 626,  594, 598,  1166, 624,  568, 626,  1164, 624,  566, 626,  568, 624,  1164, 624,  566, 626,  570, 624,  568, 626,  566};  // SONY A90`

and generated pronto is: `> $ python3 raw_to_pronto_code.py --hz 38000 -f test
Pronto code = '0000 006D 004D 0000 0059 0017 002C 0017 0015 0017 002C 0017 0016 0016 002C 0017 0015 0017 0015 0017 002C 0017 0015 0017 0015 0017 0015 0017 0015 03D2 0059 0017 002C 0017 0015 0017 002C 0017 0015 0017 002C 0017 0015 0017 0015 0017 002C 0017 0015 0017 0015 0017 0015 0017 0015 03D1 0059 0017 002C 0017 0016 0016 002C 0017 0015 0017 002C 0017 0015 0017 0015 0017 002C 0017 0015 0017 0015 0017 0015 0017 0015`

crankyoldgit, 2020-05-06:
I'll look into it. You chose a bad example to test with though. Sony doesn't use 38000Hz.
It uses 40000Hz. You'll need to adjust with `--hz 40000`
https://github.com/crankyoldgit/IRremoteESP8266/blob/ac3738285f06171b763c2cff671dd84de458ef2f/src/ir_Sony.cpp#L34

dummys, 2020-05-06:
yes I've seend that and after I changed it.

> python3 raw_to_pronto_code.py --hz 40000 -f test Pronto code

crankyoldgit, 2020-05-06:
Yeah. Just saw.

dummys, 2020-05-06:
but I'm wondering why from raw data I didn't find the same prontoCode that I found on other website. But despite this, using sendPronto I was not able to play with the sony TV.

crankyoldgit, 2020-05-06:
I found my _bug_. Working on the fix. I made one bad assumption, and one error.
Both in the python program.

dummys, 2020-05-06:
oh nice. We can check with this tool: https://github.com/bengtmartensson/IrScrutinizer

crankyoldgit, 2020-05-06:
Okay, I've updated the branch with my fixes.

When you're generating the pronto codes for Logitech, you should use: `--gap 30000` as an option to `raw_to_pronto_code.py`

Technically that will be a better result for **this** Multibrackets protocol, as the spec says it always there is always a 30000us/30ms off/space/gap at the end of a message.

I've also added some tests to confirm the pronto code generated results in basically the same `rawData` result.
e.g. The following should work.
```
  // Based on raw data:
  //   uint16_t rawData[7] = {20100, 20472, 15092, 30704, 20102, 20472, 15086};
  // and output from `raw_to_pronto_code.py --hz 38000`:
  // Pronto code = '0000 006D 0004 0000 02FB 0309 023D 048E 02FB 0309 023D 0ED8'
  uint16_t pronto_test[12] = {
      0x0000, 0x006D, 0x0004, 0x0000, 0x02FB, 0x0309, 0x023D, 0x048E,
      0x02FB, 0x0309, 0x023D, 0x0ED8};
  irsend.sendPronto(pronto_test, 12);
```

dummys, 2020-05-06:
booom working on Sony !

crankyoldgit, 2020-05-06:
> ```
> uint16_t rawData[4] = {20100, 20472,  15092, 30704};
> ```

How did you get that bit of data?! It looks wrong.
`rawData` is almost always an odd-numbered size.

dummys, 2020-05-06:
I created it myself, and decoded it successfully to a 1+ok in the PDF. Tried to send it using sendRaw, it works.

crankyoldgit, 2020-05-06:
ahh.
In that case, you've already added a "end gap". i.e. the trailing 30000us. Use `--gap 0`, I think should do the trick.

dummys, 2020-05-06:
yes but I tried both, not working. I even tried with a new capture, put it in sendRaw, working, put it in pronto --gap30000 ou without it, not working.

crankyoldgit, 2020-05-06:
what's the raw data you collected from the remote for that key?

crankyoldgit, 2020-05-06:
No, I meant the 1_ok key

dummys, 2020-05-06:
yes its this

dummys, 2020-05-06:
if we took this: uint16_t rawData[4] = {20100, 20472,  15092, 30704};

20100 - 15000 = 5000 0b 1 - 0000 - 111 code is 0x87
20472 = 4x5000 
15092 = 3x5000
30k = 30MS end

```
>> 0b10000111
135
>>> hex(0b10000111)
'0x87'
>>> 
```

crankyoldgit, 2020-05-06:
```
$ cat /tmp/one_plus_okay.txt 
uint16_t rawData[7] = {20088, 20462, 15084, 30678, 20090, 20464, 15086};
$ ./raw_to_pronto_code.py --hz 38000 --gap 30000 -f /tmp/one_plus_okay.txt 
Pronto code = '0000 006D 0004 0000 02FB 0309 023D 048D 02FB 0309 023D 0474'
```

which should yield:
```
  uint16_t pronto_test[12] = {
      0x0000, 0x006D, 0x0004, 0x0000, 0x02FB, 0x0309, 0x023D, 0x048E,
      0x02FB, 0x0309, 0x023D, 0x0474};
  irsend.sendPronto(pronto_test, 12);
```

And that doesn't work?

dummys, 2020-05-06:
```
/* IRremoteESP8266: IRsendProntoDemo
 * Copyright 2017 David Conran
 *
 * Demonstrates sending Pronto codes with IRsend.
 *
 * Version 1.0 June, 2017
 *
 * An IR LED circuit *MUST* be connected to ESP8266 pin 4 (D2), unless you
 * change the kIrLed value below.
 *
 * TL;DR: The IR LED needs to be driven by a transistor for a good result.
 *
 * Suggested circuit:
 *     https://github.com/crankyoldgit/IRremoteESP8266/wiki#ir-sending
 *
 * Common mistakes & tips:
 *   * Don't just connect the IR LED directly to the pin, it won't
 *     have enough current to drive the IR LED effectively.
 *   * Make sure you have the IR LED polarity correct.
 *     See: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity
 *   * Typical digital camera/phones can be used to see if the IR LED is flashed.
 *     Replace the IR LED with a normal LED if you don't have a digital camera
 *     when debugging.
 *   * Avoid using the following pins unless you really know what you are doing:
 *     * Pin 0/D3: Can interfere with the boot/program mode & support circuits.
 *     * Pin 1/TX/TXD0: Any serial transmissions from the ESP8266 will interfere.
 *     * Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
 *   * ESP-01 modules are tricky. We suggest you use a module with more GPIOs
 *     for your first time. e.g. ESP-12 etc.
 */

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

#define IR_LED D3

IRsend irsend(IR_LED);  // Set the GPIO to be used to sending the message.
 
 uint16_t pronto_test[12] = {
      0x0000, 0x006D, 0x0004, 0x0000, 0x02FB, 0x0309, 0x023D, 0x048E,
      0x02FB, 0x0309, 0x023D, 0x0474};

void setup() {
  irsend.begin();
    Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
}

void loop() {
  Serial.println("Sending multi12");
  irsend.sendPronto(pronto_test, 12);
  delay(2000);

  
}
```

not working

crankyoldgit, 2020-05-06:
Does sending the rawdata for that button (1+ok) work?

dummys, 2020-05-06:
yes I told you already ...

crankyoldgit, 2020-05-06:
I think I've found/fixed the problem. I've improved the accuracy of `sendPronto()` when it's dealing with _large_ values. Most protocols don't have marks or spaces anywhere near as big as this protocol, and I think it was artificially shortening/rounding down some of the values due to integer arithmetic.  I've improved it's precision by 10 times. So the values are a little closer to the original values.

Anyway. Download the branch again, and try the new/improved `sendPronto()`.

dummys, 2020-05-06:
I'm on it.

crankyoldgit, 2020-05-06:
Excellent. 1:30am down here. Time for ZZzzz.

dummys, 2020-05-06:
oh yeah you deserve it. Time 17:26 here, more easy. time for some sport ! anyway thanks again for such work, impressive.

crankyoldgit, 2020-05-06:
Reopening. When the PRs are merged, it will close issue automatically.

dummys, 2020-05-06:
ok sorry. it's you the master.

dummys, 2020-05-14:
Hi Again @crankyoldgit, Logitech engineer told me this:
```
Thank you for your efforts in providing the necessary IR-codes for your MultiBracket product. 

I have received an answer from the Harmony Datateam and the codes you have provided can not be used; they are not complete.

As a reference I give you a link with codes as they should be provided:
http://www.remotecentral.com/cgi-bin/codes/denon/avr-2400-2600
```

Which kind of code they are waiting for ?

crankyoldgit, 2020-05-14:
@dummys I am not sure what their problem is with the code and I can't tell as I don't have the output you sent them.

The only thing(s) I can tell that are different are:
1. The remotecentral link uses lowercase hex in their codes. I've updated my program to use lowercase just in case that's the issue. (It shouldn't be, but who knows)
1. The link to the protocol they suggest uses the "repeat" section of pronto codes. I was using the "initial burst" section. As you *have* to be able to support both sections for some message protocols, I'm surprised they can't accept it based on that. Either way, I've added an option to use just the "repeat" section per the protocol in their link.
1. They are sticks-in-the-mud and don't support anything that isn't linked/stored on http://www.remotecentral.com.

Assuming the main issue is the "repeat" one, I would suggest you reprocess the raw messages using the following options: `--hz 38000 --gap 30000 --repeat_code`

e.g.
```
$ tools/raw_to_pronto_code.py -h
usage: raw_to_pronto_code.py [-h] --hz HERTZ [-g USECS] [-v] [-r] [-f FILE]
                             [--stdin]
                             [rawdata]

Read an IRremoteESP8266 rawData declaration and tries to convert it in to a
Pronto code.

positional arguments:
  rawdata               A rawData line from IRrecvDumpV2. e.g. 'uint16_t
                        rawbuf[37] = {7930, 3952, 494, 1482, 520, 1482, 494,
                        1508, 494, 520, 494, 1482, 494, 520, 494, 1482, 494,
                        1482, 494, 3978, 494, 520, 494, 520, 494, 520, 494,
                        520, 520, 520, 494, 520, 494, 520, 494, 520, 494};'
                        (default: None)

optional arguments:
  -h, --help            show this help message and exit
  --hz HERTZ, --hertz HERTZ
                        Frequency of the protocol to use in code generation.
                        E.g. 38000Hz (default: None)
  -g USECS, --gap USECS, --endgap USECS
                        Nr. of uSeconds of gap to add to the end of the
                        message. (default: 100000)
  -v, --verbose         Increase output verbosity (default: False)
  -r, --repeat_code     Send using the 'repeat' section of the pronto code.
                        (default: False)
  -f FILE, --file FILE  Read in a rawData line from the file. (default: None)
  --stdin               Read in a rawData line from STDIN. (default: False)
$ cat /tmp/j
uint16_t rawData[7] = {
            20100, 20472, 15092, 30704, 20102, 20472, 15086};
$ tools/raw_to_pronto_code.py --hz 38000 -g 30000 -r -f /tmp/j 
Pronto code = '0000 006D 0000 0004 02fb 0309 023d 048e 02fb 0309 023d 0474'
```

Note: If you are going to test these "repeat"-based pronto codes using `sendPronto()` which you probably should, you will probably need to use the `repeat` argument.
e.g.
```
sendPronto(code, 1);  // Send the code with one repeat.
```

crankyoldgit, 2020-05-15:
FYI, code is in PR #1122 / Branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1103

crankyoldgit, 2020-05-15:
@dummys I've had a change of mind. The code in that branch now assumes "use repeat section" as the default., you have to use the `-i` / `--inital_burst` flag for the old behaviour.
i.e. You can ignore/remote the `-r` flag.

dummys, 2020-05-15:
ok nice thanks for all of this, I test it tonight and come back to you.

closed 2020-05-16 by crankyoldgit

dummys, 2020-05-17:
ok Tested and it works but I think we still have an issue on the length of the payload @crankyoldgit.
See this payload from remote central for a power button:
```
0000 006d 0000 0020 000b 001d 000a 0046 000a 001e 000a 001e 000a 001e 000a 0046 000a 001e 000a 001e 000a 001e 000a 001e 000a 001e 000a 0046 000a 0046 000a 001e 000a 001e 000a 06f0 000a 001e 000a 0046 000a 001e 000a 001e 000a 001e 000a 001e 000a 0046 000a 0046 000a 0046 000a 0046 000a 0046 000a 001e 000a 001e 000a 0046 000a 0046 000a 0651
```

And the a payload from a OK button generated by your tools:

```
0000 006D 0000 0006 03b9 00c1 00bf 00c2 00be 0611 03b9 00c1 00be 00c1 00be 0474
```

There is a lot of missing hex inside, so I think the need exactly the same amout of hex code for each  key. But I dont know how those are generated. And that's why they say that our code is missing a part I think.

crankyoldgit, 2020-05-17:
The your protocol doesn't have "same" number of hex entries for each message. You are not going to get the same number of hex entries without faking stuff that might cause the message to fail.

There is no 'missing hex'. Some protocols have lots of data bits, some have few. Ones with lots of "bits" will have lots of hex entries. e.g. NEC has 32 bits. Yours only uses 8 bits, plus due to the way it does things, it encodes more bits per entry than NEC does. e.g. NEC has two codes per bit, your's can have a single hex code have up to 8 bits. Hence your's is always going to be shorter.

dummys, 2020-05-17:
ok thanks for answer. I sended them the new code, lets wait and see if harmony engineers are dumb or not... Thanks for help.

crankyoldgit, 2020-05-20:
FYI, the changes mentioned above have now been included in the latest release (v2.7.7) of the library.

dummys, 2020-05-29:
Hey @crankyoldgit I don't know what to do now... See their response....

```
Thank you for your last response stating that you are really not amused with the support regarding your wish to have a new device added to the Harmony Database and your MyHarmony account in particular. 

Besides that you have strong doubts if the Harmony Engineers and the Harmony Datateam are knowing what they are doing if they are not able to convert your Pronto-code into workable IR-codes.
Believe me, they do understand their job, since they have been able to add more than 270,000 devices into the Harmony Database until today. 

In order to add a device to the Harmony Database, the Harmony Teams need either the commands learned by the Harmony user or the Pronto/HEX or RAW codes in the format desired. 
Although you are convinced that the Pronto-codes as provided by you should work and can be used to convert them into workable IR-commands, both the Harmony Engineers and Harmony Datateam have declared that it is not possible to use your codes. 

This has nothing to do with blocking the possibility to add a new device with IR-codes as what you used to do in the past.  Harmony wishes to have as many devices added as possible in order to serve the Harmony users even better. 
The possibility to add a new device is still there, but only if you are able to learn the IR-commands from the original remote to the Harmony remote. 

I hope that I have informed you sufficiently and wish you a nice day. 
```

crankyoldgit, 2020-05-29:
I personally would not have taken the approach you took, but I can understand your feelings etc on the matter.

> Unless you are able to provide the HEX/PRONTO/RAW-codes from MultiBracket there is nothing we can do with respect to add your device to the Harmony Database. 

Maybe they only will accept the codes if they originate from the manufacturer.

I'd suggest a less confrontational stance, and ask them to have an engineer explain/document what it is about (say one of) the codes that is wrong, incorrect, or what is specifically formatted poorly.

Perhaps request them to pass on links to the specs for the format(s) they do accept so you can pass it on to the manufacturer (and me ;-) so we can get the "correct" codes in a suitable format. e.g. what is _their_ raw format?

Btw, have you tried looking at https://irdb.globalcache.com/ or remotecentral to see if they have documented codes for the device?

Also, I assume your harmony device can't "learn" these codes directly?

dummys, 2020-05-29:
> Maybe they only will accept the codes if they originate from the manufacturer.
Manufacturer don't care, they respond to me "use included remote, period"

Yes I give them the link of the github issue, but I don't think they will come here to discuss with us to be honest...

> Also, I assume your harmony device can't "learn" these codes directly?
Of course not, otherwise I won't bother you with all the work.

I'm waiting their response, and ask them after to write a small document on the format they are waiting for. Because for me we already gave them the good format.
anyway thanks again helping me on this, I know you don't have to so I really appreciate.

crankyoldgit, 2020-05-29:
Maybe an approach is to get them to explain (document) to you, "what about the provided code(s) is incorrect". I am quite interested to find out what is wrong with them. (so I can "fix" it)

dummys, 2020-05-29:
Yes for sure I will start on this topic with them.

crankyoldgit, 2020-05-29:
I found this while hunting around. It may help you .. or it may screw things up etc.

https://kodi.wiki/view/Pronto_Universal_Remote_Hex_Codes

I tried converting one of the pronto codes you provided them. i.e. `0000 006D 0000 0006 0537 00c1 00bf 00c2 00bf 048e 0538 00c1 00bf 00c1 00bf 0474` (which if think is/was LEFT) via http://members.harmonyremote.com/EasyZapper/Infrared/ShowInfrared.asp (Click Manual) -> http://members.harmonyremote.com/EasyZapper/Infrared/ManualInfrared.asp?ReturnUrl=ShowInfrared%2Easp (Add the pronto code in the "Data Value" box under "Pronto") .. and I get a "successful" conversion to  `F948DP8921S13D3P139FS13EDP139FS77C5P893BS13D3P139FS13D3P139FS7519P8921S13D3P139FS13EDP139FS77C5P893BS13D3P139FS13D3P139FS7519P8921S13D3P139FS13EDP139FS77C5P893BS13D3P139FS13D3P139FS7519`

So, their "software" seems to accept those "pronto" codes you last provided to them fine. *shrug*

crankyoldgit, 2020-05-29:
Or see if they accept "Global Cache" format. See/read: https://globalcache.zendesk.com/hc/en-us/articles/360034968311-iConvert-Converting-IR-code-formats

dummys, 2020-05-31:
Ok nice, I will wait that they respond to me and talk about this as well. I think your link harmonyremote.com is for the oldest version of their software/API, because back in the days, someone has wrote a python script to add it manually, but this didn't work anymore

crankyoldgit, 2020-05-31:
I went over the spec that I could find for Pronto Codes. I'm pretty confident the original and the newer sets of codes are valid. No idea what their problem is with the codes. It seems you are not alone in having issues dealing with them. Oh well, I hope to find out what the issue is, but I won't hold my breath.

Not a lot I can do for you I'm afraid. Good luck. :-(

dummys, 2020-06-02:
pfiou nice response again:

```
Of course I have forwarded your request and the reply can be found below:

In this case we have asked for HEX/PRONTO/RAW IR codes. 
At first binary codes were submitted which we don't support. 
Subsequently you have put a lot of effort in getting the PRONTO codes to us, but those were too short an can not be used. 

We then provided an example of the codes as they should be provided: http://www.remotecentral.com/cgi-bin/codes/denon/avr-2400-2600

Upon visiting the: https://github.com/crankyoldgit/IRremoteESP8266/issues/1103 we found the same (binary) codes which were provided earlier and those codes are too short. 

As a final resort please try to learn the commands from the original remote.  

Harmony might take 3 trials to register a command.
While learning don’t press & hold the commands for too long.
Just a simple short press is enough for Harmony to capture the commands and check if that helps.

If you are unable to learn then we are helpless and can't do anything in this matter. 

As soon as you have been able to add the MultiBracket device to your Harmony set-up including the learned IR-commands, let us know and we will take the data from the device including learned commands and add your device to the Harmony Database. 

I hope to have informed you sufficiently and wish you a nice day. 
```
I'm still wondering what those codes: http://www.remotecentral.com/cgi-bin/codes/denon/avr-2400-2600 has more than our. Is it simple command just repeated several time ?

crankyoldgit, 2020-06-02:
Too short?! Sigh.
You can't help it that the protocol is short.

Then yes, I think you need to ask them what their minimum length is for a PRONTO code, and then we add repeats till it meets their requirements. I hope the actual "device" can handle receiving a code that is repeated that often.

The other option we have is to try to make a single message (pronto code) be split up arbitrarily into lots of smaller chunks that also fools your device.
e.g. Instead of a single 5000usec (or multiple there in) "pulse", break it into 5 x 1000usec pulses with a "gap" of 0 usecs or maybe just one period (26usecs) between them.

i.e. to see if we can pad it out to cover their arbitrary length limit. :-(

crankyoldgit, 2020-06-02:
> I'm still wondering what those codes: http://www.remotecentral.com/cgi-bin/codes/denon/avr-2400-2600 has more than our. Is it simple command just repeated several time ?

It's more complicated than that. Your protocol uses a very simple way of encoding "bits" than Denon (and frankly, most other protocols) use. Their bits are much "shorter" but there are far more of them. e.g. 32 to 64 bits depending. The way they encode it means for every "bit" there is a pulse and a gap. Thus it creates a pair of "entries" in the raw data/pronto code. Your protocol can merge several bits into a single pulse. e.g. I think we saw upto 5 bits being merged into a single entry (half of a pronto pair) thus it produces a numerically short (as in number of entries) code, but it's about the same total transmission time.

I'm not going to get into which is a better scheme for a protocol. Better minds than mine came up with these protocols. There are pros & cons for each, and I'm certainly not going to get into them here/now. :-P

TL;DR: I would try giving them their own "raw" format (via that web page I linked) to see if they will accept that, then try the "repeating the signal" approach and converting it to pronto (but check the repeated signal works with your actual device), and if that fails, we try the "horrible" hacks to make it seem longer without it being longer.
The first two you can do on your own, grab me for the last one if we need to resort to that.

dummys, 2020-06-02:
ok im on it, Will try to do it tonight, thanks

commit ee886a5d (2020-05-06): [bug] Incorrect length, and needs a trailing gap.
* Length stored in Pronto is the number of pairs, not the number of entries. Divide by 2.
* As the pronto code expects pairs of data, pad out the raw data with a fake trailing gap/space which is really there but we just assume it.
* Add option to customise the gap value.

* Update unit tests.

Fixes #1103
PR #1104 (2020-05-12): Add RawToPronto.py tool & improve `sendPronto()` precision
Convert a `rawData` entry into simplistic Pronto code.
Does not handle repeats etc.

* Improve precision of `sendPronto()` by 10 times.
  - Produces more accurate lengths for large values. e.g. Multibracket protocol.
  - Due to the integer maths used, some precision was lost. The lose was quite noticeable on large values.

Fixes #1103
commit 8a1a0055 (2020-05-12): Add RawToPronto.py tool & improve `sendPronto()` precision (#1104)

Convert a `rawData` entry into simplistic Pronto code.
Does not handle repeats etc.

* Improve precision of `sendPronto()` by 10 times.
  - Produces more accurate lengths for large values. e.g. Multibracket protocol.
  - Due to the integer maths used, some precision was lost. The lose was quite noticeable on large values.

Fixes #1103
PR #1106 (2020-05-12): Add support for Multibrackets protocol.
* Basic send/decode support only. It's a very simple protocol.
* Unit tests etc.

Fixes #1103