crankyoldgit/IRremoteESP8266 #1105: add support for SM5021 IR chip

ouinouin, 2020-05-06:
Hi, 
I found this chip in some post 2000 ceiling fans many times, but now I have a bit of time to help to integrate this (unfortunately Im not a skilled coder but a can use well logic analyzer , flash esps, use all the reverse enginerring tools.
the chip datasheet is here, the protocol is fairly basic , its a key encoder to IR chip , it kan encode 8 keys.

- a start word : 110
- two custom bits (used for branding purpose i suppose)
- 7 bits coding for 8 keys the table is given in the doc (....really not very efficient )
- last field is an empty field of four bits

no crc but the frame is repeated 4 times per key stroke.
i ll update the post to give the raw signal clocking.

https://pdf1.alldatasheet.com/datasheet-pdf/view/124369/ANALOGICTECH/SM5021B.html
are you interrested to support this.
if you miss time I can try to code something despite my very low skills if you orient me to a protocol that looks like this.
thanks for your nice library

ouinouin, 2020-05-06:
[image]
timing looks like 38khz

crankyoldgit, 2020-05-06:
Hey @ouinouin You can probably add the code yourself if you follow these instructions:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol

There are tools that automatically generate the code for you, or at least a good starting point.

But first step is to collect some `rawdata` via the `IRrecvDumpV2` example code.
That spec sheet is pretty good, but _"In hard data, we trust!"_

I'll try to help you along the way.

crankyoldgit, 2020-05-07:
Have you collected any `rawdata` yet?

ouinouin, 2020-05-07:
not yet, i ll try now , but i remain doubtful if my platform is the best to do the test since : it is an ESP32, (olimex ESP32-EVB) (your lib is embedded in OpenMqttgateway project) and de IR receiver is stated to be fit for 36khz freq, i try to scrape a component best fit for the purpose.

ouinouin, 2020-05-07:
hi, 
[image]
the capture matches the datasheet.
the message is repeated 4 times (in the case of my remote its 8 times (4 while pressing, 4 while releasing but in their protocol summary its stated message to be repeated  4 times).

ouinouin, 2020-05-07:
hi, managed to find an ESP8266 here is the raw value for each button : 
light on 
light off
fan speed =off
fan speed = 1
fan speed =2 
fan speed = 3
imestamp : 000376.473
Library   : v2.7.6

Protocol  : SYMPHONY
Code      : 0x1FB (11 Bits)
uint16_t rawData[191] = {1322, 374,  1322, 398,  448, 1218,  476, 1216,  478, 1218,  478, 1216,  476, 1218,  476, 1218,  1324, 398,  446, 1220,  476, 1220,  474, 8000,  1322, 400,  1292, 402,  442, 1252,  440, 1254,  440, 1254,  440, 1254,  440, 1254,  440, 1254,  1288, 408,  438, 1256,  440, 1254,  440, 8036,  1288, 406,  1288, 408,  438, 1254,  440, 1256,  440, 1254,  438, 1256,  440, 1254,  440, 1254,  1288, 408,  438, 1254,  440, 1254,  440, 8036,  1288, 408,  1288, 406,  438, 1256,  438, 1256,  438, 1256,  440, 1256,  438, 1256,  438, 1256,  1288, 406,  440, 1254,  440, 1254,  440, 8036,  1288, 408,  1286, 408,  440, 1254,  440, 1254,  440, 1254,  440, 1256,  438, 1256,  438, 1256,  1288, 406,  440, 1254,  440, 1256,  438, 8036,  1286, 408,  1286, 408,  440, 1256,  442, 1254,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 8038,  1286, 408,  1286, 408,  436, 1258,  438, 1256,  438, 1256,  438, 1282,  412, 1282,  414, 1280,  414, 1280,  414, 1280,  412, 1282,  414, 8062,  1262, 432,  1262, 432,  414, 1282,  412, 1282,  414, 1280,  414, 1280,  412, 1282,  412, 1282,  412, 1282,  412, 1282,  412, 1282,  414};  // SYMPHONY 1FB
uint64_t data = 0x1FB;

Timestamp : 000389.698
Library   : v2.7.6

Protocol  : SYMPHONY
Code      : 0x1EF (11 Bits)
uint16_t rawData[191] = {1324, 398,  1296, 398,  448, 1218,  476, 1218,  476, 1218,  476, 1218,  1324, 398,  448, 1218,  476, 1218,  476, 1218,  476, 1220,  474, 8000,  1322, 402,  1292, 402,  442, 1252,  442, 1254,  440, 1254,  440, 1254,  1288, 408,  438, 1254,  440, 1254,  440, 1254,  440, 1254,  440, 8036,  1288, 410,  1284, 408,  438, 1254,  440, 1254,  440, 1256,  438, 1254,  1288, 408,  438, 1254,  440, 1254,  440, 1254,  440, 1254,  440, 8034,  1288, 406,  1288, 408,  438, 1254,  440, 1254,  440, 1256,  440, 1254,  1288, 408,  438, 1256,  440, 1254,  440, 1254,  440, 1256,  438, 8036,  1286, 408,  1288, 406,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  1286, 408,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  440, 8036,  1288, 408,  1286, 408,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 8038,  1286, 408,  1286, 432,  414, 1282,  412, 1258,  438, 1256,  436, 1282,  414, 1280,  414, 1282,  412, 1282,  412, 1282,  412, 1282,  414, 8062,  1260, 432,  1262, 434,  412, 1282,  412, 1280,  412, 1282,  412, 1282,  412, 1282,  412, 1282,  414, 1280,  414, 1282,  412, 1282,  412};  // SYMPHONY 1EF
uint64_t data = 0x1EF;

Timestamp : 000392.119
Library   : v2.7.6

Protocol  : SYMPHONY
Code      : 0x1F7 (11 Bits)
uint16_t rawData[191] = {1324, 372,  1322, 398,  448, 1218,  476, 1218,  476, 1218,  478, 1216,  478, 1218,  1324, 398,  448, 1218,  476, 1220,  474, 1220,  474, 8000,  1320, 402,  1292, 404,  440, 1252,  442, 1254,  440, 1254,  440, 1254,  440, 1254,  1288, 406,  440, 1254,  440, 1254,  440, 1254,  440, 8036,  1288, 408,  1286, 408,  438, 1254,  440, 1256,  440, 1254,  440, 1254,  440, 1254,  1288, 408,  438, 1254,  440, 1254,  440, 1254,  440, 8036,  1288, 408,  1288, 408,  438, 1256,  440, 1254,  440, 1256,  438, 1254,  440, 1256,  1286, 408,  440, 1254,  440, 1254,  440, 1256,  438, 8036,  1288, 406,  1288, 408,  438, 1254,  440, 1256,  438, 1256,  438, 1256,  438, 1256,  1286, 406,  440, 1256,  438, 1256,  438, 1256,  438, 8036,  1288, 408,  1286, 408,  438, 1256,  440, 1254,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 8038,  1288, 406,  1286, 432,  414, 1280,  414, 1256,  438, 1280,  414, 1280,  414, 1280,  414, 1282,  412, 1282,  412, 1282,  412, 1282,  414, 8062,  1262, 432,  1262, 432,  414, 1282,  414, 1280,  412, 1282,  412, 1282,  414, 1282,  412, 1282,  414, 1280,  412, 1282,  412, 1282,  412};  // SYMPHONY 1F7
uint64_t data = 0x1F7;

Timestamp : 000394.324
Library   : v2.7.6

Protocol  : UNKNOWN
Code      : 0x827AA7B (96 Bits)
uint16_t rawData[191] = {1324, 372,  1322, 398,  448, 1218,  476, 1216,  478, 1216,  478, 1218,  478, 1218,  476, 1218,  476, 1220,  474, 1220,  474, 1220,  1346, 7128,  1320, 402,  1290, 404,  440, 1252,  442, 1254,  440, 1254,  440, 1254,  440, 1254,  440, 1254,  440, 1254,  442, 1254,  440, 1254,  1288, 7186,  1288, 408,  1286, 408,  438, 1254,  440, 1254,  440, 1254,  440, 1254,  440, 1254,  440, 1256,  440, 1254,  440, 1254,  440, 1254,  1288, 7186,  1288, 408,  1286, 408,  438, 1254,  440, 1256,  440, 1254,  440, 1254,  440, 1254,  440, 1254,  438, 1256,  438, 1256,  440, 1254,  1288, 7188,  1288, 408,  1286, 408,  438, 1256,  438, 1256,  440, 1254,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  438, 1256,  440, 1254,  1288, 7188,  1286, 408,  1286, 408,  440, 1254,  438, 1256,  438, 1256,  438, 1256,  440, 1254,  438, 1254,  440, 1256,  438, 1256,  440, 1256,  438, 8038,  1284, 408,  1286, 408,  438, 1282,  412, 1282,  414, 1280,  414, 1280,  414, 1282,  414, 1280,  412, 1276,  412, 1286,  414, 1282,  412, 8062,  1260, 434,  1262, 432,  414, 1280,  414, 1282,  412, 1282,  412, 1282,  412, 1282,  414, 1280,  414, 1282,  412, 1282,  412, 1282,  412};  // UNKNOWN 827AA7B

Timestamp : 000397.159
Library   : v2.7.6

Protocol  : SYMPHONY
Code      : 0x1FD (11 Bits)
[... cut ...] 1258,  438, 1256,  438, 1256,  438, 1256,  438, 8038,  1286, 408,  1286, 408,  438, 1256,  438, 1256,  438, 1282,  412, 1258,  438, 1280,  414, 1280,  414, 1280,  414, 1282,  412, 1282,  414, 8062,  1262, 432,  1262, 434,  412, 1282,  412, 1282,  412, 1282,  414, 1282,  412, 1280,  414, 1282,  412, 1282,  412, 1282,  414, 1280,  414, 8062,  1260, 434,  1260, 434,  412, 1282,  412, 1280,  414, 1282,  412, 1282,  412, 1282,  412, 1282,  412, 1282,  414, 1282,  412, 1282,  412};  // UNKNOWN B5AE61F

ouinouin, 2020-05-07:
hi, @crankyoldgit 
im stuck at the autoanalyze tool , its not ver yclear wich part shall i paste in  the terminal, with, without the brackets, with the final semicolon ? 

on the documents it states : 
does the output include uint16_t rawData[] = ?
Run the uint16_t rawData[] = {...}; output through the auto_analyse_raw_data.py

ouinouin, 2020-05-07:
hi, finally managed to do it 

Found 191 timing entries.
Potential Mark Candidates:
[1346, 478]
Potential Space Candidates:
[8062, 7188, 1286, 434]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 1292
kHdrSpace  = 7175
kBitMark   = 436
kOneSpace  = 1256
kZeroSpace = 407
kSpaceGap = 8050

Decoding protocol based on analysis so far:

kHdrMark+kZeroSpace(UNEXPECTED)0
  Bits: 1
  Hex:  0x0 (MSB first)
        0x0 (LSB first)
  Dec:  0 (MSB first)
        0 (LSB first)
  Bin:  0b0 (MSB first)
        0b0 (LSB first)
kHdrMark+kZeroSpace(UNEXPECTED)0111111111
  Bits: 10
  Hex:  0x1FF (MSB first)
        0x3FE (LSB first)
  Dec:  511 (MSB first)
        1022 (LSB first)
  Bin:  0b0111111111 (MSB first)
        0b1111111110 (LSB first)
kHdrMark+kHdrSpace+kHdrMark+kZeroSpace(UNEXPECTED)0
  Bits: 1
  Hex:  0x0 (MSB first)
        0x0 (LSB first)
  Dec:  0 (MSB first)
        0 (LSB first)
  Bin:  0b0 (MSB first)
        0b0 (LSB first)
kHdrMark+kZeroSpace(UNEXPECTED)0111111111
  Bits: 10
  Hex:  0x1FF (MSB first)
        0x3FE (LSB first)
  Dec:  511 (MSB first)
        1022 (LSB first)
  Bin:  0b0111111111 (MSB first)
        0b1111111110 (LSB first)
kHdrMark+kHdrSpace+kHdrMark+kZeroSpace(UNEXPECTED)0
  Bits: 1
  Hex:  0x0 (MSB first)
        0x0 (LSB first)
  Dec:  0 (MSB first)
        0 (LSB first)
  Bin:  0b0 (MSB first)
        0b0 (LSB first)
kHdrMark+kZeroSpace(UNEXPECTED)0111111111
  Bits: 10
  Hex:  0x1FF (MSB first)
        0x3FE (LSB first)
  Dec:  511 (MSB first)
        1022 (LSB first)
  Bin:  0b0111111111 (MSB first)
        0b1111111110 (LSB first)
kHdrMark+kHdrSpace+kHdrMark+kZeroSpace(UNEXPECTED)0
  Bits: 1
  Hex:  0x0 (MSB first)
        0x0 (LSB first)
  Dec:  0 (MSB first)
        0 (LSB first)
  Bin:  0b0 (MSB first)
        0b0 (LSB first)
kHdrMark+kZeroSpace(UNEXPECTED)0111111111
  Bits: 10
  Hex:  0x1FF (MSB first)
        0x3FE (LSB first)
  Dec:  511 (MSB first)
        1022 (LSB first)
  Bin:  0b0111111111 (MSB first)
        0b1111111110 (LSB first)
kHdrMark+kHdrSpace+kHdrMark+kZeroSpace(UNEXPECTED)0
  Bits: 1
  Hex:  0x0 (MSB first)
        0x0 (LSB first)
  Dec:  0 (MSB first)
        0 (LSB first)
  Bin:  0b0 (MSB first)
        0b0 (LSB first)
kHdrMark+kZeroSpace(UNEXPECTED)0111111111
  Bits: 10
  Hex:  0x1FF (MSB first)
        0x3FE (LSB first)
  Dec:  511 (MSB first)
        1022 (LSB first)
  Bin:  0b0111111111 (MSB first)
        0b1111111110 (LSB first)
kHdrMark+kHdrSpace+kHdrMark+kZeroSpace(UNEXPECTED)0
  Bits: 1
  Hex:  0x0 (MSB first)
        0x0 (LSB first)
  Dec:  0 (MSB first)
        0 (LSB first)
  Bin:  0b0 (MSB first)
        0b0 (LSB first)
kHdrMark+kZeroSpace(UNEXPECTED)0111111111GAP(8038)
  Bits: 10
  Hex:  0x1FF (MSB first)
        0x3FE (LSB first)
  Dec:  511 (MSB first)
        1022 (LSB first)
  Bin:  0b0111111111 (MSB first)
        0b1111111110 (LSB first)
kHdrMark+kZeroSpace(UNEXPECTED)0
  Bits: 1
  Hex:  0x0 (MSB first)
        0x0 (LSB first)
  Dec:  0 (MSB first)
        0 (LSB first)
  Bin:  0b0 (MSB first)
        0b0 (LSB first)
kHdrMark+kZeroSpace(UNEXPECTED)0111111111GAP(8062)
  Bits: 10
  Hex:  0x1FF (MSB first)
        0x3FE (LSB first)
  Dec:  511 (MSB first)
        1022 (LSB first)
  Bin:  0b0111111111 (MSB first)
        0b1111111110 (LSB first)
kHdrMark+kZeroSpace(UNEXPECTED)0
  Bits: 1
  Hex:  0x0 (MSB first)
        0x0 (LSB first)
  Dec:  0 (MSB first)
        0 (LSB first)
  Bin:  0b0 (MSB first)
        0b0 (LSB first)
kHdrMark+kZeroSpace(UNEXPECTED)0111111111
  Bits: 10
  Hex:  0x1FF (MSB first)
        0x3FE (LSB first)
  Dec:  511 (MSB first)
        1022 (LSB first)
  Bin:  0b0111111111 (MSB first)
        0b1111111110 (LSB first)

Total Nr. of suspected bits: 88

ouinouin, 2020-05-07:
the autoanalyse is failing.
a one bit bit is defined by emission during 1357us + space of 336 us 
a zero bit is defined by emission during 519us + space of 1195us
the end of frame period is 7952us .... 4,6 bits... note the datasheet writes that sync time is 4 bits in page 3 and 6 bits in page 4... deliciously chinese...

crankyoldgit, 2020-05-07:
You don't need to run it through the `auto_analyse_raw_data.py` program.
The library already recognises this protocol.
e.g.
```
Protocol : SYMPHONY
Code : 0x1FD (11 Bits)
```

`irsend.sendSymphony(0x1FD);` will probably work for you. Or as your captures suggest, the remote is blurting out the message, followed by 7 repeats of it, so `irsend.sendSymphony(0x1FD, 11, 7);` or `irsend.sendSymphony(0x1FD, kSymphonyBits, 7);`

Try recapturing the other key presses. You might just be having problems with them due to the frequency mismatch with the receiver.

p.s. The auto_analyse program won't work with this protocol as it uses [Manchester Code](https://en.wikipedia.org/wiki/Manchester_code) encoding and the program only works with the super-common "space encoding" protocols.

ouinouin, 2020-05-08:
but when it triggers the symphony decoding, it says there is 11 bits , instead of the 16 bits of the 5021 protocol. or this means maybe the implementation of the symphony protocol is incomplete or doesnt really knows the "shape" of a bit ?

crankyoldgit, 2020-05-08:
Yep, I can see the problem now for the trailing bit. The user supplied data never included that scenario and it worked for them.

FYI, those last "4 bits" are not really bits at all. Otherwise we are talking non-binary. ;-)
i.e. a bit is either a short "pulse" followed by approx, 4x lack of a "pulse", or a 4 x pulse followed by a  lack of short "pulse". Those "4 bits" have no pulse in them at all. Hence neither `1` or `0`.

ouinouin, 2020-05-08:
yes the 4 last arent bits, this is a silence .... but the timing of the silence is 4 bittime :-) this is described correctly in the document.

crankyoldgit, 2020-05-08:
Agreed. This is a "constant time"-type protocol. i.e. Not Manchester, as the pulses are not a 2x of of the clock freq. Nor does it alternate.

Anyway, I'm on it/looking at it.

ouinouin, 2020-05-08:
yes, unlike manchester, there is no 2/3 13 split... dont ask why. also note that with 8 bits lengths they only code... 8 buttons. :-).

crankyoldgit, 2020-05-08:
@ouinouin Hopefully this fixes everything, and it matches (at least some of) your captures that were flagged as `UNKNOWN`.

Can you please download the code in Branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1105 / PR #1107 ?
Please let me know how it goes.

ouinouin, 2020-05-09:
downloaded and tested in reception 
the results are very good : 
output : 
Timestamp : 001543.424
Library   : v2.7.6

Protocol  : SYMPHONY
Code      : 0xC20 (12 Bits)
uint16_t rawData[191] = {1326, 398,  1296, 398,  448, 1246,  448, 1246,  448, 1246,  448, 1220,  1322, 400,  446, 1246,  448, 1246,  448, 1246,  448, 1246,  450, 8002,  1320, 400,  1294, 426,  418, 1250,  444, 1250,  444, 1252,  442, 1254,  1290, 430,  414, 1254,  440, 1254,  440, 1254,  440, 1278,  416, 8034,  1290, 430,  1266, 430,  416, 1252,  442, 1252,  442, 1252,  442, 1252,  1290, 406,  440, 1254,  440, 1254,  440, 1256,  438, 1256,  440, 8036,  1290, 406,  1288, 430,  416, 1278,  416, 1280,  416, 1278,  416, 1278,  1266, 430,  416, 1278,  416, 1278,  416, 1278,  416, 1280,  414, 8062,  1264, 430,  1266, 430,  414, 1278,  416, 1280,  416, 1254,  440, 1252,  442, 1254,  442, 1252,  442, 1252,  442, 1252,  442, 1254,  440, 8034,  1290, 406,  1290, 404,  440, 1254,  442, 1252,  442, 1252,  442, 1254,  442, 1252,  442, 1254,  440, 1254,  442, 1252,  442, 1254,  442, 8034,  1290, 406,  1290, 404,  440, 1254,  442, 1254,  442, 1254,  440, 1252,  442, 1254,  442, 1252,  442, 1254,  440, 1254,  442, 1252,  442, 8036,  1288, 406,  1288, 406,  440, 1254,  440, 1254,  440, 1254,  442, 1254,  442, 1254,  440, 1254,  440, 1254,  442, 1254,  440, 1254,  440};  // SYMPHONY C20
uint64_t data = 0xC20;

now showing 12 bits , wich is the real length (better than 11 bits :-D )
I tried on the two remotes (old one and fairly recent one ) all the buttons are working (so all the possible codes since one of the two remotes has 8 buttons..
I didnt look further in the code of your project, is there a deconding of the function of each key (like on/off/vol+/vol-/ etc etc... in this case we could indicate the number of the key (key1 key2 ...key8) since their coding is shown on the datasheet.
for tinkerers who dismantle remotes to find infos, on the earliest remote , the chip is integraded under a glue spot, but an orange oscillator stamped 455 (for 455khz i suppose) might be a clue that this si the SM5021 chip.

[image]
[image]
[image]
[image]
[image]

the black remote goes with this fan : 
[image]  sold in france by castorama under their "blyss" brand... sure you ll ifnd it in other countries branded with other brand names.

i ll now test for the transmission

crankyoldgit, 2020-05-09:
Excellent.

I do tend to keep track of brand + model info of devices & remotes that are supported/known to work etc.

See https://github.com/crankyoldgit/IRremoteESP8266/blob/master/SupportedProtocols.md
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/d6ef008006f2ae16bd56465e4cc495757b42b09a/src/ir_Symphony.cpp#L7-L9

If you can supply the those details for your devices (fans & the remotes) that would be great. I'll then add them.

e.g
Brand: Blyss,  Model: BlahBlah Fan
Brand: Westinghouse,  Model: FooBar Remote.

I'm fairly confident the sending will work fine.

> I didnt look further in the code of your project, is there a deconding of the function of each key (like on/off/vol+/vol-/ etc etc... in this case we could indicate the number of the key (key1 key2 ...key8) since their coding is shown on the datasheet.

No, I don't do that level of decoding for "simple" protocol messages like this. ie. one function per message.
I only do that for "complex" messages like air conditioners, where there are multiple settings per message. It makes sense in those cases. e.g. They send the entire Aircon state in a single message each time. People want to be able to construct/change parts of those messages.

ouinouin, 2020-05-09:
hi 
Brand: Blyss,    Model: Owen-SW-5 3 speed Fan with water mist
Brand: Blyss,    Model: WP-YK8 090218 Owen-SW-5 3 speed Fan with water mist remote
Brand: Westinghouse,    Model: Unknown Ceiling fan with lights
Brand: Westinghouse,    Model: 78095 Ceiling Fan Remote 
Brand: Satellite electronic,    Model ID6 Ceiling Fan Remote

in fact while looking the [satellite electronic](https://www.fan-light.com/product.php?id=61) website [(we](url) can even add more by extension with their compatibility list, at least : 
Brand: Satellite electronic,    Model: JY199I 3speed Fan + light electronic driver
Brand: Satellite electronic,    Model: JY199I-L 3speed Fan + light electronic driver
and maybe many mmore from their catalog ...

crankyoldgit, 2020-05-09:
Thanks, I'll add them to the PR.

I tend to only add devices we know "for sure" work.

closed 2020-05-12 by crankyoldgit

crankyoldgit, 2020-05-20:
FYI, the changes mentioned above have now been included in the latest release (v2.7.7) of the library.

commit 9ac8e44a (2020-05-12): [BREAKING][BUG] Fix Symphony protocol.
* [Breaking] Per #1105 it turns out this protocol is 12 bits, and we had the 1's and 0's around the wrong way.
  - All previous values will be incorrect.
* Adjust timings slightly.
* Increase default repeat to 3, to match the spec.
* Added `matchGenericConstBitTime()` to handle constant bit time protocols correctly/better.
* Add and adjust unit tests accordingly.

FYI #1057 @soumaxetuirk
Fixes #1105
PR #1107 (2020-05-12): [BREAKING][BUG] Fix Symphony protocol.
* [Breaking] Per #1105 it turns out this protocol is 12 bits, **and** we had the 1's and 0's around the wrong way.
  - All _previous_ values will be incorrect.
* Adjust timings slightly.
* Increase default repeat to `3`, to match the spec.
* Added `matchGenericConstBitTime()` to handle constant bit time protocols correctly/better.
* Add and adjust unit tests accordingly.

FYI #1057 @soumaxetuirk
Fixes #1105
commit c9052037 (2020-05-12): [BREAKING][BUG] Fix Symphony protocol. (#1107)

* [Breaking] Per #1105 it turns out this protocol is 12 bits, **and** we had the 1's and 0's around the wrong way.
  - All _previous_ values will be incorrect.
* Adjust timings slightly.
* Increase default repeat to `3`, to match the spec.
* Added `matchGenericConstBitTime()` to handle constant bit time protocols correctly/better.
* Add and adjust unit tests accordingly.

FYI #1057 @soumaxetuirk
Fixes #1105