crankyoldgit/IRremoteESP8266 #1108: Question about Daikin remote for FTXV series, 67 bit message

petakii, 2020-05-08:
Hi!

I've been trying to reverse engineer my remotes protocol and tried to run the auto_analyse_raw_data.py program. 
Strangely, the message is 67 bits long. (The dump receives 70 bits).
The reverse engineering is in-progress, but in the meantime I tried to write the code for it to decode the signal. 

So far it seems that the encoding of modes is similar to other variants of Daikin, the difference is the number of bits. The first section is 35 bits, while the second one is 32. The extra 3 bits at the end of Section1 (and Section3) seems to be always 010.
When the timer is used, plus two sections are used, with 35+32 bits too.
This way the number of bits are not a multiple of 8. 

Any tips on how to get the code to work?

I'll post a sheet of the decoded signal when it's done.

Below is a dump:
> Timestamp : 000042.890
> Library   : v2.7.6
> 
> Protocol  : UNKNOWN
> Code      : 0x9D75D075 (70 Bits)
> uint16_t rawData[139] = {8914, 4582,  558, 648,  556, 648,  558, 646,  558, 1742,  560, 646,  558, 646,  558, 646,  558, 646,  558, 1744,  558, 644,  560, 644,  560, 1744,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  560, 646,  558, 1742,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 1744,  558, 646,  558, 1744,  558, 646,  558, 646,  558, 1744,  558, 646,  558, 20112,  560, 646,  558, 646,  558, 646,  558, 646,  558, 648,  556, 646,  558, 646,  558, 646,  558, 646,  558, 644,  560, 646,  558, 646,  558, 646,  558, 1744,  558, 648,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 646,  558, 1744,  558, 646,  558, 1744,  558, 1744,  558};  // UNKNOWN 9D75D075

Here's the result of the auto_analyse without code:

> >Found 139 timing entries.
> Potential Mark Candidates:
> [8944, 588]
> Potential Space Candidates:
> [20082, 4552, 1716, 618]
> 
> Guessing encoding type:
> Looks like it uses space encoding. Yay!
> 
> Guessing key value:
> kDaikinTestHdrMark   = 8944
> kDaikinTestHdrSpace  = 4552
> kDaikinTestBitMark   = 587
> kDaikinTestOneSpace  = 1714
> kDaikinTestZeroSpace = 616
> kDaikinTestSpaceGap = 20082
> 
> Decoding protocol based on analysis so far:
> 
> kDaikinTestHdrMark+kDaikinTestHdrSpace+00011110100100000000010000001010010GAP(20082)
>   Bits: 35
>   Hex:  0xF4802052 (MSB first)
>         0x250200978 (LSB first)
>   Dec:  4102037586 (MSB first)
>         9934211448 (LSB first)
>   Bin:  0b00011110100100000000010000001010010 (MSB first)
>         0b01001010000001000000000100101111000 (LSB first)
> kDaikinTestBitMark(UNEXPECTED)10000000000001000000000000001011
>   Bits: 32
>   Hex:  0x8004000B (MSB first)
>         0xD0002001 (LSB first)
>   Dec:  2147745803 (MSB first)
>         3489669121 (LSB first)
>   Bin:  0b10000000000001000000000000001011 (MSB first)
>         0b11010000000000000010000000000001 (LSB first)
> 
> Total Nr. of suspected bits: 67

crankyoldgit, 2020-05-09:
Have a look at the Kelvinator code. It has a similar extra three (`010`) bits.
If I recall correctly, so does Gree as well.

crankyoldgit, 2020-05-09:
e.g.
Sending:
https://github.com/crankyoldgit/IRremoteESP8266/blob/9d9ee6cbaf80ede5b5119b766d0545de80508aa3/src/ir_Kelvinator.cpp#L83-L122
or
https://github.com/crankyoldgit/IRremoteESP8266/blob/9d9ee6cbaf80ede5b5119b766d0545de80508aa3/src/ir_Gree.cpp#L57-L78

Decoding:
https://github.com/crankyoldgit/IRremoteESP8266/blob/9d9ee6cbaf80ede5b5119b766d0545de80508aa3/src/ir_Kelvinator.cpp#L438-L474
or
https://github.com/crankyoldgit/IRremoteESP8266/blob/9d9ee6cbaf80ede5b5119b766d0545de80508aa3/src/ir_Gree.cpp#L535-L563

crankyoldgit, 2020-05-18:
How are you going with this? Did those examples help you?

crankyoldgit, 2020-05-27:
Another friendly chase up. If I don't hear back from you soon, I'm going to assume you've worked it out and no longer need assistance. i.e. Close this issue.

petakii, 2020-05-27:
Sorry, didn't have time to work on it, started rewriting the code based on Kelvinator/Gree, but couldn't finish it. Probably I'll start again in a few days.

crankyoldgit, 2020-05-27:
No worries. I just hadn't had a reply.

crankyoldgit, 2020-06-03:
Can I help at all?

petakii, 2020-06-13:
Feel free to close the issue, I'll probably figure it out based on the ones you sent, when I'll have time again. Thanks for the help!

closed 2020-06-13 by petakii