crankyoldgit/IRremoteESP8266 #1109: consider to use Pattern-Rules in Makefiles

NiKiZe, 2020-05-09:
http://www.gnu.org/software/make/manual/make.html#Pattern-Rules

right now Makefile have loooots of these:
```make
ir_PROTOCOL.o : $(USER_DIR)/ir_PROTOCOL.h $(USER_DIR)/ir_PROTOCOL.cpp $(COMMON_DEPS)
    $(CXX) $(CPPFLAGS) $(CXXFLAGS) $(INCLUDES) -c $(USER_DIR)/ir_PROTOCOL.cpp

ir_PROTOCOL_test.o : ir_PROTOCOL_test.cpp $(COMMON_TEST_DEPS) $(GTEST_HEADERS)
    $(CXX) $(CPPFLAGS) $(CXXFLAGS) $(INCLUDES) -c ir_PROTOCOL_test.cpp

ir_PROTOCOL_test : $(COMMON_OBJ) ir_PROTOCOL_test.o
    $(CXX) $(CPPFLAGS) $(CXXFLAGS) -lpthread $^ -o $@
```

Have there been considerations to simplify it by using something like:
```make
ir_%.o : $(USER_DIR)/ir_%.cpp $(COMMON_DEPS)
    $(CXX) $(CPPFLAGS) $(CXXFLAGS) $(INCLUDES) -c $(USER_DIR)/ir_'$*'.cpp

ir_%_test.o : ir_%_test.cpp $(COMMON_TEST_DEPS) $(GTEST_HEADERS)
    $(CXX) $(CPPFLAGS) $(CXXFLAGS) $(INCLUDES) -c ir_'$*'_test.cpp

ir_%_test : $(COMMON_OBJ) ir_%_test.o
    $(CXX) $(CPPFLAGS) $(CXXFLAGS) -lpthread $^ -o $@
```

I'm in no way knowledgeable in Makefile language, however it seems to be a loot of duplication to me, would we be open to test this?
Or maybe there is already known disadvantages?

crankyoldgit, 2020-05-09:
> would we be open to test this?

YES!

I have very little `Makefile` foo. Less than you it seems as I didn't know about `%` in Makefiles. If you can automate it, or work out how to automatically generate it etc. That would be fabulous. 

I tried looking into tools to generate it and work out the dependencies but I gave up.

crankyoldgit, 2020-05-09:
The only tricky thing I can think of is some `ir_*.cpp` files require a dependency on their corresponding `ir_*.h` file. While some files/protocols don't have a corresponding `.h` file.

NiKiZe, 2020-05-09:
I don't think you want to generate the makefile, unless you switch to cmake.

However, it should be possible to modify the makefile to not need modifications at all when new protocols are added, but for now let's take it one step at a time, will create a PR that reduces number of lines and needed modifications, it will be far from perfect, but should still be an improvement.

crankyoldgit, 2020-05-09:
Indeed. I'm all for baby steps.
It's grown organically (and horribly) because I was mostly "Gah! Just make this thing work and re-compile all the things so that `make ir_Foo_test && ./ir_Foo_test` works!"

I hate `makefiles`! It shows, doesn't it?! Be Honest!

NiKiZe, 2020-05-09:
I might hate

> Indeed. I'm all for baby steps.
> It's grown organically (and horribly) because I was mostly "Gah! Just make this thing work and re-compile all the things so that `make ir_Foo_test && ./ir_Foo_test` works!"
> 
> I hate `makefiles`! It shows, doesn't it?! Be Honest!

I might hate them more than you :D

NiKiZe, 2020-05-09:
> The only tricky thing I can think of is some `ir_*.cpp` files require a dependency on their corresponding `ir_*.h` file. While some files/protocols don't have a corresponding `.h` file.

All protocol `.h` files are defined in `PROTOCOLS_H` which in turn goes in to `COMMON_DEPS` and later also `COMMON_TEST_DEPS`

All targets depends on `COMMON_TEST_DEPS` or `COMMON_DEPS` so they are already enforced that way, but also reading http://www.gnu.org/software/make/manual/make.html#How-Patterns-Match we should be able to have 2 targets, one referencing .h and one without, `make` should take the first target where the destination file exists.

Still testing this, so mostly a documentation of progress.

NiKiZe, 2020-05-09:
I read up a bit more on Makefiles, and introduced wildcards as well to automatically generate targets for all files that exists, hope that is reasonable as well, this should essentially remove any need to modify the Makefiles when new protocols are added

crankyoldgit, 2020-05-09:
I've also updated the wiki documentation on how to add a protocol to reflect there is no need to play with the `Makefile`s. Huzzah!

Thanks again!

PR #1110 (2020-05-09): Use pattern rules in Makefiles to reduce specific rule
Minimizes number of lines, and reduces maintenance for new protocols
This should fix #1109 

There is more work that could be done to improve things.
One observation is that RC5/RC6 is specified for two protocols in tools, but only one in tests?

This might also have added generic dependencies which wasn't there before,
but that should be a small price to pay, and should be fixable in the future.
commit 218999b2 (2020-05-09): Use pattern rules in Makefiles to reduce specific rule (#1110)

Minimises number of lines, and reduces maintenance for new protocols
This should fix #1109 

This might also have added generic dependencies which wasn't there before,
but that should be a small price to pay, and should be fixable in the future.
closed 2020-05-09 by crankyoldgit