crankyoldgit/IRremoteESP8266 #1113: Can't compile for ESP32

OscarleBG, 2020-05-11:
### Version/revision of the library used
2.7.6

1. I installed the library
2. installed the dependencies
3. opened the exemple IRMQTTServer with ArduinoIDE
4. could compile for ESP8266
5. couldn't compile for ESP32
6. tried for few hours (I'm a noob)

I always get this error:
```
Arduino: 1.8.12 (Mac OS X), Board: "ESP32 FM DevKit, 921600, None"

WARNING: library WiFiManager claims to run on esp8266 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
WARNING: library ESP8266WiFi claims to run on esp8266 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
WARNING: library ESP8266WebServer claims to run on esp8266 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from /Users/oscar/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22:0,
                 from /var/folders/15/j0bm9spx39zb1ygmy8qkny300000gn/T/arduino_modified_sketch_66146/IRMQTTServer.ino:352:
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:32:3: error: stray '\302' in program
   <title>Arduino/user_interface.h at master · esp8266/Arduino · GitHub</title>
   ^
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:32:3: error: stray '\267' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:32:3: error: stray '\302' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:32:3: error: stray '\267' in program
In file included from /Users/oscar/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22:0,
                 from /var/folders/15/j0bm9spx39zb1ygmy8qkny300000gn/T/arduino_modified_sketch_66146/IRMQTTServer.ino:352:
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:273:5: error: missing terminating ' character
     <!-- '"" --><!-- </textarea></xmp> --></option></form><form class="js-site-search-form" role="search" aria-label="Site" data-scope-type="Repository" data-scope-id="32969220" data-scoped-search-url="/esp8266/Arduino/search" data-unscoped-search-url="/search" action="/esp8266/Arduino/search" accept-charset="UTF-8" method="get">
     ^
In file included from /Users/oscar/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22:0,
                 from /var/folders/15/j0bm9spx39zb1ygmy8qkny300000gn/T/arduino_modified_sketch_66146/IRMQTTServer.ino:352:
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:320:7: error: stray '\342' in program
       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:320:7: error: stray '\206' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:320:7: error: stray '\265' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:325:7: error: stray '\342' in program
       <span class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:325:7: error: stray '\206' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:325:7: error: stray '\265' in program
In file included from /Users/oscar/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22:0,
                 from /var/folders/15/j0bm9spx39zb1ygmy8qkny300000gn/T/arduino_modified_sketch_66146/IRMQTTServer.ino:352:
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:361:7: error: stray '\342' in program
       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:361:7: error: stray '\206' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:361:7: error: stray '\265' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:366:7: error: stray '\342' in program
       <span class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:366:7: error: stray '\206' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:366:7: error: stray '\265' in program
In file included from /Users/oscar/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22:0,
                 from /var/folders/15/j0bm9spx39zb1ygmy8qkny300000gn/T/arduino_modified_sketch_66146/IRMQTTServer.ino:352:
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:393:7: error: stray '\342' in program
       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:393:7: error: stray '\206' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:393:7: error: stray '\265' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:398:7: error: stray '\342' in program
       <span class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:398:7: error: stray '\206' in program
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:398:7: error: stray '\265' in program
In file included from /Users/oscar/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22:0,
                 from /var/folders/15/j0bm9spx39zb1ygmy8qkny300000gn/T/arduino_modified_sketch_66146/IRMQTTServer.ino:352:
/Users/oscar/Documents/Arduino/libraries/WiFiEsp/src/user_interface.h:769:230: error: too many decimal points in number
               <a data-pjax="true" title="add optional espressif fw nonos-sdk 2.2.x (19.03.13) (#5873)" class="link-gray" href="/esp8266/Arduino/commit/a3ea816e65029f0d79bdc19d84971f5de3858b01">add optional espressif fw nonos-sdk 2.2.x (19.03.13) (</a><a class="issue-link js-issue-link" data-error-text="Failed to load title" data-id="420469127" data-permission-text="Title is private" data-url="https://github.com/esp8266/Arduino/issues/5873" data-hovercard-type="pull_request" data-hovercard-url="/esp8266/Arduino/pull/5873/hovercard" href="https://github.com/esp8266/Arduino/pull/5873">#5873</a><a data-pjax="true" title="add optional espressif fw nonos-sdk 2.2.x (19.03.13) (#5873)" class="link-gray" href="/esp8266/Arduino/commit/a3ea816e65029f0d79bdc19d84971f5de3858b01">)</a>
[... 449 lines cut ...]
                                                                                                                                                                                        ^
IRMQTTServer:501:28: error: expected '}' before end of line
IRMQTTServer:501:28: error: expected declaration before end of line
exit status 1
expected '}' before end of line
Invalid library found in /Users/oscar/Documents/Arduino/libraries/esp8266-2.7.1: no headers files (.h) found in /Users/oscar/Documents/Arduino/libraries/esp8266-2.7.1
Invalid library found in /Users/oscar/Documents/Arduino/libraries/esp8266-2.7.1: no headers files (.h) found in /Users/oscar/Documents/Arduino/libraries/esp8266-2.7.1

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

#### Example code used
IRMQTTServer

#### Circuit diagram and hardware used (if applicable)
I'll use an IR led with a transistor

### I have followed the steps in the [Troubleshooting Guide]
Yes, at least I tried

### Has this library/code previously worked as expected for you?
No, first time I use it.

crankyoldgit, 2020-05-12:
> WARNING: library WiFiManager claims to run on esp8266

If you read the source code for IRMQTTServer.ino around line 30ish, it tells you you need the 'development' branch for WiFiManager when using it on ESP32s. You need to go to their git page and download/install their development branch.

crankyoldgit, 2020-05-12:
https://github.com/crankyoldgit/IRremoteESP8266/blob/5fe6f991de94b578a04053de8a07e2a07c39bdaa/examples/IRMQTTServer/IRMQTTServer.ino#L36-L37

OscarleBG, 2020-05-12:
Thanks, this problem is solved for me. Unfortunately, I have another error which is rather frequent on the internet but which I could not solve, I will open another issue in case you are able to help me.

closed 2020-05-12 by OscarleBG