crankyoldgit/IRremoteESP8266 #1116: Timestamp from `tools/scrape_supported_devices.py`

NiKiZe, 2020-05-12:
### Version/revision of the library used
5fe6f991de9

### Wanted behavior
Generated timestamp in `SupportedProtocols.md` should:
* Include timezone, Also prefer using UTC
* Use timestamp of last commit to be reproducable (not cause changes to the file if run automatically)

### Actual behavior
Timestamp in `SupportedProtocols.md` becomes local current time

### Steps to reproduce the behavior
Running these commands
```
    python3 tools/scrape_supported_devices.py
    git diff
```
Always results in a changed file

if this was based on last commit (or last changed file in the case of unclean workdir) this would be reproducible and not cause unneeded changes to the documentation.

I have ideas for how to do this, and a PR will, come but might take a few days.

crankyoldgit, 2020-05-12:
Yep.
I find it annoying too. Some other source management systems (e.g. Perforce) allow variables to be expanded on commit. e.g. A timestamp or a commit number. And those can be ignored in some ways.
e.g. It knows it's just a variable name etc before the diff/commit. If I knew how to do that with git, I'd be on it like a flash.

NiKiZe, 2020-05-12:
Just saw something similar being done in a different project, so can re-implement that here.
It does need external tools, but in regards to running this script it should be doable.

crankyoldgit, 2020-05-12:
This is what I'm talking about:
https://www.perforce.com/manuals/v18.1/cmdref/Content/CmdRef/file.types.synopsis.keywords.html

In short, when it sees the _variable_ syntax in a file, it just replaces what ever it's value is (i.e. between the `$`s with what ever the current value is/will be with/at the commit.

NiKiZe, 2020-05-23:
Sorry that this took some time...
Indeed, if it did this on commit that would be perfect.
What I had in mind is from https://lists.ipxe.org/pipermail/ipxe-devel/2020-May/007051.html
essentially using: `git show -s --format=%ct HEAD`

What could be done to achieve the same is to have a script that calls this, adds the generated file, and then amends the previous commit.

I have another idea for this that I have implemented.
Which simply don't create a new file unless there is changes, however I want to use this git date as well, Hope to have a PR up in a few....

closed 2020-05-23 by NiKiZe

crankyoldgit, 2020-06-23:
FYI, the code & changes referenced above have been included in the [v2.7.8 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.7.8) of the library.

PR #1140 (2020-05-23): scrape_supported_devices.py: avoid changes to SupportedProtocols.md
* only rewrite SupportedProtocols.md if any changes are detected
* use last git commit date in header instead of current time

Fix #1116