crankyoldgit/IRremoteESP8266 #1118: Add support for Gree A/C remote button "Temp"

harrier914, 2020-05-12:
### Version/revision of the library used

Library : v2.7.1

### Adding protocol

Running the "Gree" library code in ESP8266 platform and the functions works amazing great! 

The remote unit (model YAA1FBF) has a button named "Temp" to ask the indoor A/C unit to display the temperature sensing by indoor & outdoor unit in sequential order, which can't find the function code in current library. 

Just wonder if this button function can be added in the library. Thanks.

NiKiZe, 2020-05-12:
If you run the IRrecvDumpV2 example, and record the output when you press that button, what do you get?

The data for what you want to send, needs to be provided, and can't be guessed.

harrier914, 2020-05-13:
Hi, the output from IRrecvDumpV2 recorded the "Temp" button pressing 4 times in one complete cycle of display on A/C unit:

Code      : 0x4C046050010100A0 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 20C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off
uint16_t rawData[139] = {8972, 4534,  606, 598,  608, 598,  606, 1702,  606, 1700,  606, 598,  608, 598,  606, 1700,  606, 598,  608, 600,  606, 598,  606, 1700,  608, 596,  606, 600,  606, 600,  604, 598,  608, 598,  608, 600,  606, 598,  606, 600,  604, 598,  608, 600,  608, 1698,  606, 1700,  606, 598,  608, 600,  606, 598,  608, 600,  604, 600,  606, 1700,  608, 598,  608, 1700,  606, 598,  608, 598,  606, 1704,  604, 598,  608, 20030,  606, 1700,  606, 598,  608, 598,  606, 600,  606, 598,  608, 598,  608, 598,  608, 600,  606, 1700,  608, 600,  604, 600,  606, 598,  608, 598,  606, 600,  606, 596,  606, 598,  606, 598,  608, 598,  608, 598,  606, 600,  606, 598,  608, 598,  608, 600,  606, 598,  606, 596,  608, 598,  608, 598,  606, 598,  608, 598,  606, 1700,  608, 598,  606, 1698,  610};  // GREE
uint8_t state[8] = {0x4C, 0x04, 0x60, 0x50, 0x01, 0x01, 0x00, 0xA0};

Timestamp : 000181.545
Library   : v2.7.1

Protocol  : GREE
Code      : 0x4C046050010200A0 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 20C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off
uint16_t rawData[139] = {8944, 4534,  606, 600,  604, 598,  608, 1702,  624, 1682,  604, 600,  626, 582,  624, 1682,  626, 578,  606, 598,  606, 600,  606, 1702,  604, 598,  628, 578,  606, 598,  606, 600,  606, 600,  624, 580,  604, 600,  606, 598,  606, 600,  606, 598,  608, 1700,  606, 1700,  606, 600,  604, 600,  626, 580,  606, 602,  604, 600,  606, 1702,  606, 600,  604, 1700,  606, 600,  622, 582,  606, 1702,  604, 600,  604, 20034,  604, 1702,  606, 600,  626, 578,  606, 598,  608, 600,  604, 600,  606, 600,  606, 596,  608, 600,  606, 1700,  624, 582,  606, 598,  606, 600,  606, 600,  604, 600,  606, 598,  606, 598,  608, 598,  606, 598,  610, 598,  606, 600,  606, 598,  608, 600,  606, 600,  606, 598,  606, 598,  626, 580,  606, 600,  606, 600,  626, 1682,  604, 600,  606, 1702,  606};  // GREE
uint8_t state[8] = {0x4C, 0x04, 0x60, 0x50, 0x01, 0x02, 0x00, 0xA0};

Timestamp : 000183.186
Library   : v2.7.1

Protocol  : GREE
Code      : 0x4C046050010300A0 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 20C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off
uint16_t rawData[139] = {8944, 4532,  606, 602,  604, 600,  606, 1704,  604, 1700,  606, 598,  606, 600,  606, 1702,  606, 600,  606, 598,  606, 600,  606, 1700,  608, 598,  608, 598,  606, 598,  608, 600,  606, 598,  608, 598,  606, 598,  608, 598,  606, 600,  606, 600,  604, 1702,  606, 1700,  606, 598,  606, 600,  606, 600,  604, 598,  608, 598,  606, 1702,  604, 598,  608, 1700,  608, 598,  606, 598,  608, 1698,  608, 600,  606, 20032,  606, 1702,  606, 600,  606, 598,  606, 600,  606, 598,  606, 598,  610, 598,  606, 598,  606, 1700,  606, 1700,  606, 596,  610, 598,  606, 600,  608, 596,  608, 598,  608, 598,  608, 600,  606, 598,  608, 598,  606, 600,  606, 598,  606, 600,  608, 600,  606, 598,  606, 600,  608, 598,  606, 600,  606, 600,  606, 600,  604, 1700,  606, 600,  606, 1698,  608};  // GREE
uint8_t state[8] = {0x4C, 0x04, 0x60, 0x50, 0x01, 0x03, 0x00, 0xA0};

Timestamp : 000184.998
Library   : v2.7.1

Protocol  : GREE
Code      : 0x4C046050010000A0 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 20C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off
uint16_t rawData[139] = {8970, 4536,  604, 598,  606, 602,  606, 1698,  608, 1700,  608, 600,  606, 600,  606, 1702,  606, 600,  606, 598,  606, 602,  604, 1700,  608, 598,  608, 602,  604, 600,  606, 600,  604, 600,  626, 580,  604, 600,  608, 598,  606, 600,  606, 600,  608, 1700,  608, 1700,  606, 600,  606, 598,  606, 600,  606, 598,  608, 598,  608, 1702,  606, 598,  606, 1702,  606, 600,  606, 598,  606, 1700,  624, 580,  626, 20012,  606, 1700,  608, 598,  618, 588,  606, 600,  606, 600,  604, 602,  606, 600,  606, 598,  626, 582,  604, 600,  604, 602,  604, 598,  608, 600,  604, 600,  608, 600,  606, 602,  604, 602,  606, 600,  604, 600,  606, 598,  608, 598,  606, 600,  606, 598,  606, 600,  604, 600,  606, 600,  606, 598,  606, 600,  604, 598,  608, 1700,  606, 600,  606, 1700,  606};  // GREE
uint8_t state[8] = {0x4C, 0x04, 0x60, 0x50, 0x01, 0x00, 0x00, 0xA0};

Thanks.

crankyoldgit, 2020-05-13:
@harrier914 Which ones correspond to which display mode?

harrier914, 2020-05-13:
I thought the "Temp" button sent out same code each pressed in sequential order but take a closer look noticed it actually sent out different code in last 3rd byte. A/C remote unit does a bit special than a TV Remote unit! 

I recorded again with description added in each section of code. Thanks for your reminder and help.

### **Display Set Temperature**

Timestamp : 000377.867
Library   : v2.7.1

Protocol  : GREE
Code      : 0x4C046050010100A0 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 20C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off
uint16_t rawData[139] = {8998, 4480,  662, 544,  634, 570,  660, 1646,  658, 1648,  636, 572,  658, 546,  660, 1648,  656, 550,  632, 572,  634, 570,  658, 1648,  660, 546,  660, 546,  660, 546,  658, 548,  658, 548,  656, 548,  632, 570,  660, 546,  660, 546,  660, 544,  660, 1648,  660, 1646,  662, 546,  632, 572,  658, 548,  658, 546,  662, 544,  660, 1646,  660, 548,  660, 1648,  658, 548,  662, 546,  660, 1648,  656, 548,  660, 19976,  636, 1698,  660, 546,  658, 548,  634, 572,  634, 570,  660, 546,  658, 546,  658, 548,  660, 1646,  660, 546,  658, 548,  660, 546,  636, 568,  636, 570,  658, 546,  660, 544,  658, 548,  660, 546,  656, 546,  662, 542,  660, 544,  662, 546,  660, 546,  638, 568,  660, 546,  662, 544,  662, 544,  660, 546,  660, 546,  634, 1672,  660, 546,  660, 1646,  662};  // GREE
uint8_t state[8] = {0x4C, 0x04, 0x60, 0x50, 0x01, 0x01, 0x00, 0xA0};

### **Display Indoor Ambient Temperature**

Timestamp : 000380.268
Library   : v2.7.1

Protocol  : GREE
Code      : 0x4C046050010200A0 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 20C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off
uint16_t rawData[139] = {9026, 4478,  660, 546,  660, 546,  662, 1556,  750, 1646,  660, 544,  660, 546,  658, 1646,  660, 544,  662, 544,  662, 544,  660, 1646,  662, 544,  660, 546,  658, 546,  660, 546,  662, 544,  660, 546,  662, 544,  660, 542,  662, 544,  658, 546,  660, 1646,  664, 1646,  658, 546,  660, 544,  660, 544,  662, 544,  662, 544,  662, 1646,  662, 532,  672, 1650,  658, 544,  660, 546,  662, 1644,  662, 544,  662, 19978,  660, 1644,  662, 544,  660, 546,  660, 548,  658, 546,  660, 544,  660, 544,  662, 546,  660, 546,  660, 1646,  662, 544,  662, 544,  660, 546,  660, 546,  658, 544,  658, 546,  660, 548,  660, 544,  662, 544,  660, 546,  660, 546,  658, 548,  656, 544,  662, 544,  658, 546,  662, 546,  658, 546,  658, 544,  660, 548,  658, 1646,  634, 570,  660, 1646,  662};  // GREE
uint8_t state[8] = {0x4C, 0x04, 0x60, 0x50, 0x01, 0x02, 0x00, 0xA0};

### **Display Outdoor Ambient Temperature**

Timestamp : 000382.230
Library   : v2.7.1

Protocol  : GREE
Code      : 0x4C046050010300A0 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 20C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off
uint16_t rawData[139] = {9024, 4478,  662, 544,  660, 546,  658, 1646,  660, 1646,  662, 544,  662, 544,  658, 1648,  660, 546,  662, 544,  636, 570,  658, 1648,  660, 546,  660, 544,  660, 544,  660, 546,  660, 546,  660, 544,  662, 544,  660, 546,  662, 544,  658, 548,  658, 1646,  662, 1646,  660, 544,  660, 546,  660, 548,  660, 544,  664, 546,  660, 1646,  660, 546,  660, 1648,  658, 546,  660, 546,  660, 1648,  660, 544,  660, 19980,  658, 1648,  656, 546,  660, 542,  662, 544,  660, 548,  656, 546,  660, 544,  662, 544,  660, 1646,  660, 1646,  662, 546,  658, 544,  662, 546,  658, 548,  658, 548,  658, 546,  660, 546,  658, 546,  660, 532,  674, 548,  656, 546,  660, 546,  660, 546,  634, 488,  742, 544,  660, 546,  662, 542,  664, 544,  660, 546,  662, 1646,  636, 570,  662, 1646,  658};  // GREE
uint8_t state[8] = {0x4C, 0x04, 0x60, 0x50, 0x01, 0x03, 0x00, 0xA0};

### **No Display**

imestamp : 000384.209
Library   : v2.7.1

Protocol  : GREE
Code      : 0x4C046050010000A0 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 20C, Fan: 0 (Auto), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off
uint16_t rawData[139] = {8998, 4478,  662, 546,  660, 548,  656, 1652,  654, 1650,  634, 570,  660, 546,  658, 1650,  656, 546,  658, 548,  658, 548,  658, 1646,  662, 544,  658, 548,  658, 546,  634, 570,  634, 572,  660, 534,  646, 570,  658, 548,  658, 548,  660, 544,  658, 1648,  660, 1646,  662, 546,  660, 546,  660, 546,  656, 548,  658, 548,  658, 1648,  658, 546,  658, 1648,  662, 542,  660, 546,  662, 1646,  658, 548,  656, 19982,  660, 1648,  658, 544,  664, 544,  662, 542,  640, 568,  664, 544,  658, 546,  662, 544,  660, 544,  662, 546,  660, 544,  660, 546,  662, 544,  660, 544,  662, 544,  662, 544,  660, 544,  660, 548,  658, 546,  660, 546,  658, 548,  658, 548,  660, 544,  662, 544,  662, 540,  666, 542,  662, 544,  662, 544,  662, 544,  662, 1646,  660, 548,  658, 1650,  658};  // GREE
uint8_t state[8] = {0x4C, 0x04, 0x60, 0x50, 0x01, 0x00, 0x00, 0xA0};

crankyoldgit, 2020-05-13:
Thanks for the data.
For that, you've been rewarded with PR #1120 which will implement that control for you. :-)
You can download the following branch and test it out: https://github.com/crankyoldgit/IRremoteESP8266/tree/gree_display_temp
Let us know if it works correctly.

harrier914, 2020-05-13:
Great! The testing result is positive. The A/C unit can display the corresponding Set, Indoor & Outdoor temperature sensing. Just the A/C unit need to receive the temperature display mode request in sequential order from value 0,1,2,3,0...  otherwise it will fail to show the temperature display. Enclosed the 'toSting()' message logged.

Thanks a lot for your prompt response. I can now keep the old remote unit inside the drawer and allow the integration of my home automation with your code completely and smartly :)

### **Result:**
Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 24C, Fan: 1 (Low), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off, Display Temp: 0 (Off)
Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 24C, Fan: 1 (Low), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off, Display Temp: 1 (Set)
Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 24C, Fan: 1 (Low), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off, Display Temp: 2 (Inside)
Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 24C, Fan: 1 (Low), Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Timer: Off, Display Temp: 3 (Outside)

crankyoldgit, 2020-05-14:
Thanks for confirming it's working.

closed 2020-05-14 by crankyoldgit

crankyoldgit, 2020-05-20:
FYI, the changes mentioned above have now been included in the latest release (v2.7.7) of the library.

commit 23a10d81 (2020-05-13): Add better text for the `toString()` output.
* Add comment about running `generate_irtext_h.sh` to IRtext.cpp

Fixes #1118
PR #1120 (2020-05-14): Gree: Add option to control display temp source.
* `(s|g)etDisplayTempSource()` functionality added.
* Unit tests added.

Fixes #1118
commit 0934cb3e (2020-05-14): Gree: Add option to control display temp source. (#1120)

* `(s|g)etDisplayTempSource()` functionality added.
* Unit tests added.

Fixes #1118